avrdudes/avrdude #1475: Adding the Avrdude build version to avrdude.conf

MCUdude, 2023-07-28:
Now that we're seeing new features added to avrdude.conf after every release, I think it's a good idea to have the build script(s) add the current Avrdude build number to the avrdude.conf file.

Suppose I happened to stumble across a random avrdude.conf file, it would be nice to look at it and instantly know for which version of Avrdude it was built.

mcuee, 2023-07-28:
I think this is a good idea.

stefanrueger, 2023-07-28:
> add the current Avrdude build number to the avrdude.conf file

Possibly even with a note saying Do not use this configuration file with an avrdude version before ...

MCUdude, 2023-07-29:
> Possibly even with a note saying Do not use this configuration file with an avrdude binary with a version before ...

Good idea! Would you mind creating a PR for this? 🙂

stefanrueger, 2023-07-29:
> mind creating a PR

Actually, I don't know what's involved. I know how I would like avrdude.conf to look like but not how to get there.

MCUdude, 2023-07-29:
> I know how I would like avrdude.conf to look like but not how to get there.

The same goes for me. Maybe @dl8dtl can have a look?

mcuee, 2023-08-15:
We may need to get the help from @mariusgreuel or another CMake expert.

Right now we have the full version string which is added to the avrdude binary. Now we need to attach it to avrdude.conf as well.
https://github.com/avrdudes/avrdude/blob/main/CMakeLists.txt#L92-L96

MCUdude, 2023-10-12:
@dl8dtl Sorry to ping you again, but now that Avrdude 7.3 has been mentioned, adding `VERSION` to the generated avrdude.conf would make it way easier to keep track if the file versions. Avrdude 7.x is backward compatible with older avrdude.conf files, but there is no easy way to tell if a user is using the latest one or not.

mcuee, 2023-10-16:
> We may need to get the help from @mariusgreuel or another CMake expert.
> 
> Right now we have the full version string which is added to the avrdude binary. Now we need to attach it to avrdude.conf as well. https://github.com/avrdudes/avrdude/blob/main/CMakeLists.txt#L92-L96

The trick is below for CMake to add the version string to avrdude binary. Something similar needs to be added to avrdude.conf.

```
    # If the commit is not tagged, include the date and commit hash in the full version string.
    if(NOT GIT_COMMIT_HASH STREQUAL GIT_TAG_HASH)
        set(AVRDUDE_FULL_VERSION "${CMAKE_PROJECT_VERSION}-${GIT_COMMIT_DATE} (${GIT_COMMIT_HASH})")
    endif()
endif()
```

@stefanrueger 

I do not know how to do this, but I think you can craft a bash script to do the same to add to `build.sh` to have a post processing of the `avrdude.conf` file. It may not be the best (and not good with MSVC build) but it may give an idea on how to do this with CMake later.

mcuee, 2023-10-16:
@Youw

Sorry to disturb you as the in-house CMake expert @mariusgreuel is busy and you are the only CMake expert I know. Just wondering if you can help here to see how to add the version string to a text file.

Youw, 2023-10-16:
It should be fairly simple:

The `avrdude.conf` has several passes to be generated and "the most CMake" one is [here is this script](https://github.com/avrdudes/avrdude/blob/cbac6349b02ae72ec3ed2b345efe8f121d846547/src/configure.cmake#L35).

Effectively it replaces all occurances of `@VARIABLE_NAME@` with corresponding CMake variable value.

The script I mentioned above is invoked from [here](https://github.com/avrdudes/avrdude/blob/cbac6349b02ae72ec3ed2b345efe8f121d846547/src/CMakeLists.txt#L122-L129) at `src/CMakeLists.txt`.

Because `src/` subdirectory is added at the [place](https://github.com/avrdudes/avrdude/blob/cbac6349b02ae72ec3ed2b345efe8f121d846547/CMakeLists.txt#L303) where the variable `AVRDUDE_FULL_VERSION` is known to be set (at least [this bit](https://github.com/avrdudes/avrdude/blob/cbac6349b02ae72ec3ed2b345efe8f121d846547/CMakeLists.txt#L322C21-L322C41) just a few lines below relies on it) I believe it is safe to say that only two lines of code needs to be changed:

- pass the `AVRDUDE_FULL_VERSION` variable as an additional CMake argument [here](https://github.com/avrdudes/avrdude/blob/cbac6349b02ae72ec3ed2b345efe8f121d846547/src/CMakeLists.txt#L122-L129) into `src/configure.cmake` script;
- place the `@AVRDUDE_FULL_VERSION@` anywhere inside the `src/avrdude.conf.in`;

(`DEFAULT_PAR_PORT` is a good example of exactly such thing being done.)

---
> no need to bother with the legacy auto-tools build scripts

Yes it will "break" the auto-tools-generated `avrdude.conf` but can be fairly easy fixed if the same `AVRDUDE_FULL_VERSION` variable is defined before [this](https://github.com/avrdudes/avrdude/blob/cbac6349b02ae72ec3ed2b345efe8f121d846547/src/configure.ac#L553) line.

---
> Possibly even with a note saying Do not use this configuration file with an avrdude binary with a version before ...

For as long as the compatibility version is known at CMake configuration time, it could be just another variable passed into the `src/configure.cmake` and the used in `src/avrdude.conf.in` (e.g. as part of a comment or so).

mcuee, 2023-10-16:
@Youw 

Thanks a lot for the help. I will try this out to see if I can come out with a solution for CMake.

mcuee, 2023-10-18:
>  the auto-tools-generated `avrdude.conf` can be fairly easy fixed if the same `AVRDUDE_FULL_VERSION` variable is defined before [this](https://github.com/avrdudes/avrdude/blob/cbac6349b02ae72ec3ed2b345efe8f121d846547/src/configure.ac#L553) line.

@dl8dtl 

Just wonder if you can help to upgrade `configure.ac`. Thanks.

Youw, 2023-10-18:
> Yes it will "break" the auto-tools-generated avrdude.conf but can be fairly easy fixed if the same AVRDUDE_FULL_VERSION variable is defined before [this](https://github.com/avrdudes/avrdude/blob/cbac6349b02ae72ec3ed2b345efe8f121d846547/src/configure.ac#L553) line.

> Just wonder if you can help to upgrade configure.ac. Thanks.

Yeah, but before we can "fix" autotools - first we need to "break" it by the changes I suggested earlier.

mcuee, 2023-10-28:
@Youw 

Sorry but looks like I am not up to the task to carry out this improvement. Hopefully some others can help.

Youw, 2023-10-28:
Try #1542

mcuee, 2023-10-29:
#1542 works fine for CMake. So I've merged it.

I will keep this issue open for now to see if @dl8dtl or others can fix the auto-tools build.

mcuee, 2023-10-29:
@Youw 

I am not so sure how to fix the auto-tools. I suspect it has something to do with the fact that it is building from the `src` subfolder by default.

mcuee, 2023-10-29:
@MCUdude 

Sorry for merging PR #1542 too fast. I will be more careful next time.

MCUdude, 2023-10-31:
Resolved in #1542 and #1547

PR #1542 (2023-10-29): Add Avrdude build version into avrdude.conf
Resolves: #1475
closed 2023-10-31 by MCUdude