avrdudes/avrdude #1478: Target autodetection

nyq, 2023-07-29:
It seems like avrdude is already able to internally autodetect target MCU, as it returns the following output when wrong target is specified:

```
C:\test\>avrdude.exe -c avrisp2 -p m8
avrdude: AVR device initialized and ready to accept instructions                                                                                                                                                   avrdude: device signature = 0x1e9702 (probably m128)                                                                                                                                                               avrdude error: expected signature for ATmega8 is 1E 93 07                                                                                                                                                                  double check chip or use -F to override this check 
```

So it appears that it would be relatively easy to expand this functionality so that it becomes fully available to the end user who is aiming at automated use of avrdude. I see two possible options:

1. Either expand -p command line option that would return the detected target name and nothing else so that the output can be unambiguously parsed by the batch file or any other launching process, something like this:
```

C:\test\>avrdude.exe -c avrisp2 -p ?
m128

```
2. Or expand -p command line option so that it would detect and immediately use detected target:
```
C:\test\>avrdude.exe -c avrisp2 -p auto -U flash:w:test.hex

```

Right now projects like [avrdudess](https://github.com/ZakKemble/AVRDUDESS/) apparently are using `avrdude.exe -c avrisp2 -p m8` command line call and then they are parsing avrdude output in order to fish out the target name, which is inherently unreliable, as the output format can change on the avrdude side. Would be much better if avrdude was providing autodetection service in some form of unambiguous API.

stefanrueger, 2023-07-29:
TLDR: as much as part autodetection is desirable it is out of scope.

Using the signature for id-ing doesn't work. 61 signatures in `avrdude.conf` clash:
```
$ avrdude -p*/d | grep = | cut -f2-3 -d, | sort | uniq -c | awk '$1 > 1' | wc -l
61
```

Most will be mild clashes, eg `ATmega329` and `ATmega329A` for which there is no difference in programming. But then there are serious ones: Take for example `ATxmega64B3`  and `ATmega4809`; they have the same signature but could hardly be more different: different families, different programming interfaces, different flash sizes, different flash page sizes, different EEPROM sizes and different EEPROM page sizes:

```
$ avrdude -p*/d | grep 0x96..0x51
    'ATxmega64B3' =>      [0x1E, 0x96, 0x51, 0x00800000, 0x11000, 0x100, 0x8c0000, 0x0800, 0x020, 0, 0x000, 0x0013, 'PM_SPM | PM_PDI | PM_XMEGAJTAG'], # /usr/local/etc/avrdude.conf 16605
    'ATmega4809' =>       [0x1E, 0x96, 0x51, 0x00004000, 0x0c000, 0x080, 0x001400, 0x0100, 0x040, 0, 0x000, 0x0013, 'PM_SPM | PM_UPDI'], # /usr/local/etc/avrdude.conf 20592
```

These clashes are the principal justification for not using the signature to determine the part.

That's also one reason why AVRDUDE says it's *probably* part so and so. Even if you were to accept the risk of going ahead with what AVRDUDE augurs the part to be based on the signature, AVRDUDE won't know how to read the signature memory unless it's given a part (different parts require different methods).

There is one exception to this: The `-c urclock` programmer when it uploads via an [urboot bootloader](https://github.com/stefanrueger/urboot) is told through the [urprotocol](https://github.com/stefanrueger/urboot/blob/main/urprotocol.md) for which part it was *compiled*. That's not a guarantee that the bootloader actually sits on such a part, but as all that's matters is the bootloading protocol that is OK. So, `avrdude -c urclock` is so far the only programmer for which there is no need to specify the part with `-p` *provided* it talks to an urboot bootloader.

It is a reasonable expectation that a user knows which part it wants to program. Although I have used the method you describe "at home" (where I know I only have five different parts that can be told apart by the signature), I would never encourage that method in the wild.

MCUdude, 2023-07-29:
> Take for example ATxmega64B3 and ATmega4809; they have the same signature but could hardly be more different: different families, different programming interfaces, different flash sizes, different flash page sizes, different EEPROM sizes, and different EEPROM page sizes:

Well, since these have a completely different programming interface, it wouldn't be possible for Avrdude to mix these up as the programmer (*_pdi or *_updi) is already present. The "mild" signature clashed (ATmega329, ATmega329A) doesn't really matter, as there are no differences between them in terms of programming parameters.

How about a target auto-detect flag, `-a` for instance? Then the user is taking the "risk" by explicitly specifying that he/she wants target auto-detect.

One issue I see with programmers that do several different interfaces (AVRISPmkII for instance, does ISP, TPI, and PDI) is that Avrdude doesn't know which (of the three) interfaces to use to talk to the unknown target. And switching between these programming modes automatically is something that's not currently built into Avrdude. At the moment Avrdude reads the parts information and determine which interface to use afterwards.

stefanrueger, 2023-07-31:
> Avrdude doesn't know which (of the three) interfaces to use to talk to the unknown target

And that's the crux of it. Other programmers might come along in the future offering more than these three interfaces of your example. One needs a seed part.

> fish out the target name, which is inherently unreliable, as the output format can change on the avrdude side

Correct. Error messages and the like may change without notice.

However, there is functionality in the more recent versions of AVRDUDE where you don't have to rely on the "(probably *part*)" output. The following outputs a list of all known parts with their signatures in the format of `-U signature:r:-:h`:
```
$ avrdude -p "*/St" | grep "^.pt[^m].*signature" | sed -e "s/0x0*/0x/g" -e "s/ /,/g"
```
And this line give you the signature of the attached *unknown* part given a seed part, say, m8:
```
$ avrdude -c usbtiny -p m8 -Fqqq -U sig:r:-:h
0x1e,0x95,0xf
``` 

Combining both in a bash one-liner gives you all candidates that share the same signature (note the dollar-sign at the end of the grep expression signifying end-of-line, ie, end-of-signature here)
```
$  avrdude -p "*/St" | grep "^.pt[^m].*signature" | sed -e "s/0x0*/0x/g" -e "s/ /,/g" | grep $(avrdude -c usbtiny -p m8 -Fqqq -U sig:r:-:h)$ | cut -f2
ATmega328P
ATA6614Q
LGT8F328P
```

It is then up to the user to make something out of this. How is AVRDUDE supposed to know which part might actually be connected? Depending on the circumstances it may, or may not be justified to assume it doesn't matter for the programming which one to choose. I don't think AVRDUDE can help in that.

BTW, if you use a method like this you may need to make the one-liner robust against connection errors etc.

mcuee, 2023-07-31:
@stefanrueger 

One thing that the current avrdude codes do is to print all the chip varints (including the sub-variants with regard to packaging) when the user specify the wrong part and `-v`, the output can be very long.

I am thinking we should only print out the main generic name with `-v` and only the full list with `-vv`. Just wondering how you like this idea.

stefanrueger, 2023-08-01:
> print all the chip variants [...] when the user specifies the wrong part and -v

It's documented that way, and there should be a very good reason for changing documented behaviour. A user who had relied on that behaviour, say in a shell script, would then need to rewrite that script through no fault of their own.

mcuee, 2023-08-01:
> It's documented that way, and there should be a very good reason for changing documented behaviour. A user who had relied on that behaviour, say in a shell script, would then need to rewrite that [...]

You can see the difference between v7.2 release and v7.1 release.

Actually the new output may break the shell script if they follow the old behavior.
[compare.txt](https://github.com/avrdudes/avrdude/files/12223494/compare.txt)

stefanrueger, 2023-08-01:
Sure, variants were introduced (and documented) after v7.1. Yes, possibly some scripts needed adapting following the introduction of variants. I would expect users in these cases to be accommodating because this was caused by an increase in functionality that justifies the change AVRDUDE's output.

If you or others feel strongly about printing variants with -vv only that can be done and documented, of course. The question I have is what is the increased functionality or other good reason for making the change in behaviour?

Generally, I prefer instilling a sense of stability, ie, users not having to worry about how the AVRDUDE output will look next week. I realise we carried out a major review of how messaging works for v7.1 (systematic error, warning and info display including an improved way of message quelling; systematic punctuation and spelling). That was a lot of work for you @mcuee, @MCUdude  and myself, but worth it in terms of look and feel, gradual quelling, source file and line number for debugging, reliable indication of the severity of a problem etc.  So, again, my hope is that users who needed to change their scripts will have understood the benefits of the changes.

stefanrueger, 2023-08-01:
Coming back to the topic of this issue I tend to think there is little AVRDUDE can do:
  - AVRDUDE already can figure out the set of possible connected parts given a seed part (albeit with a lengthy bash line)
  - The signature is not remotely unique (61 non-unique signatures are used by 132 parts known to AVRDUDE)
  - The set of parts with the same signature *can* differ in all sorts of parameters that are relevant for programming
 
I tend to think this issue can be closed.

mcuee, 2023-08-01:
> I tend to think this issue can be closed.

I agree.

mcuee, 2023-08-01:
> Sure, variants were introduced (and documented) after v7.1. Yes, possibly some scripts needed adapting following the introduction of variants. I would expect users in these cases to be accommodating [...]

The variants improvement is for sure a good thing.

As for the side issue I mentioned, I do not feel so strong about it. Let's see if there are users who are not happy with the long outputs with `-v`. 

Unfortunately we do not get many feedbacks yet for avrdude 7.1, let alone avrdude 7.2. This may have something to do with the fact that Arduino is still bundling avrdude 6.3.

Reference: good thing is that more and more distros are using avrdue 7.1 and some of them even start to bundle avrdude 7.2.
https://github.com/avrdudes/avrdude/discussions/1477#discussioncomment-6585202

stefanrueger, 2023-08-04:
AVRDUDE cannot guess the part from the signature as it is not unique and the set of parts with the same signature can differ in all sorts of parameters that are relevant for programming.

closed 2023-08-05 by MCUdude