avrdudes/avrdude #1483: Using linuxspi on raspberry pi leaves reset pin set low

merlinblack, 2023-08-07:
This is a minor issue as after running avrdude you can 'reset' the reset with for example:

`gpioset $(gpiofind GPIO25)=1`

I've looked into this - it's not a bug in avrdude really. Here's what I found:

Prior to some changes in the raspberry pi gpio driver, the code as it is would have worked and I guess might be still working on other hardware. (Anyone with an OrangePi able to comment?)

However the raspberry driver has been updated meaning that changes to GPIO via libgpiod are now persistent, and no longer are 'reset' to the default when the device file is closed.

Here's the change in the driver:
https://github.com/raspberrypi/linux/commit/022689f0973d87956b2e5e8aaa0c29803cdb2a71

And a good explaination of how it affects things:
https://raspberrypi.stackexchange.com/a/143016/132519

I'm thinking an option to set the reset line high again explicitly for linuxspi would be useful. What are peoples thoughts?

Unless someone happens to be looking at this already - or at the same spot in the code, I can take a look at making a patch myself in the next week or so.

MCUdude, 2023-08-07:
I think Avrdude should drive the reset pin high by default, so that linuxgpio can be used with Raspberry Pi without having to think about how the reset pin bahaves. Maybe we could utilize -E (exitspec, see the user manual) to either let the driver do what it wants with the pin when finished programming, or forcing it low?

merlinblack, 2023-08-07:
Yes, I was thinking this too after opening the issue. 

We don't need to make this optional. Just make the pin high once done with talking to the chip, before closing the device file. The driver will do what the driver does - possibly reset the pin to INPUT and make it high or leave it alone. It does not matter at that point.

merlinblack, 2023-08-07:
I was just poking around in the code and noticed in `linuxspi_close` (linuxspi.c:267) it does set the reset line high again if you tell it to with '-Enoreset' on the command line.

I thought this would be the default?

stefanrueger, 2023-08-09:
As avrdude can leave the reset line in a defined state with the`-E` option all should be OK, not? Is there something the documentation could clarify? I am usually hesitant to change the behaviour of AVRDUDE unless it is to address a clear-cut bug,

mcuee, 2023-08-11:
So I think this issue can be closed, any objections?

MCUdude, 2023-08-11:
> So this issue can be closed, any objections?

We can. @merlinblack could use `-Ereset` or `-Enoreset`. It it mention in the documentation that `linuxspi` supports this, but is isn't very obvious if you're not looking for it.

MCUdude, 2023-08-11:
Maybe we could add `-E help` (similar to `-x help)` so that users can see what kind of -E options that's available for a particular programmer?

merlinblack, 2023-08-12:
Using -Enoreset is a good workaround.

However I think without any -E option, the reset line should be returned to how it was before hand. Looking at a comment in the code, this is the intention.

merlinblack, 2023-08-12:
I had a bit of an experiment session, but hit a road block.

I tried out this code (copied and modified from an example), just after the code that opens the gpio character device.
```
/** Get state of the reset pin, so we can put it back later */
    struct gpioline_info line_info;
    line_info.line_offset = pgm->pinno[PIN_AVR_RESET] & PIN_MASK;
    ret = ioctl(fd_gpiochip, GPIO_GET_LINEINFO_IOCTL, &line_info);
    if (ret == -1)
    {
        printf("Unable to get line info from offset %d: %s", line_info.line_offset, strerror(errno));
    }
    else
    {
        printf("offset: %d, name: %s, consumer: %s. Flags:\t[%s]\t[%s]\t[%s]\t[%s]\t[%s]\n",
               line_info.line_offset,
               line_info.name,
               line_info.consumer,
               (line_info.flags & GPIOLINE_FLAG_IS_OUT) ? "OUTPUT" : "INPUT",
               (line_info.flags & GPIOLINE_FLAG_ACTIVE_LOW) ? "ACTIVE_LOW" : "ACTIVE_HIGH",
               (line_info.flags & GPIOLINE_FLAG_OPEN_DRAIN) ? "OPEN_DRAIN" : "...",
               (line_info.flags & GPIOLINE_FLAG_OPEN_SOURCE) ? "OPENSOURCE" : "...",
               (line_info.flags & GPIOLINE_FLAG_KERNEL) ? "KERNEL" : "");
    }
```
It gets the INPUT vs OUTPUT correct, however always reports the line as ACTIVE_HIGH.

It seems that the 'gpioinfo' utility does the same thing. If the line is set low - by using avrdude or gpioset - it is always reported as high. I a LED attached to pin 25, so I can physically see when the line is high or low.

Looks like using the -E flag is the way to go.

mcuee, 2023-08-12:
In this case, I think we can close this issue.

mcuee, 2023-08-12:
> Maybe we could add `-E help` (similar to `-x help)` so that users can see what kind of -E options that's available for a particular programmer?

@MCUdude 

I think this is a good idea.

merlinblack, 2023-08-13:
TLDR; Use '-Enoreset' on the command line to return the reset line to HIGH.

closed 2023-08-13 by merlinblack

stefanrueger, 2023-08-13:
> Maybe we could add -E help (similar to -x help) so that users can see what kind of -E options that's available 

@MCUdude Great idea!