avrdudes/avrdude #1490: Support ATmega8/16/32/64/128(A) using FTDI JTAG -- remaining work from #1324

mcuee, 2023-08-15:
#1324 has been merged. It is a great addition to avrdude.
* #1324

It will be good that the issues with older ATmega8/16/32/64/128(A) parts can be fixed in the future.

mcuee, 2023-08-15:
Yes, they are well supported by the other JTAG capable programmers, like JTAG1/2/3. I myself have tested them with JTAG ICE 1, AVR Dragon, SNAP and PICKit 4.

dl8dtl, 2023-08-16:
ATmega8 is not an ATmega in that respect: it doesn't have JTAG. Technically, it's a tiny device. Later variants (ATmega88 etc.) offer debugWIRE.

mcuee, 2023-08-16:
> ATmega8 is not an ATmega in that respect: it doesn't have JTAG. Technically, it's a tiny device. Later variants (ATmega88 etc.) offer debugWIRE.

Good Point. I have updated the title and first post to remove ATmega8/8A.

mcuee, 2024-03-03:
Close this one as not planned until someone is actibely working on this issue.

This issue is also linked here.
https://github.com/avrdudes/avrdude/discussions/1710

closed 2024-03-03 by mcuee