avrdudes/avrdude #1493: Atmega8 signature read command is incorrect

nejohnson, 2023-08-25:
According to Atmel/Microchip's Atmega8 documentation the command to read the signature byte is:
0011.0000--00xx.xxxx--xxxx.xxaa--oooo.oooo
whereas in AvrDude is it currently
0011.0000--xxxx.xxxx--xxxx.xxaa--oooo.oooo

The error is in the top two bits of the second byte: they should be 00 not xx.

stefanrueger, 2023-08-25:
Thanks for noting and reporting. This is true but luckily inconsequential as AVRDUDE initialises the `x` bits actually with `0`. I'll create a PR, nevertheless.

stefanrueger, 2023-08-25:
I've looked at 10 randomly selected parts and found a couple of further discrepancies of the same kind (`x` vs `0`) between the `avrdude.conf` file and the data sheet. Neither matters for programming as `x` are replaced by `0` when creating the actual command. This might be owing to changes in the data sheets or through inheritance errors. I have fixed these, but not gone to the trouble of reviewing all 186 parts (too much trouble for no actual gain).

I noted, however, that some calibration commands are wrongly set as read signature commands:

```
$ avrdude -p*/St | grep ptmmop | grep 0011\.0000 | grep -v signat
.ptmmop	ATmega64M1	calibration	read	0011.0000--000x.xxxx--xxxx.xxaa--oooo.oooo
.ptmmop	ATmega32M1	calibration	read	0011.0000--000x.xxxx--xxxx.xxaa--oooo.oooo
.ptmmop	ATmega16M1	calibration	read	0011.0000--000x.xxxx--xxxx.xxaa--oooo.oooo
.ptmmop	ATmega32C1	calibration	read	0011.0000--000x.xxxx--xxxx.xxaa--oooo.oooo
.ptmmop	ATmega64C1	calibration	read	0011.0000--000x.xxxx--xxxx.xxaa--oooo.oooo
.ptmmop	ATA5505	calibration	read	0011.0000--000x.xxxx--xxxx.xxaa--oooo.oooo
.ptmmop	ATA6612C	calibration	read	0011.0000--000x.xxxx--xxxx.xxaa--oooo.oooo
.ptmmop	ATA6613C	calibration	read	0011.0000--000x.xxxx--xxxx.xxaa--oooo.oooo
.ptmmop	ATA6614Q	calibration	read	0011.0000--000x.xxxx--xxxx.xxaa--oooo.oooo
.ptmmop	ATA6616C	calibration	read	0011.0000--000x.xxxx--xxxx.xxaa--oooo.oooo
.ptmmop	ATA6617C	calibration	read	0011.0000--000x.xxxx--xxxx.xxaa--oooo.oooo
.ptmmop	ATA664251	calibration	read	0011.0000--000x.xxxx--xxxx.xxaa--oooo.oooo
.ptmmop	ATmega16HVA	calibration	read	0011.0000--000x.xxxx--xxxx.xxaa--oooo.oooo
.ptmmop	ATmega8HVA	calibration	read	0011.0000--000x.xxxx--xxxx.xxaa--oooo.oooo
.ptmmop	ATmega16HVB	calibration	read	0011.0000--000x.xxxx--xxxx.xxaa--oooo.oooo
.ptmmop	ATmega16HVBrevB	calibration	read	0011.0000--000x.xxxx--xxxx.xxaa--oooo.oooo
.ptmmop	ATmega32HVB	calibration	read	0011.0000--000x.xxxx--xxxx.xxaa--oooo.oooo
.ptmmop	ATmega32HVBrevB	calibration	read	0011.0000--000x.xxxx--xxxx.xxaa--oooo.oooo
.ptmmop	ATmega64HVE2	calibration	read	0011.0000--000x.xxxx--xxxx.xxaa--oooo.oooo
.ptmmop	ATmega32HVE2	calibration	read	0011.0000--000x.xxxx--xxxx.xxaa--oooo.oooo
.ptmmop	AT90PWM1	calibration	read	0011.0000--000x.xxxx--xxxx.xxaa--oooo.oooo
.ptmmop	AT90PWM161	calibration	read	0011.0000--000x.xxxx--xxxx.xxaa--oooo.oooo
.ptmmop	AT90PWM81	calibration	read	0011.0000--000x.xxxx--xxxx.xxaa--oooo.oooo
```
I also noted that some calibration read commands have an address bit but the memory has size 1.

Hence, a more substantive PR that reviews the calibration commands.

mcuee, 2023-08-26:
> I've looked at 10 randomly selected parts and found a couple of further discrepancies of the same kind (x vs 0) between the avrdude.conf file and the data sheet. Neither matters for programming as x [...]

I agree since it does not really matter as you mentioned that  `x` are replaced by `0` when creating the actual command.

nejohnson, 2023-08-26:
As the quote goes: "luck is the residue of design".  It may be that in _the current version_ avrdude replaces `x` with `0`, but that's not a guarantee.

As a user I see `x` as "don't care" so it could just as easily be a `1`, or a random choice.  Perhaps some future avrdude programmer decides to truly let `x` be uninitialized to whatever random rubbish was in memory at the time.  Still consistent with the spec (because "don't care" means we don't care, right?), but now some devices will exhibit weird random programming failures.

In fact, that would be an interesting fuzzing test: insert random `0` and `1` into the `x` bits and test avrdude still successfully programs devices.

nejohnson, 2023-08-26:
In the case of the Atmega8, I'm looking at version 10/04 of the datasheet (which I printed out 20 years ago) and the most recent version on Microchip's website, and they are both consistent with regard to the signature read command.

My last avrdude bug report was http://savannah.nongnu.org/bugs/?3835. It seems I missed the signature read command.

stefanrueger, 2023-08-26:
@nejohnson You are 100% correct in your analysis. Issue #942 recognised the potential problem of `x` bits being left truly uninitialised: debugging wrongly set `x` bits in a command might be hard owing to the random nature that one would expect. PR #943 makes AVRDUDE deterministic in this respect by design. You are also right to suggest that setting `x` randomly could be a good test to see whether the `x` bit in `avrdude.conf` is actually *don't care*.  And yes, if a future PR reverted setting `x` randomly it would become necessary to check the data sheets against `avrdude.conf` at that point.

> My last avrdude bug report was http://savannah.nongnu.org/bugs/?3835. 

Welcome back! The project benefits from these contributions

closed 2023-08-27 by stefanrueger

nejohnson, 2023-08-28:
Thanks, happy to help such an awesome tool!

PR #1495 (2023-08-27): Review calibration read command in avrdude.conf
Fixes #1493