avrdudes/avrdude #1496: windows amd64 binary in "releases" doesn't include USB support

WestfW, 2023-08-28:
The windows binary in the releases area does not seem to have been compiled with USB support.
I'm not sure whether that's intentional, or an oversight.
```
"c:\Program Files\avrdude\avrdude.exe" -cch341a
avrdude error: no usb support, please compile again with libusb installed
avrdude error: unable to open programmer ch341a on port usb
```

Hmm.  I get different errors with other USB programmers, so this could be more complex than I thought?  Maybe it's just a bogus error message, since I have neither ch341a or arduinoisp actually connected...

```
"c:\Program Files\avrdude\avrdude.exe" -carduinoisp
avrdude error: cannot find USBtiny device (0x2341/0x49)
avrdude error: unable to open programmer arduinoisp on port usb
```

mcuee, 2023-08-28:
For CH341 it is a known issue that MSVC build does not work.
* https://github.com/avrdudes/avrdude/discussions/1440

mcuee, 2023-08-28:
As for USBTiny, it should work. What is the driver you are using?

If the MSVC build does not work, please try my MinGW64 build here.
https://github.com/mcuee/avrdude/releases/tag/v7.2 (avrdude-v7.2_mingw64.zip)(https://github.com/mcuee/avrdude/releases/download/v7.2/avrdude-v7.2_mingw64.zip)

Is your device USB Composite Device? MSVC build has a known problem with USB Composite Device.
* https://github.com/avrdudes/avrdude/issues/968

mcuee, 2023-09-04:
@WestfW 

Just wondering if you have any updates? Thanks.

WestfW, 2023-09-04:
I was just curious, and following up on another user's question.
I don't actually have any hardware that would enable me to provide useful feedback :-(

mcuee, 2023-09-04:
I see. I will close this issue for now. Please re-open if you have more info. Thanks.

closed 2023-09-04 by mcuee