avrdudes/avrdude #1499: Cannot program fuses/flash memory/erase with avrdude on Ubuntu 22.04 - bug?

Nefarious19, 2023-09-05:
Hi, I'm working on small project featuring tiny13. I'm using avrisp mk 2 clone with latest LUFA fw. I works fine on windows with libusbK and with avrdude and with atmel studio 7 (Microchip studio actually). However, I have problem under linux. I working on Ubuntu 22.04 with latest updates. 

`avrdude -Pusb -cavrisp2 -pt13 -U flash:w:ATTinyController.hex`

`avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9007 (probably t13)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip

avrdude: processing -U flash:w:ATTinyController.hex:i
avrdude: reading input file ATTinyController.hex for flash
         with 374 bytes in 1 section within [0, 0x175]
         using 12 pages and 10 pad bytes
avrdude: writing 374 bytes flash ...
Writing | -------------------------------------------------- | 0% 0.02 s 
avrdude error: wrote 0 out of 1 bytes, err = could not detach kernel driver from interface 0: No data available
avrdude error: unable to send command to serial port
avrdude error: wrote 0 out of 1 bytes, err = could not detach kernel driver from interface 0: No data available
avrdude error: unable to send command to serial port
avrdude error: wrote 0 out of 1 bytes, err = could not detach kernel driver from interface 0: No data available
avrdude error: unable to send command to serial port
avrdude error: wrote 0 out of 1 bytes, err = could not detach kernel driver from interface 0: No data available
avrdude error: unable to send command to serial port
avrdude error: wrote 0 out of 1 bytes, err = could not detach kernel driver from interface 0: No data available
avrdude error: unable to send command to serial port
avrdude error: wrote 0 out of 1 bytes, err = could not detach kernel driver from interface 0: No data available
avrdude error: unable to send command to serial port
avrdude error: unable to communicate with programmer (0)
avrdude error: wrote 0 out of 5 bytes, err = could not detach kernel driver from interface 0: No data available
avrdude error: unable to send command to serial port
avrdude error: wrote 0 out of 1 bytes, err = could not detach kernel driver from interface 0: No data available
avrdude error: unable to send command to serial port
avrdude error: wrote 0 out of 1 bytes, err = could not detach kernel driver from interface 0: No data available
avrdude error: unable to send command to serial port
avrdude error: wrote 0 out of 1 bytes, err = could not detach kernel driver from interface 0: No data available
avrdude error: unable to send command to serial port
avrdude error: wrote 0 out of 1 bytes, err = could not detach kernel driver from interface 0: No data available
avrdude error: unable to send command to serial port
avrdude error: wrote 0 out of 1 bytes, err = could not detach kernel driver from interface 0: No data available
avrdude error: unable to send command to serial port
avrdude error: wrote 0 out of 1 bytes, err = could not detach kernel driver from interface 0: No data available
avrdude error: unable to send command to serial port
avrdude error: unable to communicate with programmer (0)
avrdude error: wrote 0 out of 42 bytes, err = could not detach kernel driver from interface 0: No data available
avrdude error: unable to send command to serial port
avrdude error: wrote 0 out of 1 bytes, err = could not detach kernel driver from interface 0: No data available
avrdude error: unable to send command to serial port
avrdude error: wrote 0 out of 1 bytes, err = could not detach kernel driver from interface 0: No data available
avrdude error: unable to send command to serial port
avrdude error: wrote 0 out of 1 bytes, err = could not detach kernel driver from interface 0: No data available
avrdude error: unable to send command to serial port
avrdude error: wrote 0 out of 1 bytes, err = could not detach kernel driver from interface 0: No data available
avrdude error: unable to send command to serial port
avrdude error: wrote 0 out of 1 bytes, err = could not detach kernel driver from interface 0: No data available
avrdude error: unable to send command to serial port
avrdude error: wrote 0 out of 1 bytes, err = could not detach kernel driver from interface 0: No data available
avrdude error: unable to send command to serial port
avrdude error: unable to communicate with programmer (0)
avrdude error: wrote 0 out of 5 bytes, err = could not detach kernel driver from interface 0: No data available
avrdude error: unable to send command to serial port
avrdude error: wrote 0 out of 1 bytes, err = could not detach kernel driver from interface 0: No data available
avrdude error: unable to send command to serial port
avrdude error: wrote 0 out of 1 bytes, err = could not detach kernel driver from interface 0: No data available
avrdude error: unable to send command to serial port
avrdude error: wrote 0 out of 1 bytes, err = could not detach kernel driver from interface 0: No data available
avrdude error: unable to send command to serial port
avrdude error: wrote 0 out of 1 bytes, err = could not detach kernel driver from interface 0: No data available
avrdude error: unable to send command to serial port
avrdude error: wrote 0 out of 1 bytes, err = could not detach kernel driver from interface 0: No data available
[... 214 lines cut ...]
avrdude error: unable to send command to serial port
avrdude error: wrote 0 out of 1 bytes, err = could not detach kernel driver from interface 0: No data available
avrdude error: unable to send command to serial port
avrdude error: wrote 0 out of 1 bytes, err = could not detach kernel driver from interface 0: No data available
avrdude error: unable to send command to serial port
avrdude error: wrote 0 out of 1 bytes, err = could not detach kernel driver from interface 0: No data available
avrdude error: unable to send command to serial port
avrdude error: wrote 0 out of 1 bytes, err = could not detach kernel driver from interface 0: No data available
avrdude error: unable to send command to serial port
avrdude error: unable to communicate with programmer (0)
avrdude: 374 bytes of flash written
avrdude: verifying flash memory against ATTinyController.hex
Reading | -------------------------------------------------- | 0% 0.00 s 
avrdude error: wrote 0 out of 5 bytes, err = could not detach kernel driver from interface 0: No data available
avrdude error: unable to send command to serial port
avrdude error: wrote 0 out of 1 bytes, err = could not detach kernel driver from interface 0: No data available
avrdude error: unable to send command to serial port
avrdude warning: verification mismatch
        device 0xff != input 0x09 at addr 0x0000 (error)
avrdude error: verification mismatch

avrdude done.  Thank you.`

The same hex is uploading fine with USBasp. I tried with sudo, no change, I checked udev rules, everything looks ok. I changed usb cable, the same. But the same setup works on windows so it must be some linux driver/libusb issue.
Can someone test it and confirm?

mcuee, 2023-09-06:
@Nefarious19 

Ubuntu may come with very old version of avrdude. Please use the 7.2 release.

You can build from the 7.2 release source code.
https://github.com/avrdudes/avrdude/releases/tag/v7.2

Or if you are not willing to build from source, you can download the binary here.
https://github.com/mcuee/avrdude/releases/tag/v7.2

MCUdude, 2023-09-06:
Isn't there a bug in the LUFA firmware on the AVRISP mkII clones that causes issues like this?

https://github.com/avrdudes/avrdude/issues/321#issuecomment-1697772510

MCUdude, 2023-09-11:
@Nefarious19, I have no idea how to fix the LUFA firmware. Post the Avrdude output over at the [LUFA Github repo](https://github.com/abcminiuser/lufa/issues), and see what the developers over these suggest.

mcuee, 2023-09-12:
@Nefarious19 

Try to manually detach the kernel driver to see if that helps.
https://lwn.net/Articles/143397/

stefanrueger, 2023-09-24:
@Nefarious19 How is this going? Have you been able to solve your problem? Is there anything the team can do? Otherwise I'd suggest to close the issue.

mcuee, 2023-09-27:
@Nefarious19 
I will close this issue for now. Please re-open if you can demonstrate that it is an avrdude issue.

closed 2023-09-27 by mcuee