avrdudes/avrdude #1502: Progress report in scripting

DRSDavidSoft, 2023-09-22:
Hi there @mcuee and @stefanrueger, sorry to bother you, I was wondering if I could take your time for a quick moment.

I'm browsing the code for the progress report functionality of `avrdude`, in order to understand it better, so I could extract its value from the terminal and use it in my script.

I had two questions about this:
1. It seems that the progress report will only be initialized if the `verbose` switch is supplied.  Assuming that this is correct, can something please be added to make the progress bar be displayed when verbose mode is not requested, in order to avoid displaying the avr part information on each invocation of the executable?
2. It also appears that avrdude makes useof `iatty()` function to check if the `STDERR` is being redirected to a file or a terminal.  I'm having some difficulty in regards to making this return true under Windows where my script is running.  Could you please be open to accepting a new switch such as `--tty` or `-no-tty` that would force one behavior instead of checking the `iatty()` return value?

I'm terribly sorry for bothering you for this, but I couldn't find anything for my use case here.  When the `no_ttty` mode of the progress report is used, I only get the `#` character for each step of the progress and the final `100%` value.  Instead, I'd rather to get the full `\r <hdr> .... <value> %` line each time so I can extract the value.

Fortunately, it appears that the hashes are always 50 characters in length, so for now I can count those as well, but I'd hugely appreciate it if you could please be open in implementing these couple of new switches.

Thank you!

dl8dtl, 2023-09-22:
Please, don't wrap scripts around AVRDUDE. It is meant as a human interface, not as a machine-readable API.
Exactly for your purpose, we segregated the actual backend in a `libavrdude` library years ago. There, you can simply provide your own callback function for progress reporting. If you don't want to use C but e.g. Python, it should be possible to create a Python wrapper using something like SWIG. Might be a bit of work at the beginning, but in the end, you'll get a much more seamless integration.

DRSDavidSoft, 2023-09-22:
@dl8dtl Thank you, I was especially wondering if an API like this existed! I'm pretty sure the C calls can be consumed using ffi in PHP (which is what I'm using at the moment).

Great at what providing a library for this is, I still would appreciate it a ton if you could consider the initial request.  The reasoning is that `avrdude` is called like how Arduino IDE uses it, i.e. the raw output is shown to the remote user (also like how AVRDUDESS does this locally).  I kind of want to implement a visual progress bar for when the firmware is being deployed to the flash memory of devices that take some time to finish, such as the ATmega2560. I'd prefer to display any generated errors directly to the user, at least until I switch to `libavrdude` as it is appropriate here.

There is a small utility called XLoader which is a minimal implementation of how AVRDUDESS and Arduino IDE work, but this tool doesn't provide a "terminal" to display the actual output of `avrdude` to the user. In such cases I believe the only correct approach is to use `libavrdude`.

In any case, I appreciate your help and guidance and I would understand if the switches that I'm asking wouldn't be implemented in `avrdude`, even though it would be a bit sad.

stefanrueger, 2023-09-23:
> I'm having some difficulty in regards to making [`isatty()`] return true under Windows where my script is running.

The only reason I can think of is that stderr is not output on a terminal when the Windows script is running,

I agree with @dl8dtl it's best to use libavrdude to create one's own API. The simplest way for this is to create a bespoke `main.c` for own applications. Note that it's not only progress reporting that can be substituted but also `main.c`'s core reporting function `avrdude_message2()`.

AVRDUDE doesn't have many options, and there should be a *very good* reason for introducing new options.

dl8dtl, 2023-09-23:
> AVRDUDE doesn't have many options

This should probably read "has already too many options". ;-)

DRSDavidSoft, 2023-09-24:
Fair enough, thanks for the help! I'll go with the `libavrdude` path then!

closed 2023-09-24 by stefanrueger