avrdudes/avrdude #1503: 7e94ed4442b breaks flashing GT2560 boards

diresi, 2023-09-25:
Hi,

I have a 3d printer running a GT2560 [1] board, which is essentially an ATmega2560 with a CH340C as USB-to-Serial converter.
I found that flashing works with avrdude 6.3 but not with 7.2.

It took me a while to trace that to DTR handling that was changed in https://github.com/avrdudes/avrdude/pull/1309.
Flashing works again when reverting commit 7e94ed4442b in wiring.c.  

[1]
https://www.geeetech.com/wiki/images/d/d3/Hardware_GT2560_RevA%2B.pdf

stefanrueger, 2023-09-25:
@diresi Yes, C9 is a bit high with 1 µF, so the 50 ms dis/charge time in the `usleep()` calls might be a tad bit low for this board?  Could you play with either value in lines 175 and 183 to see if you can get the board to reset and synchronise?

mcuee, 2023-09-26:
Another way is to change C9 to something like 100nF, and then see if avrdude 7.2 version works or not.

diresi, 2023-09-26:
That was solid advice :)
It works with 80ms discharge time, I created a pull request for that w/o reading any contribution guidelines ... please feel free to reject.

I was playing with the idea of replacing C9 already (since I broke the original SMD one when attaching probes) but then again, maybe someone else is also using that board, right?

stefanrueger, 2023-09-26:
OK, further tests have demonstrated that the board design and too high out-of-specs Vcc led to the board being put into HVPP mode. Tinkering with `usleep()` has ostensibly solved the symptoms but the usleep() times were not wrong per se.

closed 2023-09-28 by stefanrueger