avrdudes/avrdude #1514: pgm_led not activated for flash write or eeprom write - only for chip erase

Ho-Ro, 2023-10-12:
During preparing a patch for my ScratchMonkey programmer (based on Arduino Nano) that can handle its 4 status LEDs `rdy_led`, `err_led`, `pgm_led` and `vfy_led` I detected that the `pgm_led` is only activated during chip erase, but not during the flash or eeprom write. This is caused by missing calls to `pgm->pgm_led()` from `avrdude`.

My working ScratchMonkey implementation is located in my branch [ScrMon](https://github.com/Ho-Ro/avrdude/tree/ScrMon).

**How to detect this issue**
For debugging the pgm_led behaviour I've created the branch [pgm_led](https://github.com/Ho-Ro/avrdude/tree/pgm_led) that just prints the led status changes to `stderr`.

**Flash programming**
```sh
$ avrdude_Ho-Ro -pt85 -U flash:w:t85_flash.hex -q -c scratchmonkey
LED RDY ON
avrdude_Ho-Ro: AVR device initialized and ready to accept instructions
avrdude_Ho-Ro: device signature = 0x1e930b (probably t85)
avrdude_Ho-Ro: Note: flash memory has been specified, an erase cycle will be performed.
               To disable this feature, specify the -D option.
avrdude_Ho-Ro: erasing chip
LED PGM ON
LED PGM OFF

avrdude_Ho-Ro: processing -U flash:w:t85_flash.hex:i
avrdude_Ho-Ro: reading input file t85_flash.hex for flash
               with 8168 bytes in 1 section within [0, 0x1fe7]
               using 128 pages and 24 pad bytes
avrdude_Ho-Ro: writing 8168 bytes flash ...
avrdude_Ho-Ro: 8168 bytes of flash written
LED VFY ON
avrdude_Ho-Ro: verifying flash memory against t85_flash.hex
avrdude_Ho-Ro: 8168 bytes of flash verified
LED VFY OFF
LED RDY OFF

avrdude_Ho-Ro done.  Thank you.
```

**EEPROM programming**
```sh
$ avrdude_Ho-Ro -pt85 -U eeprom:w:t85_eeprom.hex -q -c scratchmonkey
LED RDY ON
avrdude_Ho-Ro: AVR device initialized and ready to accept instructions
avrdude_Ho-Ro: device signature = 0x1e930b (probably t85)

avrdude_Ho-Ro: processing -U eeprom:w:t85_eeprom.hex:i
avrdude_Ho-Ro: reading input file t85_eeprom.hex for eeprom
               with 512 bytes in 1 section within [0, 0x1ff]
               using 128 pages and 0 pad bytes
avrdude_Ho-Ro: writing 512 bytes eeprom ...
avrdude_Ho-Ro: 512 bytes of eeprom written
LED VFY ON
avrdude_Ho-Ro: verifying eeprom memory against t85_eeprom.hex
avrdude_Ho-Ro: 512 bytes of eeprom verified
LED VFY OFF
LED RDY OFF

avrdude_Ho-Ro done.  Thank you.
```

But even when using a programmer that doesn't provide status LEDs you can see that calls to activate the `pgm_led` are missing. I implemented own `pgm_default_XXX_led()` functions in `pgm.c` that are called as default when the programmer doesn't provide its own `XXX_led()` function. This default function simply writes a status message to `stderr`, as seen in my example below:

**Flash programming**
```sh
$ avrdude_Ho-Ro -pt85 -U flash:w:t85_flash.hex:i -q -c stk500v2
pgm_default_rdy_led( 0 )
pgm_default_err_led( 0 )
pgm_default_pgm_led( 0 )
pgm_default_vfy_led( 0 )
pgm_default_rdy_led( 1 )
avrdude_Ho-Ro: AVR device initialized and ready to accept instructions
avrdude_Ho-Ro: device signature = 0x1e930b (probably t85)
avrdude_Ho-Ro: Note: flash memory has been specified, an erase cycle will be performed.
               To disable this feature, specify the -D option.
avrdude_Ho-Ro: erasing chip
pgm_default_pgm_led( 1 )
pgm_default_pgm_led( 0 )

avrdude_Ho-Ro: processing -U flash:w:t85_flash.hex:i
avrdude_Ho-Ro: reading input file t85_flash.hex for flash
               with 8168 bytes in 1 section within [0, 0x1fe7]
               using 128 pages and 24 pad bytes
avrdude_Ho-Ro: writing 8168 bytes flash ...
pgm_default_err_led( 0 )
avrdude_Ho-Ro: 8168 bytes of flash written
pgm_default_vfy_led( 1 )
avrdude_Ho-Ro: verifying flash memory against t85_flash.hex
avrdude_Ho-Ro: 8168 bytes of flash verified
pgm_default_vfy_led( 0 )
pgm_default_rdy_led( 0 )

avrdude_Ho-Ro done.  Thank you.
```

If this `pgm_led` issue is fixed I would like to provide a PR for the full *"ScratchMonkey with LEDs"* support, not only the *"ScratchMonkey is just another name for STK500"* solution as it is currently.

mcuee, 2023-10-12:
@Ho-Ro 

Looks like an avrdude bug to me. Can you create a PR to address this?

MCUdude, 2023-10-12:
The Scratchmonkey programmer was discussed in #989 and applied in #1429.
LED support was deliberately left out.

mcuee, 2023-10-13:
> The Scratchmonkey programmer was discussed in #989 and applied in #1429. LED support was deliberately left out.

Ah, I remember now. I have removed the bug label.

Ho-Ro, 2023-10-13:
@mcuee I do not agree that it is not a bug. The general functionality of avrdude intends to operate the four LEDs, if the specific programmer is set up to do so and provides its own function call for it.  If not, the stub from `pgm.c` is used, which does nothing. The programmer of the specific module must be able to rely on all higher-level calls from the 'generic' avrdude being done correctly.
This quick hack below shows when LED actions would be triggered if the selected programmer model supports it. All calls during flash or EEPROM writing are missing, see the 3rd example call from my issue report above.

```c
diff --git a/src/pgm.c b/src/pgm.c
index 7863906..34f1d03 100644
--- a/src/pgm.c
+++ b/src/pgm.c
@@ -51,11 +51,53 @@ static void pgm_default_enable(PROGRAMMER *pgm, const AVRPART *p) {
   pgm_default();
 }
 
+
+/**************************************************************************
+ * Ho-Ro:
+ * Test setup for debugging strange PGM_LED behaviour of avrdude
+ * It switches the PGM_LED on during chip erase
+ * but not during flash write or eeprom write
+ * see https://github.com/avrdudes/avrdude/issues/1514
+ **************************************************************************/
+#define DEBUG_XXX_LED
+#ifdef DEBUG_XXX_LED
+static int pgm_default_rdy_led(const PROGRAMMER *pgm, int value) {
+  // Ho-Ro: "catch" all unset calls to rdy_led()
+  fprintf( stderr, "pgm_default_rdy_led( %d )\n", value );
+   // If programmer has no LEDs, just do nothing
+  return 0;
+}
+
+static int pgm_default_err_led(const PROGRAMMER *pgm, int value) {
+  // Ho-Ro: "catch" all unset calls to err_led()
+  fprintf( stderr, "pgm_default_err_led( %d )\n", value );
+   // If programmer has no LEDs, just do nothing
+  return 0;
+}
+
+static int pgm_default_pgm_led(const PROGRAMMER *pgm, int value) {
+  // Ho-Ro: "catch" all unset calls to pgm_led()
+  fprintf( stderr, "pgm_default_pgm_led( %d )\n", value );
+   // If programmer has no LEDs, just do nothing
+  return 0;
+}
+
+static int pgm_default_vfy_led(const PROGRAMMER *pgm, int value) {
+  // Ho-Ro: "catch" all unset calls to vfy_led()
+  fprintf( stderr, "pgm_default_vfy_led( %d )\n", value );
+   // If programmer has no LEDs, just do nothing
+  return 0;
+}
+
+#else
+
 static int pgm_default_led(const PROGRAMMER *pgm, int value) {
    // If programmer has no LEDs, just do nothing
   return 0;
 }
 
+#endif
+
 
 static void pgm_default_powerup_powerdown(const PROGRAMMER *pgm) {
    // If programmer does not support powerup/down, just do nothing
@@ -117,10 +159,17 @@ PROGRAMMER *pgm_new(void) {
    * implementation. Hence, they don't need to be defined in
    * the programmer.
    */
+#ifdef DEBUG_XXX_LED
+  pgm->rdy_led        = pgm_default_rdy_led;
+  pgm->err_led        = pgm_default_err_led;
+  pgm->pgm_led        = pgm_default_pgm_led;
+  pgm->vfy_led        = pgm_default_vfy_led;
+#else
   pgm->rdy_led        = pgm_default_led;
   pgm->err_led        = pgm_default_led;
   pgm->pgm_led        = pgm_default_led;
   pgm->vfy_led        = pgm_default_led;
+#endif
   pgm->read_byte_cached  = avr_read_byte_cached;
   pgm->write_byte_cached = avr_write_byte_cached;
   pgm->chip_erase_cached = avr_chip_erase_cached;
```

mcuee, 2023-10-13:
Okay, change the label back to "bug".

Ho-Ro, 2023-10-13:
#1517 does not close this issue (`pgm_led` is not activated during programming, i.e. write to flash or EEPROM - only during chip erase).

But @stefanrueger's PR adds perfect LED support for programmer type `scratchmonkey`. :+1:
 
I also encountered **another issue** with the `vfy_led`, that should be solved together with the `pgm_led` issue:
If the input file does not exist, the led is switched on before verifying, but it's not switched off at the end (see output using my test patch):

```
$ avrdude_Ho-Ro -pt85 -q -U flash:v:file_does_not_exist:r -cstk500v2
avrdude_Ho-Ro OS error: file file_does_not_exist is not readable: No such file or directory
pgm_default_rdy_led( 0 )
pgm_default_err_led( 0 )
pgm_default_pgm_led( 0 )
pgm_default_vfy_led( 0 )
pgm_default_rdy_led( 1 )
avrdude_Ho-Ro: AVR device initialized and ready to accept instructions
avrdude_Ho-Ro: device signature = 0x1e930b (probably t85)

avrdude_Ho-Ro: processing -U flash:v:file_does_not_exist:r
pgm_default_vfy_led( 1 )
avrdude_Ho-Ro: verifying flash memory against file_does_not_exist
avrdude_Ho-Ro OS error: cannot open input file file_does_not_exist: No such file or directory
avrdude_Ho-Ro error: read from file file_does_not_exist failed
pgm_default_rdy_led( 0 )

avrdude_Ho-Ro done.  Thank you.
```
Strange that avrdude detects that the file does not exist, but does not exit and then tries to verify, switching the LEDs on and off (and forgets the switch off the `vfy_led`). 

Obviously there are not so many users of SMo out in the wild that like to have *Blinkenlights*.

And then I also have to grumble and report errors. :)

Ho-Ro, 2023-10-13:
Attached a quick fix for the `vfy_led` issue, but keep the `err_led` untouched to show a possible error condition:

```diff
diff --git a/src/main.c b/src/main.c
index 5b5eb69..88599ea 100644
--- a/src/main.c
+++ b/src/main.c
@@ -1609,6 +1609,8 @@ main_exit:
     pgm->disable(pgm);

     pgm->rdy_led(pgm, OFF);
+    pgm->pgm_led(pgm, OFF);
+    pgm->vfy_led(pgm, OFF);

     pgm->close(pgm);
   }
```

stefanrueger, 2023-10-13:
TLDR; AVRDUDE currently operates a mixture of approaches to drive LEDs; these need reviewing and streamlining. I suggest to remove LED twiddling from callee individual programmer functions to the caller in the AVRDUDE logic for consistency and ease of implementation. Some other design questions below.

The long of it.

This is an interesting issue in that it is not documented how avrdude drives the four LEDs. At least a little grep wasn't able to find *anything of substance* in the 80-page pdf, web site or man page.

So, what *should* happen? There are 4 LEDs: rdy, pgm, vfy and err. Here my suggestion 
  - Rdy is relatively uncontroversial: set once the programmer is initialised and off before AVRDUDE exits (done)
  - Err when a fatal error happened, stays after exit; but what is a fatal error? Only those that are related to programming? 
  - Pgm? On during read, write and erase attempts? How about caches?
  - Vfy: on whilst a verify attempt is made; should it stay on when an error was encountered to indicate it was a verification error rather as opposed to a programming error?

The biggest design question is: Should the LEDs be operated by the AVRDUDE overall programmer-independent logic or by individual programmers, ie, be part of the many, many programmer-implemented paged_load(), paged_write(), byte_read(), byte_write() and chip_erase() attempts?

Right now it is a mixture:  the generic bytewise read and write functions in avr.c operate LEDs (so programmer-independent); some (most, all?) individual programmer chip_erase() routines operate LEDs; some (but not all) individual programmer paged_load() and paged_write() calls operate LEDs (this is why flash write/read with ScratchMonkey does not show the pgm LED); rarely any bytewise read/write routines in programmers operate LEDs (probably b/c the LED logic is programmer-independent).

I suggest to remove all programmer-specific ..._led() calls and move them instead to the main AVRDUDE logic:, 
 - If a -c programmer code provides ..._led() access functions AVRDUDE sets the lights in the same way throughout
 - Applications written using `libavrdude` can implement a different LED logic as they please - the programmer code does not get in the way
 - Usually there are a few reasons why programming a byte/page can go wrong, and each `return -1` would need to assess which LEDs should be set/cleared; multiply this by some 50 programmer types and 5 functions per programmer and it is clear it is a Herculean task to get this right
 - There are only a few places where the AVRDUDE main logic actually calls the programming functions: managable

Ho-Ro, 2023-10-15:
> So, what should happen? There are 4 LEDs: rdy, pgm, vfy and err. Here my suggestion

@stefanrueger Please find my LED proposals below:

> Rdy is relatively uncontroversial: set once the programmer is initialised and off before AVRDUDE exits (done)

My idea would be to use it as a kind of OK indicator (coloured green) also after programming access. I have changed the behaviour of avrdude so that rdy remains active after exit to indicate progress (I have commented out the one line at the end of main.c).
I always press reset on my nanoSTK (Arduino Nano with ScratchMonkey firmware) to start a new avrdude call with a freshly initialised programmer. Previously, after successful writing, I had encountered unstable communication on the next avrdude call - apparently caused by SMo - which I finally *"fixed "* by starting the watchdog with 8s timeout when exiting progmode. That way the status stays on long enough for me to notice it, and I can start the next programming with a *brand new* programmer setup.
SMo is unfortunately no longer actively developed, and I don't have the time to go deep there, I *actually* just need a small ISP programming adapter because my original STK500 takes up too much desk space for me. 

> Err when a fatal error happened, stays after exit; but what is a fatal error? Only those that are related to programming?

Also, if the detection of the device to be programmed failed, this should be displayed as err.

> Pgm? On during read, write and erase attempts? How about caches?

Should remain ON if read/write/erase attempts were not finished successfully.

> Vfy: on whilst a verify attempt is made; should it stay on when an error was encountered to indicate it was a verification error rather as opposed to a read/write error?

I think it should remain ON if verification does not complete successfully.

Just my 0.02 €

stefanrueger, 2023-10-15:
@Ho-Ro Thanks for your proposal!

I can relate to the idea that PGM/VFY should remain on in case of errors. So, after AVRDUDE exits

| PGM | VFY | ERR | Semantics|
| :-: | :-: | :-: | :-- |
| off | off | off | OK: all tasks done w/o errors |
| off | off | on | Some error not related to read/write/erase |
| on | off | on | read/write/erase error |
| off | on | on | verification error *but no* read/write/erase error |
| on | on | on | read/write/erase error *and* verification error |

Other combinations should not show after exit. During programming LEDS are twiddled to instill a sense of progress (and it might well be that an error first triggers an ERR LED but once healing the error that it has the previous value)

Interpreting the LEDs only makes sense once AVRDUDE was able to read the conf files, understand what the user wants, and successfully initialise, open end enable the programmer, ie, if RDY has shone once. After all, this is a prerequisite to setting LEDs.

@mcuee @MCUdude @dl8dtl @mariusgreuel What do the other dudes think?

> [Use RDY LED as] OK indicator (coloured green)

That differs from the original intention and millennia-old practice (well probably 20 years). I wouldn't want to touch the RDY semantics and operation.

> watchdog with 8s

To keep all set LEDs on for this period before resetting the programmer? The SMo firmware could switch on the RDY LED and set the watchdog to 8s so it's clear when the programmer has reset itself. You would loose the info whether or not AVRDUDE crossed the AVR device initialized and ready to accept instructions line. You would still be told the OK info by the absence of ERR.

mcuee, 2023-10-21:
Looks like a good idea. 

I do not seem to have good programmer to carry out the tests but I will see what I can do.

closed 2023-10-23 by stefanrueger