avrdudes/avrdude #1527: JTAG signature write

stefanrueger, 2023-10-17:
There are a couple of ..._write_byte() routines that contain code for writing the signature page, eg,
https://github.com/avrdudes/avrdude/blob/cbac6349b02ae72ec3ed2b345efe8f121d846547/src/jtagmkII.c#L2374-L2378

Is that a thing? For which parts could `jtagmkII` programmers *actually* write the signature (or calibration)?

mcuee, 2023-10-18:
@stefanrueger 

This is the issue for Calibration. Supposedly AVRJTAG mkII is able to carry out calibration but it is not supported by avrdude.

I have closed that issue. We can re-open if needed.
* https://github.com/avrdudes/avrdude/issues/71

mcuee, 2023-10-18:
Other than OSCCAL_BYTE, I think `XMEGA_USER_SIGNATURE` can be read/write as well.

mcuee, 2023-10-18:
For JTAGICE 1: 
https://ww1.microchip.com/downloads/en/Appnotes/doc2524.pdf

Page 12:
> Memory type EEPROM_JTAG uses page write or byte write and byte read. Memtypes FUSE_JTAG, LOCK_JTAG, SIGN_JTAG and OSCAL_JTAG use byte read and (write). 
> The address sent to JTAG ICE is a word address for Program Memory and byte address for all other memory types. Only the start address and the number of bytes to be read/written must be supplied.

mcuee, 2023-10-18:
The above documents are not so clear whether the calibration OSCCAL_BYTE can be written or not. However, from AVR055, it seems the calibration does not really write to the byte, rather write to another place (EEPROM or Flash) and later can be loaded to the OSCCAL register.
https://ww1.microchip.com/downloads/en/Appnotes/doc8002.pdf
https://onlinedocs.microchip.com/pr/GUID-CD53F55D-3184-4CEB-B4FB-0C4E57125F5B-en-US-2/index.html?GUID-54F94268-CAD4-4FDA-B3DF-ECF4F980D1D1

mcuee, 2023-10-18:
> **The above documents are not so clear whether the calibration OSCCAL_BYTE or OSCAL_JTAG byte can be written or not.**

Maybe @xedbg can help here. Thanks.

xedbg, 2023-10-18:
No - calibration sections can only be written in the Microchip fab, sorry.
(And not even I know how to do that ;)

dl8dtl, 2023-10-18:
> And not even I know how to do that

They are written on the chip tester, using tester bit streams. The device is "sealed" afterwards, which is (IIRC) an OTP fuse that cannot be erased later on, thus protecting the production signature row.

Early AVRs (notably the AT90S1200) could indeed accidentally erase or overwrite the signature bytes; devices with signature 0xFF, 0xFF, 0xFF (this is documented in AVR910) and even 0x00, 0x00, 0x00 have been observed. That's been the original reason for adding the `-F` option to AVRDUDE.

mcuee, 2023-10-18:
@dl8dtl 

Thanks for the bakground info. That is quite interesting.

Do you think the codes mentioned by @stefanrueger need to be changed?

stefanrueger, 2023-10-18:
> OTP fuse that cannot be erased later on, thus protecting the production signature row

Thanks all round. I suggest treating `signature` , `calibration`, `prodsig` and all sub memories such a `tempsense` as read-only in AVRDUDE, ie, the code should not attempt to write to device, and any intention to write something *different* to what is there should yield an error. (This allows writing the backup of a read-only memory without triggering an error.)

mcuee, 2023-10-18:
> Thanks all round. I suggest treating `signature` , `calibration`, `prodsig` and all sub memories such a `tempsense` as read-only in AVRDUDE, ie, the code should not attempt to write to device, and [...]

I agree.

mcuee, 2023-10-23:
Just noticed that the datasheet is clear about sigrow for the AVR DD and EA family (probably also other families).
https://ww1.microchip.com/downloads/aemDocuments/documents/MCU08/ProductDocuments/DataSheets/AVR64DD32-28-Prelim-DataSheet-DS40002315B.pdf

https://ww1.microchip.com/downloads/aemDocuments/documents/MCU08/ProductDocuments/DataSheets/AVR64EA-28-32-48-DataSheet-DS40002443.pdf

> The content of the Signature Row (SIGROW) fuses is preprogrammed and read-only. SIGROW contains information
such as device ID, serial number, and calibration values.

PR #1554 (2023-11-11): Review write_byte() functions
Fixes #1527 

This PR allows read-only memories to appear written successfully provided the data to be written is already there. This is useful so that "writing" a backup to read-only memories works without generating an error and without actually trying to write to the part.

Trying to write a read-only memory with a byte that is different to the one in the memory will cause a write error. This is useful, if for example, one tries to write a backup of a part to another part with different signature. If the signature is written back first, this constitutes a sanit [... cut]
closed 2023-11-11 by stefanrueger