avrdudes/avrdude #1529: Report: Incorrect SerialUPDI control of AVR64EA32

askn37, 2023-10-21:
This is a known bug as of commit 481a91d.

1. You are trying to read the "Chip Silicon Revision" register without checking the status of the target device. This will fail if the target device is "resetting" or "sleeping".
2. The address width used to access the flash area is incorrectly set to 16 bits. Therefore, the IO memory area is accessed instead of the flash area. The correct address width must be 24 bits.(Shouldn't we send 0x55 0x6A 0x00 0x00 0x80 here?)

case.1 Debug log

```
avrdude: sending 2 bytes [0x55, 0xe6]
avrdude: send: U [55] . [e6] 
avrdude: recv: U [55] . [e6] 
avrdude: recv: A [41] V [56] R [52]   [20]   [20]   [20]   [20]   [20] P [50] : [3a] 3 [33] D [44] : [3a] 1 [31] - [2d] 3 [33] M [4d] 2 [32]   [20] ( [28] B [42] 1 [31] . [2e] 5 [35] 9 [39] F [46] 0 [30] 2 [32] . [2e] 0 [30] ) [29] . [00] 
avrdude: received 32 bytes [0x41, 0x56, 0x52, 0x20, 0x20, 0x20, 0x20, 0x20, 0x50, 0x3a, 0x33, 0x44, 0x3a, 0x31, 0x2d, 0x33, 0x4d, 0x32, 0x20, 0x28, 0x42, 0x31, 0x2e, 0x35, 0x39, 0x46, 0x30, 0x32, 0x2e, 0x30, 0x29, 0x00]
avrdude: received SIB: [AVR     P:3D:1-3M2 (B1.59F02.0)]
avrdude: Device family ID: AVR     
avrdude: NVM interface: P:3
avrdude: Debug interface: D:1
avrdude: PDI oscillator: 3M2 
avrdude: Extra information: (B1.59F02.0)
avrdude: NVM type 3: 16-bit, page oriented
avrdude: reading 1 bytes from 0x000F01
avrdude: ST_PTR to 0x000F01
avrdude: sending 4 bytes [0x55, 0x69, 0x01, 0x0f]
avrdude: send: U [55] i [69] . [01] . [0f] 
avrdude: recv: U [55] i [69] . [01] . [0f] 
avrdude: ser_recv(): programmer is not responding
avrdude: serialupdi_recv(): programmer is not responding
avrdude: UPDI ST_PTR recv failed on ACK
avrdude: ST_PTR operation failed
avrdude serialupdi_initialize() [serialupdi.c:636] error: Reading chip silicon revision failed
avrdude main() [main.c:1404] error: initialization failed, rc=-1
        - double check the connections and try again
        - use -b to set lower baud rate, e.g. -b 57600
        - use -F to override this check
```

case.2 Debug log

```
avrdude: processing -U flash:r:avr64ea32/Flash.hex:i
avrdude: reading flash memory ...
avrdude: reading 128 bytes from 0x800000
avrdude: ST_PTR to 0x800000
avrdude: sending 4 bytes [0x55, 0x69, 0x00, 0x00]
avrdude: send: U [55] i [69] . [00] . [00] 
avrdude: recv: U [55] i [69] . [00] . [00] 
avrdude: recv: @ [40] 
avrdude: received 1 bytes [0x40]
avrdude: repeat 128
avrdude: sending 3 bytes [0x55, 0xa0, 0x7f]
avrdude: send: U [55] . [a0] . [7f] 
avrdude: recv: U [55] . [a0] . [7f] 
avrdude: LD8 from ptr++
avrdude: sending 2 bytes [0x55, 0x24]
avrdude: send: U [55] $ [24] 
avrdude: recv: U [55] $ [24] 
avrdude: recv: . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [ff] . [7f] . [02]   [20] . [00] . [07] . [00]   [20] . [00] . [07] . [00]   [20] . [00] . [07] . [00]   [20] . [00] . [07] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [07] . [00] . [00] . [00] . [00] . [00] . [00] . [05] . [00] . [00] . [00] . [00] * [2a] . [00] . [06] . [00] . [00] . [00] . [00] . [00] . [00] . [10] . [00] . [00] . [00] . [2e] . [00] . [00] . [00] . [00] . [00] . [00] 
avrdude: received 128 bytes [0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xff, 0x7f, 0x02, 0x20, 0x00, 0x07, 0x00, 0x20, 0x00, 0x07, 0x00, 0x20, 0x00, 0x07, 0x00, 0x20, 0x00, 0x07, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x07, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x05, 0x00, 0x00, 0x00, 0x00, 0x2a, 0x00, 0x06, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x10, 0x00, 0x00, 0x00, 0x2e, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]
```

stefanrueger, 2023-10-21:
@dbuchwald Do you want to have a look? I suspect that 24-bit addresses are needed for the 64k device because the effective address after adding the flash offset exceeds 16 bit.

dbuchwald, 2023-10-21:
I will take a look, but might not be able to test it though, as I don't have the AVR64EA32 chip at hand. Anyway, will keep you posted!

dbuchwald, 2023-10-21:
OK, so this is what I came up with: according to the logs above, NVM link mode is version 3, which is 16-bit address/page oriented write. I checked it again in the original Microchip's pymcuprog implementation, and this is what they do:

https://github.com/microchip-pic-avr-tools/pymcuprog/blob/main/pymcuprog/serialupdi/application.py#L130

And this is what I do:

https://github.com/avrdudes/avrdude/blob/main/src/serialupdi.c#L163

So, as long as the chip itself reports to be NVM link version 3, we should and will use 16-bit addresses.

The only thing I can suggest now is for @askn37 to try to program the original device using pymcuprog software and see if it works. If it fails, then probably there is some bug in interface/documentation provided by Microchip. 
If it does work, then please, run pymcuprog with the highest verbosity level and post the output here - it will produce almost identical trace (with all the NVM commands listed), and I can use it to trace the issue back to the source.

mcuee, 2023-10-21:
I tested this before avrdude 7.2 release and it was okay.
https://github.com/avrdudes/avrdude/discussions/1257#discussioncomment-6203438

Just tested again with latest git main and it is still okay. I am using Microchip AVR64EA48 Curiosity Nano bought from Mouser SIngapore.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude
Usage: avrdude [options]
...
...
avrdude version 7.2-20231018 (481a91dd), https://github.com/avrdudes/avrdude

P> .\avrdude -c pkobn_updi -p avr64ea48 -U .\median_Release_1.0.0.hex
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e961e (probably avr64ea48)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip

avrdude: processing -U flash:w:.\median_Release_1.0.0.hex:i
avrdude: reading input file .\median_Release_1.0.0.hex for flash
         with 5122 bytes in 1 section within [0, 0x1401]
         using 41 pages and 126 pad bytes
avrdude: writing 5122 bytes flash ...
Writing | ################################################## | 100% 2.49 s
avrdude: 5122 bytes of flash written
avrdude: verifying flash memory against .\median_Release_1.0.0.hex
Reading | ################################################## | 100% 1.24 s
avrdude: 5122 bytes of flash verified

avrdude done.  Thank you.
```

Full debug log here.

<details>
<summary> click for the big full debug log with -vvvv</summary>

```
PS>.\avrdude -c pkobn_updi -p avr64ea48 -U .\median_Release_1.0.0.hex -vvvv

avrdude: Version 7.2-20231018 (481a91dd)
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is C:\work\avr\avrdude_test\avrdude_bin\avrdude.conf

         Using Port                    : usb
         Using Programmer              : pkobn_updi
avrdude: jtag3_open_updi()
avrdude: HV UPDI support: 1
avrdude: usbhid_open(): probing for max packet size
avrdude: usbhid_open(): setting max_xfer from DAP_Info response to 64
avrdude: found CMSIS-DAP compliant device, using EDBG protocol
avrdude: jtag3_getsync()

avrdude: jtag3_edbg_prepare()
avrdude: jtag3_edbg_prepare(): connection status 0x01
avrdude: sending sign-on command:
avrdude: jtag3_edbg_send(): sending 3 bytes
avrdude: jtag3_edbg_recv():
avrdude: jtag3_recv(): Got message seqno 0 (command_sequence == 0)

Raw message:
01  80  00
[general] OK
avrdude: defaulting memtype in -U w:.\median_Release_1.0.0.hex option to "flash"
avrdude: input file .\median_Release_1.0.0.hex auto detected as Intel Hex
         AVR Part                      : AVR64EA48
         RESET disposition             : dedicated
[... 2429 lines cut ...]
[AVR] OK
avrdude: jtag3_close()
avrdude: sending AVR sign-off command:
avrdude: jtag3_edbg_send(): sending 3 bytes
avrdude: jtag3_edbg_recv():
avrdude: jtag3_recv(): Got message seqno 101 (command_sequence == 101)

Raw message:
12  80  00
[AVR] OK
avrdude: sending sign-off command:
avrdude: jtag3_edbg_send(): sending 4 bytes
avrdude: jtag3_edbg_recv():
avrdude: jtag3_recv(): Got message seqno 102 (command_sequence == 102)

Raw message:
01  80  00
[general] OK

avrdude: jtag3_edbg_signoff()

avrdude done.  Thank you.
```

</details>

mcuee, 2023-10-21:
@askn37 

I have changed the label to `unconfirmed` for now. Please follow the suggestion from @dbuchwald. 

I will change the lable back to bug if another user or I can confirm the issue. Thanks.

mcuee, 2023-10-21:
@stefanrueger and @dbuchwald 

I can not test this right now but I am not sure if you can get some useful info from my debug log using the `-c pkobn_updi` programmer.

stefanrueger, 2023-10-21:
@askn37 Thanks for reporting and for expanding your [submission](https://github.com/avrdudes/avrdude/issues/1529#issue-1955163642) with additional information. Could I ask you to give the complete line, including which `-c` programmer etc where you get `IO` memory instead of `flash`, please. So, we can try to reproduce. You appear to use `-U flash:r:avr64ea32/Flash.hex:i` as one element.

dbuchwald, 2023-10-22:
@mcuee if you have access to Curiosity Nano board with AVR64EA32 (or AVR64EA48) chip, you can connect your SerialUPDI programmer like so:

[image]

I know it looks flimsy AF, but it does work, good enough to run some tests.

That being said, I want to clarify something - my implementation of SerialUPDI in AVRDUDE was always meant to be just a port of Python-based Microchip reference implementation. It was never built directly based on chip datasheets, an for a reason. I could have never dreamt of testing my code against statistically significant sample of the actual microcontrollers, so following manufacturer's reference implementation was the only "natural" thing to do.
This is why I need comparison between **pymcuprog** and **AVRDUDE**, I need to know how the reference implementation handles support for these processors.
Sure, I can add new parameter to the programmer to enable 24-bit operation on NVM v3 and offer it as an experimental feature, but I would be hesitant to changing the core implementation based on datasheet of a chip I can't get my hands on in reasonable timeframe.

Bottom line: I really need to know two things:

1. How does **pymcuprog** handle scenarios with "resetting"/"sleeping" chip (as far as I recall, it doesn't), and how to reproduce this behavior (how to put one of the Dx chips I do have in these states),
2. How does **pymcuprog** handle programming of the AVR64EA32 chips.

Please note: English is not my first language, and now rereading the above it sounds like I don't want to handle this issue, but that would be wrong impression. I really do want to help, and I'm happy to experiment with different approaches, but I really need to know that whatever we are trying to do here will not break existing functionality.

mcuee, 2023-10-22:
@dbuchwald 

Thanks a lot for the tip. I will try to get some compariosons done tomorrow.between pymcuprog and avrdude.

No worry, you have been very helpful to avrdude project.

askn37, 2023-10-22:
@mcuee
 
> 1. How does **pymcuprog** handle scenarios with "resetting"/"sleeping" chip (as far as I recall, it doesn't), and how to reproduce this behavior (how to put one of the Dx chips I do have in these [...]

Unfortunately, it appears that pymcuprog has not been updated since October 2022. Microchip published his AVR16EAxx and AVR32EAxx repositories in April 2023 and began shipping engineering samples to customers. Therefore, it is unlikely that pymcuprog can expect full compatibility with his NVMCTRL version 3.

In fact, the first release of Microchip's AVR-LIBC library for his AVR32EAxx had an EEPROM control bug that incorrectly used version 0 code for NVMCTRL version 3. I first discovered it and reported it to Microchip. The repository was properly modified and published in July.

I have never tested pymcuprog before. Because debugging Python is not my main job. However, if new issues are discovered, you will need to report them to Microchip again.

(By the way, my native language is Japanese. Conversation with a microchip is quite troublesome.)

mcuee, 2023-10-22:
@xedbg 

Just wondering if you can chime help to check here. Thanks.

stefanrueger, 2023-10-22:
> I'm happy to experiment with different approaches

Thank you, @dbuchwald. This is great. Given the choice between a working implementation and one that's conform to a possibly outdated reference implementation, I'd go for the former in a heartbeat. It would be good if you could continue to look after the UPDI implementation @dbuchwald. Many thanks all round, @mcuee for testing and @askn37 for making us aware of this problem. (@askn37 if you have specific code suggestions to try, please feel free to share).

dbuchwald, 2023-10-22:
@mcuee if you get some time to replicate the issue, you can also grab the version I just provided via PR above - it basically replaces 16-bit addressing mode for NVM v3 controllers by 24-bit mode. You can test if it works any better.
The main problem I see now is that I don't know if there is any reasonable way to get list of devices using NVM v3 and test them against this implementation.
As for Microchip dropping support for pymcuprog/serialupdi, this is unfortunate, but somehow consistent with their way of operation. Let's leave it there.

If it works, then we need to look into this "sleep" mode, and as far as I can tell from the datasheet, it might not be trivial thing to do:

_The UPDI PHY layer runs independently of all sleep modes, and the UPDI is always accessible for a connected debugger independent of the device’s sleep state. If the system enters a sleep mode that turns the system clock off, the UPDI cannot access the system bus and read memories and peripherals. When enabled, the UPDI will request the system clock so that the UPDI always has contact with the rest of the device. Thus, the UPDI PHY layer clock is unaffected by the sleep mode’s settings. By reading the System Domain in Sleep (INSLEEP) bit in the ASI System Status (UPDI.ASI_SYS_STATUS) register, it is possible to monitor if the system domain is in a sleep mode. It is possible to prevent the system clock from stopping when going into a sleep mode by writing to the Request System Clock (CLKREQ) bit in the ASI System Control A (UPDI.ASI_SYS_CTRLA) register. If this bit is set, the system’s sleep mode state is emulated, and the UPDI can access the system bus and read the peripheral registers even in the deepest sleep modes._

mcuee, 2023-10-22:
I will label this as bug for now based on the reports by @askn37. But I will try to carry out the test to confirm later.

mcuee, 2023-10-23:
@dbuchwald 

I can confirm there is an issue in git main and your PR #1531 helps, at least for reading. For writing I still encounfer faliures but I do not have a steady hand now to hold the pin headers long enough. I will do that later.

1) git main failed to verify flash contents (I used the on-board programmer to program the flash first and then verify with serialupdi.
```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git -c serialupdi -P COM9 -p avr64ea48 -U flash:v:median_Release_1.0.0.hex:i
avrdude_git: AVR device initialized and ready to accept instructions
avrdude_git: device signature = 0x1e961e (probably avr64ea48)

avrdude_git: processing -U flash:v:median_Release_1.0.0.hex:i
avrdude_git: verifying flash memory against median_Release_1.0.0.hex
Reading | ################################################## | 100% 1.19 s
avrdude_git warning: verification mismatch
            device 0x00 != input 0x5c at addr 0x0000 (error)
avrdude_git error: verification mismatch

avrdude_git done.  Thank you.
```

PR #1531 is good in terms of verification.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_pr1531 -C .\avrdude_pr1531.conf -c serialupdi -P COM9 -p avr64ea48 -U flash:v:median_Release_1.0.0.hex:i
avrdude_pr1531: AVR device initialized and ready to accept instructions
avrdude_pr1531: device signature = 0x1e961e (probably avr64ea48)

avrdude_pr1531: processing -U flash:v:median_Release_1.0.0.hex:i
avrdude_pr1531: verifying flash memory against median_Release_1.0.0.hex
Reading | ################################################## | 100% 1.19 s
avrdude_pr1531: 5122 bytes of flash verified

avrdude_pr1531 done.  Thank you.
```

mcuee, 2023-10-23:
> For writing I still encounfer faliures but I do not have a steady hand now to hold the pin headers long enough. I will do that later.

@dbuchwald 

I managed to hold the pin header and PR #1531 is good as well.

```
PS > .\avrdude_pr1531 -C .\avrdude_pr1531.conf -c serialupdi -P COM9 -p avr64ea48 -U median_Release_1.0.0.hex
avrdude_pr1531: AVR device initialized and ready to accept instructions
avrdude_pr1531: device signature = 0x1e961e (probably avr64ea48)
avrdude_pr1531: Note: flash memory has been specified, an erase cycle will be performed.
                To disable this feature, specify the -D option.
avrdude_pr1531: erasing chip

avrdude_pr1531: processing -U flash:w:median_Release_1.0.0.hex:i
avrdude_pr1531: reading input file median_Release_1.0.0.hex for flash
                with 5122 bytes in 1 section within [0, 0x1401]
                using 41 pages and 126 pad bytes
avrdude_pr1531: writing 5122 bytes flash ...
Writing | ################################################## | 100% 1.91 s
avrdude_pr1531: 5122 bytes of flash written
avrdude_pr1531: verifying flash memory against median_Release_1.0.0.hex
Reading | ################################################## | 100% 0.93 s
avrdude_pr1531: 5122 bytes of flash verified

avrdude_pr1531 done.  Thank you.
```

git main will fail.
```
PS > .\avrdude_git -c serialupdi -P COM9 -p avr64ea48 -U median_Release_1.0.0.hex
avrdude_git: AVR device initialized and ready to accept instructions
avrdude_git: device signature = 0x1e961e (probably avr64ea48)
avrdude_git: Note: flash memory has been specified, an erase cycle will be performed.
             To disable this feature, specify the -D option.
avrdude_git: erasing chip

avrdude_git: processing -U flash:w:median_Release_1.0.0.hex:i
avrdude_git: reading input file median_Release_1.0.0.hex for flash
             with 5122 bytes in 1 section within [0, 0x1401]
             using 41 pages and 126 pad bytes
avrdude_git: writing 5122 bytes flash ...
Writing | ##------------------------------------------------ | 4% 2.16 s
avrdude_git error: commit data command failed
avrdude_git error: wait NVM ready timed out
avrdude_git error: updi_nvm_wait_ready() failed
 ***failed;
avrdude_git error: wait NVM ready timed out
avrdude_git error: updi_nvm_wait_ready() failed
 ***failed;
avrdude_git error: wait NVM ready timed out
avrdude_git error: updi_nvm_wait_ready() failed
 ***failed;
avrdude_git error: wait NVM ready timed out
avrdude_git error: updi_nvm_wait_ready() failed
 ***failed;
avrdude_git error: wait NVM ready timed out
avrdude_git error: updi_nvm_wait_ready() failed
 ***failed;
avrdude_git error: wait NVM ready timed out
avrdude_git error: updi_nvm_wait_ready() failed
 ***failed;
(I hit CTRL-C to exit)
```

mcuee, 2023-10-23:
BTW, I need to mention that the nice setial2updi adapter (with pogo-pins or pin headers) is courtesy of @SpenceKonde (aka Dr. Azzy).

He is very kind and sent me the following at the end of last year.
1) Nice serialupdi adapter with pogo pins and pin headers
2) AVR64DD32 board
3) AVR32DD14 breakout board
4) ATTiny3227 breakout board

mcuee, 2023-10-23:
@askn37 

Please give PR #1531 a try. Thanks. My testing shows that it works fine.

askn37, 2023-10-23:
I would like to thank you all for your great work.

The attached code is a program that allows you to experience sleep failures (rather than address width issues). This applies to tinyAVR-2 and most of its AVR_DA/DB/DD/EA/EB. It can be constructed with AVR-LIBC.

When I run it, the LED keeps blinking using PA7, but the CPU remains in a STANDBY sleep state with interrupts disabled. Also, F_CPU is 512Hz.

(TCB0, CCL, and EVOUT are used for this LED blinking operation, but tinyAVR-0/1 and megaAVR-0 are not supported because the difference in CCL is troublesome.)

Please upload your code to the device using a method other than the problematic SerialUPDI and test it with SerialUPDI. It is important not to reset the CPU at this point. The CPU startup start time is affected by the SUT setting value of FUSE_SYSCFG1.

To test without going to sleep, run a SerialUPDI with the reset pad LOW, wait a second or two, then release the reset. SerialUPDI should be able to recognize the CPU. A power-on reset is required if the chip's reset pad functionality is disabled in FUSE_SYSCFG0.

Engineering samples of the AVR16EBxx will begin arriving to stakeholders soon. It seems that the application for free distribution has ended.
[moderAVR_sleep_c.txt](https://github.com/avrdudes/avrdude/files/13065838/moderAVR_sleep_c.txt)

xedbg, 2023-10-23:
> Just wondering if you can help to check here. Thanks.

Sure, i can take a closer look today.
But something does looks off here - the 'default' should use 24-bit UPDI going forward for all parts.

dbuchwald, 2023-10-23:
> But something does looks off here - the 'default' should use 24-bit UPDI going forward for all parts.

And that was the change I made - chips reporting NVM v3 are now handled with 24-bit addressing. That being said, it would be great to find some sort of list of all the devices using this NVM mode to test it with more than one or two...

mcuee, 2023-10-23:
> If it works, then we need to look into this "sleep" mode, and as far as I can tell from the datasheet, it might not be trivial thing to do:
>
> _The UPDI PHY layer runs independently of all sleep [...]

If this is not trivial to do, I think we can just document this as a  limitation.

dbuchwald, 2023-10-23:
> If this is not trivial to do, I think we can just document this as a limitation.

I will try to reproduce the issue using the code provided by @askn37, and if it occurs on a hardware that I do have at hand (Curiosity Nano with AVR128DA), I can see if there is any procedure that would "wake up" the chip. So far I haven't found any suggestion in the datasheet, but I need to look deeper. At the very least I can try to read the status register to check if device is asleep and display some sort of information. 

> Do you know other AVR families which use NVM v3?

Nope. I actually found my original PR with first implementation of SerialUPDI interface https://github.com/avrdudes/avrdude/pull/772 and found this part of the comment:

> NVM mode 3 has not been tested, since I don't have EA-series chips either. I don't even know if they can be obtained

So yeah, testing against another NVM v3 device would be great...

xedbg, 2023-10-23:
> > > But something does looks off here - the 'default' should use 24-bit UPDI going forward for all parts.
> > 
> > 
> > And that was the change I made - chips reporting NVM v3 are now handled with 24-bit addressing. That being said, it would be great to find some sort of list of all the devices using this NVM mode to test it with more than one or two...
> 
> Do you know other AVR families which use NVM v3?

I think its just EA.  (DU uses P:4 and EB uses P:5)

askn37, 2023-10-24:
@dbuchwald: The easiest way is to get information other than UPDI_SIB after "entering NVM programming mode". It is now guaranteed to be unaffected by the execution state of the target CPU.

@mcuee: The AVR_EA and AVR_EB series are currently known to use NVMCTRL version 3. As proof, look for the "ioavr16eb32.h" file in his PACK, available at [Microchip Packs Repository](https://packs.download.microchip.com). And check the NVMCTRL_CMD_enum information. Let's also take a look at "#define NVMCTRL_ADDR". This is where the differences between the versions of NVMCTRL become apparent.

mcuee, 2023-10-25:
> @mcuee: The AVR_EA and AVR_EB series are currently known to use NVMCTRL version 3. As proof, look for the "ioavr16eb32.h" file in his PACK, available at [Microchip Packs [...]

Thanks a lot for the tip. I have just refreshed my atdf repo. The only missing pack seems to be AVR32 DFP.

```
 MINGW64 /c/work/avr/avrdude_test/atdf
$ ls -la *.atdf | wc -l
387

$ ls *.atdf -laG --time-style=long-iso | awk '{print $4, $5, $6, $7}'
399906 2020-09-18 15:57 AT32UC3A0128.atdf
399907 2020-09-18 15:57 AT32UC3A0256.atdf
399907 2020-09-18 15:57 AT32UC3A0512.atdf
381293 2020-09-18 15:57 AT32UC3A1128.atdf
381293 2020-09-18 15:57 AT32UC3A1256.atdf
381293 2020-09-18 15:57 AT32UC3A1512.atdf
515771 2020-09-18 15:57 AT32UC3A3128.atdf
524960 2020-09-18 15:57 AT32UC3A3128S.atdf
515771 2020-09-18 15:57 AT32UC3A3256.atdf
524960 2020-09-18 15:57 AT32UC3A3256S.atdf
513912 2020-09-18 15:57 AT32UC3A364.atdf
523101 2020-09-18 15:57 AT32UC3A364S.atdf
515790 2020-09-18 15:57 AT32UC3A4128.atdf
524979 2020-09-18 15:57 AT32UC3A4128S.atdf
515790 2020-09-18 15:57 AT32UC3A4256.atdf
524979 2020-09-18 15:57 AT32UC3A4256S.atdf
513931 2020-09-18 15:57 AT32UC3A464.atdf
523120 2020-09-18 15:57 AT32UC3A464S.atdf
353307 2020-09-18 15:57 AT32UC3B0128.atdf
353306 2020-09-18 15:57 AT32UC3B0256.atdf
357446 2020-09-18 15:57 AT32UC3B0512.atdf
353305 2020-09-18 15:57 AT32UC3B064.atdf
337678 2020-09-18 15:57 AT32UC3B1128.atdf
337678 2020-09-18 15:57 AT32UC3B1256.atdf
356551 2020-09-18 15:57 AT32UC3B1512.atdf
337674 2020-09-18 15:57 AT32UC3B164.atdf
841945 2020-09-18 15:57 AT32UC3C0128C.atdf
838821 2020-09-18 15:57 AT32UC3C0256C.atdf
838918 2020-09-18 15:57 AT32UC3C0512C.atdf
838821 2020-09-18 15:57 AT32UC3C064C.atdf
827160 2020-09-18 15:57 AT32UC3C1128C.atdf
827161 2020-09-18 15:57 AT32UC3C1256C.atdf
827244 2020-09-18 15:57 AT32UC3C1512C.atdf
827161 2020-09-18 15:57 AT32UC3C164C.atdf
815056 2020-09-18 15:57 AT32UC3C2128C.atdf
815057 2020-09-18 15:57 AT32UC3C2256C.atdf
815140 2020-09-18 15:57 AT32UC3C2512C.atdf
815057 2020-09-18 15:57 AT32UC3C264C.atdf
607660 2020-09-18 15:57 AT32UC3L0128.atdf
489228 2020-09-18 15:57 AT32UC3L016.atdf
607429 2020-09-18 15:57 AT32UC3L0256.atdf
489351 2020-09-18 15:57 AT32UC3L032.atdf
512152 2020-09-18 15:57 AT32UC3L064.atdf
102904 2023-08-15 15:08 AT90CAN128.atdf
101282 2023-08-15 15:08 AT90CAN32.atdf
101402 2023-08-15 15:08 AT90CAN64.atdf
75537 2023-08-15 15:08 AT90PWM1.atdf
70469 2023-08-15 15:08 AT90PWM161.atdf
79625 2023-08-15 15:08 AT90PWM216.atdf
81495 2023-08-15 15:08 AT90PWM2B.atdf
88682 2023-08-15 15:08 AT90PWM3.atdf
86968 2023-08-15 15:08 AT90PWM316.atdf
87613 2023-08-15 15:08 AT90PWM3B.atdf
71631 2023-08-15 15:08 AT90PWM81.atdf
85865 2023-08-15 15:08 AT90USB1286.atdf
91745 2023-08-15 15:08 AT90USB1287.atdf
62628 2023-08-15 15:08 AT90USB162.atdf
91741 2023-08-15 15:08 AT90USB646.atdf
91741 2023-08-15 15:08 AT90USB647.atdf
62627 2023-08-15 15:08 AT90USB82.atdf
[... 304 lines cut ...]
309619 2022-12-16 13:03 AVR32DD14.atdf
319441 2022-12-16 13:03 AVR32DD20.atdf
327340 2022-12-16 13:03 AVR32DD28.atdf
331275 2022-12-16 13:03 AVR32DD32.atdf
298103 2023-10-02 14:18 AVR32EA28.atdf
302718 2023-10-02 14:18 AVR32EA32.atdf
321484 2023-10-02 14:18 AVR32EA48.atdf
325064 2022-12-16 13:03 AVR64DA28.atdf
332430 2022-12-16 13:03 AVR64DA32.atdf
399272 2022-12-16 13:03 AVR64DA48.atdf
428356 2022-12-16 13:03 AVR64DA64.atdf
350851 2022-12-16 13:03 AVR64DB28.atdf
357181 2022-12-16 13:03 AVR64DB32.atdf
432881 2022-12-16 13:03 AVR64DB48.atdf
456871 2022-12-16 13:03 AVR64DB64.atdf
309621 2022-12-16 13:03 AVR64DD14.atdf
319443 2022-12-16 13:03 AVR64DD20.atdf
327342 2022-12-16 13:03 AVR64DD28.atdf
331277 2022-12-16 13:03 AVR64DD32.atdf
298105 2023-10-02 14:18 AVR64EA28.atdf
302720 2023-10-02 14:18 AVR64EA32.atdf
321486 2023-10-02 14:18 AVR64EA48.atdf
428426 2020-09-18 16:00 WLR089U0.atdf

```

mcuee, 2023-10-25:
> I would like to thank you all for your great work.
>
> The attached code is a program that allows you to experience sleep failures (rather than address width issues). This applies to tinyAVR-2 and most [...]

I think I can reproduce the issue just using serialupdi.

You can see programming is okay as it resets the MCU.

But then running simple query command will fail as that does not reset the MCU.

```
PS C:\work\avr\avrdude_test\avrdude_bin> cat .\Makefile
CFLAGS ?= -Os -DF_APU=512UL -mmcu=avr32dd14
LDFLAGS ?= -mmcu=avr32dd14

all: sleep.hex

sleep.hex: sleep.elf
        avr-objcopy -O ihex sleep.elf sleep.hex

sleep.elf: sleep.o
        avr-gcc $(LDFLAGS) -o $@ $^

sleep.o: sleep.c
        avr-gcc $(CFLAGS) -c -o $@ $<

clean: FRC
        rm -f sleep.elf sleep.hex sleep.o

FRC:

PS C:\work\avr\avrdude_test\avrdude_bin> make
avr-gcc -Os -DF_APU=512UL -mmcu=avr32dd14 -c -o sleep.o sleep.c
avr-gcc -mmcu=avr32dd14 -o sleep.elf sleep.o
avr-objcopy -O ihex sleep.elf sleep.hex

PS C:\work\avr\avrdude_test\avrdude_bin> function prompt {$null}

PS>.\avrdude_git -c serialupdi -P COM11 -p avr32dd14 -U .\sleep.hex
avrdude_git: AVR device initialized and ready to accept instructions
avrdude_git: device signature = 0x1e953b (probably avr32dd14)
avrdude_git: Note: flash memory has been specified, an erase cycle will be performed.
             To disable this feature, specify the -D option.
avrdude_git: erasing chip

avrdude_git: processing -U flash:w:.\sleep.hex:i
avrdude_git: reading input file .\sleep.hex for flash
             with 276 bytes in 1 section within [0, 0x113]
             using 1 page and 236 pad bytes
avrdude_git: writing 276 bytes flash ...
Writing | ################################################## | 100% 0.16 s
avrdude_git: 276 bytes of flash written
avrdude_git: verifying flash memory against .\sleep.hex
Reading | ################################################## | 100% 0.07 s
avrdude_git: 276 bytes of flash verified

avrdude_git done.  Thank you.

PS>.\avrdude_git -c serialupdi -P COM11 -p avr32dd14 -vvvv

avrdude_git: Version 7.2-20231018 (481a91dd)
             Copyright the AVRDUDE authors;
             see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

             System wide configuration file is C:\work\avr\avrdude_test\avrdude_bin\avrdude.conf

             Using Port                    : COM11
             Using Programmer              : serialupdi
avrdude_git: opening serial port ...
avrdude_git: sending 1 bytes [0x00]
[... 123 lines cut ...]
avrdude_git: check failed
avrdude_git: restoring datalink failed
avrdude_git serialupdi_initialize() [serialupdi.c:569] error: UPDI link initialization failed
avrdude_git main() [main.c:1404] error: initialization failed, rc=-1
            - double check the connections and try again
            - use -b to set lower baud rate, e.g. -b 57600
            - use -F to override this check
avrdude_git: leaving NVM programming mode
avrdude_git: sending reset request
avrdude_git: STCS 0x59 to address 0x08
avrdude_git: sending 3 bytes [0x55, 0xc8, 0x59]
avrdude_git: send: U [55] . [c8] Y [59]
avrdude_git: ser_recv(): programmer is not responding
avrdude_git: sending release reset request
avrdude_git: STCS 0x00 to address 0x08
avrdude_git: sending 3 bytes [0x55, 0xc8, 0x00]
avrdude_git: send: U [55] . [c8] . [00]
avrdude_git: ser_recv(): programmer is not responding
avrdude_git: STCS 0x0C to address 0x03
avrdude_git: sending 3 bytes [0x55, 0xc3, 0x0c]
avrdude_git: send: U [55] . [c3] . [0c]
avrdude_git: ser_recv(): programmer is not responding

avrdude_git done.  Thank you.
```

mcuee, 2023-10-25:
Hmm, strange. Using `-v` will fail. Without `-v` it will be good.

```
PS>.\avrdude_git -c serialupdi -P COM11 -p avr32dd14 -U .\sleep.hex
avrdude_git: AVR device initialized and ready to accept instructions
avrdude_git: device signature = 0x1e953b (probably avr32dd14)
avrdude_git: Note: flash memory has been specified, an erase cycle will be performed.
             To disable this feature, specify the -D option.
avrdude_git: erasing chip

avrdude_git: processing -U flash:w:.\sleep.hex:i
avrdude_git: reading input file .\sleep.hex for flash
             with 276 bytes in 1 section within [0, 0x113]
             using 1 page and 236 pad bytes
avrdude_git: writing 276 bytes flash ...
Writing | ################################################## | 100% 0.16 s
avrdude_git: 276 bytes of flash written
avrdude_git: verifying flash memory against .\sleep.hex
Reading | ################################################## | 100% 0.07 s
avrdude_git: 276 bytes of flash verified

avrdude_git done.  Thank you.

PS>.\avrdude_git -c serialupdi -P COM11 -p avr32dd14 -U .\sleep.hex
avrdude_git: AVR device initialized and ready to accept instructions
avrdude_git: device signature = 0x1e953b (probably avr32dd14)
avrdude_git: Note: flash memory has been specified, an erase cycle will be performed.
             To disable this feature, specify the -D option.
avrdude_git: erasing chip

avrdude_git: processing -U flash:w:.\sleep.hex:i
avrdude_git: reading input file .\sleep.hex for flash
             with 276 bytes in 1 section within [0, 0x113]
             using 1 page and 236 pad bytes
avrdude_git: writing 276 bytes flash ...
Writing | ################################################## | 100% 0.16 s
avrdude_git: 276 bytes of flash written
avrdude_git: verifying flash memory against .\sleep.hex
Reading | ################################################## | 100% 0.07 s
avrdude_git: 276 bytes of flash verified

avrdude_git done.  Thank you.

PS>.\avrdude_git -c serialupdi -P COM11 -p avr32dd14 -U .\sleep.hex -vvvv

avrdude_git: Version 7.2-20231018 (481a91dd)
             Copyright the AVRDUDE authors;
             see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

             System wide configuration file is C:\work\avr\avrdude_test\avrdude_bin\avrdude.conf

             Using Port                    : COM11
             Using Programmer              : serialupdi
avrdude_git: opening serial port ...
avrdude_git: sending 1 bytes [0x00]
avrdude_git: send: . [00]
avrdude_git: recv: . [00]
avrdude_git: defaulting memtype in -U w:.\sleep.hex option to "flash"
avrdude_git: input file .\sleep.hex auto detected as Intel Hex
             AVR Part                      : AVR32DD14
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               fuse0       wdtcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
[... 182 lines cut ...]
            - use -F to override this check
avrdude_git: leaving NVM programming mode

avrdude_git done.  Thank you.

PS>.\avrdude_git -c serialupdi -P COM11 -p avr32dd14 -U .\sleep.hex
avrdude_git: AVR device initialized and ready to accept instructions
avrdude_git: device signature = 0x1e953b (probably avr32dd14)
avrdude_git: Note: flash memory has been specified, an erase cycle will be performed.
             To disable this feature, specify the -D option.
avrdude_git: erasing chip

avrdude_git: processing -U flash:w:.\sleep.hex:i
avrdude_git: reading input file .\sleep.hex for flash
             with 276 bytes in 1 section within [0, 0x113]
             using 1 page and 236 pad bytes
avrdude_git: writing 276 bytes flash ...
Writing | ################################################## | 100% 0.16 s
avrdude_git: 276 bytes of flash written
avrdude_git: verifying flash memory against .\sleep.hex
Reading | ################################################## | 100% 0.07 s
avrdude_git: 276 bytes of flash verified

avrdude_git done.  Thank you.
```

stefanrueger, 2023-10-25:
> Using -v will fail. Without -v it will be good.

I suspect just one -v is OK, but -vvvv might ruin the timing of commands by the time it takes to print. Try to log on RAM, eg, -vvvvl/dev/shm/error.log

stefanrueger, 2023-10-25:
>  parse the atdf files to show that only the AVR EA and EB family AVRs are using NVM v3

Do you know how ATDF encodes this? `xml_grep` is useful

```
xml_grep -g --pretty_print indented --cond 'module[@name="NVMCTRL"]' *.atdf
```

reduces the amount to look at

mcuee, 2023-10-25:
> > Using -v will fail. Without -v it will be good.
> 
> I suspect just one -v is OK, but -vvvv might ruin the timing of commands by the time it takes to print. Try to log on RAM, eg, -vvvvl/dev/shm/error.log

One `-v` is already causing problems. Please refer to the log posted earlier.

```
PS>.\avrdude_git -c serialupdi -P COM11 -p avr32dd14 -U .\sleep.hex -v

avrdude_git: Version 7.2-20231018 (481a91dd)
             Copyright the AVRDUDE authors;
             see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

             System wide configuration file is C:\work\avr\avrdude_test\avrdude_bin\avrdude.conf

             Using Port                    : COM11
             Using Programmer              : serialupdi
avrdude_git: input file .\sleep.hex auto detected as Intel Hex
             AVR Part                      : AVR32DD14
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               fuse0       wdtcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuse1       bodcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuse2       osccfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuse5       syscfg0     0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuse6       syscfg1     0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuse7       codesize    0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuse8       bootsize    0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuses                   0     0     0    0 no         16   16      0     0     0 0x00 0x00
               lock                    0     0     0    0 no          4    1      0     0     0 0x00 0x00
               prodsig     sigrow      0     0     0    0 no        128  128      0     0     0 0x00 0x00
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
               tempsense               0     0     0    0 no          4    1      0     0     0 0x00 0x00
               sernum                  0     0     0    0 no         16    1      0     0     0 0x00 0x00
               userrow     usersig     0     0     0    0 no         32   32      0     0     0 0x00 0x00
               data                    0     0     0    0 no          0    1      0     0     0 0x00 0x00
               io                      0     0     0    0 no       4160    1      0     0     0 0x00 0x00
               sib                     0     0     0    0 no         32    1      0     0     0 0x00 0x00
               eeprom                  0     0     0    0 no        256    1      0     0     0 0x00 0x00
               flash                   0     0     0    0 no      32768  512      0     0     0 0x00 0x00

             Programmer Type : serialupdi
             Description     : SerialUPDI
avrdude_git serialupdi_initialize() error: UPDI link initialization failed
avrdude_git main() error: initialization failed, rc=-1
            - double check the connections and try again
            - use -b to set lower baud rate, e.g. -b 57600
            - use -F to override this check
avrdude_git: leaving NVM programming mode

avrdude_git done.  Thank you.
```

mcuee, 2023-10-25:
Hmm, I can not reproduce the issue on ATtiny3227.

```
PS>make -f .\Makefile_t3227
avr-gcc -Os -DF_APU=512UL -mmcu=attiny3227 -c -o sleep.o sleep.c
avr-gcc -mmcu=attiny3227 -o sleep.elf sleep.o
avr-objcopy -O ihex sleep.elf sleep.hex
PS>.\avrdude_git -c serialupdi -P COM11 -p t3227 -U .\sleep.hex
avrdude_git: AVR device initialized and ready to accept instructions
avrdude_git: device signature = 0x1e9526 (probably t3227)
avrdude_git: Note: flash memory has been specified, an erase cycle will be performed.
             To disable this feature, specify the -D option.
avrdude_git: erasing chip

avrdude_git: processing -U flash:w:.\sleep.hex:i
avrdude_git: reading input file .\sleep.hex for flash
             with 252 bytes in 1 section within [0, 0xfb]
             using 2 pages and 4 pad bytes
avrdude_git: writing 252 bytes flash ...
Writing | ################################################## | 100% 0.11 s
avrdude_git: 252 bytes of flash written
avrdude_git: verifying flash memory against .\sleep.hex
Reading | ################################################## | 100% 0.04 s
avrdude_git: 252 bytes of flash verified

avrdude_git done.  Thank you.

PS>.\avrdude_git -c serialupdi -P COM11 -p t3227
avrdude_git: AVR device initialized and ready to accept instructions
avrdude_git: device signature = 0x1e9526 (probably t3227)

avrdude_git done.  Thank you.

PS>.\avrdude_git -c serialupdi -P COM11 -p t3227 -v

avrdude_git: Version 7.2-20231018 (481a91dd)
             Copyright the AVRDUDE authors;
             see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

             System wide configuration file is C:\work\avr\avrdude_test\avrdude_bin\avrdude.conf

             Using Port                    : COM11
             Using Programmer              : serialupdi
             AVR Part                      : ATtiny3227
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               fuse0       wdtcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuse1       bodcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuse2       osccfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuse5       syscfg0     0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuse6       syscfg1     0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuse7       append      0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuse8       bootend     0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuses                   0     0     0    0 no         10   10      0     0     0 0x00 0x00
               lock                    0     0     0    0 no          1    1      0     0     0 0x00 0x00
               prodsig     sigrow      0     0     0    0 no         64   64      0     0     0 0x00 0x00
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
               tempsense               0     0     0    0 no          2    1      0     0     0 0x00 0x00
               sernum                  0     0     0    0 no         10    1      0     0     0 0x00 0x00
               osccal16                0     0     0    0 no          2    1      0     0     0 0x00 0x00
               osccal20                0     0     0    0 no          2    1      0     0     0 0x00 0x00
               osc16err                0     0     0    0 no          2    1      0     0     0 0x00 0x00
               osc20err                0     0     0    0 no          2    1      0     0     0 0x00 0x00
[... 292 lines cut ...]
avrdude_git: sending 4 bytes [0x55, 0x04, 0x02, 0x11]
avrdude_git: send: U [55] . [04] . [02] . [11]
avrdude_git: recv: U [55] . [04] . [02] . [11]
avrdude_git: recv: & [26]
avrdude_git: received 1 bytes [0x26]
Reading | ################################################## | 100% 0.05 s
avrdude_git: device signature = 0x1e9526 (probably t3227)
avrdude_git: leaving NVM programming mode
avrdude_git: sending reset request
avrdude_git: STCS 0x59 to address 0x08
avrdude_git: sending 3 bytes [0x55, 0xc8, 0x59]
avrdude_git: send: U [55] . [c8] Y [59]
avrdude_git: recv: U [55] . [c8] Y [59]
avrdude_git: sending release reset request
avrdude_git: STCS 0x00 to address 0x08
avrdude_git: sending 3 bytes [0x55, 0xc8, 0x00]
avrdude_git: send: U [55] . [c8] . [00]
avrdude_git: recv: U [55] . [c8] . [00]
avrdude_git: STCS 0x0C to address 0x03
avrdude_git: sending 3 bytes [0x55, 0xc3, 0x0c]
avrdude_git: send: U [55] . [c3] . [0c]
avrdude_git: recv: U [55] . [c3] . [0c]

avrdude_git done.  Thank you.
```

mcuee, 2023-10-25:
> > > Using -v will fail. Without -v it will be good.
> > 
> > 
> > I suspect just one -v is OK, but -vvvv might ruin the timing of commands by the time it takes to print. Try to log on RAM, eg, -vvvvl/dev/shm/error.log
> 
> One `-v` is already causing problems. Please refer to the log posted earlier.
> 

Same for another AVR64DD32 board. One `-v` and it will fail. This one has the LED and I can see the Sleep FW is working.

```
PS>make -f .\Makefile_avr64dd32
avr-gcc -Os -DF_APU=512UL -mmcu=avr64dd32 -c -o sleep.o sleep.c
avr-gcc -mmcu=avr64dd32 -o sleep.elf sleep.o
avr-objcopy -O ihex sleep.elf sleep.hex
PS>.\avrdude_git -c serialupdi -P COM11 -p avr64dd32 -U .\sleep.hex
avrdude_git: AVR device initialized and ready to accept instructions
avrdude_git: device signature = 0x1e961a (probably avr64dd32)
avrdude_git: Note: flash memory has been specified, an erase cycle will be performed.
             To disable this feature, specify the -D option.
avrdude_git: erasing chip

avrdude_git: processing -U flash:w:.\sleep.hex:i
avrdude_git: reading input file .\sleep.hex for flash
             with 276 bytes in 1 section within [0, 0x113]
             using 1 page and 236 pad bytes
avrdude_git: writing 276 bytes flash ...
Writing | ################################################## | 100% 0.16 s
avrdude_git: 276 bytes of flash written
avrdude_git: verifying flash memory against .\sleep.hex
Reading | ################################################## | 100% 0.07 s
avrdude_git: 276 bytes of flash verified

avrdude_git done.  Thank you.

PS>.\avrdude_git -c serialupdi -P COM11 -p avr64dd32
avrdude_git: AVR device initialized and ready to accept instructions
avrdude_git: device signature = 0x1e961a (probably avr64dd32)

avrdude_git done.  Thank you.

PS>.\avrdude_git -c serialupdi -P COM11 -p avr64dd32 -v

avrdude_git: Version 7.2-20231018 (481a91dd)
             Copyright the AVRDUDE authors;
             see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

             System wide configuration file is C:\work\avr\avrdude_test\avrdude_bin\avrdude.conf

             Using Port                    : COM11
             Using Programmer              : serialupdi
             AVR Part                      : AVR64DD32
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               fuse0       wdtcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuse1       bodcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuse2       osccfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuse5       syscfg0     0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuse6       syscfg1     0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuse7       codesize    0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuse8       bootsize    0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuses                   0     0     0    0 no         16   16      0     0     0 0x00 0x00
               lock                    0     0     0    0 no          4    1      0     0     0 0x00 0x00
               prodsig     sigrow      0     0     0    0 no        128  128      0     0     0 0x00 0x00
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
               tempsense               0     0     0    0 no          4    1      0     0     0 0x00 0x00
               sernum                  0     0     0    0 no         16    1      0     0     0 0x00 0x00
               userrow     usersig     0     0     0    0 no         32   32      0     0     0 0x00 0x00
               data                    0     0     0    0 no          0    1      0     0     0 0x00 0x00
               io                      0     0     0    0 no       4160    1      0     0     0 0x00 0x00
               sib                     0     0     0    0 no         32    1      0     0     0 0x00 0x00
               eeprom                  0     0     0    0 no        256    1      0     0     0 0x00 0x00
               flash                   0     0     0    0 no      65536  512      0     0     0 0x00 0x00

             Programmer Type : serialupdi
             Description     : SerialUPDI
avrdude_git serialupdi_initialize() error: UPDI link initialization failed
avrdude_git main() error: initialization failed, rc=-1
            - double check the connections and try again
            - use -b to set lower baud rate, e.g. -b 57600
            - use -F to override this check
avrdude_git: leaving NVM programming mode

avrdude_git done.  Thank you.
```

mcuee, 2023-10-25:
MSYS2 terminal has /dev/shm but it does not help.

```
MINGW64 /c/work/avr/avrdude_test/avrdude_bin
$ ./avrdude_git -c serialupdi -P COM11 -p avr64dd32 -vvvv > /dev/shm/errorvvvv.log 2>&1

$ cat /dev/shm/errorvvvv.log

avrdude_git: Version 7.2-20231018 (481a91dd)
             Copyright the AVRDUDE authors;
             see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

             System wide configuration file is C:\work\avr\avrdude_test\avrdude_bin\avrdude.conf

             Using Port                    : COM11
             Using Programmer              : serialupdi
avrdude_git: opening serial port ...
avrdude_git: sending 1 bytes [0x00]
avrdude_git: send: . [00]
avrdude_git: recv: . [00]
             AVR Part                      : AVR64DD32
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               fuse0       wdtcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               fuse1       bodcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               fuse2       osccfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               fuse5       syscfg0     0     0     0    0 no          1    1      0     0     0 0x00 0x00
                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               fuse6       syscfg1     0     0     0    0 no          1    1      0     0     0 0x00 0x00
                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               fuse7       codesize    0     0     0    0 no          1    1      0     0     0 0x00 0x00
                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               fuse8       bootsize    0     0     0    0 no          1    1      0     0     0 0x00 0x00
                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               fuses                   0     0     0    0 no         16   16      0     0     0 0x00 0x00
                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               lock                    0     0     0    0 no          4    1      0     0     0 0x00 0x00
                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               prodsig     sigrow      0     0     0    0 no        128  128      0     0     0 0x00 0x00
                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
[... cut ...]
avrdude_git: send: U [55] . [c8] Y [59]
avrdude_git: ser_recv(): programmer is not responding
avrdude_git: sending release reset request
avrdude_git: STCS 0x00 to address 0x08
avrdude_git: sending 3 bytes [0x55, 0xc8, 0x00]
avrdude_git: send: U [55] . [c8] . [00]
avrdude_git: ser_recv(): programmer is not responding
avrdude_git: STCS 0x0C to address 0x03
avrdude_git: sending 3 bytes [0x55, 0xc3, 0x0c]
avrdude_git: send: U [55] . [c3] . [0c]
avrdude_git: ser_recv(): programmer is not responding

avrdude_git done.  Thank you.
```

mcuee, 2023-10-25:
> > parse the atdf files to show that only the AVR EA and EB family AVRs are using NVM v3
> 
> Do you know how ATDF encodes this? `xml_grep` is useful
> 
> ```
> xml_grep -g --pretty_print indented --cond 'module[@name="NVMCTRL"]' *.atdf
> ```
> 
> reduces the amount to look at

Thanks for the tip. I tried to install xml_grep by using the command `cpan install XML::Twig` under MSYS2 but it failed. 

Guess it is easier use Linux or macOS for this task.

But I will try under Windows using Strawberry Perl.

mcuee, 2023-10-25:
> Thanks for the tip. I tried to install xml_grep by using the command `cpan install XML::Twig` under MSYS2 but it failed.
> 
> Guess it is easier use Linux or macOS for this task.
> 
> But I will try under Windows using Strawberry Perl.

Strawberry Perl has the module with the default installation, but it does not seem to work.

```
PS C:\work\avr\avrdude_test\atdf> xml_grep -g --pretty_print indented --cond 'module[@name="NVMCTRL"]' *.atdf
Couldn't open *.atdf:
Invalid argument at C:\Strawberry\perl\bin\xml_grep.bat line 149.
 at C:\Strawberry\perl\bin\xml_grep.bat line 149.
<!-- error parsing file '*.atdf' -->
```

dbuchwald, 2023-10-25:
Hello, I just wanted to apologise for not contributing to the investigation - I have a very tough week at work and no energy to look into it in my spare time.

@mcuee Thank you very much for trying to replicate the issue and for providing detailed instructions regarding build and testing of the code in question. I will look into as soon as I have the time and energy to do so. As for the actual resolution - looking at the logs I can see that the UPDI controller doesn't seem to be responding at all, so this might be tough nut to crack, but I will do my best to find out how to handle this.

mcuee, 2023-10-25:
@dbuchwald 

Take your time and thanks for your great contribution to avrdude project. We all do this at our spare time and paid job is always more important.

dbuchwald, 2023-10-26:
Guys, I need some help/guidance here. I downloaded the sleep.c module:

```c
/**
 * @file moderAVR_sleep.c
 * @author askn (K.Sato) multix.jp
 * @brief
 * @version 0.1
 * @date 2023-10-23
 *
 * @copyright Copyright (c) 2023
 *
 */

/* Note: Does not work with tinyAVR-0/1 and megaAVR-0 */
/*       Please run on tinyAVR-2 and AVR_Dx/Ex        */

/* Build with only AVR-LIBC header files */
#include <avr/io.h>
#include <avr/sleep.h>
#include <util/atomic.h>

int main (void) {

  /*
   * Set F_CPU to 512Hz
   */

  #if defined(CLKCTRL_TIMEBASE_gm)
    /* This CLKCTRL_TIMEBASE_gm symbol is currently only in AVR_EA/EB */
    #define EVSYS_CHANNEL0_CCL_LUT0_gc EVSYS_CHANNEL_CCL_LUT0_gc
    #define CLKSEL_OSC32K_gc           CLKCTRL_CLKSEL_OSC32K_gc
    #define CLKCTRL_PDIV_64X_gc        CLKCTRL_PDIV_DIV64_gc
  #elif (LOCKBITS_SIZE == 1)
    /* Symbol alias for tinyAVR-2 */
    #define CLKSEL_OSC32K_gc           CLKCTRL_CLKSEL_XOSC32K_gc
  #endif

  _PROTECTED_WRITE(CLKCTRL_MCLKCTRLA, CLKSEL_OSC32K_gc);
  _PROTECTED_WRITE(CLKCTRL_MCLKCTRLB, CLKCTRL_PDIV_64X_gc | CLKCTRL_PEN_bm);

  /*
   * Capture PWM WOB0 output with LUT0 input
   */

  CCL_TRUTH0 = CCL_TRUTH_1_bm;
  CCL_LUT0CTRLB = CCL_INSEL0_TCB0_gc;               // <-- in  WOB0
  CCL_LUT0CTRLA = CCL_ENABLE_bm;                    // --> out LUT0_OUT
  CCL_CTRLA = CCL_ENABLE_bm;

  /*
   * Copy LUT0_OUT output to EVOUTA_ALT1 input
   * Connect the LED to PA7
   */

  PORTMUX_EVSYSROUTEA = PORTMUX_EVOUTA_ALT1_gc;     // --> in  EVOUTA_ALT1 (PA7)
  EVSYS_CHANNEL0 = EVSYS_CHANNEL0_CCL_LUT0_gc;      // <-- out LUT0_OUT
  EVSYS_USEREVSYSEVOUTA = EVSYS_USER_CHANNEL0_gc;   // Binding

  /*
   * Set the PWM (WOB0) output period to 1Hz
   */

  TCB0_CCMP = ((127) << 8) | 255;                   // PWM duty cycle 50:50
  TCB0_CTRLB = TCB_CNTMODE_PWM8_gc;                 // --> out WOB0
  TCB0_CTRLA = TCB_ENABLE_bm | TCB_RUNSTDBY_bm |    // Runs TCB0 even if the CPU is STANDBY
               TCB_CLKSEL_DIV2_gc;                  // Runs TCB0 counter on 1/2 of F_CPU

  /*
   * Sleeps the CPU in STANDBY mode
   */

  set_sleep_mode(SLEEP_MODE_STANDBY);
  sleep_enable();
  sleep_cpu();

  /* Since there are no interrupts, this will never be executed again. */
}

// end of code
```

as well as the makefile:

```makefile
DEVICE=avr128db64
CFLAGS ?= -Os -DF_APU=512UL -mmcu=${DEVICE} -I/usr/local/avr/include
LDFLAGS ?= -mmcu=${DEVICE} -L/usr/local/avr/lib/avrxmega4

all: sleep.hex

sleep.hex: sleep.elf
        avr-objcopy -O ihex sleep.elf sleep.hex

sleep.elf: sleep.o
        avr-gcc $(LDFLAGS) -o $@ $^

sleep.o: sleep.c
        avr-gcc $(CFLAGS) -c -o $@ $<

clean: FRC
        rm -f sleep.elf sleep.hex sleep.o

FRC:
```

Since I'm using Fedora 38 for development, I had to build avr-libc from sources. The one that is provided in fedora packages is 2.0, which doesn't support Dx devices.

I have also downloaded the latest avr-libc from here https://github.com/avrdudes/avr-libc/tree/main and built it according to instructions in the INSTALL file.

Now, when I try to build the hex file I get:

```
make clean all
rm -f sleep.elf sleep.hex sleep.o
avr-gcc -Os -DF_APU=512UL -mmcu=avr128db64 -I/usr/local/avr/include -c -o sleep.o sleep.c
In file included from /usr/local/avr/include/avr/io.h:759,
                 from sleep.c:16:
/usr/local/avr/include/avr/lock.h:234: warning: "LOCKBITS_DEFAULT" redefined
  234 | #define LOCKBITS_DEFAULT (0xFF)
      | 
In file included from /usr/local/avr/include/avr/io.h:709:
/usr/local/avr/include/avr/ioavr128db64.h:8273: note: this is the location of the previous definition
 8273 | #define LOCKBITS_DEFAULT  (0x5CC5C55C)
      | 
avr-gcc -mmcu=avr128db64 -L/usr/local/avr/lib/avrxmega4 -o sleep.elf sleep.o
/usr/lib/gcc/avr/13.2.0/../../../../avr/bin/ld: cannot find crtavr128db64.o: No such file or directory
collect2: error: ld returned 1 exit status
make: *** [makefile:11: sleep.elf] Error 1
```

This is weird, because:
```
ls -l /usr/local/avr/lib/avrxmega4/crtavr128db64.o
-rw-r--r--. 1 root root 69132 Oct 26 08:37 /usr/local/avr/lib/avrxmega4/crtavr128db64.o
```

What also surprises me is that with avr-libc build from sources I have to provide this weird linker path: **-L/usr/local/avr/lib/avrxmega4** 

I tried also building on MacOS, running into completely different set of issues there. 

And the worst part is that I have had it working just fine, the whole toolchain, and now it seems I have two computers where I can't build a single hex file and very little motivation to actually keep going with it...

mcuee, 2023-10-26:
@dbuchwald 

FYI I use the prebuild toolchain from @SpenceKonde. You can give it a try.
https://spencekondetoolchains.s3.amazonaws.com/

For Windows I use the one mentioned in DxCore README.
https://github.com/SpenceKonde/DxCore/blob/master/megaavr/bootloaders/optiboot_dx/README.md
https://spencekondetoolchains.s3.amazonaws.com/avr-gcc-7.3.0-atmel3.6.1-azduino6-i686-w64-mingw32.tar.bz2

For Linux, you can use the following.
https://spencekondetoolchains.s3.amazonaws.com/avr-gcc-7.3.0-atmel3.6.1-azduino6-x86_64-pc-linux-gnu.tar.bz2

For macOS, you can use the following.
https://spencekondetoolchains.s3.amazonaws.com/avr-gcc-7.3.0-atmel3.6.1-azduino6-x86_64-apple-darwin14.tar.bz2

I think he as built some new versions which you can also try.

mcuee, 2023-10-26:
@dbuchwald 

If you want to build your own toolchain, you can also use the one from Microchip as the base.
https://www.microchip.com/en-us/tools-resources/develop/microchip-studio/gcc-compilers

And then you need to download the relevant DFP packs from Microchip.
https://packs.download.microchip.com/

dbuchwald, 2023-10-26:
@askn37 

Hello, I tried replicating the sleep issue, but I think I might be doing something wrong. I built the `sleep.c` file for my Curiosity Nano AVR128DA48 board, uploaded it using the onboard programmer (pkobn_updi), and then tried contacting the device using serialupdi programmer - and it seems to be responding fine, it just reports that the device was in SLEEP mode, but works as if everything was OK.

What am I doing wrong here?

```
avrdude -c pkobn_updi -p avr128da48 -U flash:w:sleep.hex
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9708 (probably avr128da48)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip

avrdude: processing -U flash:w:sleep.hex:i
avrdude: reading input file sleep.hex for flash
         with 364 bytes in 1 section within [0, 0x16b]
         using 1 page and 148 pad bytes
avrdude: writing 364 bytes flash ...
Writing | ################################################## | 100% 0.25 s
avrdude: 364 bytes of flash written
avrdude: verifying flash memory against sleep.hex
Reading | ################################################## | 100% 0.09 s
avrdude: 364 bytes of flash verified

avrdude done.  Thank you.

❯ avrdude -c serialupdi -P /dev/cu.usbserial-1420 -p avr128da48 -v

avrdude: Version 7.2-20231024 (d6c61f5f)
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is /usr/local/etc/avrdude.conf
         User configuration file is /Users/dawid/.avrduderc
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/cu.usbserial-1420
         Using Programmer              : serialupdi
         AVR Part                      : AVR128DA48
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           fuse0       wdtcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse1       bodcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse2       osccfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse5       syscfg0     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse6       syscfg1     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse7       codesize    0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse8       bootsize    0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuses                   0     0     0    0 no         16   16      0     0     0 0x00 0x00
           lock                    0     0     0    0 no          4    1      0     0     0 0x00 0x00
           prodsig     sigrow      0     0     0    0 no        128  128      0     0     0 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           tempsense               0     0     0    0 no          4    1      0     0     0 0x00 0x00
           sernum                  0     0     0    0 no         16    1      0     0     0 0x00 0x00
           userrow     usersig     0     0     0    0 no         32   32      0     0     0 0x00 0x00
           data                    0     0     0    0 no          0    1      0     0     0 0x00 0x00
           io                      0     0     0    0 no       4160    1      0     0     0 0x00 0x00
           sib                     0     0     0    0 no         32    1      0     0     0 0x00 0x00
           eeprom                  0     0     0    0 no        512    1      0     0     0 0x00 0x00
           flash                   0     0     0    0 no     131072  512      0     0     0 0x00 0x00

         Programmer Type : serialupdi
         Description     : SerialUPDI
avrdude: device is in SLEEP mode
avrdude: NVM type 2: 24-bit, word oriented write
avrdude: Chip silicon revision: 1.6
avrdude: entering NVM programming mode
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9708 (probably avr128da48)
avrdude: leaving NVM programming mode

avrdude done.  Thank you.
```

mcuee, 2023-10-26:
@dbuchwald 

Interestingly I also can not reproduce the issue under Linux for the AVR64DD32 board (which I have issues under Windows). There are no issues with `-v` or `-vvvv`.

```
mcuee@UbuntuSwift3 ~/build/avr/avrdude_bin $ make
avr-gcc -Os -DF_APU=512UL -mmcu=avr64dd32 -c -o sleep.o sleep.c
avr-gcc -mmcu=avr64dd32 -o sleep.elf sleep.o
avr-objcopy -O ihex sleep.elf sleep.hex
mcuee@UbuntuSwift3 ~/build/avr/avrdude_bin $ ./avrdude_git -c serialupdi -P /dev/ttyUSB0 -p avr64dd32 -U sleep.hex 
avrdude_git: AVR device initialized and ready to accept instructions
avrdude_git: device signature = 0x1e961a (probably avr64dd32)
avrdude_git: Note: flash memory has been specified, an erase cycle will be performed.
             To disable this feature, specify the -D option.
avrdude_git: erasing chip

avrdude_git: processing -U flash:w:sleep.hex:i
avrdude_git: reading input file sleep.hex for flash
             with 276 bytes in 1 section within [0, 0x113]
             using 1 page and 236 pad bytes
avrdude_git: writing 276 bytes flash ...
Writing | ################################################## | 100% 0.17 s 
avrdude_git: 276 bytes of flash written
avrdude_git: verifying flash memory against sleep.hex
Reading | ################################################## | 100% 0.07 s 
avrdude_git: 276 bytes of flash verified

avrdude_git done.  Thank you.

mcuee@UbuntuSwift3 ~/build/avr/avrdude_bin $ ./avrdude_git -c serialupdi -P /dev/ttyUSB0 -p avr64dd32 -v

avrdude_git: Version 7.2-20231024 (d6c61f5f)
             Copyright the AVRDUDE authors;
             see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

             System wide configuration file is /home/mcuee/build/avr/avrdude_bin/avrdude.conf
             User configuration file is /home/mcuee/.avrduderc
             User configuration file does not exist or is not a regular file, skipping

             Using Port                    : /dev/ttyUSB0
             Using Programmer              : serialupdi
             AVR Part                      : AVR64DD32
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               fuse0       wdtcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuse1       bodcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuse2       osccfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuse5       syscfg0     0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuse6       syscfg1     0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuse7       codesize    0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuse8       bootsize    0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuses                   0     0     0    0 no         16   16      0     0     0 0x00 0x00
               lock                    0     0     0    0 no          4    1      0     0     0 0x00 0x00
               prodsig     sigrow      0     0     0    0 no        128  128      0     0     0 0x00 0x00
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
               tempsense               0     0     0    0 no          4    1      0     0     0 0x00 0x00
               sernum                  0     0     0    0 no         16    1      0     0     0 0x00 0x00
               userrow     usersig     0     0     0    0 no         32   32      0     0     0 0x00 0x00
               data                    0     0     0    0 no          0    1      0     0     0 0x00 0x00
               io                      0     0     0    0 no       4160    1      0     0     0 0x00 0x00
               sib                     0     0     0    0 no         32    1      0     0     0 0x00 0x00
               eeprom                  0     0     0    0 no        256    1      0     0     0 0x00 0x00
               flash                   0     0     0    0 no      65536  512      0     0     0 0x00 0x00
[... 252 lines cut ...]
avrdude_git: recv: U [55] . [08] . [02] . [11] . [00] 
avrdude_git: recv: . [1a] 
avrdude_git: received 1 bytes [0x1a]
Reading | ################################################## | 100% 0.02 s 
avrdude_git: device signature = 0x1e961a (probably avr64dd32)
avrdude_git: leaving NVM programming mode
avrdude_git: sending reset request
avrdude_git: STCS 0x59 to address 0x08
avrdude_git: sending 3 bytes [0x55, 0xc8, 0x59]
avrdude_git: send: U [55] . [c8] Y [59] 
avrdude_git: recv: U [55] . [c8] Y [59] 
avrdude_git: sending release reset request
avrdude_git: STCS 0x00 to address 0x08
avrdude_git: sending 3 bytes [0x55, 0xc8, 0x00]
avrdude_git: send: U [55] . [c8] . [00] 
avrdude_git: recv: U [55] . [c8] . [00] 
avrdude_git: STCS 0x0C to address 0x03
avrdude_git: sending 3 bytes [0x55, 0xc3, 0x0c]
avrdude_git: send: U [55] . [c3] . [0c] 
avrdude_git: recv: U [55] . [c3] . [0c] 

avrdude_git done.  Thank you.
```

</details>

mcuee, 2023-10-27:
I can not reproduce the issue under my Mac Mini M1 either.

<details>
<summary> Click for the run log, including `-v` and `-vvvv`</summary>

```
mcuee@mcuees-Mac-mini avrdude_bin % ./avrdude_git -c serialupdi -P /dev/tty.usbserial-22430 -p avr64dd32 
avrdude_git: AVR device initialized and ready to accept instructions
avrdude_git: device signature = 0x1e961a (probably avr64dd32)

avrdude_git done.  Thank you.

mcuee@mcuees-Mac-mini avrdude_bin % ./avrdude_git -c serialupdi -P /dev/tty.usbserial-22430 -p avr64dd32 -v

avrdude_git: Version 7.1-20230108
             Copyright the AVRDUDE authors;
             see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

             System wide configuration file is /Users/mcuee/build/avr/avrdude_test/avrdude_bin/avrdude.conf
             User configuration file is /Users/mcuee/.avrduderc
             User configuration file does not exist or is not a regular file, skipping

             Using Port                    : /dev/tty.usbserial-22430
             Using Programmer              : serialupdi
             AVR Part                      : AVR64DD32
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               fuse0       wdtcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuse1       bodcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuse2       osccfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuse5       syscfg0     0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuse6       syscfg1     0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuse7       codesize    0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuse8       bootsize    0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuses                   0     0     0    0 no         16   16      0     0     0 0x00 0x00
               lock                    0     0     0    0 no          4    1      0     0     0 0x00 0x00
               prodsig     sigrow      0     0     0    0 no        128  128      0     0     0 0x00 0x00
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
               tempsense               0     0     0    0 no          4    1      0     0     0 0x00 0x00
               sernum                  0     0     0    0 no         16    1      0     0     0 0x00 0x00
               userrow     usersig     0     0     0    0 no         32   32      0     0     0 0x00 0x00
               data                    0     0     0    0 no          0    1      0     0     0 0x00 0x00
               io                      0     0     0    0 no       4160    1      0     0     0 0x00 0x00
               sib                     0     0     0    0 no         32    1      0     0     0 0x00 0x00
               eeprom                  0     0     0    0 no        256    1      0     0     0 0x00 0x00
               flash                   0     0     0    0 no      65536  512      0     0     0 0x00 0x00

             Programmer Type : serialupdi
             Description     : SerialUPDI
avrdude_git: device is in SLEEP mode
avrdude_git: NVM type 2: 24-bit, word oriented write
avrdude_git: Chip silicon revision: 1.3
avrdude_git: entering NVM programming mode
avrdude_git: AVR device initialized and ready to accept instructions
avrdude_git: device signature = 0x1e961a (probably avr64dd32)
avrdude_git: leaving NVM programming mode

avrdude_git done.  Thank you.

mcuee@mcuees-Mac-mini avrdude_bin % ./avrdude_git -c serialupdi -P /dev/tty.usbserial-22430 -p avr64dd32 -vvvv

avrdude_git: Version 7.1-20230108
             Copyright the AVRDUDE authors;
[... 230 lines cut ...]
avrdude_git: recv: U [55] . [08] . [02] . [11] . [00] 
avrdude_git: recv: . [1a] 
avrdude_git: received 1 bytes [0x1a]
Reading | ################################################## | 100% 0.02 s 
avrdude_git: device signature = 0x1e961a (probably avr64dd32)
avrdude_git: leaving NVM programming mode
avrdude_git: sending reset request
avrdude_git: STCS 0x59 to address 0x08
avrdude_git: sending 3 bytes [0x55, 0xc8, 0x59]
avrdude_git: send: U [55] . [c8] Y [59] 
avrdude_git: recv: U [55] . [c8] Y [59] 
avrdude_git: sending release reset request
avrdude_git: STCS 0x00 to address 0x08
avrdude_git: sending 3 bytes [0x55, 0xc8, 0x00]
avrdude_git: send: U [55] . [c8] . [00] 
avrdude_git: recv: U [55] . [c8] . [00] 
avrdude_git: STCS 0x0C to address 0x03
avrdude_git: sending 3 bytes [0x55, 0xc3, 0x0c]
avrdude_git: send: U [55] . [c3] . [0c] 
avrdude_git: recv: U [55] . [c3] . [0c] 

avrdude_git done.  Thank you.
```

</details>

mcuee, 2023-10-27:
So it may seems to be a timing issue under Windows, at least for my side.

@askn37 
Please give the latest git main a try to see if you can reproduce the Sleep issue or not under macOS. Thanks.

mcuee, 2023-10-27:
@stefanrueger 

Just wondering if this is related to #1249. Thanks.
* https://github.com/avrdudes/avrdude/issues/1249

askn37, 2023-10-27:
[image]

Since I'm an AVR chip collector, I've been experimenting with about 20 models of UPDI AVR bare chips from different family and revisions, along with several different SerialUPDI implementations. Unfortunately, I don't have the recalled chip.

One implementation that showed interesting behavior was using DTR to toggle UART passthrough and a combination of RTS and capacitors to perform a hard reset. This circuit is used in conjunction with the Arduino IDE's serial plotter.

This type cannot use SerialUPDI without the "-xrtsdtr=high" option. Also, the target AVR wakes up more slowly than circuits that don't use it.

When used on the latest main branch (NVMCTRL improved), it stops with "error: Reading chip silicon revision failed". However, disconnecting the RTS reset fixes it without any problems. So it's clear that the capacitor in the reset circuit was interfering. Either the design is wrong or avrdude is impatient.

Therefore, I created a build with "Silicon Check" commented out and experimented with RTS reset, and this time it proceeded normally even though "error: key was not accepted" was displayed. What did you say?

```
![IMG_3608](https://github.com/avrdudes/avrdude/assets/1407370/53d2d4f5-918c-4f98-89b9-c0754bdc2f1d)

+./avrdude_test/build_darwin/src/avrdude -xrtsdtr=high -p avr64ea32 -c serialupdi -P /dev/cu.usbserial-230 -U userrow:r:avr64ea32/USERROW.hex:i -v

avrdude: Version 7.2-20231024 (d6c61f5f)
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is /Users/askn/Arduino/UPDI_TEST/avrdude_test/build_darwin/src/avrdude.conf
         User configuration file is /Users/askn/.avrduderc

         Using Port                    : /dev/cu.usbserial-230
         Using Programmer              : serialupdi
         AVR Part                      : AVR64EA32
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           fuse0       wdtcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse1       bodcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse2       osccfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse5       syscfg0     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse6       syscfg1     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse7       codesize    0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse8       bootsize    0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuses                   0     0     0    0 no         16   16      0     0     0 0x00 0x00
           lock                    0     0     0    0 no          4    1      0     0     0 0x00 0x00
           prodsig     sigrow      0     0     0    0 no        128  128      0     0     0 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           tempsense               0     0     0    0 no          4    1      0     0     0 0x00 0x00
           sernum                  0     0     0    0 no         16    1      0     0     0 0x00 0x00
           userrow     usersig     0     0     0    0 no         64   64      0     0     0 0x00 0x00
           data                    0     0     0    0 no          0    1      0     0     0 0x00 0x00
           io                      0     0     0    0 no       4160    1      0     0     0 0x00 0x00
           sib                     0     0     0    0 no         32    1      0     0     0 0x00 0x00
           eeprom                  0     0     0    0 no        512    8      0     0     0 0x00 0x00
           flash                   0     0     0    0 no      65536  128      0     0     0 0x00 0x00

         Programmer Type : serialupdi
         Description     : SerialUPDI
avrdude: forcing serial DTR/RTS handshake lines HIGH
avrdude: NVM type 3: 24-bit, page oriented
avrdude: entering NVM programming mode
avrdude serialupdi_enter_progmode() error: key was not accepted
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e961f (probably avr64ea32)

avrdude: processing -U userrow:r:avr64ea32/USERROW.hex:i
avrdude: reading userrow/usersig memory ...
Reading | ################################################## | 100% 0.02s
avrdude: writing output file avr64ea32/USERROW.hex
avrdude: leaving NVM programming mode
avrdude: releasing DTR/RTS handshake lines

avrdude done.  Thank you.
```

This indicates that the UPDI layer has not yet fully woken up when reading "UPDI_CS_ASI_KEY_STATUS" after sending "UPDI_SIB_128", or that key activation took longer than usual. I'm sure avrdude is impatient. There probably aren't any other concerns. thank you.

mcuee, 2023-10-27:
@askn37 

Thanks a lot for sharing your insight. Nice AVR boards!

Please help to post the full debug log with `-vvvv` and maybe that can help @dbuchwald and maybe @stefanrueger to debug the issue to make avrdude more robust.

mcuee, 2023-10-27:
@askn37 

BTW, I see that you have alternative Arduino Core in your repos (in Japanese)
https://github.com/askn37
https://github.com/askn37/multix-zinnia-sdk-modernAVR
https://github.com/askn37/multix-zinnia-sdk-megaAVR

Just wondering if you compare your Cores with MegaCoreX, megaTinyCore and DxCore. Thanks.
https://github.com/MCUdude/MegaCoreX
https://github.com/SpenceKonde/megaTinyCore
https://github.com/SpenceKonde/DxCore

mcuee, 2023-10-27:
@stefanrueger and @dbuchwald 

Looks like there is a regression under Windows in git main from avrdude 7.2 release, under Windows.

You can see I have no issues with avrdude 7.2 release.

```
PS > .\avrdude72 -C .\avrdude72.conf -c serialupdi -P COM16 -p avr64dd32 -vvvv

avrdude72: Version 7.2
           Copyright the AVRDUDE authors;
           see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

           System wide configuration file is C:\work\avr\avrdude_test\avrdude_bin\avrdude72.conf

           Using Port                    : COM16
           Using Programmer              : serialupdi
avrdude72: opening serial port ...
avrdude72: sending 1 bytes [0x00]
avrdude72: send: . [00]
avrdude72: recv: . [00]
           AVR Part                      : AVR64DD32
           RESET disposition             : dedicated
           RETRY pulse                   : SCK
           Serial program mode           : yes
           Parallel program mode         : yes
           Memory Detail                 :

                                             Block Poll               Page                       Polled
             Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
             ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
             fuse0       wdtcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
                                             Block Poll               Page                       Polled
             Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
             ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
             fuse1       bodcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
                                             Block Poll               Page                       Polled
             Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
             ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
             fuse2       osccfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
                                             Block Poll               Page                       Polled
             Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
             ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
             fuse5       syscfg0     0     0     0    0 no          1    1      0     0     0 0x00 0x00
                                             Block Poll               Page                       Polled
             Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
             ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
             fuse6       syscfg1     0     0     0    0 no          1    1      0     0     0 0x00 0x00
                                             Block Poll               Page                       Polled
             Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
             ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
             fuse7       codesize    0     0     0    0 no          1    1      0     0     0 0x00 0x00
                                             Block Poll               Page                       Polled
             Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
             ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
             fuse8       bootsize    0     0     0    0 no          1    1      0     0     0 0x00 0x00
                                             Block Poll               Page                       Polled
             Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
             ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
             fuses                   0     0     0    0 no          9   16      0     0     0 0x00 0x00
                                             Block Poll               Page                       Polled
             Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
             ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
             lock                    0     0     0    0 no          4    1      0     0     0 0x00 0x00
                                             Block Poll               Page                       Polled
             Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
             ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
             tempsense               0     0     0    0 no          2    1      0     0     0 0x00 0x00
                                             Block Poll               Page                       Polled
             Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
[... 145 lines cut ...]
avrdude72: sending 5 bytes [0x55, 0x08, 0x02, 0x11, 0x00]
[... cut ...]
avrdude72: sending 3 bytes [0x55, 0xc8, 0x59]
avrdude72: send: U [55] . [c8] Y [59]
avrdude72: recv: U [55] . [c8] Y [59]
avrdude72: sending release reset request
avrdude72: STCS 0x00 to address 0x08
avrdude72: sending 3 bytes [0x55, 0xc8, 0x00]
avrdude72: send: U [55] . [c8] . [00]
avrdude72: recv: U [55] . [c8] . [00]
avrdude72: STCS 0x0C to address 0x03
avrdude72: sending 3 bytes [0x55, 0xc3, 0x0c]
avrdude72: send: U [55] . [c3] . [0c]
avrdude72: recv: U [55] . [c3] . [0c]

avrdude72 done.  Thank you.
```

mcuee, 2023-10-27:
Looking at the histories, there are not many changes. I will try to find the commit which causes the regression.
https://github.com/avrdudes/avrdude/commits/main/src/serialupdi.c

The following commit is already causing problems: Added support for chip silicon revision number
https://github.com/avrdudes/avrdude/commit/ac760ff49064c5de53abed709057341ec4a78b64

mcuee, 2023-10-27:
Hmm, no, it seems to be a MinGW vs MSVC issue.

Official MSVC build is okay but not my MinGW build, even for avrdude 7.2 release version. I will check out git main later.

```
PS>.\avrdude72_mingw64 -C .\avrdude72.conf -c serialupdi -P COM16 -p avr64dd32 -v

avrdude72_mingw64: Version 7.2
...
avrdude72_mingw64 serialupdi_initialize() error: UPDI link initialization failed
avrdude72_mingw64 main() error: initialization failed, rc=-1
                  - double check the connections and try again
                  - use -b to set lower baud rate, e.g. -b 57600
                  - use -F to override this check
avrdude72_mingw64: leaving NVM programming mode

avrdude72_mingw64 done.  Thank you.

PS>.\avrdude72_msvc -C .\avrdude72.conf -c serialupdi -P COM16 -p avr64dd32 -v

avrdude72_msvc: Version 7.2
 ...
avrdude72_msvc: device is in SLEEP mode
avrdude72_msvc: NVM type 2: 24-bit, word oriented write
avrdude72_msvc: entering NVM programming mode
avrdude72_msvc: AVR device initialized and ready to accept instructions
avrdude72_msvc: device signature = 0x1e961a (probably avr64dd32)
avrdude72_msvc: leaving NVM programming mode

avrdude72_msvc done.  Thank you.
```

mcuee, 2023-10-27:
Same for git main, using the github action binary of git main (not my build). 

MSVC build is okay, but not MinGW build.
```
PS>pwd

Path
----
C:\work\avr\avrdude_test\avrdude_bin\github\avrdude-mingw-x86_64

PS>.\avrdude -c serialupdi -P COM16 -p avr64dd32 -v

avrdude: Version 7.2-20231024 (d6c61f5)
...
avrdude serialupdi_initialize() error: UPDI link initialization failed
avrdude main() error: initialization failed, rc=-1
        - double check the connections and try again
        - use -b to set lower baud rate, e.g. -b 57600
        - use -F to override this check
avrdude: leaving NVM programming mode

avrdude done.  Thank you.

PS>cd .\avrdude-msvc-x64\
PS>pwd

Path
----
C:\work\avr\avrdude_test\avrdude_bin\github\avrdude-msvc-x64

PS>.\avrdude -c serialupdi -P COM16 -p avr64dd32 -v

avrdude: Version 7.2-20231024 (d6c61f5)
...
avrdude: device is in SLEEP mode
avrdude: NVM type 2: 24-bit, word oriented write
avrdude: Chip silicon revision: 1.3
avrdude: entering NVM programming mode
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e961a (probably avr64dd32)
avrdude: leaving NVM programming mode

avrdude done.  Thank you.
```

mcuee, 2023-10-27:
Then I checked MSVC build:
1) avrdude 7.1 and avrdude 7.2 official MSVC binary works with both `-v` and `-vvvv`.
2) avrdude 7.0 official MSVC binary works with `-v`, `-vv` and `-vvv` but not `-vvvv`.

askn37, 2023-10-27:
@mcuee

> BTW, I see that you have alternative Arduino Core in your repos (in Japanese)
> https://github.com/askn37
> https://github.com/askn37/multix-zinnia-sdk-modernAVR [...]

Of course, the cross-check has been completed. The avrdude used when each project is set up with Arduino IDE can be obtained from the link below.

My project (multix-zinnia-sdk) is the only one that has 7.2 deployed as of today. This is because we are dealing with AVR_EA (NVMCTRLv3 chip). MCUdude supports all NVMCTRLv0 chips, so there is nothing to worry about. SpenceKonde (Dr.Azzy) uses "prog.py" and does not use avrdude's built-in SerialUPDI, so it is avrdude 6.3.

That's why it was my duty to launch this issue.

## https://github.com/askn37/

http://downloads.arduino.cc/tools/avrdude_7.2-arduino.1_Linux_ARMv6.tar.gz
http://downloads.arduino.cc/tools/avrdude_7.2-arduino.1_Linux_ARM64.tar.gz
http://downloads.arduino.cc/tools/avrdude_7.2-arduino.1_macOS_64bit.tar.gz
http://downloads.arduino.cc/tools/avrdude_7.2-arduino.1_Linux_64bit.tar.gz
http://downloads.arduino.cc/tools/avrdude_7.2-arduino.1_Linux_32bit.tar.gz
http://downloads.arduino.cc/tools/avrdude_7.2-arduino.1_Windows_32bit.tar.gz

## https://github.com/MCUdude/

http://downloads.arduino.cc/tools/avrdude_7.1-arduino.1_Linux_ARMv6.tar.gz
http://downloads.arduino.cc/tools/avrdude_7.1-arduino.1_Linux_ARM64.tar.gz
http://downloads.arduino.cc/tools/avrdude_7.1-arduino.1_macOS_64bit.tar.gz
http://downloads.arduino.cc/tools/avrdude_7.1-arduino.1_Linux_64bit.tar.gz
http://downloads.arduino.cc/tools/avrdude_7.1-arduino.1_Linux_32bit.tar.gz
http://downloads.arduino.cc/tools/avrdude_7.1-arduino.1_Windows_32bit.tar.gz

## https://github.com/SpenceKonde/

http://downloads.arduino.cc/tools/avrdude-6.3.0-arduino18-armhf-pc-linux-gnu.tar.bz2
http://downloads.arduino.cc/tools/avrdude-6.3.0-arduino18-aarch64-pc-linux-gnu.tar.bz2
http://downloads.arduino.cc/tools/avrdude-6.3.0-arduino18-x86_64-apple-darwin12.tar.bz2
http://downloads.arduino.cc/tools/avrdude-6.3.0-arduino18-x86_64-pc-linux-gnu.tar.bz2
http://downloads.arduino.cc/tools/avrdude-6.3.0-arduino18-i686-w64-mingw32.zip

mcuee, 2023-10-27:
@askn37 

Thanks for adopting avrdude 7.2 release!

@MCUdude will for sure adopt avrdude 7.2 for his MegaCoreX for his next release. The current 1.1.2 release was on 2023-05-01 so it was before avrdude 7.2 release.

As for DxCore and megaTinyCore, there are issues created to adopt avrdude 7.2. I think @SpenceConde will adopt avrdude 7.2 release in the near future. He actually wanted to adopt earlier but there were no suitable binaries at that time.
* https://github.com/SpenceKonde/DxCore/issues/330
* https://github.com/SpenceKonde/megaTinyCore/issues/919
* https://github.com/avrdudes/avrdude/discussions/1477

askn37, 2023-10-27:
@mcuee 

> The following commit is already causing problems: Added support for chip silicon revision number
https://github.com/avrdudes/avrdude/commit/ac760ff49064c5de53abed709057341ec4a78b64

I believe the code in question is placed in "static int serialupdi_initialize" because it requires "const AVRPART *p". But in reality, if you could just "enter NVM programming mode" and run it, you would have been fine. However, the "static int serialupdi_enter_progmode" that does this does not take "const AVRPART *p" as an argument. Therefore, this change still requires consideration.

The current AVR64EA32-I/PT (TQFP Revision B1) may have unknown errata. After writing the ERASE/NVMPROG/USERROW key, the UPDI_CS_ASI_KEY_STATUS register may be unreadable or unchanged. This is especially true in high voltage mode. Is there a problem with UPDI clock synchronization? But I don't have an oscilloscope that can decode his UPDI communications, so I can't find any evidence yet. There has been no response from Microchip either.

mcuee, 2023-10-27:
Errata for AVR EA family. Indeed there are issues related to UPDI and NVM programming.
https://ww1.microchip.com/downloads/aemDocuments/documents/MCU08/ProductDocuments/Errata/AVR64EA-28-32-48-SilConErrataClarif-DS80001048.pdf

**Device** 
2.2.1. NVM Programming Does Not Work Below 2.7V
2.2.2. Reduced Flash Endurance for VDD Below BODLEVEL3
2.2.3. Write Operation Lost if Consecutive Writes to Specific Address Spaces

**CRCSCAN** 
2.3.1. Running CRC Scan on Part of The Flash is Non-Functional X

**NVMCTRL** 
2.4.1. Flash Multi Page Erase Non-Functional from UPDI
2.4.2. Flash-Self Programming Failing When Flash Read During Programming

**USART** 
2.5.1. Receiver Non-Functional after Detection of Inconsistent Synchronization Field

mcuee, 2023-10-27:
Further debug under Windows: `-b 57600` works with MinGW build.

<details>
<summary> -b 57600 works for MinGW build with -v or -vvvv</summary>

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git -c serialupdi -P COM11 -p avr64dd32 -v -b 57600

avrdude_git: Version 7.2-20231018 (481a91dd)
             Copyright the AVRDUDE authors;
             see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

             System wide configuration file is C:\work\avr\avrdude_test\avrdude_bin\avrdude.conf

             Using Port                    : COM11
             Using Programmer              : serialupdi
             Overriding Baud Rate          : 57600
             AVR Part                      : AVR64DD32
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               fuse0       wdtcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuse1       bodcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuse2       osccfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuse5       syscfg0     0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuse6       syscfg1     0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuse7       codesize    0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuse8       bootsize    0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuses                   0     0     0    0 no         16   16      0     0     0 0x00 0x00
               lock                    0     0     0    0 no          4    1      0     0     0 0x00 0x00
               prodsig     sigrow      0     0     0    0 no        128  128      0     0     0 0x00 0x00
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
               tempsense               0     0     0    0 no          4    1      0     0     0 0x00 0x00
               sernum                  0     0     0    0 no         16    1      0     0     0 0x00 0x00
               userrow     usersig     0     0     0    0 no         32   32      0     0     0 0x00 0x00
               data                    0     0     0    0 no          0    1      0     0     0 0x00 0x00
               io                      0     0     0    0 no       4160    1      0     0     0 0x00 0x00
               sib                     0     0     0    0 no         32    1      0     0     0 0x00 0x00
               eeprom                  0     0     0    0 no        256    1      0     0     0 0x00 0x00
               flash                   0     0     0    0 no      65536  512      0     0     0 0x00 0x00

             Programmer Type : serialupdi
             Description     : SerialUPDI
avrdude_git: device is in SLEEP mode
avrdude_git: NVM type 2: 24-bit, word oriented write
avrdude_git: Chip silicon revision: 1.3
avrdude_git: entering NVM programming mode
avrdude_git: AVR device initialized and ready to accept instructions
avrdude_git: device signature = 0x1e961a (probably avr64dd32)
avrdude_git: leaving NVM programming mode

avrdude_git done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git -c serialupdi -P COM11 -p avr64dd32 -vvvv -b 57600

avrdude_git: Version 7.2-20231018 (481a91dd)
             Copyright the AVRDUDE authors;
             see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

             System wide configuration file is C:\work\avr\avrdude_test\avrdude_bin\avrdude.conf

             Using Port                    : COM11
             Using Programmer              : serialupdi
             Overriding Baud Rate          : 57600
[... 243 lines cut ...]
avrdude_git: recv: U [55] . [08] . [02] . [11] . [00]
avrdude_git: recv: . [1a]
avrdude_git: received 1 bytes [0x1a]
Reading | ################################################## | 100% 0.05 s
avrdude_git: device signature = 0x1e961a (probably avr64dd32)
avrdude_git: leaving NVM programming mode
avrdude_git: sending reset request
avrdude_git: STCS 0x59 to address 0x08
avrdude_git: sending 3 bytes [0x55, 0xc8, 0x59]
avrdude_git: send: U [55] . [c8] Y [59]
avrdude_git: recv: U [55] . [c8] Y [59]
avrdude_git: sending release reset request
avrdude_git: STCS 0x00 to address 0x08
avrdude_git: sending 3 bytes [0x55, 0xc8, 0x00]
avrdude_git: send: U [55] . [c8] . [00]
avrdude_git: recv: U [55] . [c8] . [00]
avrdude_git: STCS 0x0C to address 0x03
avrdude_git: sending 3 bytes [0x55, 0xc3, 0x0c]
avrdude_git: send: U [55] . [c3] . [0c]
avrdude_git: recv: U [55] . [c3] . [0c]

avrdude_git done.  Thank you.
```

</details>

stefanrueger, 2023-10-27:
>  or avrdude is impatient

@askn37 Is there a good place in `serialupdi_initialize()` to put an `usleep()` to solve this problem and how long for?

Also, I wonder whether some OS open() calls twiddle RTS while other OS don't touch RTS when opening a serial connection.

[36 more replies]

closed 2023-10-28 by mcuee