avrdudes/avrdude #1536: TPI using jtag3: which memories can be read/written?

stefanrueger, 2023-10-26:
The following code 
https://github.com/avrdudes/avrdude/blob/d6c61f5fab4d0d061fb85233acee6ecaa85dd9c0/src/jtag3.c#L2779-L2795
in that it does not match the full range of memory types available under TPI
https://github.com/avrdudes/avrdude/blob/d6c61f5fab4d0d061fb85233acee6ecaa85dd9c0/src/jtag3_private.h#L348-L355

I realise TPI parts might not have EEPROM, but they do have a calibration byte. So is that code above correct?

Can someone try to program TPI parts via jtag3 and read the `calibration` memory. I suspect it'll read the lockbits instead. But who knowns? Also `signature` looks fishy...

MCUdude, 2023-10-26:
ATtiny104 using the Xplained Mini TPI:

```
$ ./avrdude -cxplainedmini_tpi -pattiny104 -qq -t
avrdude> part
AVR Part                      : ATtiny104
RESET disposition             : dedicated
RETRY pulse                   : SCK
Serial program mode           : yes
Parallel program mode         : yes
Memory Detail                 :

                                  Block Poll               Page                       Polled
  Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
  ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
  fuse                    0     0     4    0 no          1   16      0     0     0 0x00 0x00
  lockbits                0     0     0    0 no          1   16      0     0     0 0x00 0x00
  signature               0     0     0    0 no          3   16      0     0     0 0x00 0x00
  calibration             0     0     0    0 no          1   16      0     0     0 0x00 0x00
  flash                   0     0   128    0 no       1024   16      0     0     0 0x00 0x00
  prodsig     sigrow      0     0     0    0 no         16   16      0     0     0 0x00 0x00
  io                      0     0     0    0 no         64    1      0     0     0 0x00 0x00
avrdude> read fuse
0000  fd                                                |.               |
avrdude> read lockbits
0000  ff                                                |.               |
avrdude> read signature
0000  1e 90 0b                                          |..              |
avrdude> read calibration
0000  49                                                |I               |
avrdude> read prodsig
0000  1e 90 0b ff ff ff 35 59  30 38 34 34 2f 18 28 38  |.. ...5Y0844/.(8|
avrdude> read io 0 64
0000  22 00 00 00 00 00 00 00  00 00 00 00 06 00 20 00  |"............. .|
0010  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
0020  c0 44 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |.D..............|
0030  00 00 00 00 00 00 03 00  00 49 00 0b 00 00 00 00  |.........I. ....|
avrdude> quit
```

MCUdude, 2023-10-26:
ATtiny13 using PICkit4 in TPI mode:

```
$ ./avrdude -cpickit4_tpi -pattiny10 -qq -t
avrdude> part
AVR Part                      : ATtiny10
RESET disposition             : dedicated
RETRY pulse                   : SCK
Serial program mode           : yes
Parallel program mode         : yes
Memory Detail                 :

                                  Block Poll               Page                       Polled
  Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
  ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
  fuse                    0     0     4    0 no          1   16      0     0     0 0x00 0x00
  lockbits                0     0     0    0 no          1   16      0     0     0 0x00 0x00
  signature               0     0     0    0 no          3   16      0     0     0 0x00 0x00
  calibration             0     0     0    0 no          1   16      0     0     0 0x00 0x00
  flash                   0     0   128    0 no       1024   16      0     0     0 0x00 0x00
  io                      0     0     0    0 no         64    1      0     0     0 0x00 0x00
avrdude> read fuse
0000  fd                                                |.               |
avrdude> read lockbits
0000  ff                                                |.               |
avrdude> read signature
0000  1e 90 03                                          |...             |
avrdude> read calibration
0000  7f                                                |.               |
avrdude> read io
0000  01 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
0010  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
0020  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
0030  00 00 00 00 00 00 03 00  00 7f 00 0b 00 00 00 00  |........... ....|
avrdude> q
```

stefanrueger, 2023-10-26:
Ahh; got it. The offset in the avrdude.conf file is used to discriminate the memories, not the `mtype`. OK, thanks for checking!

closed 2023-10-26 by stefanrueger