avrdudes/avrdude #1537: jtagmkI.c and jtagmkII.c rely on OCDEN being bit 7 of hfuse

stefanrueger, 2023-10-26:
... but on the ATmega406 it's bit 1
[image]

And here an example where it is used:
https://github.com/avrdudes/avrdude/blob/d6c61f5fab4d0d061fb85233acee6ecaa85dd9c0/src/jtagmkII.c#L98-L107

https://github.com/avrdudes/avrdude/blob/d6c61f5fab4d0d061fb85233acee6ecaa85dd9c0/src/jtagmkII.c#L1350-L1352

Not a biggie (false alerts) but should be fixed at some point.

mcuee, 2023-10-26:
I believe JTAG ICE (jtag1) does not support ATmega406, at least not officially. But I do not have the chip to carry out the test now.

I just checked the help file of Atmel Studio 4.19.730 (last version to support JTAG ICE).
https://www.microchip.com/en-us/tools-resources/archives/avr-sam-mcus
ATmega16(L),  ATmega162(L), ATmega169(L or V), ATmega32(L), ATmega323(L), ATmega64(L), ATmega128(L).

@dl8dtl 
Just wondering if you have ATmega406 to confirm. Thanks.

mcuee, 2023-10-26:
BTA, JTAG ICE HW capability is there but the firmware is quite limited since Atmel dropped the support many years ago. I think there are quite some people who want to improve the FW to support more functionalities. However, I have not seen something concrete. The clones of JTAG ICE are really cheap (quality varies) and it is easy to DIY as well, unlike the clones for JTAG ICE mkII which are still quite expensive.
* https://github.com/avrdudes/avrdude/discussions/1080
* https://github.com/avrdudes/avrdude/issues/1054

PR #1577 (2023-11-26): Provide and use libavrude functions to access part config values
This PR provides `avr_get_config_value()` and `avr_set_config_value()` functions

Example usage:
```    
  int ocden = 0;
  if(avr_get_config_value(pgm, p, "ocden", &ocden) == 0 && ocden) // ocden == 1 means disabled
    pmsg_warning("OCDEN fuse not programmed, single-byte EEPROM updates not possible\n");
```

Fixes #1537 

Best way to test this PR is to program a classic part with jtag mkI or jtag mkII programmers. There should be no warning when the OCDEN fuse is programmed, and there should be a warning if it isn't.
closed 2023-11-26 by stefanrueger