avrdudes/avrdude #1541: MSYS2 clang64 build warnings

mcuee, 2023-10-28:
There are a few warnings when building using MSYS2 clang64.

```
[50/72] Building C object src/CMakeFiles/libavrdude.dir/ser_win32.c.obj
C:/work/avr/avrdude_test/avrdude/src/ser_win32.c:484:3: warning: comparison of integers of different signs: 'SOCKET' (aka 'unsigned long long') and 'const int' [-Wsign-compare]
  484 |                 FD_SET(fd->ifd, &rfds);
      |                 ^~~~~~~~~~~~~~~~~~~~~~
C:/msys64/clang64/include/psdk_inc/_fd_types.h:77:40: note: expanded from macro 'FD_SET'
   77 |                 if (((fd_set *)(set))->fd_array[__i] == (fd)) {         \
      |                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ ^   ~~
C:/work/avr/avrdude_test/avrdude/src/ser_win32.c:643:3: warning: comparison of integers of different signs: 'SOCKET' (aka 'unsigned long long') and 'const int' [-Wsign-compare]
  643 |                 FD_SET(fd->ifd, &rfds);
      |                 ^~~~~~~~~~~~~~~~~~~~~~
C:/msys64/clang64/include/psdk_inc/_fd_types.h:77:40: note: expanded from macro 'FD_SET'
   77 |                 if (((fd_set *)(set))->fd_array[__i] == (fd)) {         \
      |                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ ^   ~~
2 warnings generated.
```

stefanrueger, 2023-10-28:
According to `man FD_SET` the function is called as it should be. I suspect Windows has a different internal representation and therefore triggers the warning. I suspect using `FD_SET((unsigned long long) fd->ifd, &rfds);` might silence the warning and still work, but it could be that different compilers have different macros and that you get warnings in other compilers. As far as I am concerned we could leave this as is.

mcuee, 2023-10-29:
Okay, I will close this issue.

closed 2023-10-29 by mcuee