avrdudes/avrdude #1544: Using an old "avrdude.conf" will result in a segmentation fault.

askn37, 2023-10-29:
As the title suggests, after one hardship, another one.

This issue currently occurs when the following combinations are met:

- git main
- Use "jtagmkii_updi". (Is it just jtag2updi?)
- Target AVR_Dx. (probably all known models)
- "memory "io"" directive is not written.

The issue does not occur in the currently released 7.2.

Discovered OS:

```
% uname -a
Darwin alicia.local 23.0.0 Darwin Kernel Version 23.0.0: Fri Sep 15 14:41:34 PDT 2023; root:xnu-10002.1.13~1/RELEASE_ARM64_T8103 arm64
```

### A case where a problem occurs.

```
% /Users/askn/Collaborator/avrdude_pr/build_darwin/src/avrdude -C avrdude.conf.fail -p avr32da32 -c jtag2updi -P /dev/cu.usbmodem2301 -qvvvv

avrdude: Version 7.2-20231029 (4be0f37)
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is /Users/askn/Arduino/UPDI_TEST/avrdude.conf.fail
         User configuration file is /Users/askn/.avrduderc
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/cu.usbmodem2301
         Using Programmer              : jtag2updi
avrdude: jtagmkII_open_pdi()
avrdude: jtagmkII_getsync()
avrdude: jtagmkII_getsync() attempt 1 of 10: sending sign-on command: 
avrdude: jtagmkII_send(): sending 1 bytes
avrdude: send: . [1b] . [00] . [00] . [01] . [00] . [00] . [00] . [0e] . [01] . [f3] . [97] 
avrdude: jtagmkII_recv():
avrdude: ser_recv(): programmer is not responding
avrdude: jtagmkII_recv(): timeout receiving packet
avrdude jtagmkII_getsync() [jtagmkII.c:686] warning: attempt 1 of 10: sign-on command: status -1
avrdude: jtagmkII_getsync() attempt 2 of 10: sending sign-on command: 
avrdude: jtagmkII_send(): sending 1 bytes
avrdude: send: . [1b] . [00] . [00] . [01] . [00] . [00] . [00] . [0e] . [01] . [f3] . [97] 
avrdude: jtagmkII_recv():
avrdude: recv: . [1b] 
avrdude: recv: . [00] 
avrdude: recv: . [00] 
avrdude: recv: . [1d] 
avrdude: recv: . [00] 
avrdude: recv: . [00] 
avrdude: recv: . [00] 
avrdude: recv: . [0e] 
avrdude: recv: . [86] . [01] . [01] . [00] . [06] . [01] . [01] . [00] . [06] . [01] . [00] . [00] . [00] . [00] . [00] . [00] J [4a] T [54] A [41] G [47] I [49] C [43] E [45]   [20] m [6d] k [6b] I [49] I [49] . [00] 
avrdude: recv: < [3c] 
avrdude: recv: . [7f] 

avrdude: jtagmkII_recv(): got message seqno 0 (command_sequence == 0)
avrdude: recv: . [86] . [01] . [01] . [00] . [06] . [01] . [01] . [00] . [06] . [01] . [00] . [00] . [00] . [00] . [00] . [00] J [4a] T [54] A [41] G [47] I [49] C [43] E [45]   [20] m [6d] k [6b] I [49] I [49] . [00] 

Raw message:
0x86 0x01 0x01 0x00 0x06 0x01 0x01 0x00 0x06 0x01 0x00 0x00 0x00 0x00 0x00 0x00
0x4a 0x54 0x41 0x47 0x49 0x43 0x45 0x20 0x6d 0x6b 0x49 0x49 0x00 
Sign-on succeeded

JTAG ICE mkII sign-on message:
Communications protocol version: 1
M_MCU:
  boot-loader FW version:        1
[... 843 lines cut ...]
     ;
 
+    memory "io"
+        size               = 4160;
+        readsize           = 1;
+    ;
+
     memory "sib"
         size               = 32;
         readsize           = 1;
```

[avrdude_conf_fail.txt](https://github.com/avrdudes/avrdude/files/13197368/avrdude_conf_fail.txt)
[avrdude_conf_good.txt](https://github.com/avrdudes/avrdude/files/13197369/avrdude_conf_good.txt)

### Additional Information

I mainly use a self-made implementation using "jtagmkii_updi", and that's where I first discovered the problem. Naturally, I had doubts about my implementation at first, so I also tried his jtag2updi for comparison, and it reproduced the problem.

The difference between the two is whether S_MCU_FW is 6 (jtag2updi) or 7 (updi4avr). The former uses "jtagmkII_set_devdescr" and the latter uses "jtagmkII_set_xmega_params". I don't know of any other implementation that uses this with "jtagmkii_updi"/"jtagmkii_pdi". It probably doesn't matter.

### Important things

- Are there any other implementations, chip definitions, or OSes that reproduce this?
- Is the problem only with "jtagmkII.c"?

mcuee, 2023-10-29:
I tend to think this belongs to `won't fix` type of issues. Basically I do not think this is a problem myself.

But maybe @dl8dtl, @stefanrueger and @MCUdude can chime in and offer their opinion.

askn37, 2023-10-29:
@mcuee

I am confident that the combination of "avrdude" itself and the updated "avrdude.conf" that comes with it does not cause this problem. However, be careful when combining "~/.avrduderc". I can only imagine what would have happened if someone had added their own "programmers" and "parts" in the past. (Except for some weirdo who switches home directories and tests different versions of his avrdude)

stefanrueger, 2023-10-29:
Whenever I have had a seg fault in avrdude, it was because of make not realising some files were changed (it goes by date, not checksum). Going back and forth between versions using `git checkout` or similar can be the cause of this. Of course, it could also be some real problem but AVRDUDE rarely has these.

@askn37 Please could I ask you to `make clean` (or whatever effects removing of all compiled object files) after switching between versions and before compiling. Here is what I do: I copy the following `Makefile` into the cloned avrdude version
```
all:
	cmake --build build_linux

install:
	cmake --build build_linux; sudo cmake --build build_linux --target install

new:
	sudo rm -rf build_linux
	cmake -D BUILD_DOC=1 -D CMAKE_BUILD_TYPE=RelWithDebInfo -D HAVE_LINUXGPIO=1 -D HAVE_LINUXSPI=1 -D HAVE_PARPORT=1 -B build_linux
	cmake --build build_linux
```

Then I carry out `make new` whenever I switch versions.

stefanrueger, 2023-10-29:
@askn37 But you are right. I spotted code that would cause the segfault in the way you describe. I will create a PR for this. Thanks for reporting!

mcuee, 2023-10-29:
> @askn37 But you are right. I spotted code that would cause the segfault in the way you describe. I will create a PR for this. Thanks for reporting!

I see. Changed the label to bug.

askn37, 2023-10-29:
Supplementary exam. 
The build in question was a simpler method and stopped at another failure. Caution is required.

```
$ git clone https://github.com/avrdudes/avrdude.git avrdude_main
$ cd avrdude_main
$ pwd
/Users/askn/Collaborator/avrdude_main
$ sh build.sh
$ ./build_darwin/src/avrdude -c dryrun -v

avrdude: Version 7.2-20231029 (9dfb6164)
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is /Users/askn/Collaborator/avrdude_main/build_darwin/src/avrdude.conf
avrdude main() error: unable to process system wide configuration file /Users/askn/Collaborator/avrdude_main/build_darwin/src/avrdude.conf
```

mcuee, 2023-10-29:
@askn37 

I am sorry. This is an issue in git main caused by PR #1542. I made a mistake and did not test it properly. This will be fixed.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git -c usbasp -p m2560
avrdude_git error: unable to process system wide configuration file C:\work\avr\avrdude_test\avrdude_bin\avrdude.conf
```

askn37, 2023-10-29:
@mcuee

We've found some other strange issues with "dump sernum", but they're a lower priority, so we'll hold off on retesting and reporting until the build issues are resolved.

mcuee, 2023-10-29:
> @mcuee
> 
> We've found some other strange issues with "dump sernum", but they're a lower priority, so we'll hold off on retesting and reporting until the build issues are resolved.

@askn37 
Yes, that is perfectly fine.

And thanks a lot for collecting the potential issues in your repo. You are very helpful.
* https://github.com/askn37/avrdude/issues/2

stefanrueger, 2023-10-30:
> I  will create a PR for this. 

On second thoughts I pushed a commit onto PR #1538 to avoid merge conflicts. @askn37 Please try PR #1538. You should now get an error message rather than a seg fault.

askn37, 2023-10-30:
@stefanrueger 

> > I  will create a PR for this.
> 
> On second thoughts I pushed a commit onto PR #1538 to avoid merge conflicts. @askn37 Please try PR #1538. You should now get an error message rather than a seg fault.

got it. Segment violation resolved.

```
askn@alicia avrdude_main % git branch                  
  main
* memory-types
askn@alicia avrdude_main % git log --oneline -n 5
54eacc51 (HEAD -> memory-types) Convert avr_locate_mem(p, string) to avr_locate_mem_by_type()
2be7f7e5 Refer to memory rather than memory type
03db6e5c Use memory type for fuse memories instead of fuse names
c8d11f65 Replace str_eq(m->desc, "(usersig|userrow|prodsig|sigrow") with macros
cd53766d Replace str_eq(m->desc, "...") with mem_is_...(m)
askn@alicia avrdude_main % ./build_darwin/src/avrdude -P /dev/cu.usbserial-230 -c jtag2updi -p avr64dd32 -vT "dump sernum"

avrdude: Version 7.2-20231030 (54eacc51)
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is /Users/askn/Collaborator/avrdude_main/build_darwin/src/avrdude.conf
         User configuration file is /Users/askn/.avrduderc

         Using Port                    : /dev/cu.usbserial-230
         Using Programmer              : jtag2updi
JTAG ICE mkII sign-on message:
Communications protocol version: 1
M_MCU:
  boot-loader FW version:        1
  firmware version:              2.01
  hardware version:              1
S_MCU:
  boot-loader FW version:        1
  firmware version:              7.53
  hardware version:              1
Serial number:                   4a:34:94:37:55:3e
Device ID:                       JTAGICE mkII
         AVR Part                      : AVR64DD32
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           fuse0       wdtcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse1       bodcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse2       osccfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse5       syscfg0     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse6       syscfg1     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse7       codesize    0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse8       bootsize    0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuses                   0     0     0    0 no         16   16      0     0     0 0x00 0x00
           lock                    0     0     0    0 no          4    1      0     0     0 0x00 0x00
           prodsig     sigrow      0     0     0    0 no        128  128      0     0     0 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           tempsense               0     0     0    0 no          4    1      0     0     0 0x00 0x00
           sernum                  0     0     0    0 no         16    1      0     0     0 0x00 0x00
           userrow     usersig     0     0     0    0 no         32   32      0     0     0 0x00 0x00
           data                    0     0     0    0 no          0    1      0     0     0 0x00 0x00
           io                      0     0     0    0 no       4160    1      0     0     0 0x00 0x00
           sib                     0     0     0    0 no         32    1      0     0     0 0x00 0x00
           eeprom                  0     0     0    0 no        256    1      0     0     0 0x00 0x00
           flash                   0     0     0    0 no      65536  512      0     0     0 0x00 0x00

         Programmer Type : JTAGMKII_UPDI
         Description     : JTAGv2 to UPDI bridge
         M_MCU HW version: 1
         M_MCU FW version: 2.01
         S_MCU HW version: 1
         S_MCU FW version: 7.53
         Serial number   : 4a:34:94:37:55:3e
avrdude: silicon revision: 1.3
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e961a (probably avr64dd32)

avrdude: processing -T dump sernum
>>> dump sernum 0x0 0x10
Reading | -------------------------------------------------- | 0% 0.00 s 
avrdude jtagmkII_read_byte() error: unknown memory sernum
avrdude cmd_dump() error: (dump) error reading sernum address 0x00000 of part AVR64DD32
               read operation not supported on memory sernum

avrdude done.  Thank you.
```

stefanrueger, 2023-10-30:
> avrdude jtagmkII_read_byte() error: unknown memory sernum

@askn37 I have pushed a commit to PR #1538 to handle sub-memories of `sigrow`. Please update form the PR and test reading `sernum`, `osccal20` etc memories.

askn37, 2023-10-31:
@stefanrueger 

> @askn37 I have pushed a commit to PR #1538 to handle sub-memories of `sigrow`. Please update from the PR and test reading `sernum`, `osccal20` etc memories.

JTAG2UPDI results are negative.

Added debug print to check if the condition does not hold at the relevant location.

```
diff --git a/src/jtagmkII.c b/src/jtagmkII.c
index 6dcec577..db22150b 100644
--- a/src/jtagmkII.c
+++ b/src/jtagmkII.c
@@ -2239,13 +2239,20 @@ static int jtagmkII_read_byte(const PROGRAMMER *pgm, const AVRPART *p, const AVR
     }
   } else if (mem_is_in_sigrow(mem)) {
     AVRMEM *sr = avr_locate_sigrow(p);
-    int doff;
-    if ((p->prog_modes & (PM_PDI | PM_UPDI)) && sr && (doff = mem->offset-sr->offset) >= 0 &&
+    int doff = mem->offset-sr->offset;
+    if ((p->prog_modes & (PM_PDI | PM_UPDI)) && sr && doff >= 0 &&
       (int) (addr + doff) < sr->size) {
       cmd[1] = MTYPE_PRODSIG;
       addr += doff;
     } else {
       pmsg_error("unable to handle memory %s\n", mem->desc);
+      pmsg_error("check: p->prog_modes => %x\n", p->prog_modes);
+      pmsg_error("check: (PM_PDI | PM_UPDI) => %x\n", (PM_PDI | PM_UPDI));
+      pmsg_error("check: bool=> %x\n", (p->prog_modes & (PM_PDI | PM_UPDI)));
+      pmsg_error("check: sr->size => %x\n", sr->size);
+      pmsg_error("check: addr => %x\n", addr);
+      pmsg_error("check: doff => %x\n", doff);
+      pmsg_error("check: (addr + doff) < sr->size => %x\n", (int) (addr + doff) < sr->size);
       return -1;
     }
   } else if (mem_is_io(mem)) {
```

And I got the following information.

```
sh -x ./J_READ.sh -qvvvv | grep "\(error|conf\)"

+ /Users/askn/Collaborator/avrdude_main/build_darwin/src/avrdude -p avr64dd32 -c jtag2updi -P /dev/cu.usbserial-230 -U fuses:r:avr64dd32/Fuse.hex:i -U lock:r:avr64dd32/Lock.hex:i -U sernum:r:avr64dd32/SERNUM.bin:r -U eeprom:r:avr64dd32/EEPROM.hex:i -U userrow:r:avr64dd32/USERROW.hex:i -U flash:r:avr64dd32/Flash.hex:i -qvvvv
         System wide configuration file is /Users/askn/Collaborator/avrdude_main/build_darwin/src/avrdude.conf
         User configuration file is /Users/askn/.avrduderc
         User configuration file does not exist or is not a regular file, skipping
avrdude jtagmkII_read_byte() [jtagmkII.c:2248] error: unable to handle memory sernum
avrdude jtagmkII_read_byte() [jtagmkII.c:2249] error: check: p->prog_modes => 11
avrdude jtagmkII_read_byte() [jtagmkII.c:2250] error: check: (PM_PDI | PM_UPDI) => 18
avrdude jtagmkII_read_byte() [jtagmkII.c:2251] error: check: bool=> 10
avrdude jtagmkII_read_byte() [jtagmkII.c:2252] error: check: sr->size => 80
avrdude jtagmkII_read_byte() [jtagmkII.c:2253] error: check: addr => 1110
avrdude jtagmkII_read_byte() [jtagmkII.c:2254] error: check: doff => 10
avrdude jtagmkII_read_byte() [jtagmkII.c:2255] error: check: (addr + doff) < sr->size => 0
avrdude avr_read_mem() [avr.c:484] error: unable to read byte at address 0x0000
avrdude avr_read_mem() [avr.c:486] error: read operation not supported for memory sernum
avrdude do_op() [update.c:411] error: unable to read all of sernum memory, rc=-2
```

```(int) (addr + doff) < sr->size```  showed false.

please refer.

stefanrueger, 2023-10-31:
@askn37 Vvv useful; the addr already contains the offset. It is not necessary to figure out where the sub-memory sits in sigrow.

I made change in PR #1538 to reflect this.

stefanrueger, 2023-10-31:
@askn37 Thanks for testing. Your results look like `sigrow`/`prodsig` is *not* read correctly. I would expect this to *contain* `signature`, `sernum`, `tempsense`, etc but it doesn't look like it in your examples. `-c dryrun` shows where which sub-memories should go:
```
$ avrdude -qqp t824 -c dryrun -U prodsig:r:-:I
:200000001E93295554535251504F4E4D4CFFFFFFFFFFFFFFFFFFFFFF6F6F4F4FFFFFFFFF74 // 00000> ..)UTSRQPONML...........ooOO....
:20002000545465654545FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFDE // 00020> TTeeEE..........................
:00000001FF
```

`T` is where the `tempsense` calibration values should be, `o` the `osccal` values, `UTS...` where `sernum` should be.

The `jtagmkII.c` code is not crisp and clear as it has grown organically to cater for very different interfaces over the last 20 years. I think https://github.com/avrdudes/avrdude/blob/2bb36b9b0b828c8fe3c92e308c5e40f56e6aa9aa/src/jtagmkII.c#L1006 and https://github.com/avrdudes/avrdude/blob/2bb36b9b0b828c8fe3c92e308c5e40f56e6aa9aa/src/jtagmkII.c#L2169 has the effect of adding offset twice and we are reading from `0x2200` rather than `0x1100`. Can someone confirm, please?

askn37, 2023-10-31:
@stefanrueger 

Unlike JTAG2UPDI, which uses the device_descriptor structure, UPDI4AVR uses the xmega_device_desc structure. The branch condition to switch is ```PDATA(pgm)->fwver >= 0x700```.

This requires that the byte chunks passed in STK500v2 are zero-based relative addresses. If the address value is doubled, then both the structure and the READ/WRITE instruction can be considered to have the same address.

UPDI4AVR was fine in the 6.3 to 7.2 implementation, so this fact suggests a break in compatibility.

To double check this behavior you will need another JTAG debugger that uses jtagmkII_set_xmega_params. (who is that?)

askn37, 2023-10-31:
I will add a little explanation to deepen your knowledge.

'xmega_device_desc', as the name suggests, contains a memory layout enhanced with the XMEGA architecture. It does not pass the model number of the working chip or his NVMCTRL version number. Instead, pass this structure to the STK500v2 adapter.

```
/* New Xmega device descriptor, for firmware version 7 and above */
struct xmega_device_desc {
    unsigned char whatever[2];		// cannot guess; must be 0x0002
    unsigned char datalen;		// length of the following data, = 47
    unsigned char nvm_app_offset[4];	// NVM offset for application flash
    unsigned char nvm_boot_offset[4];	// NVM offset for boot flash
    unsigned char nvm_eeprom_offset[4]; // NVM offset for EEPROM
    unsigned char nvm_fuse_offset[4];	// NVM offset for fuses
    unsigned char nvm_lock_offset[4];	// NVM offset for lock bits
    unsigned char nvm_user_sig_offset[4]; // NVM offset for user signature row
    unsigned char nvm_prod_sig_offset[4]; // NVM offset for production sign. row
    unsigned char nvm_data_offset[4];	// NVM offset for data memory (SRAM + IO)
    unsigned char app_size[4];		// size of application flash
    unsigned char boot_size[2];		// size of boot flash
    unsigned char flash_page_size[2];	// flash page size
    unsigned char eeprom_size[2];	// size of EEPROM
    unsigned char eeprom_page_size;	// EEPROM page size
    unsigned char nvm_base_addr[2];	// IO space base address of NVM controller
    unsigned char mcu_base_addr[2];	// IO space base address of MCU control
};
#endif /* JTAGMKII_PRIVATE_EXPORTED */
```

Seeing this, the STK500v2 adapter can deduce which series the target chip belongs to even if it cannot read the target chip's PDI/UPDI. In other words, you can get hints on how to switch the circuits and signal lines required for high voltage control. Regarding UPDI, even if you can't read the SIB from the chip, you can infer the version of NVMCTRL and therefore decide whether to give 12V to the UPDI pad or 8.2V to the RESET pad. In particular, nvm_lock_offset, nvm_user_sig_offset, and nvm_prod_sig_offset are significantly different. Additionally, the SRAM starting address needed to rewrite his USERROW of the lock device can also be obtained through nvm_data_offset. (Since it is not normally used, a dummy value is set.)

The problem is that subsequent READ/WRITE instructions mix relative addressing to the structure and absolute addressing from the beginning of the memory map. It's probably an old bug, but it's a convention kept for backwards compatibility. Additionally, terminal mode also has non-uniform addressing. The complex conditional branching in the code below is the result of trial and error to accommodate the differences. At least, this is how I interpreted it from 6.3 to 7.2.

https://github.com/askn37/multix-zinnia-updi4avr-firmware-builder/blob/main/libraries/UPDI4AVR/examples/UPDI4AVR_FW753B/src/NVM.cpp#L104

stefanrueger, 2023-10-31:
> complex conditional branching in the code below is the result of trial and error to accommodate the differences.

It is better controlling both sides the FW of the programmer HW *and* the `-c` module of AVRDUDE. Assuming there is a published open hardware design for a programmer HW, a competently written FW for the programmer, then I am sure that the AVRDUDE team would consider adding a corresponding `-c` programmer module.

askn37, 2023-10-31:
test case plan to me

Prepare the following shell script

-c *** -p t824 -U fuses:r:-:I -U lock:r:-:I -U prodsig:r:-:I

Try this 2x2 pattern

Two types of configuration files
Old type (7.0 or earlier?): jtagmkII_pdi+has_updi
New type (main): jtagmkII_updi+PM_UPDI

Build the latest JTGA2UPDI main and write it to UNO (mine hasn't been touched in a long time)
Prepare the public UPDI4AVR main (return to before AVR_EB modification)

stefanrueger, 2023-11-01:
| UPDI4AVR was fine in the 6.3 to 7.2 implementation, so this fact suggests a break in compatibility.

This may well be. I suspect the root problem is that updi4avr project is (so far) not catered for: AVRDUDE has no way of knowing which unsupported projects utilise which behaviour!

Whilst we develop AVRDUDE further we try hard to ensure that *once supported* and *known* programmers keep working with AVRDUDE. These are often programmers someone in the team has and that we can test on. We welcome reports on regressions.

Also, please note, it is generally *discouraged* to use outdated .conf files with newer AVRDUDE versions. Always try to use the matching .conf file: AVRDUDE expects it can rely on a configuration model of a part/programmer that matches the code base.

Having said this, once you figure out what needs changing in AVRDUDE so that updi4avr keeps working well, feel free to submit a PR. We will try to see if that breaks supported programmers. If it does not, we will consider merging. The best way forward, however, is to make sure that AVRDUDE knows about a project/programmer and has a dedicated `-c` programmer for it that works in tandem with it. If you can control both sides of the programming, host downloader/uploader and programmer firmware, then both `-c` programmer and the FW have a sporting chance to be crisp, clear, simple and easily maintainable.

askn37, 2023-11-01:
@stefanrueger 

UPDI4AVR also provides upward compatibility with JTAG2UPDI. This means you can safely create a bootloader using "-c jtag2updi". We are also approaching AVR_Dx/Ex, which JATG2UPDI has not followed before. This functionality is available out-of-the-box for DxCore and MCUdude projects. (Their "avrdude" is still old)

What I specifically want to do is cross-check whether the old and new "avrdude"(and the pair of configuration files) and their adapters are backwards compatible. Generate a report if any discrepancies are found.

askn37, 2023-11-01:
## Testcase JTAG2UPDI --- STK500v2 FW_V=6

Script

```
#!/bin/bash

#
# STK500v2 cross-check
# 

avrdude_main=~/Collaborator/avrdude_main/build_darwin/src/avrdude
avrdude_72=~/Library/Arduino15/packages/MultiX-Zinnia/tools/avrdude/7.2-arduino.1/bin/avrdude
avrdude_71=~/Library/Arduino15/packages/MegaCoreX/tools/avrdude/7.1-arduino.1/bin/avrdude

conf=
part=t824
port_jtag2updi=/dev/cu.usbmodem1201
port_updi4avr=/dev/cu.usbserial-230
avrdude_prog_force=jtag2updi
verbose=-qvvvv
test_prog=jtag2updi
test_fwv=6

# testcase userconfig for avrdude_63
(cat <<-EOL
programmer
    id                     = "jtag2updi";
    desc                   = "JTAGv2 to UPDI bridge";
    type                   = "jtagmkii_pdi";
    connection_type        = serial;
    baudrate               = 115200;
EOL
) > _avrduderc_old

test_run_part () {
  avrdude_name=$1
  avrdude_prog=$2
  avrdude_cmd=$(eval echo \$$1)
  avrdude_port=$(eval echo \$port_${2})
  sigrow=
  echo
  echo -----------------------
  echo Test case: $avrdude_name
  echo Test prog: $avrdude_prog
  echo Test port: $avrdude_port
# rm -f ~/.avrduderc 2>&1 > /dev/null
  if [ $avrdude_name = "avrdude_main" ]
  then
    sigrow="-U sigrow:r:sigrow_${avrdude_prog}_${avrdude_name}.hex:i"
  fi
  cmd=$(echo $avrdude_cmd $verbose -p $part -c $avrdude_prog_force -P $avrdude_port $conf \
    -l ${avrdude_name}_${avrdude_prog}.log \
    -U fuses:r:fuses_${avrdude_prog}_${avrdude_name}.hex:i \
    -U lock:r:lock_${avrdude_prog}_${avrdude_name}.hex:i \
    -U prodsig:r:prodsig_${avrdude_prog}_${avrdude_name}.hex:i \
    -U sernum:r:sernum_${avrdude_prog}_${avrdude_name}.hex:i \
    $sigrow \
  )
  echo +$cmd
  $cmd
}

test_compare () {
  cmd=$(echo diff -u ${1}_${2}_avrdude_71.hex ${1}_${2}_avrdude_72.hex)
  echo
  echo +$cmd
  echo
  $cmd

[... 157 lines cut ...]
;
;
;
avrdude: processing -U sernum:r:sernum_jtag2updi_avrdude_main.hex:i
avrdude: reading sernum memory ...
avrdude: jtagmkII_read_byte(.., sernum, 0x0, ...)
avrdude: jtagmkII_read_byte(): sending read memory command: 
avrdude: jtagmkII_send(): sending 10 bytes
avrdude: send: . [1b] . [11] . [00] . [0a] . [00] . [00] . [00] 
. [0e] . [05] . [c6] 
. [01] . [00] . [00] . [00] 
. [03] . [11] . [00] . [00] 
d [64] , [2c] 
```

This is different

```
v7.2 sernum [0e] [05] [c0] MTYPE_FLASH and [04] [11] [00] [00] start_addr 0x1104 !!!
main sernum [0e] [05] [c6] MTYPE_PRODSIG
```

## Conclusion

What you found was a bug in the old version.

askn37, 2023-11-02:
I made one mistake.
In the previous comment he wrote the label STK500v2 (AVR068).
JTAG2UPDI and UPDI4AVR are actually implementations of JTAGICEmkII (AVR067).
The test result is correct, but the label is incorrect. I apologize.

stefanrueger, 2023-11-02:
Thanks @askn37 for thorough testing. I was wondering whether the test still works OK in PR #1538?

If so, I take it that once PR #1538 is merged this Issue can be closed.

askn37, 2023-11-02:
@stefanrueger 

My main job is as a debugger for hire. We are also familiar with building networks and audio/video distribution. Examples include old satellite links, disaster helicopter telemetry, and secret nuclear reactor hotlines.
We can also help you translate AVR documentation into your native language (https://avr.jp). The disadvantage of not being a native English speaker is the need to understand all of his publicly available AVR documents and identify differences from the originals. We are also currently working on translating the AVR_EB documentation.

That's why I'm always concerned about the smell of bug...

The rest is a comparison of how JTAGICEmkII's FWV=7 has changed from before. Issues that were previously ignored may be brought to light. However, FWV=7 is not documented. I think AVRDragon is probably implemented, but I don't have it, so my evaluation will be a guess.

askn37, 2023-11-02:
# Testcase JTAGICEmkII FWV=7

This document reveals what avrdude is serially outputting in the JTAGICEmkII protocol, especially when FWV=7 and PM_UPDI. Probably no one would have imagined such a combination. Either it wasn't there or it was hidden by the USB connection.

I happened to have created such an UPDI implementation here, so I can now interact directly with avrdude.

This document reveals what avrdude is serially outputting in the JTAGICEmkII protocol, especially when FWV=7 and PM_UPDI. Probably no one would have imagined such a combination. Either it wasn't there or it was hidden by the USB connection.

I happened to have created such an UPDI implementation here, so I can now interact directly with avrdude.

JTAG2UPDI has FWV=6, so it does not match this document. Originally the XMEGA PDI protocol, it has absolute addressing up to 32 bits wide. Previous versions of avrdude with FWV=6 required a trick in avrdude.conf to enable the 17th bit to be able to handle AVR_DA/DB 128KiB.

## target=t202 (P:0)

avrdude v6.3 does not support tinyAVR-2, so this test uses tinyAVR-0 instead. This is a 16bit address chip.
In the end, v6.8, v7.1, and v7.2 exhibited the same behavior, so this consideration was unnecessary.
(The v6.8 configuration is missing "sernum")

## acrdude v7.2

### v7.2 '-U' BLOCK READ

#### v7.2 jtagmkII_getparm(): Sending get parameter command

```
avrdude: Version 7.2-arduino.1

S_MCU:
  boot-loader FW version:        1
  firmware version:              7.53
  hardware version:              1
```

OK, are you ready?

#### v7.2 jtagmkII_set_xmega_params(): Sending set Xmega params command

```
avrdude: jtagmkII_set_xmega_params(): Sending set Xmega params command:
avrdude: jtagmkII_send(): sending 51 bytes
avrdude: send: . [1b] . [07] . [00]
3 [33] . [00] . [00] . [00]
. [0e] 6 [36]                   //  struct xmega_device_desc {
. [02] . [00]                   //      unsigned char whatever[2];            // cannot guess; must be 0x0002
/ [2f]                          //      unsigned char datalen;                // length of the following data, = 47
. [00] . [00] . [00] . [00]     //      unsigned char nvm_app_offset[4];        // NVM offset for application flash
. [00] . [00] . [00] . [00]     //      unsigned char nvm_boot_offset[4];     // NVM offset for boot flash
. [00] . [14] . [00] . [00]     //      unsigned char nvm_eeprom_offset[4];     // NVM offset for EEPROM
. [80] . [12] . [00] . [00]     //      unsigned char nvm_fuse_offset[4];     // NVM offset for fuses
. [8a] . [12] . [00] . [00]     //      unsigned char nvm_lock_offset[4];     // NVM offset for lock bits
. [00] . [13] . [00] . [00]     //      unsigned char nvm_user_sig_offset[4];   // NVM offset for user signature row
. [03] . [11] . [00] . [00]     //      unsigned char nvm_prod_sig_offset[4];   // NVM offset for production sign. row
. [00] . [00] . [00] . [01]     //      unsigned char nvm_data_offset[4];     // NVM offset for data memory (SRAM + IO)
. [00] . [00] . [00] . [00]     //      unsigned char app_size[4];            // size of application flash
. [00] . [00]                   //      unsigned char boot_size[2];           // size of boot flash
@ [40] . [00]                   //      unsigned char flash_page_size[2];     // flash page size
@ [40] . [00]                   //      unsigned char eeprom_size[2];         // size of EEPROM
  [20]                          //      unsigned char eeprom_page_size;         // EEPROM page size
. [00] . [10]                   //      unsigned char nvm_base_addr[2];         // IO space base address of NVM controller
. [00] . [00]                   //      unsigned char mcu_base_addr[2];         // IO space base address of MCU control
. [8c] . [1c]                   //  }   // crc16-ccitt
```

- [0e] [36] -> CMND_SET_XMEGA_PARAMS
- nvm_prod_sig_offset is 0x1103 <- but top address in oldstyle config
- nvm_data_offset -> dummy address $0100:0000

'application' and 'boot' have no memory settings, so nvm_app_offset and nvm_boot_offset will be zero. It has been ignored in avrdude.conf for many years as it is not particularly needed in the XMEGA generation. However, the UPDI generation is different. This will cause problems later....

#### v7.2 jtagmkII_program_enable(): Sending enter progmode command
[... 608 lines cut ...]

## avrdude_main Summary

This is a test result using avrdude_main with UPDI4AVR and may differ from other JTAGICEmkII + FWV=7 + UPDI implementations. However, compatibility was broken in the following items. It's quite possible that other implementations are similar.

```
avrdude: Version 7.2-20231031 (76cbf91a)

S_MCU:
  boot-loader FW version:        1
  firmware version:              7.53
  hardware version:              1
```

- **reading chip silicon revision** : Although it is in byte mode with MTYPE_XMEGA_FLASH, nvm_data_offset was already added to the address field. Unable to read from correct IO memory due to address overflow.

  - *Side note: nvm_data_offset is one of the long-reserved items that allows the user to pass arbitrary "data" address parameters to the adapter. Is this really necessary now?*

- **reading fuses memory** : Since MTYPE_XMEGA_FLASH and block mode were ordered, nvm_fuse_offset cannot be added correctly. Either bytes/blocks should be MTYPE_FUSE_BITS. As with 'dump fuses', byte-oriented and absolute addressing is recommended.

- **dump eeprom** : Although it is MTYPE_EEPROM and block mode, no relative address is specified. The result will be twice the address.

Please refer to the above.

- *P.S: "-vvvv" character dump in terminal mode breaks above 0x80. Must be replaced with "."*

stefanrueger, 2023-11-02:
@MCUdude @mcuee: Which programmer expects `-c jtag2updi` and has FW 7 or above?

MCUdude, 2023-11-02:
@stefanrueger this is the output from my official Arduino Nano Every:

```
$ ./avrdude -cjtag2updi -patmega4809 -P /dev/cu.usbmodem14101 -r -v

avrdude: Version 7.2-20230720
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is /Users/hans/Downloads/avrdude-terminal/src/avrdude.conf
         User configuration file is /Users/hans/.avrduderc

avrdude: touching serial port /dev/cu.usbmodem14101 at 1200 baud
avrdude: waiting for new port... using same port /dev/cu.usbmodem14101
         Using Port                    : /dev/cu.usbmodem14101
         Using Programmer              : jtag2updi
JTAG ICE mkII sign-on message:
Communications protocol version: 1
M_MCU:
  boot-loader FW version:        1
  firmware version:              6.00
  hardware version:              1
S_MCU:
  boot-loader FW version:        1
  firmware version:              6.00
  hardware version:              1
Serial number:                   00:00:00:00:00:00
Device ID:                       JTAGICE mkII
         AVR Part                      : ATmega4809
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Memory Detail                 :
[...]
         Programmer Type : JTAGMKII_UPDI
         Description     : JTAGv2 to UPDI bridge
         M_MCU HW version: 1
         M_MCU FW version: 6.00
         S_MCU HW version: 1
         S_MCU FW version: 6.00
         Serial number   : 00:00:00:00:00:00
avrdude: silicon revision: 0.1
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9651 (probably m4809)

avrdude done.  Thank you.
```

And this is the output from from the [Thinary Nano Every](https://www.aliexpress.com/item/4001060953783.html) (Atmega4808 based):
```
$ ./avrdude -cjtag2updi -patmega4808 -P /dev/cu.usbserial-1410 -r -v

avrdude: Version 7.2-20230720
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is /Users/hans/Downloads/avrdude-terminal/src/avrdude.conf
         User configuration file is /Users/hans/.avrduderc

avrdude: touching serial port /dev/cu.usbserial-1410 at 1200 baud
avrdude: waiting for new port... using same port /dev/cu.usbserial-1410
         Using Port                    : /dev/cu.usbserial-1410
         Using Programmer              : jtag2updi
JTAG ICE mkII sign-on message:
Communications protocol version: 1
M_MCU:
  boot-loader FW version:        1
  firmware version:              6.00
  hardware version:              1
S_MCU:
  boot-loader FW version:        1
  firmware version:              6.00
  hardware version:              1
Serial number:                   0a:0b:0c:0d:0e:0f
Device ID:                       Thinary mkII
         AVR Part                      : ATmega4808
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Memory Detail                 :
[...]
         Programmer Type : JTAGMKII_UPDI
         Description     : JTAGv2 to UPDI bridge
         M_MCU HW version: 1
         M_MCU FW version: 6.00
         S_MCU HW version: 1
         S_MCU FW version: 6.00
         Serial number   : 0a:0b:0c:0d:0e:0f
avrdude: silicon revision: 0.1
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9650 (probably m4808)

avrdude done.  Thank you.
```

And this is from a DIY jtag2updi I don't know when I flashed with the latest firmware available. Probably a year ago:

```
$ ./avrdude -cjtag2updi -patmega4808 -P /dev/cu.usbserial-1410 -v

avrdude: Version 7.2-20230720
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is /Users/hans/Downloads/avrdude-terminal/src/avrdude.conf
         User configuration file is /Users/hans/.avrduderc

         Using Port                    : /dev/cu.usbserial-1410
         Using Programmer              : jtag2updi
JTAG ICE mkII sign-on message:
Communications protocol version: 1
M_MCU:
  boot-loader FW version:        1
  firmware version:              6.00
  hardware version:              1
S_MCU:
  boot-loader FW version:        1
  firmware version:              6.00
  hardware version:              1
Serial number:                   00:00:00:00:00:00
Device ID:                       JTAGICE mkII
         AVR Part                      : ATmega4808
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Memory Detail                 :
[...]
         Programmer Type : JTAGMKII_UPDI
         Description     : JTAGv2 to UPDI bridge
         M_MCU HW version: 1
         M_MCU FW version: 6.00
         S_MCU HW version: 1
         S_MCU FW version: 6.00
         Serial number   : 00:00:00:00:00:00
avrdude: silicon revision: 0.1
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9650 (probably m4808)

avrdude done.  Thank you.
```

So I don't know of any jtag2updi programmers out there that report themselves as being v7.

And when looking at the "official" jtag2updi firmware, there are no traces of v7 being mentioned anywhere:

https://github.com/ElTangas/jtag2updi/blob/07be876105e0b9cfedf2723b0ac88780bcae50d8/source/JTAG2.h#L62-L72

mcuee, 2023-11-02:
> So I don't know of any jtag2updi programmers out there that report themselves as being FW 7.

Same here. It does not exist as of now.

mcuee, 2023-11-02:
Basically jtag2updi FW needs to be updated to support AVR EA/EB. Maybe then it will be called FW 7.

stefanrueger, 2023-11-02:
@mcuee @MCUdude Thanks! Now the other question is which programmers use -c jtag2**pdi** (so XMEGA parts) use FW 7 or above?

stefanrueger, 2023-11-02:
OK, summarising: there are no programmers that use jtag2 for UPDI parts with FW > 6. The *only* jtag2 programmers with FW > 6 cater for XMEGA parts. So, the following lines
https://github.com/avrdudes/avrdude/blob/96b081b2f9800b119a88c27bcaba11c5aa32fa16/src/jtagmkII.c#L1288-L1291

*really* should be replaced by

```
  (PDATA(pgm)->fwver>=0x700 && (p->prog_modes&PM_PDI)? jtagmkII_set_xmega_params: jtagmkII_set_devdescr)(pgm, p);
```

Comments? @askn37 @mcuee @MCUdude

MCUdude, 2023-11-02:
I'm fine with that as long as it doesn't break anything.

In the really old days, the jtag2updi project just used "vanilla" avrdude 6.3 using the `jtagmkii_pdi` implementation.

```
#------------------------------------------------------------
# jtag2updi programmer definitions
#------------------------------------------------------------

programmer
  id    = "jtag2updi";
  desc  = "JTAGv2 to UPDI bridge";
  type  = "jtagmkii_pdi";
  connection_type = serial;
  baudrate = 115200;
;
```

MCUdude, 2023-11-02:
@stefanrueger I only have a mac at hand:

> Usbhid send and receive (-vvvvv)

https://gist.github.com/MCUdude/4f6e5609928fda35da15ba3c1797b7d5

> Usbdev send and reveive (-vvvvv) 

https://gist.github.com/MCUdude/dd9e41e04e37cd6168e49b1837f4192b

stefanrueger, 2023-11-02:
@MCUdude Brilliant. Will wait for @mcuee's windows runs.

stefanrueger, 2023-11-03:
@askn37 I do not see how eeprom access differs between v7.2 and PR #1538 for FW=7 + UPDI. How so?

| reading fuses memory : [...] byte-oriented and absolute addressing is recommended

I think this has happened in the last commit to PR #1538. Please check

|  reading chip silicon revision : Although it is in byte mode with MTYPE_XMEGA_FLASH, nvm_data_offset was already added

| dump eeprom : [...] The result will be twice the address.

What do you suggest? Please tell us in a git diff.

mcuee, 2023-11-03:
I think this communication trace feature is very useful improvement to debug low level protocol.

The following debug logs are generated using the github action binaries (mingw64 and MSVC64).
https://github.com/avrdudes/avrdude/actions/runs/6739037907

**Usbdev send and reveive (-vvvvv)**

**1) usbasp MinGW build with libusb-1.0**

<details>
<summary> Click for full debug log </summary>

```
PS>pwd

Path
----
C:\work\avr\avrdude_test\avrdude_bin\github_bin\pr1538\avrdude-mingw-x86_64

PS>.\avrdude -c usbasp -p m328pb -vvvv

avrdude: Version 7.2-20231102 (3eaa9f2)
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is C:\work\avr\avrdude_test\avrdude_bin\github_bin\pr1538\avrdude-mingw-x86_64\avrdude.conf

         Using Port                    : usb
         Using Programmer              : usbasp
avrdude: usbasp_open("usb")
avrdude: seen device from vendor >www.fischl.de<
avrdude: seen product >USBasp<
         AVR Part                      : ATmega328PB
         Chip Erase delay              : 10500 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : possible i/o
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                 65    20     4    0 no       1024    4      0  3600  3600 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           flash                  65    10   128    0 yes     32768  128    256  4500  4500 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           efuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
[... 150 lines cut ...]
         <= [00]
avrdude: AVR device initialized and ready to accept instructions
Reading |                                                    | 0% 0.00 s
avrdude: usbasp_spi_cmd(0x30, 0x00, 0x00, 0x00) ...
avrdude: usbasp_transmit("USBASP_FUNC_TRANSMIT", 0x30, 0x00, 0x00, 0x00)
         <= [00] [30] [00] [1e]
avrdude: usbasp_spi_cmd() => 0x00, 0x30, 0x00, 0x1e
avrdude: usbasp_spi_cmd(0x30, 0x00, 0x01, 0x00) ...
avrdude: usbasp_transmit("USBASP_FUNC_TRANSMIT", 0x30, 0x00, 0x01, 0x00)
         <= [00] [30] [00] [95]
avrdude: usbasp_spi_cmd() => 0x00, 0x30, 0x00, 0x95
Reading | #################                                  | 33% 0.01 s
avrdude: usbasp_spi_cmd(0x30, 0x00, 0x02, 0x00) ...
avrdude: usbasp_transmit("USBASP_FUNC_TRANSMIT", 0x30, 0x00, 0x02, 0x00)
         <= [00] [30] [00] [16]
avrdude: usbasp_spi_cmd() => 0x00, 0x30, 0x00, 0x16
Reading | ################################################## | 100% 0.01 s
avrdude: device signature = 0x1e9516 (probably m328pb)
avrdude: usbasp_close()
avrdude: usbasp_transmit("USBASP_FUNC_DISCONNECT", 0x00, 0x00, 0x00, 0x00)

avrdude done.  Thank you.
```

</details>

askn37, 2023-11-03:
avrdude: Built Version 7.2-20231102 (9c894e15) builting now...
Therefore, this information is already outdated.

```
-avrdude: Version 7.2-arduino.1
+avrdude: Version 7.2-20231031 (76cbf91a)
 avrdude> dump eeprom
 >>> dump eeprom 0x0 0x40
 avrdude: jtagmkII_paged_load(.., eeprom, 32, 32)
 avrdude: jtagmkII_paged_load(): block_size at addr 0 is 32
 avrdude: jtagmkII_paged_load(): sending read memory command:
 avrdude: jtagmkII_send(): sending 10 bytes
-avrdude: send: . [1b] . [0d] . [00] 
-. [0a] . [00] . [00] . [00] 
-. [0e] . [05] " [22] 
-  [20] . [00] . [00] . [00] 
-. [00] . [00] . [00] . [00] 
-b [62] � [c3] 
+avrdude: send: . [1b] . [0e] . [00]
+. [0a] . [00] . [00] . [00]
+. [0e] . [05] " [22]
+  [20] . [00] . [00] . [00]
+. [00] . [14] . [00] . [00]
+. [87] . [15]
```

76cbf91a: This version incorrectly wrote the EEPROM address value directly in the offset field.

```
-avrdude: Version 7.2-20231031 (76cbf91a)
+avrdude: Version 7.2-20231102 (9c894e15)
 avrdude> dump eeprom
 >>> dump eeprom 0x0 0x40
 avrdude: jtagmkII_paged_load(.., eeprom, 32, 32)
 avrdude: jtagmkII_paged_load(): block_size at addr 0 is 32
 avrdude: jtagmkII_paged_load(): sending read memory command: 
 avrdude: jtagmkII_send(): sending 10 bytes
-avrdude: send: . [1b] . [0e] . [00]
-. [0a] . [00] . [00] . [00]
-. [0e] . [05] " [22]
-  [20] . [00] . [00] . [00]
-. [00] . [14] . [00] . [00]
-. [87] . [15]
+avrdude: ser_send: . [1b] . [0d] . [00] 
+. [0a] . [00] . [00] . [00] 
+. [0e] . [05] " [22] 
+  [20] . [00] . [00] . [00] 
+. [00] . [00] . [00] . [00] 
+b [62] . [c3]
```

9c894e15: This version reverted to the same behavior as v7.2. It's not a bug anymore.

mcuee, 2023-11-03:
**Serial send and receive in windows (-vvvv)**

1) MinGW build with libserialport, using `-c urclock`.

<details>
<summary> Click for full debug log </summary>

```
PS>.\avrdude -c urclock -P ch340 -p m328pb -xshowall -vvvv

avrdude: Version 7.2-20231102 (3eaa9f2)
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is C:\work\avr\avrdude_test\avrdude_bin\github_bin\pr1538\avrdude-mingw-x86_64\avrdude.conf

avrdude: compiled without readline library, cannot use avrdude -t -c urclock interactively
         but it is still possible to pipe: echo "d fl 0 32; quit" | tr \; \\n | avrdude -t -curclock
avrdude: showall set
         Using Port                    : COM4
         Using Programmer              : urclock
avrdude:  251 ms: enter urclock_getsync()
avrdude: ser_send: 0 [30]   [20]
avrdude: ser_recv(): programmer is not responding
avrdude:  335 ms: sleeping for 32 ms
avrdude: ser_send:   [20]   [20]
avrdude: ser_recv: . [a0] x [78]
avrdude: ser_send:   [20]   [20]
avrdude: ser_recv: . [a0] x [78]
avrdude: ser_send:   [20]
avrdude: ser_recv(): programmer is not responding
avrdude: ser_send:   [20]
avrdude:  562 ms: all good, ready to rock
         AVR Part                      : ATmega328PB
         Chip Erase delay              : 10500 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : possible i/o
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                 65    20     4    0 no       1024    4      0  3600  3600 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           flash                  65    10   128    0 yes     32768  128    256  4500  4500 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           efuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
                                           Block Poll               Page                       Polled
[... 159 lines cut ...]
         Urboot protocol for ATmega328PB
avrdude: ser_send: P [50]   [20]
avrdude: ser_recv: . [a0]
avrdude: ser_recv: x [78]
avrdude: AVR device initialized and ready to accept instructions
Reading |                                                    | 0% 0.00 s
avrdude: ur_readEF(urclock, ATmega328PB, flash, 000000A25C8FEA20, 0x007ffa, 6, F)
avrdude: ser_send: . [03] . [fa] . [7f] . [06]   [20]
avrdude: ser_recv: . [a0]
avrdude: ser_recv: . [04] . [00] . [92] . [cf] . [e1] ? [3f]
avrdude: ser_recv: x [78]
avrdude: ur_readEF(urclock, ATmega328PB, flash, 000000A25C8FEA20, 0x007df5, 11, F)
avrdude: ser_send: . [03] . [f5] } [7d] . [0b]   [20]
avrdude: ser_recv: . [a0]
avrdude: ser_recv: . [ff] . [ff] . [ff] . [ff] . [ff] . [ff] . [ff] . [ff] . [ff] . [ff] . [ff]
avrdude: ser_recv: x [78]
avrdude: ur_readEF(urclock, ATmega328PB, eeprom, 000000A25C8FE940, 0x000101, 6, E)
avrdude: ser_send: . [01] . [01] . [01] . [06]   [20]
avrdude: ser_recv: . [a0]
avrdude: ser_recv: . [ff] . [ff] . [ff] . [ff] . [ff] . [ff]
avrdude: ser_recv: x [78]
0000ffffffff 0000-00-00 00.00  application 0 store 0 meta 1 boot 512 u7.7 weu-hprac vector 0 (RESET) ATmega328PB
```

</details>

mcuee, 2023-11-03:
Unfortunately I can not test this one.

** Serial send and receive in windows over ethernet (-vvvv) --- these are the net_send() and net_recv() functions **

mcuee, 2023-11-03:
**Usbhid send and receive (-vvvvv) --- one more v the code uses msg_trace2()**

1) MinGW build with hidapi

<details>
<summary> Click for full debug log </summary>

```
PS>cd .\avrdude-mingw-x86_64\
PS>.\avrdude -c pkobn_updi -p avr64ea48 -vvvv

avrdude: Version 7.2-20231102 (3eaa9f2)
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is C:\work\avr\avrdude_test\avrdude_bin\github_bin\pr1538\avrdude-mingw-x86_64\avrdude.conf

         Using Port                    : usb
         Using Programmer              : pkobn_updi
avrdude: jtag3_open_updi()
avrdude: HV UPDI support: 1
avrdude: usbhid_open(): probing for max packet size
avrdude: usbhid_open(): setting max_xfer from DAP_Info response to 64
avrdude: found CMSIS-DAP compliant device, using EDBG protocol
avrdude: jtag3_getsync()

avrdude: jtag3_edbg_prepare()
avrdude: jtag3_edbg_prepare(): connection status 0x01
avrdude: sending sign-on command:
avrdude: jtag3_edbg_send(): sending 3 bytes
avrdude: jtag3_edbg_recv():
avrdude: jtag3_recv(): Got message seqno 0 (command_sequence == 0)

Raw message:
01  80  00
[general] OK
         AVR Part                      : AVR64EA48
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           fuses                   0     0     0    0 no         16    1      0     0     0 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           fuse0       wdtcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           fuse1       bodcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           fuse2       osccfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           fuse5       syscfg0     0     0     0    0 no          1    1      0     0     0 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           fuse6       syscfg1     0     0     0    0 no          1    1      0     0     0 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
[... 562 lines cut ...]
[AVR] OK
avrdude: jtag3_close()
avrdude: sending AVR sign-off command:
avrdude: jtag3_edbg_send(): sending 3 bytes
avrdude: jtag3_edbg_recv():
avrdude: jtag3_recv(): Got message seqno 17 (command_sequence == 17)

Raw message:
12  80  00
[AVR] OK
avrdude: sending sign-off command:
avrdude: jtag3_edbg_send(): sending 4 bytes
avrdude: jtag3_edbg_recv():
avrdude: jtag3_recv(): Got message seqno 18 (command_sequence == 18)

Raw message:
01  80  00
[general] OK

avrdude: jtag3_edbg_signoff()

avrdude done.  Thank you.
```

</details>

askn37, 2023-11-03:
```
avrdude: Version 7.2-20231102 (9c894e15)

avrdude: jtagmkII_read_byte(): sending read memory command: 
avrdude: jtagmkII_send(): sending 10 bytes
avrdude: ser_send: . [1b] . [09] . [00] 
. [0a] . [00] . [00] . [00] 
. [0e] . [05] . [c0] 
. [01] . [00] . [00] . [00] 
. [01] . [0f] . [00] . [01] 
. [93] . [bd]
avrdude: jtagmkII_recv():
avrdude: ser_recv: . [1b]
avrdude: ser_recv: . [09]
avrdude: ser_recv: . [00]
avrdude: ser_recv: . [02]
avrdude: ser_recv: . [00]
avrdude: ser_recv: . [00]
avrdude: ser_recv: . [00]
avrdude: ser_recv: . [0e]
avrdude: ser_recv: . [82] . [02]
avrdude: ser_recv: l [6c]
avrdude: ser_recv: . [da]

avrdude: jtagmkII_recv(): got message seqno 9 (command_sequence == 9)
avrdude: jtagmkII_recv: . [82] . [02]

Raw message:
0x82 0x02 
memory contents:
0x02  

avrdude: jtagmkII_read_chip_rev(): received chip silicon revision: 0x02
avrdude: silicon revision: 0.2
avrdude: AVR device initialized and ready to accept instructions
```

This is still a disadvantage for me.

```
/* Description of AVR067 */
MTYPE_IO_SHADOW    = 0x30  // cached IO registers?
MTYPE_SRAM         = 0x20  // target's SRAM or [ext.] IO registers
MTYPE_EEPROM       = 0x22  // EEPROM, what way?
MTYPE_EVENT        = 0x60  // ICE event memory
MTYPE_SPM          = 0xA0  // flash through LPM/SPM
MTYPE_FLASH_PAGE   = 0xB0  // flash in programming mode
MTYPE_EEPROM_PAGE  = 0xB1  // EEPROM in programming mode
MTYPE_FUSE_BITS    = 0xB2  // fuse bits in programming mode
MTYPE_LOCK_BITS    = 0xB3  // lock bits in programming mode
MTYPE_SIGN_JTAG    = 0xB4  // signature in programming mode
MTYPE_OSCCAL_BYTE  = 0xB5  // osccal cells in programming mode
MTYPE_CAN          = 0xB6  // CAN mailbox
MTYPE_XMEGA_FLASH  = 0xC0  // xmega (app.) flash
MTYPE_BOOT_FLASH   = 0xC1  // xmega boot flash
MTYPE_EEPROM_XMEGA = 0xC4  // xmega EEPROM in debug mode
MTYPE_USERSIG      = 0xC5  // xmega user signature
MTYPE_PRODSIG      = 0xC6  // xmega production signature
```

Can someone clearly explain why the IO data memory bus is configured with MTYPE_XMEGA_FLASH=0xC0? This label is originally intended to refer to the code address bus after `APPCODE`. It must be a coincidence that I happen to be able to read the data bus.

mcuee, 2023-11-03:
@stefanrueger 

Just want to add some debug logs with similar testing devices as @MCUdude, but tested using latest PR #1538 under Windows with MSYS2 mingw64 build wth traces.

> > Which programmer expects -c jtag2updi and has FW 7 or above?
> 
> @stefanrueger none that I'm aware of. For reference, I've gathered the FW version of the Arduino Nano Every, Thinary Nano Every and a DIY jtag2updi:
> 
> This is the output from my official Arduino Nano Every:

My debug log. I got slightly newer FW version compared to Hans: `S_MCU FW version: 6.07`. 

You can also see that `-r` function works fine.

<details>
<summary> Click for full debug log for official Arduino Nano Every</summary>

```
PS>.\avrdude -c jtag2updi -P COM12 -p atmega4809 -vvvv -r

avrdude: Version 7.2-20231102 (3eaa9f2)
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is C:\work\avr\avrdude_test\avrdude_bin\github_bin\pr1538\avrdude-mingw-x86_64\avrdude.conf

avrdude: touching serial port COM12 at 1200 baud
avrdude: waiting for new port... using same port COM12
         Using Port                    : COM12
         Using Programmer              : jtag2updi
avrdude: jtagmkII_open_pdi()
avrdude: jtagmkII_getsync()
avrdude: jtagmkII_getsync() attempt 1 of 10: sending sign-on command:
avrdude: jtagmkII_send(): sending 1 bytes
avrdude: ser_send: . [1b] . [00] . [00] . [01] . [00] . [00] . [00] . [0e] . [01] . [f3] . [97]
avrdude: jtagmkII_recv():
avrdude: ser_recv: . [1b]
avrdude: ser_recv: . [00]
avrdude: ser_recv: . [00]
avrdude: ser_recv: . [1d]
avrdude: ser_recv: . [00]
avrdude: ser_recv: . [00]
avrdude: ser_recv: . [00]
avrdude: ser_recv: . [0e]
avrdude: ser_recv: . [86] . [01] . [01] . [07] . [01] . [01] . [01] . [07] . [06] . [01] . [00] . [00] . [00] . [00] . [00] . [00] J [4a] T [54] A [41] G [47] I [49] C [43] E [45]   [20] m [6d] k [6b] I [49] I [49] . [00]
avrdude: ser_recv: o [6f]
avrdude: ser_recv: V [56]

avrdude: jtagmkII_recv(): got message seqno 0 (command_sequence == 0)
avrdude: jtagmkII_recv: . [86] . [01] . [01] . [07] . [01] . [01] . [01] . [07] . [06] . [01] . [00] . [00] . [00] . [00] . [00] . [00] J [4a] T [54] A [41] G [47] I [49] C [43] E [45]   [20] m [6d] k [6b] I [49] I [49] . [00]

Raw message:
0x86 0x01 0x01 0x07 0x01 0x01 0x01 0x07 0x06 0x01 0x00 0x00 0x00 0x00 0x00 0x00
0x4a 0x54 0x41 0x47 0x49 0x43 0x45 0x20 0x6d 0x6b 0x49 0x49 0x00
Sign-on succeeded

JTAG ICE mkII sign-on message:
Communications protocol version: 1
M_MCU:
  boot-loader FW version:        1
  firmware version:              1.07
  hardware version:              1
S_MCU:
  boot-loader FW version:        1
  firmware version:              6.07
  hardware version:              1
Serial number:                   00:00:00:00:00:00
Device ID:                       JTAGICE mkII
avrdude: jtagmkII_getsync(): using a 298-byte device descriptor
avrdude: jtagmkII_setparm()
[... 503 lines cut ...]
avrdude: ser_recv: . [00]
avrdude: ser_recv: . [01]
avrdude: ser_recv: . [00]
avrdude: ser_recv: . [00]
avrdude: ser_recv: . [00]
avrdude: ser_recv: . [0e]
avrdude: ser_recv: . [80]
avrdude: ser_recv: . [b5]
avrdude: ser_recv: . [d8]

avrdude: jtagmkII_recv(): got message seqno 17 (command_sequence == 17)
avrdude: jtagmkII_recv: . [80]

Raw message:
0x80
OK

avrdude done.  Thank you.
```

</details>

> And this is the output from from the [Thinary Nano Every](https://www.aliexpress.com/item/4001060953783.html) (Atmega4808 based):

> And this is from a DIY jtag2updi I don't know when I flashed with the latest firmware available. Probably a year ago:

mcuee, 2023-11-03:
One more searial trace for Windows with MSYS2 mingw64 build (with libserialport) using `-c serialupdi`.

Take note I have to use `-b 57600` as this is a known issue with MSYS2 mingw compiler.

<details>
<summary>Click for full debug log for serialupdi with AVR64DD32</summary>

```
PS>.\avrdude -c serialupdi -P Ch340 -p avr64dd32 -vvvv -b 57600

avrdude: Version 7.2-20231102 (3eaa9f2)
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is C:\work\avr\avrdude_test\avrdude_bin\github_bin\pr1538\avrdude-mingw-x86_64\avrdude.conf

         Using Port                    : COM11
         Using Programmer              : serialupdi
         Overriding Baud Rate          : 57600
avrdude: opening serial port ...
avrdude: sending 1 bytes [0x00]
avrdude: ser_send: . [00]
avrdude: ser_recv: . [00]
         AVR Part                      : AVR64DD32
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           fuses                   0     0     0    0 no         16    1      0     0     0 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           fuse0       wdtcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           fuse1       bodcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           fuse2       osccfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           fuse5       syscfg0     0     0     0    0 no          1    1      0     0     0 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           fuse6       syscfg1     0     0     0    0 no          1    1      0     0     0 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           fuse7       codesize    0     0     0    0 no          1    1      0     0     0 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           fuse8       bootsize    0     0     0    0 no          1    1      0     0     0 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           lock                    0     0     0    0 no          4    1      0     0     0 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           prodsig     sigrow      0     0     0    0 no        128  128      0     0     0 0x00 0x00
[... 192 lines cut ...]
avrdude: ser_recv: U [55] . [08] . [02] . [11] . [00]
avrdude: ser_recv: . [1a]
avrdude: received 1 bytes [0x1a]
Reading | ################################################## | 100% 0.05 s
[... cut ...]
avrdude: sending 3 bytes [0x55, 0xc8, 0x59]
avrdude: ser_send: U [55] . [c8] Y [59]
avrdude: ser_recv: U [55] . [c8] Y [59]
avrdude: sending release reset request
avrdude: STCS 0x00 to address 0x08
avrdude: sending 3 bytes [0x55, 0xc8, 0x00]
avrdude: ser_send: U [55] . [c8] . [00]
avrdude: ser_recv: U [55] . [c8] . [00]
avrdude: STCS 0x0C to address 0x03
avrdude: sending 3 bytes [0x55, 0xc3, 0x0c]
avrdude: ser_send: U [55] . [c3] . [0c]
avrdude: ser_recv: U [55] . [c3] . [0c]

avrdude done.  Thank you.
```

</details>

mcuee, 2023-11-03:
One more usbdev traces for Windows with MSYS2 mingw64 build (with libusb-1.0+libusb-compat-0.1 API) using `-c avrispmkii` and ATxemga32A4u.

<details>
<summary>Click for full debug log using AVRISP mkII and ATxmega32A4U</summary>

```
PS>.\avrdude -c avrispmkii -p x32a4u -vvvv

avrdude: Version 7.2-20231102 (3eaa9f2)
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is C:\work\avr\avrdude_test\avrdude_bin\github_bin\pr1538\avrdude-mingw-x86_64\avrdude.conf

         Using Port                    : usb
         Using Programmer              : avrispmkII
avrdude: usbdev_open(): found AVRISP mkII, serno: 00000000193B
avrdude: usbdev_send: . [01]
avrdude: usbdev_recv_frame: . [01] . [00] . [0a] A [41] V [56] R [52] I [49] S [53] P [50] _ [5f] M [4d] K [4b] 2 [32]
avrdude: stk500v2_getsync(): found AVRISP mkII programmer
         AVR Part                      : ATxmega32A4U
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           fuse1                   0     0     0    0 no          1    1      0     0     0 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           fuse2                   0     0     0    0 no          1    1      0     0     0 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           fuse4                   0     0     0    0 no          1    1      0     0     0 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           fuse5                   0     0     0    0 no          1    1      0     0     0 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           lock                    0     0     0    0 no          1    1      0     0     0 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           prodsig     sigrow      0     0     0    0 no         52    1      0     0     0 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           data                    0     0     0    0 no          0    1      0     0     0 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           io                      0     0     0    0 no       4096    1      0     0     0 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           fuse0                   0     0     0    0 no          1    1      0     0     0 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
[... cut ...]
avrdude: usbdev_recv_frame: P [50] . [05] . [00] . [95]
Reading | #################                                  | 33% 0.01 s
avrdude: usbdev_send: P [50] . [05] . [01] . [01] . [00] . [00] . [92] . [00] . [01]
avrdude: usbdev_recv_frame: P [50] . [05] . [00] A [41]
Reading | ################################################## | 100% 0.02 s
avrdude: device signature = 0x1e9541 (probably x32a4u)
avrdude: usbdev_send: P [50] . [02]
avrdude: usbdev_recv_frame: P [50] . [02] . [00]

avrdude done.  Thank you.
```

</details>

mcuee, 2023-11-03:
> Can someone clearly explain why the IO data memory bus is configured with MTYPE_XMEGA_FLASH=0xC0? This label is originally intended to refer to the code address bus after `APPCODE`. It must be a coincidence that I happen to be able to read the data bus.
>

@askn37 
I think you might have a point.

@stefanrueger 
How do I know if the reading of `io` is correct. The following data look suspicious.
`0090  1e 95 41` --> looks like signature.

```
PS>.\avrdude -c avrispmkii -p x32a4u -qqt
avrdude> part

AVR Part                      : ATxmega32A4U
RESET disposition             : dedicated
RETRY pulse                   : SCK
Serial program mode           : yes
Parallel program mode         : yes
Memory Detail                 :

                                  Block Poll               Page                       Polled
  Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
  ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
  fuse1                   0     0     0    0 no          1    1      0     0     0 0x00 0x00
  fuse2                   0     0     0    0 no          1    1      0     0     0 0x00 0x00
  fuse4                   0     0     0    0 no          1    1      0     0     0 0x00 0x00
  fuse5                   0     0     0    0 no          1    1      0     0     0 0x00 0x00
  lock                    0     0     0    0 no          1    1      0     0     0 0x00 0x00
  prodsig     sigrow      0     0     0    0 no         52    1      0     0     0 0x00 0x00
  signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
  data                    0     0     0    0 no          0    1      0     0     0 0x00 0x00
  io                      0     0     0    0 no       4096    1      0     0     0 0x00 0x00
  fuse0                   0     0     0    0 no          1    1      0     0     0 0x00 0x00
  eeprom                  0     0     0    0 no       1024   32      0     0     0 0x00 0x00
  flash                   0     0     0    0 no      36864  256      0     0     0 0x00 0x00
  application             0     0     0    0 no      32768  256      0     0     0 0x00 0x00
  apptable                0     0     0    0 no       4096  256      0     0     0 0x00 0x00
  boot                    0     0     0    0 no       4096  256      0     0     0 0x00 0x00
  usersig                 0     0     0    0 no        256  256      0     0     0 0x00 0x00
avrdude> read fuse1
0000  00                                                |.               |
avrdude> read fuse2
0000  63                                                |c               |
avrdude> read fuse4
0000  1e                                                |.               |
avrdude> read fuse5
0000  3f                                                |?               |
avrdude> read fuse0
0000  55                                                |U               |
avrdude> read lock
0000  ff                                                |.               |
avrdude> read sigrow
0000  0d 40 7b 0a 40 7f ff 01  38 55 30 31 35 32 ff ff  | @{ @...8U0152..|
0010  03 ff 05 00 16 00 ff ff  ff ff cf 07 24 40 ff ff  |............$@..|
0020  00 00 00 ff ff ff ff ff  ff ff ff ff ff ff 5e 09  |..............^ |
0030  ff ff 02 15                                       |....            |
avrdude> read prodsig
0000  0d 40 7b 0a 40 7f ff 01  38 55 30 31 35 32 ff ff  | @{ @...8U0152..|
0010  03 ff 05 00 16 00 ff ff  ff ff cf 07 24 40 ff ff  |............$@..|
0020  00 00 00 ff ff ff ff ff  ff ff ff ff ff ff 5e 09  |..............^ |
0030  ff ff 02 15                                       |....            |
avrdude> read signature
0000  1e 95 41                                          |..A             |
avrdude> read eeprom 0 0x10
0000  54 68 65 20 71 75 69 63  6b 20 62 72 6f 77 6e 20  |The quick brown |
avrdude> read flash 0 0x10
0000  fd c0 00 00 18 c1 00 00  16 c1 00 00 14 c1 00 00  |................|
avrdude> read application 0 0x10
0000  fd c0 00 00 18 c1 00 00  16 c1 00 00 14 c1 00 00  |................|
[... 32 lines cut ...]
00c0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00d0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00e0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00f0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
avrdude> read data
avrdude error: (read) cannot read memory data of size 0
avrdude> read io
0000  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
0010  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
0020  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
0030  00 00 00 00 00 00 00 00  00 00 00 00 00 ff 2f 00  |............../.|
0040  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
0050  01 00 00 00 7b 02 00 00  00 00 00 00 00 00 00 01  |....{...........|
0060  00 00 40 0a 00 12 7a 00  00 00 40 0d 00 a1 07 00  |..@ ..z...@ ....|
0070  00 00 00 00 00 00 00 00  13 00 00 00 00 00 00 00  |................|
0080  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
0090  1e 95 41 04 00 00 00 00  00 00 00 00 00 00 00 00  |..A.............|
00a0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00b0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00c0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00d0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00e0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00f0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
avrdude> quit
```

mcuee, 2023-11-03:
> My Dragon uses the JTAG2 protocol use FW 7. And[ it seems like the JTAGmkII does as well](https://onlinedocs.microchip.com/pr/GUID-73C92233-8EC5-497C-92C3-D52ED257761E-en-US-1/index.html?GUID-60701625-2CF9-43AA-943E-D0F78B165880):

@MCUdude 

I can confirm that AVR Drago uses the JTAG2 protocol with FW 7. But I can not connect to ATxmega32A4U. I remember AVR Dragon has some limitations when it comes to PDI programming.

```
JTAG ICE mkII sign-on message:
Communications protocol version: 1
M_MCU:
  boot-loader FW version:        255
  firmware version:              7.39
  hardware version:              1
S_MCU:
  boot-loader FW version:        255
  firmware version:              7.39
  hardware version:              2
```

Run log with latest PR #1538 with trace function, MSYS2 mingw64 build with libusb-1.0+libusb-compat-0.1 (WinUSB driver).
```
PS>.\avrdude -c dragon_pdi -p x32a4u -vvvv

avrdude: Version 7.2-20231102 (3eaa9f2)
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is C:\work\avr\avrdude_test\avrdude_bin\github_bin\pr1538\avrdude-mingw-x86_64\avrdude.conf

         Using Port                    : usb
         Using Programmer              : dragon_pdi
avrdude: jtagmkII_dragon_open_pdi()
avrdude: usbdev_open(): found AVRDRAGON, serno: 00A200001487
avrdude: jtagmkII_getsync()
avrdude: jtagmkII_getsync() attempt 1 of 10: sending sign-on command:
avrdude: jtagmkII_send(): sending 1 bytes
avrdude: usbdev_send: . [1b] . [00] . [00] . [01] . [00] . [00] . [00] . [0e] . [01] . [f3] . [97]
avrdude: jtagmkII_recv():

avrdude: jtagmkII_recv(): got message seqno 0 (command_sequence == 0)
avrdude: jtagmkII_recv: . [86] . [01] . [ff] ' [27] . [07] . [01] . [ff] ' [27] . [07] . [02] . [00] . [a2] . [00] . [00] . [14] . [87] A [41] V [56] R [52] D [44] R [52] A [41] G [47] O [4f] N [4e] . [00]

Raw message:
0x86 0x01 0xff 0x27 0x07 0x01 0xff 0x27 0x07 0x02 0x00 0xa2 0x00 0x00 0x14 0x87
0x41 0x56 0x52 0x44 0x52 0x41 0x47 0x4f 0x4e 0x00
Sign-on succeeded

JTAG ICE mkII sign-on message:
Communications protocol version: 1
M_MCU:
  boot-loader FW version:        255
  firmware version:              7.39
  hardware version:              1
S_MCU:
  boot-loader FW version:        255
  firmware version:              7.39
  hardware version:              2
Serial number:                   00:a2:00:00:14:87
Device ID:                       AVRDRAGON
avrdude: jtagmkII_getsync(): using a 298-byte device descriptor
avrdude: jtagmkII_setparm()
avrdude: jtagmkII_setparm() sending set parameter command (parm emulator_mode 0x03, 1 bytes):
avrdude: jtagmkII_send(): sending 3 bytes
avrdude: usbdev_send: . [1b] . [01] . [00] . [03] . [00] . [00] . [00] . [0e] . [02] . [03] . [06] $ [24] 1 [31]
avrdude: jtagmkII_recv():

avrdude: jtagmkII_recv(): got message seqno 1 (command_sequence == 1)
avrdude: jtagmkII_recv: . [ae]

Raw message:
0xae
unknown message 0xae

avrdude jtagmkII_setparm() [jtagmkII.c:2542] error: bad response to set parameter emulator_mode: Unknown JTAG ICE mkII result code 0xae
avrdude main() [main.c:1280] error: unable to open port usb for programmer dragon_pdi

avrdude done.  Thank you.
```

askn37, 2023-11-03:
> How do I know if the reading of io is correct. The following data look suspicious.
> 0090  1e 95 41 --> looks like signature.

ATxmega32A4U [Production Signature Row] PDI Address base is **$08E0200** (25-bit address 32MiB memory map)

Why there? I'm sure the address offset is wrong...
(I only have the datasheet)

mcuee, 2023-11-03:
@MCUdude 

Since I cannot get my AVR Drago to connect to my ATxmega32A4U breakout board, please help to test your AVR Dragon with your xMega target and see if there ar anything strange (eg: to dump `io` and other memory types, like what I did using AVRISP mkII). Thanks.

askn37, 2023-11-03:
I've never thought about it before, but the "part" command doesn't help. In the Xmega generation "x32a4u" example, I would like to know the start address definition for each section, but it does not exist. There is even less information about the UPDI generation. What do "mode", "delay" and "polling index" mean? Where can I change them? Of course they were important to older generations of programmers. I feel a generation gap....

stefanrueger, 2023-11-03:
| How do I know if the reading of io is correct. The following data look suspicious.
| 0090  1e 95 41 --> looks like signature.

The `io` memory is the internal I/O register file. It was introduced in PR #1474. It is likely that not all programmers have been extended to be able to read/write it. And some cannot (for example ISP programming has no access; we have not yet found a way to write to `io` for XMEGAs). But in this case the result looks correct. The ATxmega32A4U has the MCU peripheral at address 0x90 in `io`, which in turn starts with the three device id bytes followed by the revision ID (4 in this case). See below.

[image]

And here the details of the MCU register:

[image]

stefanrueger, 2023-11-03:
| the "part" command doesn't help

Please! This is a venerable command as old as the terminal introduced at a time when there were only classic parts around. 

But yes, it is right to say that the AVRDUDE project should probably review the part command and extend it to XMEGAs and UPDI parts as to display relevant properties of these types. @MCUdude, this sounds right up your road? :wink:  The same is true for debug messages that ought to be updated with more relevant info (and perhaps consider removal of the repeated headlines).
```
         AVR Part                      : AVR32DA32
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           fuse0       wdtcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           fuse1       bodcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           fuse2       osccfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
```

| would like to know the start address definition for each section

BTW, you can always ask avrdude:
```
$ avrdude -px32a4u/At | grep offset
.ptmm	ATxmega32A4U	eeprom	offset	0x8c0000
.ptmm	ATxmega32A4U	flash	offset	0x800000
.ptmm	ATxmega32A4U	application	offset	0x800000
.ptmm	ATxmega32A4U	apptable	offset	0x807000
.ptmm	ATxmega32A4U	boot	offset	0x808000
.ptmm	ATxmega32A4U	fuse0	offset	0x8f0020
.ptmm	ATxmega32A4U	fuse1	offset	0x8f0021
.ptmm	ATxmega32A4U	fuse2	offset	0x8f0022
.ptmm	ATxmega32A4U	fuse4	offset	0x8f0024
.ptmm	ATxmega32A4U	fuse5	offset	0x8f0025
.ptmm	ATxmega32A4U	lock	offset	0x8f0027
.ptmm	ATxmega32A4U	prodsig	offset	0x8e0200
.ptmm	ATxmega32A4U	signature	offset	0x1000090
.ptmm	ATxmega32A4U	usersig	offset	0x8e0400
.ptmm	ATxmega32A4U	data	offset	0x1000000
.ptmm	ATxmega32A4U	io	offset	0x0
```

MCUdude, 2023-11-03:
The `part` command could need some love. The Memory detail table it prints is verbose, and IMO contains a lot of information you usually don't need to know (and can rather use the developer options instead).

If you could give me a list of what _should_ be there, (what what should be removed), I can see if I can create a more useful implementation.

```
         AVR Part                      : AVR32DA32
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Memory Detail                 :
```

* I've seen several AVRs with dedicated reset pins that say `RESET disposition: IO`. I don't see why we should even print this.
* What does RETRY pulse even mean?
* The AVR32DA32 doesn't have a parallel programming interface at all. Not sure why Avrdude thinks this...

mcuee, 2023-11-03:
> The part command could need some love. The Memory detail table it prints is verbose, and IMO contains a lot of information you usually don't need to know (and can rather use the developer options instead).

Totally agree.

stefanrueger, 2023-11-03:
@MCUdude Thanks! Yes, you are right. If it was up to me I would keep the info printed for the classic parts with the exception of
 - Memory poll index (not used by AVRDUDE and actually 0 in avrdude.conf for all memories of all parts)
 - Repeated headlines above a certain verbosity (was that ever useful to anyone?)

For the `PM_PDI`, `PM_UPDI` and (rare, only half-heartedly supported 32-bit) `PM_aWire` parts I trust that you have *way* more experience for what might be useful (eg, offset, size, page_size, readsize etc) than I; it is clear that the unused bits (mode, delay, readback etc) should not be displayed. 

> AVR32DA32 doesn't have a parallel programming interface at all. Not sure why Avrdude thinks this...

Haha, well spotted! I guess it's the *default* for classic parts (so parallel unless some bit is set in some flag), and hence this is incorrectly asserted for UPDI parts. 

Yes, this needs a bit TLC. Thank you for taking that on.

stefanrueger, 2023-11-03:
@askn37 This issue will be closed shortly. Let me reflect on some of the things observed:

> bug in the old version [...]  This version reverted to the same behavior as v7.2. It's not a bug anymore. [...]

We tend to talk about undocumented behaviour, when an *unsupported* programmer sends info back to AVRDUDE that makes it behave in a certain way. When AVRDUDE does not behave like an *unsupported* programmer expects we tend *not* to think that this is a bug. This may be unfortunate for the *unsupported* programmer, but it is not a bug.

> UPDI4AVR was fine in the 6.3 to 7.2 implementation, so this fact suggests a break in compatibility.

`updi4avr` is not on the list of currently 162 *supported* programmers (`avrdude -c\?`). It probably should be, but it it isn't. The updi4avr project uses undocumented behaviour by AVRDUDE: sending a FW version to AVRDUDE that makes it think it is connected to a supported programmer that might be connected to an XMEGA and therefore sets XMEGA parameters (eg, offsets for the application section or bootloader section). This is not a great way to establish a protocol between host downloader/uploader and the programmer FW:
 - Using an XMEGA structure for UPDI parts does not read well nor is it easy to maintain
 - AVRDUDE might decide to change undocumented behaviour for any or no reason. For example, an update of a supported programmer might require AVRDUDE follows suit. (How should the AVRDUDE project know that some programmer out there uses *undocumented* behaviour?)

A better way to support a new programmer is to submit to the AVRDUDE project a PR for an independent programmer module, eg, `updi4avr.c` that exports a `-c updi4avr` programmer. This way the AVRDUDE project *knows* about that programmer.

>  [mode, delay etc] were important to older generations of programmers

They *are* important to older generations of programmers! Just because new parts and programmers came along does not mean that all hardware with classic parts vanish in thin air. The AVRDUDE project prides itself with being able to handle 20-year old parts and programmers!

closed 2023-11-03 by stefanrueger

askn37, 2023-11-03:
@stefanrueger 

I also agree with you.

I have already started gathering information to create updi4avr.c. (I still don't know how to add new files to the project)
For the time being, the JTAGICEmkII protocol (AVR067) is the reference, so using jtagmkII.c is almost the same, but we will clean up unnecessary code that is not related to PM_UPDI. The only additional functionality will probably be high voltage support.
In the future JTAG2UPDI and its variants may also probably be taken care of. (The main difference between FWV=6 and FWV=7 is the presence or absence of address offsets for processing block chunks, so it doesn't require much effort.)