avrdudes/avrdude #1551: Programmer in .avrduderc not recognised

j00st, 2023-11-01:
On version 7.2, it seems that custom defined programmer configurations in .avrduderc are not recognised by the avrdude command. This works however on version 7.1. I couldn't find any related change in the changelog.

- I'm using a Raspberry Pi to flash a custom board with AtMega328PB chip;
- I've compiled `avrdude` with `linuxspi` enabled;
- I've defined the custom programmer in `/root/.avrduderc`;
- I've installed `avrdude` with `sudo` (trying without didn't help);
- I'm using Raspbian GNU/Linux 12 (bookworm)

**Content of the `.avrduderc` file:**
```
programmer
    id = "my_programmer";
    desc = "Use Linux SPI device in /dev/spidev*";
    type = "linuxspi";
    connection_type = spi;
    reset = 6;    # Pi GPIO number
;
```

**When running with 7.2:**
```
$ sudo avrdude -c my_programmer -p m328pb -P /dev/spidev0.0:/dev/gpiochip0 -U lfuse:w:0xe2:m -U hfuse:w:0xd9:m -U efuse:w:0xff:m Flash binary

avrdude error: cannot find programmer id my_programmer

Valid programmers are:
  2232hio            = 2232hio based on FT2232H with buffer and LEDs
  4232h              = FT4232H based generic programmer
  adafruit_gemma     = Adafruit Trinket Gemma bootloader disguised as USBtiny
  arduino            = Arduino for bootloader using STK500 v1 protocol
  arduino-ft232r     = Arduino: FT232R connected to ISP
  diecimila          = Arduino: FT232R connected to ISP
  arduino_as_isp     = Arduino board as programmer using arduino as ISP firmware
  arduino_gemma      = Arduino Gemma bootloader disguised as USBtiny
  arduinoisp         = Arduino ISP Programmer
  arduinoisporg      = Arduino ISP Programmer
etc
```

**Result with 7.1:**
```
$ sudo avrdude -c my_programmer -p m328pb -P /dev/spidev0.0:/dev/gpiochip0 -U lfuse:w:0xe2:m -U hfuse:w:0xd9:m -U efuse:w:0xff:m Flash binary

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9516 (probably m328pb)
avrdude: reading input file 0xe2 for lfuse
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte lfuse ...
avrdude: 1 byte of lfuse written
avrdude: verifying lfuse memory against 0xe2
avrdude: 1 byte of lfuse verified
avrdude: reading input file 0xd9 for hfuse
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte hfuse ...
avrdude: 1 byte of hfuse written
avrdude: verifying hfuse memory against 0xd9
avrdude: 1 byte of hfuse verified
avrdude: reading input file 0xff for efuse
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte efuse ...
avrdude: 1 byte of efuse written
avrdude: verifying efuse memory against 0xff
avrdude: 1 byte of efuse verified

avrdude done.  Thank you.
```

stefanrueger, 2023-11-02:
The only way I can reproduce this behaviour is by creating an, eg, empty file `~/.config/avrdude/avrdude.rc` which takes precedence over `~/.avrduderc` and hence the latter won't be opened. 

Here the excerpt from the manual (page 54 or thereabouts of git main):

AVRDUDE first looks for a system wide configuration file in a platform dependent location. On Unix, this is usually /usr/local/etc/avrdude.conf, whilst on Windows it is usually in the same location as the executable file. The full name of this file can be
specified using the -C command line option. After parsing the system wide configuration file, AVRDUDE looks for a per-user configuration file to augment or override the system wide defaults. On Unix, the per-user file is ${XDG_CONFIG_HOME}/avrdude/avrdude.rc, whereas if ${XDG_CONFIG_HOME} is either not set or empty, ${HOME}/.config/ is used instead. If that does not exists .avrduderc within the user’s home directory is used. On Windows, this file is the avrdude.rc file located in the same directory as the executable.

j00st, 2023-11-02:
I don't think this is what happened in my case. If I run `avrdude` with the verbose `-v` option, it actually shows that it's using the file at `/root/.avrduderc`:

```
$ sudo avrdude -c base -p m328pb -F -P /dev/spidev0.0:/dev/gpiochip0 -U lfuse:w:0xe2:m -U hfuse:w:0xd9:m -U efuse:w:0xff:m Flash binary -v

avrdude: Version 7.2-20231030 (2bb36b9b)
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is /usr/local/etc/avrdude.conf
         User configuration file is /root/.avrduderc

avrdude programmer_not_found() error: cannot find programmer id base

Valid programmers are:
  2232hio            = 2232hio based on FT2232H with buffer and LEDs [/usr/local/etc/avrdude.conf:687]
  4232h              = FT4232H based generic programmer [/usr/local/etc/avrdude.conf:769]
  adafruit_gemma     = Adafruit Trinket Gemma bootloader disguised as USBtiny [/usr/local/etc/avrdude.conf:1576]
  arduino            = Arduino for bootloader using STK500 v1 protocol [/usr/local/etc/avrdude.conf:575]
  arduino-ft232r     = Arduino: FT232R connected to ISP [/usr/local/etc/avrdude.conf:1372]
  diecimila          = Arduino: FT232R connected to ISP [/usr/local/etc/avrdude.conf:1372]
  arduino_as_isp     = Arduino board as programmer using arduino as ISP firmware 
```
Also I cannot find the file in the location in the home folder you mentioned, nor anywhere else on the system:
```
$ cat ~/.config/avrdude/avrdude.rc
cat: /home/tai/.config/avrdude/avrdude.rc: No such file or directory
$ sudo find / -name .avrduderc
/root/.avrduderc
```

To answer your previous question, `linuxspi` appears to be defined in `avrdude.conf`:
```
$ cat /usr/local/etc/avrdude.conf | grep linuxspi
# linuxspi
programmer # linuxspi
    id                     = "linuxspi";
    type                   = "linuxspi";
```

I can actually use version 7.1 for my purpose, but wanted to report my findings anyway.

stefanrueger, 2023-11-02:
> shows that it's using the file at /root/.avrduderc

Ahh, of course! Notice you are *running* avrdude with sudo, ie, as user root. There may be a subtle difference in 7.2 and 7.1 as to  how AVRDUDE figures out what the home directory `~` is, eg, whether the ${HOME} variable is used or `~`, which is likely to result in different paths depending whether sudo is used or not. 

`avrdude` does not need to (and actually should not!) run as root. My recommendation: Try running `avrdude` as yourself (no sudo). If you run into problems that AVRDUDE cannot read the port or files ensure you are either in the group that has access to the port or change permissions accordingly. These things are firmly outside of avrdude and depend on your OS and how you set things up.

j00st, 2023-11-02:
I can indeed run as non-root. But that still results in the same problem.

If I copy the `.avrduderc` file to `/home/pi/.config/avrdude/avrdude.rc` and try running it as `pi`:

```
pi@SIM1:~ $ whoami
pi
pi@SIM1:~ $ ls -l .config/avrdude/avrdude.rc
-rw-r--r-- 1 pi pi 185 Nov  2 11:53 .config/avrdude/avrdude.rc
pi@SIM1:~ $ cat .config/avrdude/avrdude.rc
programmer
   id = "my_programmer";
   desc = "Use Linux SPI device in /dev/spidev*";
   type = "linuxspi";
   connection_type = spi;
   reset = 6;    # Pi GPIO number - this is J8:22
;
pi@SIM1:~ $ avrdude -c my_programmer -p m328pb -P /dev/spidev0.0:/dev/gpiochip0 -U lfuse:w:0xe2:m -U hfuse:w:0xd9:m -U efuse:w:0xff:m Flash binary -v

avrdude: Version 7.2-20231102 (96b081b2)
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is /usr/local/etc/avrdude.conf
         User configuration file is /home/pi/.config/avrdude/avrdude.rc

avrdude programmer_not_found() error: cannot find programmer id my_programmer
```

Like I said, I'm fine with using the previous version. But if I understand the documentation right it seems like the programmer is simply not recognised. Maybe something else also changed between versions that I'm not yet aware of.

stefanrueger, 2023-11-02:
OK, I got it now and can reproduce it. The error message is wrong; the programmer was recognised but rejected owing to a missing `prog_modes` parameter. Use the following entry and it will work
```
#------------------------------------------------------------
# my_programmer
#------------------------------------------------------------

programmer # my_programmer
    id                     = "my_programmer";
    desc                   = "Use Linux SPI device in /dev/spidev*";
    type                   = "linuxspi";
    prog_modes             = PM_TPI | PM_ISP;
    connection_type        = spi;
    reset                  = 6;    # Pi GPIO number
;
```

I'll raise a PR so that future versions will say in this case
```
avrdude error: programmer my_programmer lacks prog_modes setting
```

Thanks, @j00st, for reporting and for helping in figuring out what was the problem.

j00st, 2023-11-02:
Ah check! That indeed fixes the issue. Glad we could sort this out, thanks for the active support!

PR #1557 (2023-11-11): Test programmers for prog_modes and type
Fixes #1551 

This corrects the error message when AVRDUDE cannot use a programmer to say it is owing to lack of `prog_modes` or `type`.
closed 2023-11-11 by stefanrueger