avrdudes/avrdude #1552: To add libserialport support to Windows MSVC build

mcuee, 2023-11-02:
As of now, Windows MSVC build does not support libserialport.

It will be good to add the support of libserialport for the Windows MSVC build.

mcuee, 2023-11-02:
One approach is to create a fork avrdude-libserialport under the avrdude organization. Change it and then integrated into MSVC build, same as the approach developed by @mariusgreuel.

mcuee, 2023-11-02:
**Known Windows specific issues.**

**Bug**
* https://github.com/avrdudes/avrdude/issues/968 (MSVC build only)
* https://github.com/avrdudes/avrdude/issues/1414

**Lack of features compared to Linux and macOS**
MSVC build
* https://github.com/avrdudes/avrdude/issues/1552
* https://github.com/avrdudes/avrdude/discussions/1440

MinGW build
* https://github.com/avrdudes/avrdude/issues/1271

mcuee, 2024-02-20:
@dl8dtl or @mariusgreuel or @MCUdude or @stefanrueger 

Just wondering if you can help to fork the following repo under the avrdudes organization. I do not have the permission to do that.
https://github.com/sigrokproject/libserialport

The idea is to add CMake files to libserialport fork and then try to integrate that into avrdude MSVC build.

MCUdude, 2024-02-20:
@mcuee here you go!
https://github.com/avrdudes/libserialport

mcuee, 2024-02-21:
> @mcuee here you go! https://github.com/avrdudes/libserialport

@MCUdude 

Thanks. Can you add me as a collaborator to the repo? I do not have the access to that repo now.

mcuee, 2024-02-24:
To do:
1) improve https://github.com/avrdudes/libserialport
2) to make it compatible with MSVC
3) adding CMake build script
4) integrate into avrdude MSVC build

No 2 and No 3 may require quite some efforts.

mcuee, 2024-03-17:
Remove this from avrdude 7.4 milestone since there is no active development on this topic.