avrdudes/avrdude #1567: usbasp not work in 7.2, but work in 5.1 !!!

Ku100vEvgen, 2023-11-11:
Hi !

found trouble, usbasp not work in 7.2, but work in 5.1 !!!

SET PROG_PATH7="hardware\avrdude\avrdude-v7.2-windows-windows-x86\"
SET PROG_PATH5="hardware\avrdude\avrdude-v5\"

SET PROG_CFG1=-p m168p -c usbasp       -v -P usb -U flash:r:flash_dump.hex:i
SET PROG_CFG2=-p m168p -c usbasp-clone -v -P usb -U flash:r:flash_dump.hex:i

rem WORK !
%PROG_PATH5%\avrdude.exe %PROG_CFG1%
pause

rem FAIL1
%PROG_PATH7%\avrdude.exe %PROG_CFG1%
pause

rem FAIL2
%PROG_PATH7%\avrdude.exe %PROG_CFG2%
pause

WORK:
avrdude.exe: Version 5.10, compiled on Sep  9 2010 at 15:36:47
avrdude.exe: AVR device initialized and ready to accept instructions
Reading | ################################################## | 100% 0.02s
avrdude.exe: Device signature = 0x1e940b
avrdude.exe: safemode: lfuse reads as EF
avrdude.exe: safemode: hfuse reads as D4
avrdude.exe: safemode: efuse reads as 1
avrdude.exe: reading flash memory:
....
avrdude.exe done.  Thank you.

FAIL1:
avrdude: Version 7.2
         Using Port                    : usb
         Using Programmer              : usbasp
avrdude usbasp_open() error: cannot find USB device with vid=0x16c0 pid=0x5dc ve
ndor='www.fischl.de' product='USBasp'
avrdude main() error: unable to open programmer usbasp on port usb
avrdude done.  Thank you.

FAIL2
avrdude: Version 7.2
         Using Port                    : usb
         Using Programmer              : usbasp-clone
avrdude usbasp_open() error: cannot find USB device with vid=0x16c0 pid=0x5dc
avrdude main() error: unable to open programmer usbasp-clone on port usb
avrdude done.  Thank you.

Ku100vEvgen, 2023-11-11:
[image]
[dbg_flash_read_avrdude2-ful_log.txt](https://github.com/avrdudes/avrdude/files/13324737/dbg_flash_read_avrdude2-ful_log.txt)

mcuee, 2023-11-11:
Which usbasp FW are you using and which Windows device driver are you using? 

It is recommend to use the following FW and WinUSB driver for avrdude 7.0 and later. If you want to use libusb0.sys driver (libusb-win32), please upgrade to 1.2.7.3 release or 1.3.0.1 snapshot.
https://github.com/nerdralph/usbasp

The following FW will have more functionalities but you will have to use MinGW build and not the official MSVC build if you use avrdude 7.2 binary.
https://github.com/avrdudes/avrdude/issues/968

Ku100vEvgen, 2023-11-11:
yes, it helped. Thank you for your promptness
[image]

closed 2023-11-11 by Ku100vEvgen

mcuee, 2023-11-12:
Thanks for the updates. Glad that you have sorted out the issue.