avrdudes/avrdude #1575: The commit message is wrong for https://github.com/avrdudes/avrdude/commit/71bf4f72a9ac327e6821a8f0c4797c784ce35b6d

mcuee, 2023-11-18:
The commit message is wrong for https://github.com/avrdudes/avrdude/commit/71bf4f72a9ac327e6821a8f0c4797c784ce35b6d.

mcuee, 2023-11-18:
It is probably better to fix the commit message.

stefanrueger, 2023-11-19:
The joys of a squash merge without manually and carefully editing the commit message.

The message *looks wrong* but it accurately describes the elements of the squash merge in that the branch that was ultimately squash/merged had merged changes of main into it (probably for avoidance of merge conflicts) that had happened during the development, so the additions to main will appear again as (squashed) commits... I normally edit squash-merge commit messages, but was too fast on this occasion

I think it's OK to leave as is.

mcuee, 2023-11-19:
Okay, I will close this issue.

closed 2023-11-19 by mcuee