avrdudes/avrdude #1578: Upload port required when using `-cdryrun` on MacOS

MCUdude, 2023-11-23:
Why do I have to specify an upload port (I've been using `-PUSB`) when using the dryrun programmer on MacOS?
Is it a bug, or does it just have to be this way?

```
$ ./avrdude -cdryrun -patmega1284p

avrdude error: no port has been specified on the command line or in the config file
        specify a port using the -P option and try again
```

stefanrueger, 2023-11-26:
When AVRDUDE is compiled, a default serial port is set for the OS but apparently not for MacOS. You can set `default_serial = "USB";` in ~/.avrduderc.

MCUdude, 2023-11-26:
But why is this different on MacOS than on other operating systems? And why does it happens only with `-c dryrun`? Is it because the missing `connection_type` in avrdude.conf`

in this code snippet, `pgm->conntype`is 0 when using dryrun.

https://github.com/avrdudes/avrdude/blob/9464b0c35a7e6b7d2c95cc00e6d66403201093cb/src/main.c#L1184-L1223

stefanrueger, 2023-11-26:
> missing `connection_type`

Hmm, yes, that would probably default to 0, which is `CONNTYPE_PARALLEL`. Better to change the code to ignore port setting on `-c dryrun`

PR #1581 (2023-12-01): Ignore port settings when using -c dryrun
Fixes #1578