avrdudes/avrdude #1584: Display string alignment instead of "space counting"

Ho-Ro, 2023-11-30:
Just a quick thought: Let the computer do this nasty work. Could a format like this help to maintain an uniform appearance more easily instead of manual space counting? 
```c
const int msg_space = 21;
msg_info("%s%-*s : 0x%02x\n", p, msg_space, "ICE HW version", hw);
```

_Originally posted by @Ho-Ro in https://github.com/avrdudes/avrdude/pull/1574#discussion_r1410374668_

stefanrueger, 2023-12-01:
It's a good idea, in principle. Would probably require sth like a `#define MSG_PROG_DISPLAY_COL 21` in some suitable header file so that each source file has access to the magic 21. I have done a quick search for strings containing ` : `. Turns out just one spacing is misbehaving:
```
$  grep '"[^"]* : [^"]*"' -rh src | tr "()," \\n | sed "s/^ *//" | grep ^'"%s'
"%sPin assignment        : 0..7 = DBUS0..7\n"
"%sAVR Part              : %s\n"
"%sProgramming modes     : %s\n"
"%sICE HW version        : %d\n"
"%sICE FW version        : %d.%02d 
"%sSerial number         : %s\n"
"%sVtarget               : %.2f V\n"
"%sJTAG clk Xmega        : %u kHz\n"
"%sJTAG clk prog.        : %u kHz\n"
"%sJTAG clk debug        : %u kHz\n"
"%sPDI/UPDI clk          : %u kHz\n"
"%sVout set              : %.2f V\n"
"%sVout measured         : %.02f V\n"
"%sCh A voltage          : %.03f V\n"
"%sCh A current          : %.3f mA\n"
"%sCh B voltage          : %.03f V\n"
"%sCh B current          : %.3f mA\n"
"%sMain MCU HW version   : %d\n"
"%sMain MCU FW version   : %d.%02d\n"
"%sSec. MCU HW version   : %d\n"
"%sSec. MCU FW version   : %d.%02d\n"
"%sSerial number         : %02x:%02x:%02x:%02x:%02x:%02x\n"
"%sVtarget               : %.1f V\n"
"%sJTAG clock            : %s 
"%sICE HW version        : 0x%02x\n"
"%sICE FW version        : 0x%02x\n"
"%sVtarget               : %.1f V\n"
"%sJTAG clock            : %s 
"%sHW Version            : %d\n"
"%sFW Version            : %d.%d\n"
"%sTopcard               : %s\n"
"%sVtarget               : %.1f V\n"
"%sVaref                 : %.1f V\n"
"%sOscillator            : "
"%sSCK period            : %.1f us\n"
"%sXTAL frequency        : %.*f %s\n"
"%sPin assignment        : 0..7 = DBUS0..7
"%sProgrammer Type       : %s\n"
"%sDescription           : %s\n"
"%sProgrammer model      : %s\n"
"%sHW version            : %d\n"
"%sSerial number         : %s\n"
"%sFW Version Controller : %d.%02d\n"
"%sTopcard               : %s\n"
"%sRouting card          : %s\n"
"%sSocket card           : %s\n"
"%sRC_ID table rev       : %d\n"
"%sEC_ID table rev       : %d\n"
"%sVtarget               : %.1f V\n"
"%sVaref                 : %.1f V\n"
"%sOscillator            : "
"%sSCK period            : %.1f us\n"
"%sXTAL frequency        : %.*f %s\n"
"%sSCK period            : %.1f us\n"
"%sSCK period            : %.1f us\n"
"%sVaref 0               : %.2f V\n"
"%sVaref 1               : %.2f V\n"
"%sSCK period            : %.1f us\n"
"%sOscillator            : %.3f %s\n"
"%sSCK period            : %.1f us\n"
"%sPin assignment          : /sys/class/gpio/gpio{n}\n"
"%sPin assignment        : libgpiod\n"
```

I have fixed that spacing by committing to git main straight away. 

Maybe this issue can be put on the back-burner for after the v7.3 release or perhaps closed for the time being (as it only becomes an issue again when we want to move the column arrangement for the programmer display).

Ho-Ro, 2023-12-01:
Ok, your grep & co magic solves the problem perfectly, no need for a complete reformatting - I will close.

closed 2023-12-01 by Ho-Ro