avrdudes/avrdude #1591: Make programmer -c name checking more forgiving

MCUdude, 2023-12-08:
Avrdude supports a bunch of different programmers, and sometimes if can be a bit hard to remember what the correct name for it is. Some programmers have dashes in them, like `digilent-hs2`. Others have underscores, like `dragon_pp`, and other has none, like `stk600pp`.

There are no good pattern or rule to follow; the names are what they are, some with underscores, some without.
But what if Avrdude could be a little more forgiving, and "understand" that the user probably wants to use `stk600pp` when using `-cstk600_pp` or `-cstk600-pp`, and `atmelice_isp` when using `-catmeliceisp` or `-catmelice-isp`?

dl8dtl, 2023-12-08:
Well, it reminds me: back in those old days, sendmail also tried to be forgiving in slightly misspelled names.

As a sysadmin, someday I noticed that one of the pseudo accounts named "yp" (for "yellow pages", later renamed "network information services" by Sun) had a mail in its mailbox. It contained a private mail from one of our employees, obviously by no means intended for that pseudo account at all.

I was a bit surprised, until one other sysadmin got me the clue: sendmail tries to deliver misspelled addressees.

Elm (a famous MUA then), by default asked at the end: "Copies?". It was meant to give the user a chance to add a Cc addressee. That employee, however, assumed it were asking him whether he'd like to get a copy, and he answered "y".

Sendmail then found the pseudo user "yp" to be the closest match for an addressee named "y", and delivered …

Bottom line: better no automatism. However, giving the user a hint might be good, so if the user specifies `-cstk600_pp`, they can be hinted that `Did you mean -c stk600pp?`.

MCUdude, 2023-12-08:
> However, giving the user a hint might be good, so if the user specifies -cstk600_pp, they can be hinted Did you mean -c stk600pp?.

Yes, that's just as good.
Currently, Avrdude prints out all 140+ supported programmer options if the programmer name doesn't match perfectly. It would be great if Avrdude could provide a suggestion, and explicitly mention that -c? will print all compatible programmers.

stefanrueger, 2023-12-09:
The set of programmers is small enough, finding a few close ones could be done with sth like the [Damerau-Levenshtein](https://en.wikipedia.org/wiki/Damerau%E2%80%93Levenshtein_distance) distance.  By someone else than me, I hasten to add :) The implementation could give insertion/deletion/mixup of hyphen, underscore a really small penalty (smaller than that of ordinary characters) so that merely misremembering these would most likely give the smallest Damerau-Levenshtein distance and thus the closest recommendation.

If implemented, for symmetry one might want to do sth similar for the parts, too.

I agree it is much better to ask Did you mean ... than to do something else than user specified. Just imagine someone with an older avrdude that did not yet have a certain programmer tried a command line that works for a newer one.

MCUdude, 2023-12-09:
Great suggestion @stefanrueger!

I found a C implementation of the Levenshtein distance. It does what it should:

``` c
#include <string.h>
#include <stdlib.h>
#include <stdint.h>
#include <stdio.h>

// Returns a size_t, depicting the difference between `a` and `b`.
// See <https://en.wikipedia.org/wiki/Levenshtein_distance> for more information.
size_t levenshtein_n(const char *a, const size_t length, const char *b, const size_t bLength) {
  // Shortcut optimizations / degenerate cases.
  if (a == b) {
    return 0;
  }

  if (length == 0) {
    return bLength;
  }

  if (bLength == 0) {
    return length;
  }

  size_t *cache = calloc(length, sizeof(size_t));
  size_t index = 0;
  size_t bIndex = 0;
  size_t distance;
  size_t bDistance;
  size_t result;
  char code;

  // initialize the vector.
  while (index < length) {
    cache[index] = index + 1;
    index++;
  }

  // Loop.
  while (bIndex < bLength) {
    code = b[bIndex];
    result = distance = bIndex++;
    index = SIZE_MAX;

    while (++index < length) {
      bDistance = code == a[index] ? distance : distance + 1;
      distance = cache[index];

      cache[index] = result = distance > result
        ? bDistance > result
          ? result + 1
          : bDistance
        : bDistance > distance
          ? distance + 1
          : bDistance;
    }
  }

  free(cache);

  return result;
}

size_t levenshtein(const char *a, const char *b) {
  const size_t length = strlen(a);
  const size_t bLength = strlen(b);

  return levenshtein_n(a, length, b, bLength);
}

int main()
{
    char* str;
    printf("Enter programmer: ");
    scanf("%ms", &str);
    const char *database[] = {"2232hio", "4232h", "adafruit_gemma", "arduino", "arduino-ft232r", "diecimila", "arduino_as_isp", "arduino_gemma", "arduinoisp", "arduinoisporg", "atmelice", "atmelice_jtag", "atmelice_dw", "atmelice_isp", "atmelice_pdi", "atmelice_tpi", "atmelice_updi", "avr109", "avr910", "avr911", "avrftdi", "2232h", "avrisp", "avrisp-u", "avrispmkII", "avrisp2", "avrispv2", "buspirate", "buspirate_bb", "butterfly", "butterfly_mk", "mkbutterfly", "bwmega", "c232hm", "c2n232i", "ch341a", "dasa", "dasa3", "digilent-hs2", "dragon_dw", "dragon_hvsp", "dragon_isp", "dragon_jtag", "dragon_pdi", "dragon_pp", "dryrun", "ehajo-isp", "flip1", "flip2", "flyswatter2", "ft2232h", "ft2232h_jtag", "ft232h", "ft232h_jtag", "ft232r", "ft245r", "ft4232h", "iseavrprog", "jtag1slow", "jtag2dw", "jtag2fast", "jtag2", "jtag2isp", "jtag2pdi", "jtag2slow", "jtag2updi", "jtag3", "jtag3dw", "jtag3isp", "jtag3pdi", "jtag3updi", "jtagkey", "jtagmkI", "jtag1", "jtagmkII", "jtagmkII_avr32", "jtag2avr32", "ktlink", "lm3s811", "mib510", "micronucleus", "nibobee", "o-link", "openmoko", "pavr", "pickit2", "pickit4", "pickit4_jtag", "pickit4_isp", "pickit4_pdi", "pickit4_tpi", "pickit4_updi", "pkobn_updi", "ponyser", "powerdebugger", "powerdebugger_jtag", "powerdebugger_dw", "powerdebugger_isp", "powerdebugger_pdi", "powerdebugger_tpi", "powerdebugger_updi", "serialupdi", "siprog", "snap", "snap_jtag", "snap_isp", "snap_pdi", "snap_tpi", "snap_updi", "stk500", "stk500hvsp", "scratchmonkey_hvsp", "stk500pp", "scratchmonkey_pp", "stk500v1", "stk500v2", "scratchmonkey", "stk600", "stk600hvsp", "stk600pp", "tc2030", "teensy", "tigard", "ttl232r", "tumpa", "tumpa-b", "tumpa_jtag", "um232h", "uncompatino", "urclock", "usbasp", "usbasp-clone", "usbtiny", "wiring", "xbee", "xplainedmini", "xplainedmini_isp", "xplainedmini_dw", "xplainedmini_tpi", "xplainedmini_updi", "xplainedpro", "xplainedpro_pdi", "xplainedpro_updi"};
    
    for(int i = 0; i < (sizeof(database) / sizeof(database[0])); i++) {
        int l = levenshtein(str, database[i]);
        if(l < 4) {
            printf("%d: %s\n", l, database[i]);
        }
    }
    
    free(str);
    return 0;
}
```

```
Enter programmer: ysbasp
3: snap
1: usbasp
```

```
Enter programmer: stk500hvpp
1: stk500hvsp
2: stk500pp
3: stk500v1
3: stk500v2
2: stk600hvsp
3: stk600pp
```

Do you think using Levenshtein alone is good enough, or do you think we should utilize `str_starts` in combination with Levenshtein when a larger part of the string is missing?

MCUdude, 2023-12-09:
@stefanrueger I haven't been able to find a working C implementation of Damerau-Levenshtein. Do you think Levenshtein is good enough? If not, I would love to get some help finding a working DL implementation.

stefanrueger, 2023-12-09:
You might strike lucky searching for weighted edit distance (the emphasis being on *weighted*, which allows you to assign different weights for different symbols, eg, underscore and hyphen carrying less penalty). Edit distance is a less posh name for the DL distance.

MCUdude, 2023-12-10:
I found a c implementation where swapping, insertion, substitution, and deletion can be weighted. I still have to figure out how to tweak the weighting to make it do what I want. I haven't fully understood yet _how_ the weighting affects the result.

```
#include <string.h>
#include <stdlib.h>
#include <stdio.h>

/*
 * This function implements the Damerau-Levenshtein algorithm to
 * calculate a distance between strings.
 *
 * Basically, it says how many letters need to be swapped, substituted,
 * deleted from, or added to string1, at least, to get string2.
 *
 * The idea is to build a distance matrix for the substrings of both
 * strings.  To avoid a large space complexity, only the last three rows
 * are kept in memory (if swaps had the same or higher cost as one deletion
 * plus one insertion, only two rows would be needed).
 *
 * At any stage, "i + 1" denotes the length of the current substring of
 * string1 that the distance is calculated for.
 *
 * row2 holds the current row, row1 the previous row (i.e. for the substring
 * of string1 of length "i"), and row0 the row before that.
 *
 * In other words, at the start of the big loop, row2[j + 1] contains the
 * Damerau-Levenshtein distance between the substring of string1 of length
 * "i" and the substring of string2 of length "j + 1".
 *
 * All the big loop does is determine the partial minimum-cost paths.
 *
 * It does so by calculating the costs of the path ending in characters
 * i (in string1) and j (in string2), respectively, given that the last
 * operation is a substitution, a swap, a deletion, or an insertion.
 *
 * This implementation allows the costs to be weighted:
 *
 * - w (as in "sWap")
 * - s (as in "Substitution")
 * - a (for insertion, AKA "Add")
 * - d (as in "Deletion")
 *
 * Note that this algorithm calculates a distance _iff_ d == a.
 */
 
int damerau_levenshtein(const char *str1, const char *str2, int w, int s, int a, int d) {
  int len1 = strlen(str1), len2 = strlen(str2);
  int i, j;
  int *row0 = calloc(len2 + 1, sizeof(int*));
  int *row1 = calloc(len2 + 1, sizeof(int*));
  int *row2 = calloc(len2 + 1, sizeof(int*));

  for (j = 0; j <= len2; j++)
    row1[j] = j * a;
  for (i = 0; i < len1; i++) {
    int *dummy;
    row2[0] = (i + 1) * d;
    for (j = 0; j < len2; j++) {
      /* substitution */
      row2[j + 1] = row1[j] + s * (str1[i] != str2[j]);
      /* swap */
      if (i > 0 && j > 0 && str1[i - 1] == str2[j] && str1[i] == str2[j - 1] && row2[j + 1] > row0[j - 1] + w)
        row2[j + 1] = row0[j - 1] + w;
      /* deletion */
      if (row2[j + 1] > row1[j + 1] + d)
        row2[j + 1] = row1[j + 1] + d;
      /* insertion */
      if (row2[j + 1] > row2[j] + a)
        row2[j + 1] = row2[j] + a;
    }
    dummy = row0;
    row0 = row1;
    row1 = row2;
    row2 = dummy;
  }
  i = row1[len2];
  free(row0);
  free(row1);
  free(row2);
  return i;
}

int main() {
  const char *pgm[] = {"2232hio", "4232h", "adafruit_gemma", "arduino", "arduino-ft232r", "diecimila", "arduino_as_isp", "arduino_gemma", "arduinoisp", "arduinoisporg", "atmelice", "atmelice_jtag", "atmelice_dw", "atmelice_isp", "atmelice_pdi", "atmelice_tpi", "atmelice_updi", "avr109", "avr910", "avr911", "avrftdi", "2232h", "avrisp", "avrisp-u", "avrispmkII", "avrisp2", "avrispv2", "buspirate", "buspirate_bb", "butterfly", "butterfly_mk", "mkbutterfly", "bwmega", "c232hm", "c2n232i", "ch341a", "dasa", "dasa3", "digilent-hs2", "dragon_dw", "dragon_hvsp", "dragon_isp", "dragon_jtag", "dragon_pdi", "dragon_pp", "dryrun", "ehajo-isp", "flip1", "flip2", "flyswatter2", "ft2232h", "ft2232h_jtag", "ft232h", "ft232h_jtag", "ft232r", "ft245r", "ft4232h", "iseavrprog", "jtag1slow", "jtag2dw", "jtag2fast", "jtag2", "jtag2isp", "jtag2pdi", "jtag2slow", "jtag2updi", "jtag3", "jtag3dw", "jtag3isp", "jtag3pdi", "jtag3updi", "jtagkey", "jtagmkI", "jtag1", "jtagmkII", "jtagmkII_avr32", "jtag2avr32", "ktlink", "lm3s811", "mib510", "micronucleus", "nibobee", "o-link", "openmoko", "pavr", "pickit2", "pickit4", "pickit4_jtag", "pickit4_isp", "pickit4_pdi", "pickit4_tpi", "pickit4_updi", "pkobn_updi", "ponyser", "powerdebugger", "powerdebugger_jtag", "powerdebugger_dw", "powerdebugger_isp", "powerdebugger_pdi", "powerdebugger_tpi", "powerdebugger_updi", "serialupdi", "siprog", "snap", "snap_jtag", "snap_isp", "snap_pdi", "snap_tpi", "snap_updi", "stk500", "stk500hvsp", "scratchmonkey_hvsp", "stk500pp", "scratchmonkey_pp", "stk500v1", "stk500v2", "scratchmonkey", "stk600", "stk600hvsp", "stk600pp", "tc2030", "teensy", "tigard", "ttl232r", "tumpa", "tumpa-b", "tumpa_jtag", "um232h", "uncompatino", "urclock", "usbasp", "usbasp-clone", "usbtiny", "wiring", "xbee", "xplainedmini", "xplainedmini_isp", "xplainedmini_dw", "xplainedmini_tpi", "xplainedmini_updi", "xplainedpro", "xplainedpro_pdi", "xplainedpro_updi"};
  char* str;
  printf("Enter programmer: ");
  scanf("%ms", &str);

  for(int i = 0; i < (sizeof(pgm) / sizeof(pgm[0])); i++) {
    int l = damerau_levenshtein(str, pgm[i], 1, 1, 1, 1);
    if(l < 5)
      printf("%d: %s\n", l, pgm[i]);
  }

  free(str);
  return 0;
}
```

closed 2023-12-18 by stefanrueger