avrdudes/avrdude #1594: avrdude git main issue with SNAP and AVR DD

mcuee, 2023-12-18:
From the comments here by @avrfreak
* https://github.com/avrdudes/avrdude/discussions/1473#discussioncomment-7873906

I tested v7.3, it just hangs like avrdude cannot find the com port.
(just like in older versions if the programmer was not connected)
V7.0 & V7.2 both work well with my programmer.

FYI I use a snap debugger connected to the USB port.
The debugger is in avr mode.

I am using : avrdude -C avrdude.conf -c snap_updi -p avr32DD20 -B 4.0 -e -U flash:w:flash.hex:a -U eeprom:w:eep.hex:a

I have a few AVR16EB32's to test with if anyone has any ideas why it's not working with my programmer.

mcuee, 2023-12-18:
avrdude 7.2 release is okay as per @avrfreak.
https://github.com/avrdudes/avrdude/files/13696709/v72_LOG.txt

```

avrdude: Version 7.2
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is G:\avrdude-v7.2-windows-x86\avrdude.conf

         Using Port                    : usb
         Using Programmer              : snap_updi
         Setting bit clk period        : 4.0
avrdude: jtag3_open_updi()
avrdude: HV UPDI support: 1
avrdude: usbhid_open(): probing for max packet size
avrdude: usbhid_open(): setting max_xfer from DAP_Info response to 64
avrdude: found CMSIS-DAP compliant device, using EDBG protocol
avrdude: jtag3_getsync()

avrdude: jtag3_edbg_prepare()
avrdude: jtag3_edbg_prepare(): connection status 0x01
avrdude: sending sign-on command: 
avrdude: jtag3_edbg_send(): sending 3 bytes
avrdude: jtag3_edbg_recv():
avrdude: jtag3_recv(): Got message seqno 0 (command_sequence == 0)

Raw message:
01  80  00  
[general] OK
... (skipped) ...
```

mcuee, 2023-12-18:
@avrfreak

Please try the following binaries to see if there are any differences in the behavior. The issue seems to be related to HIDAPI.

Latest git main github action binaries.
1) Windows 32bit mingw32 cross-build using Arduino avrdude-packing script
https://github.com/avrdudes/avrdude/actions/runs/7242320665 
(download dist.zip and then get the 32bit Windows binary)

2) Windows 64bit mingw64 and msvc64 binaries
https://github.com/avrdudes/avrdude/actions/runs/7242320659
avrdude-mingw-x86_64.zip --> avrdude mingw64 binary with MSYS2 bundled hidapi library
avrdude-msvc-x64.zip --> avrdude msvc64 binary with [avrdude bundled hidapi library](https://github.com/avrdudes/libhidapi)

avrfreak, 2023-12-18:
I tested both versions msvc & mingw on windows 10 with a snap debugger as the programmer. They are both working with older devices but not with the new AVREB.
I will post the log files here & the other thread.

[msvc_logfile.txt](https://github.com/avrdudes/avrdude/files/13704759/msvc_logfile.txt)
[mingw_logfile.txt](https://github.com/avrdudes/avrdude/files/13704767/mingw_logfile.txt)

mcuee, 2023-12-19:
> I tested both versions msvc & mingw on windows 10 with a snap debugger as the programmer. They are both working with older devices but not with the new AVREB. I will post the log files here & the other thread.

Thanks for the updates.

As per @MCUdude, SNAP does not support AVR EB yet in AVR mode.
* https://github.com/avrdudes/avrdude/discussions/1473#discussioncomment-7888231

mcuee, 2023-12-19:
@avrfreak 

Just want to confirm that the issue is only with AVR EB. 

avrdude git main is working with AVR DD, right?

avrfreak, 2023-12-19:
V7.3 does not work with the SNAP debugger/programmer regardless of the device that is being programmed. It only works with V7.2. I mostly use avrdudeV7.0 with the snap programmer (all day today, never any problems)

[mingw_logfile.txt](https://github.com/avrdudes/avrdude/files/13711177/mingw_logfile.txt)
[msvc_logfile.txt](https://github.com/avrdudes/avrdude/files/13711180/msvc_logfile.txt)

avrfreak, 2023-12-19:
I would like to know how avrdude V7.3 was tested? What programmer was used? Will jtag2updi work with V7.3?
Thanks

mcuee, 2023-12-19:
> I would like to know how avrdude V7.3 was tested? What programmer was used? Will jtag2updi work with V7.3? Thanks

There is no avrdude 7.3 release yet. jtag2updi should work with avrdude git main (which is close to avrdude 7.3 if no big issues were found).

Please help to provide a clear description of the issue you encountered -- basically avrdude 7.2 works with the devices you tested (please list them) but not avrdude git main.

mcuee, 2023-12-19:
@avrfreak 

Still thanks a lot for carrying out the tests on git main. If this issue is confirmed to be a regression from avrdude 7.2, it will be a high priority to fix before avrdude 7.3 release.

mcuee, 2023-12-21:
I will need to set up my SNAP to test with an AVR Dx chip.

Just a first test with a classic ATmega32A first, using SNAP in JTAG mode and there is no issue (same for PICKit 4).

```
PS> .\avrdude_git -c snap -p m32a -U flash:v:m32a_blink_read.hex:i
avrdude_git: AVR device initialized and ready to accept instructions
avrdude_git: device signature = 0x1e9502 (probably m32a)

avrdude_git: processing -U flash:v:m32a_blink_read.hex:i
avrdude_git: verifying flash memory against m32a_blink_read.hex
Reading | ################################################## | 100% 0.08 s
avrdude_git: 526 bytes of flash verified

avrdude_git done.  Thank you.
```

Full debug log with `-vvvv`.

```
PS> .\avrdude_git -c snap -p m32a -U flash:v:m32a_blink_read.hex:i -vvvv

avrdude_git: Version 7.2-20231218 (67792ca8)
             Copyright the AVRDUDE authors;
             see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

             System wide configuration file is C:\work\avr\avrdude_test\avrdude_bin\avrdude.conf
             User configuration file is C:\work\avr\avrdude_test\avrdude_bin\avrdude.rc

             Using port            : usb
             Using programmer      : snap
             Default baud rate     : 57600
avrdude_git: jtag3_open()
avrdude_git: usbhid_open(): probing for max packet size
avrdude_git: usbhid_open(): setting max_xfer from DAP_Info response to 64
avrdude_git: found CMSIS-DAP compliant device, using EDBG protocol
avrdude_git: jtag3_getsync()

avrdude_git: jtag3_edbg_prepare()
avrdude_git: jtag3_edbg_prepare(): connection status 0x01
avrdude_git: sending sign-on command:
avrdude_git: jtag3_edbg_send(): sending 3 bytes
avrdude_git: jtag3_edbg_recv():
avrdude_git: jtag3_recv(): Got message seqno 0 (command_sequence == 0)

Raw message:
01  80  00
[general] OK
             AVR Part              : ATmega32A
             Programming modes     : ISP, HVPP, JTAG, JTAGmkI, SPM

             Memory        Size  Pg size
             ---------------------------
             eeprom        1024        4
             flash        32768      128
             hfuse            1        1
             lfuse            1        1
             lock             1        1
             calibration      4        1
             signature        3        1
             io              64        1
             sram          2048        1

             Variants       Package  F max   T range         V range
             --------------------------------------------------------------
             ATmega32A-AN   TQFP44   16 MHz  [-40 C, 105 C]  [2.7 V, 5.5 V]
             ATmega32A-ANR  TQFP44   16 MHz  [N/A,     N/A]  [2.7 V, 5.5 V]
             ATmega32A-AU   TQFP44   16 MHz  [-40 C,  85 C]  [2.7 V, 5.5 V]
             ATmega32A-AUR  TQFP44   16 MHz  [-40 C,  85 C]  [2.7 V, 5.5 V]
             ATmega32A-MN   VQFN44   16 MHz  [-40 C,  85 C]  [2.7 V, 5.5 V]
[... 571 lines cut ...]
Raw message:
12  80  00
[AVR] OK
avrdude_git: jtag3_close()
avrdude_git: sending AVR sign-off command:
avrdude_git: jtag3_edbg_send(): sending 3 bytes
avrdude_git: jtag3_edbg_recv():
avrdude_git: jtag3_recv(): Got message seqno 22 (command_sequence == 22)

Raw message:
12  80  00
[AVR] OK
avrdude_git: sending sign-off command:
avrdude_git: jtag3_edbg_send(): sending 4 bytes
avrdude_git: jtag3_edbg_recv():
avrdude_git: jtag3_recv(): Got message seqno 23 (command_sequence == 23)

Raw message:
01  80  00
[general] OK

avrdude_git: jtag3_edbg_signoff()

avrdude_git done.  Thank you.
```

stefanrueger, 2024-01-01:
@mcuee Do we have more intell on this to either confirm this issue or, failing that, close it?

avrfreak, 2024-01-01:
I just tested it again. Using a snap debugger. Here is the log file.

[logfile.txt](https://github.com/avrdudes/avrdude/files/13806216/logfile.txt)

avrfreak, 2024-01-01:
here is the exact same setup using avrdude v7.2. It is working.
[logfile.txt](https://github.com/avrdudes/avrdude/files/13806227/logfile.txt)

stefanrueger, 2024-01-01:
@avrfreak We have no doubt that you experience this issue. The problem we have is that @mcuee who has access to *a lot* of platforms has been unable to reproduce the issue, which is kind of important for us to investigate further.

The other peculiarity is that [some of your runs appear to have worked earlier on](https://github.com/avrdudes/avrdude/issues/1594#issuecomment-1860606717). Begs the question what was the *difference* in the setup/OS/libraries/machine comms on your side between when those binaries worked on your machine at the time and when they did not work just now. Knowing the answer to that might help us reproduce the issue and investigate further.

avrfreak, 2024-01-02:
I will try it on other computers. I have several. I will let you know.
Thank you

avrfreak, 2024-01-02:
It appears there are two drivers for the snap programmer/debugger.
Maybe this is the reason why avrdude soes not see the programmer.
I attached some pictures.
[image]
[image]

avrfreak, 2024-01-02:
I tried it on another PC. It looks like the driver is different.
It appears to be working with an avr16EA28.

[image]

[image]

mcuee, 2024-01-02:
> @mcuee Do we have more intell on this to either confirm this issue or, failing that, close it?

I can not produce the issue myself.

I can only think this is an issue with the HID interface of the SNAP. It could be an issue with HIDAPI, or a specific issue with @avrfreak's system, or it could be the way how avrdude is using hidapi.

mcuee, 2024-01-02:
@Youw 

The issue is quite strange to me. Just wondering if you have some idea to further debug this issue.

avrfreak, 2024-01-02:
I think something changed in avrdude because V7.2 always works with the snap programmer, no matter what driver version is installed on my computer. It's just this version that does not work with the snap programmer:
**Version 7.2-20231218 (67792ca)**

stefanrueger, 2024-01-02:
As long as we don't have a reproducible way to observe the described issue we are a bit stuck. @avrfreak You can help by isolate the conditions when the issue occurs for you and when it does not.

mcuee, 2024-01-02:
Last time it was reported that Microchip PICKit 2 (USB HID) may have some issues with some Windows applications. 

I am not so sure it is a similar issue here -- some Windows applications may interfere with Microchip SNAP or PICKit 4 (using USB HID interface for avrdude).

avrfreak, 2024-01-02:
> As long as we don't have a reproducible way to observe the described issue we are a bit stuck. @avrfreak You can help by isolate the conditions when the issue occurs for you and when it does not.

Yes I can help. The PC that does not work is a W7 x64 machine.
I have tested it on a laptop with W10 it appears to be working.

Again, previous avrdude versions before this are working so something must have changed in avrdude.

stefanrueger, 2024-01-03:
> The PC that does not work is a W7 x64 machine.

@mcuee Is that something you can confirm?

mcuee, 2024-01-03:
> > The PC that does not work is a W7 x64 machine.
> 
> @mcuee Is that something you can confirm?

Unfortunately I do not have legacy Windows installations (Windows 7/8/8.1). I only have Windows 10/11. Hopefully someone else with Windows 7 machine can chime in and help. I do not think Windows 7 is the real issue here though as HIDAPI should work fine under Windows 7 (it may even work in older version of Windows as it has not dropped the support of Windows XP yet).

@avrfreak 

Still I do not think Windows 7 is the real issue. If possible, please test in another Windows 7 machine.

avrfreak, 2024-01-04:
I tested in another windows 7 machine, still not working, however Avrdude v7.2 is working.
Something must have changed in the newer version of avrdude.

avrfreak, 2024-01-04:
Tested avrdude on a PC with windows 10 using the snap programmer with an AVR16EB32. Not working. Using the logic analyzer, it appears that the SIB is read but then it stops. Here is a verbose log:
[logfile.txt](https://github.com/avrdudes/avrdude/files/13830505/logfile.txt)

Another note, I can't seem to make the code tags work on this forum (when trying to post the log text)
It formats only the first portion. Is this a bug or maybe I am doing something wrong?
here is how the log appears:
`
avrdude: Version 7.2-20231218 (67792ca)
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is G:\avrdude\msvc\avrdude.conf

         Using port            : usb
         Using programmer      : snap_updi
         Setting bit clk period: 4.0 us
avrdude: jtag3_open_updi()
avrdude: HV UPDI support: 1
avrdude: usbhid_open(): probing for max packet size
avrdude: usbhid_open(): setting max_xfer from DAP_Info response to 64
avrdude: found CMSIS-DAP compliant device, using EDBG protocol
avrdude: jtag3_getsync()

avrdude: jtag3_edbg_prepare()
avrdude: jtag3_edbg_prepare(): connection status 0x01
avrdude: sending sign-on command: 
avrdude: jtag3_edbg_send(): sending 3 bytes
avrdude: jtag3_edbg_recv():
avrdude: jtag3_recv(): Got message seqno 0 (command_sequence == 0)

Raw message:
01  80  00  
[general] OK
avrdude: input file flash.hex auto detected as Intel Hex
avrdude: input file eep.hex auto detected as Intel Hex
         AVR Part              : AVR16EB32
         Programming modes     : UPDI, SPM

         Memory            Size  Pg size    Offset
         -----------------------------------------
         eeprom             512        8    0x1400 
         flash            16384       64  0x800000 
         fuses               16        1    0x1050 
         fuse0/wdtcfg         1        1    0x1050 
         fuse1/bodcfg         1        1    0x1051 
         fuse2/osccfg         1        1    0x1052 
         fuse5/syscfg0        1        1    0x1055 
         fuse6/syscfg1        1        1    0x1056 
         fuse7/codesize       1        1    0x1057 
         fuse8/bootsize       1        1    0x1058 
         fusea/pdicfg         2        1    0x105a 
         lock                 4        1    0x1040 
         prodsig/sigrow     128      128    0x1080 
         signature            3        1    0x1080 
         tempsense            4        1    0x1084 
         sernum              16        1    0x1090 
         bootrow             64       64    0x1100 
         userrow/usersig     64       64    0x1200 
         io                4160        1         0 
         sram              2048        1    0x7800 
         sib                 32        1         0 

         Variants             Package  F max   T range         V range       
         --------------------------------------------------------------------
         AVR16EB32-VQFN/TQFP  VQFN32   20 MHz  [-40 C, 125 C]  [1.8 V, 5.5 V]

         Programmer Type       : JTAGICE3_UPDI
         Description           : MPLAB(R) SNAP in UPDI mode
avrdude: jtag3_getparm()
avrdude: sending get parameter (scope 0x01, section 0, parm 0) command: 
[... 141 lines cut ...]
        - use -B to set lower the bit clock frequency, e.g. -B 125kHz
        - use -F to override this check
avrdude: jtag3_close()
avrdude: sending AVR sign-off command: 
avrdude: jtag3_edbg_send(): sending 3 bytes
avrdude: jtag3_edbg_recv():
avrdude: jtag3_recv(): Got message seqno 14 (command_sequence == 14)

Raw message:
12  80  00  
[AVR] OK
avrdude: sending sign-off command: 
avrdude: jtag3_edbg_send(): sending 4 bytes
avrdude: jtag3_edbg_recv():
avrdude: jtag3_recv(): Got message seqno 15 (command_sequence == 15)

Raw message:
01  80  00  
[general] OK

avrdude: jtag3_edbg_signoff()

avrdude done.  Thank you.

`

mcuee, 2024-01-04:
> ok so you edited my post and changed the title of the thread. The main subject was dealing with the SNAP programmer in avrdude which was working until the newest beta release. The problem is NOT with windows 7, it is with avrdude.

The problem is avrdude under Windows 7.

But if you also encounter problems under Windows 10 or Windows 11, let me know and I will change the title back.

I edit your post to properly use the code tag.
```
This is an example.
```

mcuee, 2024-01-04:
Anyway, I can post your deleted post back here.

+++++++++++++
Tested avrdude on a PC with windows 10 using the snap debugger/programmer with an AVR16EB32. Not working. Anyone else tested an AVREB device with avrdude with any programmer?

```
avrdude: Version 7.2-20231218 (67792ca)
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is G:\avrdude\msvc\avrdude.conf

         Using port            : usb
         Using programmer      : snap_updi
         Setting bit clk period: 4.0 us
avrdude: jtag3_open_updi()
avrdude: HV UPDI support: 1
avrdude: usbhid_open(): probing for max packet size
avrdude: usbhid_open(): setting max_xfer from DAP_Info response to 64
avrdude: found CMSIS-DAP compliant device, using EDBG protocol
avrdude: jtag3_getsync()

avrdude: jtag3_edbg_prepare()
avrdude: jtag3_edbg_prepare(): connection status 0x01
avrdude: sending sign-on command: 
avrdude: jtag3_edbg_send(): sending 3 bytes
avrdude: jtag3_edbg_recv():
avrdude: jtag3_recv(): Got message seqno 0 (command_sequence == 0)

Raw message:
01  80  00  
[general] OK
avrdude: input file flash.hex auto detected as Intel Hex
avrdude: input file eep.hex auto detected as Intel Hex
         AVR Part              : AVR16EB32
         Programming modes     : UPDI, SPM

         Memory            Size  Pg size    Offset
         -----------------------------------------
         eeprom             512        8    0x1400 
         flash            16384       64  0x800000 
         fuses               16        1    0x1050 
         fuse0/wdtcfg         1        1    0x1050 
         fuse1/bodcfg         1        1    0x1051 
         fuse2/osccfg         1        1    0x1052 
         fuse5/syscfg0        1        1    0x1055 
         fuse6/syscfg1        1        1    0x1056 
         fuse7/codesize       1        1    0x1057 
         fuse8/bootsize       1        1    0x1058 
         fusea/pdicfg         2        1    0x105a 
         lock                 4        1    0x1040 
         prodsig/sigrow     128      128    0x1080 
         signature            3        1    0x1080 
         tempsense            4        1    0x1084 
         sernum              16        1    0x1090 
         bootrow             64       64    0x1100 
         userrow/usersig     64       64    0x1200 
         io                4160        1         0 
         sram              2048        1    0x7800 
         sib                 32        1         0 

         Variants             Package  F max   T range         V range       
         --------------------------------------------------------------------
         AVR16EB32-VQFN/TQFP  VQFN32   20 MHz  [-40 C, 125 C]  [1.8 V, 5.5 V]

         Programmer Type       : JTAGICE3_UPDI
         Description           : MPLAB(R) SNAP in UPDI mode
avrdude: jtag3_getparm()
avrdude: sending get parameter (scope 0x01, section 0, parm 0) command: 
avrdude: jtag3_edbg_send(): sending 6 bytes
[... 139 lines cut ...]
        - double check the connections and try again
        - use -B to set lower the bit clock frequency, e.g. -B 125kHz
        - use -F to override this check
avrdude: jtag3_close()
avrdude: sending AVR sign-off command: 
avrdude: jtag3_edbg_send(): sending 3 bytes
avrdude: jtag3_edbg_recv():
avrdude: jtag3_recv(): Got message seqno 14 (command_sequence == 14)

Raw message:
12  80  00  
[AVR] OK
avrdude: sending sign-off command: 
avrdude: jtag3_edbg_send(): sending 4 bytes
avrdude: jtag3_edbg_recv():
avrdude: jtag3_recv(): Got message seqno 15 (command_sequence == 15)

Raw message:
01  80  00  
[general] OK

avrdude: jtag3_edbg_signoff()

avrdude done.  Thank you.
```

mcuee, 2024-01-04:
@avrfreak 

Please help to try the mingw32 binary from the following github action build under your Windows 7 machines. The binary is inside the file `dist.zip`
https://github.com/avrdudes/avrdude/actions/runs/7397687787

avrfreak, 2024-01-04:
> @avrfreak
> 
> Please help to try the mingw32 Windows 32bit binary from the following github action build under your Windows 7 machines. The binary is inside the file `dist.zip`. https://github.com/avrdudes/avrdude/actions/runs/7397687787

`
G:\Jan4>avrdude -C avrdude.conf -c snap_
updi -p avr64EA28 -B 4.0 -e -U flash:w:flash.hex:a -U eeprom:w:eep.hex:a

avrdude warning: cannot determine file format for flash.hex, specify explicitly
avrdude warning: cannot determine file format for eep.hex, specify explicitly
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9620 (probably avr64ea28)
avrdude: erasing chip

avrdude: processing -U flash:w:flash.hex:a
avrdude error: cannot determine file format for flash.hex, specify explicitly
avrdude error: read from file flash.hex failed

avrdude done.  Thank you.

`

mcuee, 2024-01-05:
@avrfreak 

The code tag works -- you have to use triple ` and not single `.

mcuee, 2024-01-05:
@avrfreak 

Please post the hex files. Thanks.

Your latest issue is totally different from the previous posted error logs.

* https://github.com/avrdudes/avrdude/issues/1594#issuecomment-1873512395
```

avrdude: Version 7.2-20231218 (67792ca)
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is G:\avrdude7.3_beta\avrdude.conf

         Using port            : usb
         Using programmer      : snap_updi
         Setting bit clk period: 4.0 us
avrdude: jtag3_open_updi()
avrdude: HV UPDI support: 1
avrdude: usbhid_open(): probing for max packet size
avrdude: usbhid_open(): setting max_xfer from DAP_Info response to 64
avrdude: found CMSIS-DAP compliant device, using EDBG protocol
avrdude: jtag3_getsync()

avrdude: jtag3_edbg_prepare()
avrdude: jtag3_edbg_prepare(): connection status 0x01
avrdude: sending sign-on command: 
avrdude: jtag3_edbg_send(): sending 3 bytes
avrdude: Version 7.2-20231218 (67792ca)
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is G:\avrdude7.3_beta\avrdude.conf

         Using port            : usb
         Using programmer      : snap_updi
         Setting bit clk period: 4.0 us
avrdude: jtag3_open_updi()
avrdude: HV UPDI support: 1
avrdude: usbhid_open(): probing for max packet size
avrdude: usbhid_open(): setting max_xfer from DAP_Info response to 64
avrdude: found CMSIS-DAP compliant device, using EDBG protocol
avrdude: jtag3_getsync()

avrdude: jtag3_edbg_prepare()
avrdude: jtag3_edbg_prepare(): connection status 0x01
avrdude: sending sign-on command: 
avrdude: jtag3_edbg_send(): sending 3 bytes
```

stefanrueger, 2024-01-05:
> avrdude warning: cannot determine file format for flash.hex, specify explicitly

Is the problem that when compiled under Windows 7 avrdude can no longer recognise Intel Hex files? OK, *this* is info we can work with. OK, first of all you could do what avrdude suggests: specify the file type explicitly. Use `:i` instead of `:a`. 

So, perhaps The Windows 7 compiler does not understand the `"%2zx"` scanf format (`z` means the variable to be set is of type `size_t`)? Has been around for a long time.

mcuee, 2024-01-05:
@avrfreak and @stefanrueger 

The situation about Windows binaries is a bit complicated. We have three types of Windows binary from github actions, for latest git main.

1) MSVC build (using VS2022, x86, x64 and arm64 build), without libserialport, using avrdude's own libusb/libftdi, using avrdude's own GNU readline wrapper, using hidapi 0.13.0 release
https://github.com/avrdudes/avrdude/blob/main/.github/workflows/build.yml
https://github.com/avrdudes/avrdude/actions/runs/7397687783 (latest run for git main)

2) MSYS2 mingw32/64 (x86 and x64 build), with libserialport, with GNU Readline, with libusb-1.0.26 and libusb-compat-0.1, with libftdi1, with hidapi 0.14.0.
https://github.com/avrdudes/avrdude/blob/main/.github/workflows/build.yml
https://github.com/avrdudes/avrdude/actions/runs/7397687783 (latest run for git main)

3) using Arduino avrdude-packing script (32bit x86 build only), using a Linux container with cross compiler, without libserialport, with libusb-1.0.26 and libusb-compat-0.1.7, libftdi1-1.4, hidapi 0.12.0, GNU Readline 8.0.
https://github.com/avrdudes/avrdude/blob/main/.github/workflows/arduino_packing.yml
https://github.com/avrdudes/avrdude/actions/runs/7397687787 (latets run for git main)

More info about the docker container and cross compiler as well as the libraries used
https://github.com/arduino/crossbuild/blob/main/Dockerfile
https://github.com/arduino/crossbuild/tree/main/deps

mcuee, 2024-01-05:
@avrfreak 

Please try the other two Windows binaries on your Windows 7 machine as well to see if there are behavior differences. To make it more consistent, please use the 32bit x86 MSVC build and MSYS2 mingw32 build.
https://github.com/avrdudes/avrdude/actions/runs/7397687783

File names:
1) avrdude-msvc-x86 (13.7 MB)
2) avrdude-mingw-i686 (6.9 MB)

stefanrueger, 2024-01-05:
> situation about Windows 

I *suspect* the observed problem where avrdude fails to recognise the INtel hex file stems from a missing C99 flag (or the compiler not implementing ANSI C standard C99) when compiling for Windows 7.

Test with the following C file
```
#include <stdio.h>

int main() {
  size_t n;

  printf("type a number\n");
  if(1 == scanf("%zi", &n))
    printf("you entered %zu\n", n);
  else
    printf("cannot process %%zi format\n");
}
```

stefanrueger, 2024-01-05:
@avrfreak Thanks for reporting a problem you observed. Pro tips:
 - Create a separate issue for each problem
 - Tell us what exactly the problem is and how you came to experience it
     + OS
     + Programmer
     + Part
     + How connected/rigged up
     + AVRDUDE command line
     + Output of the command (sometimes it helps us if you upload a file with the `-vv` or `-vvv` output)
     + What you would have expected to happen

avrfreak, 2024-01-05:
It was the ":a" in the command line. I changed it to ":i" and it works.
I have used :a all along with no problems. I just copied the command line (with the ":a") from an avrdude GUI I used. I have been using this for years without problems.

If I get some time I will try the other binaries.
Thank you

mcuee, 2024-01-05:
@avrfreak 

Thanks for the update.

So at least the Arduino_packing 32bit Windows binary works under Windows 7 now once you use `:i` and not `:a`. 
Please try the other two binaries as well under Windows 7 when you got the time. 

@stefanrueger can probably fix this hex file issue with `:a` under Windows 7.

mcuee, 2024-01-05:
> @avrfreak Thanks for reporting a problem you observed. Pro tips:
> 
> * Create a separate issue for each problem
> * Tell us what exactly the problem is and how you came to experience it
>   
>   * OS
>   * Programmer
>   * Part
>   * How connected/rigged up
>   * AVRDUDE command line
>   * Output of the command
>   * Sometimes it also helps us if you upload a file with the `-vv` or `-vvv` output
>   * What you would have expected to happen

I have included this in the FAQ with minor addition. Feel free to update.
https://github.com/avrdudes/avrdude/wiki/FAQ#how-to-report-an-issue-to-avrdude-project

stefanrueger, 2024-01-05:
It is not clear to me what went wrong. I *suspect* the format `"%zx"` does not work as intended in the function `couldbe()`. The `z` length modifier was introduced in C99 (ie, made standard in 2000).

@mcuee can you find out using the CI whether this is so? Either compile the following C source and execute the `a.out` in the CI or add the body of `main()` to AVRDUDE's `main()` and execute `avrdude` in Windows 7; maybe @avrfreak could help and try this out, too.
```
#include <stdio.h>

int main() {
  size_t n;
  printf("%%zi format %s\n", 1 == sscanf("42", "%zi", &n) && n == 42? "OK": "fails");
}
```

*If* that turns out to be the problem then there could be several reasons:
 - The C compiler  for Windows is unable to compile C99 code (considered "the C language" from 2000!)
 - We miss a compile-time flag to ask the compiler to compile C99 standard
 - Should the executable be dynamically linked, it might be that the run-time system utilises a vvv old C-library (pre 2000?) that fails to understand the `"%zx"` format; in this case this might be unique to @avrfreak's system

It could still be a different reason why @avrfreak observes this behaviour.

mcuee, 2024-01-05:
@stefanrueger 

As mentioned, there are three scenarios for the Windows avrdude binaries -- different compiler toolchain and different libraries used.

1) MSVC toolchain in CI -- pretty up-to-date VS2022 v143 toolchain

Dependancies under my Windows 11 machine.
```
MINGW64 /c/work/avr/avrdude_test/avrdude_bin/github_bin/avrdude-msvc-x64
$ ldd avrdude.exe
        ntdll.dll => /c/WINDOWS/SYSTEM32/ntdll.dll (0x7ff8efaf0000)
        KERNEL32.DLL => /c/WINDOWS/System32/KERNEL32.DLL (0x7ff8ef260000)
        KERNELBASE.dll => /c/WINDOWS/System32/KERNELBASE.dll (0x7ff8ed1d0000)
        apphelp.dll => /c/WINDOWS/SYSTEM32/apphelp.dll (0x7ff8e9df0000)
        SETUPAPI.dll => /c/WINDOWS/System32/SETUPAPI.dll (0x7ff8ee030000)
        msvcrt.dll => /c/WINDOWS/System32/msvcrt.dll (0x7ff8edf80000)
        WS2_32.dll => /c/WINDOWS/System32/WS2_32.dll (0x7ff8edad0000)
        RPCRT4.dll => /c/WINDOWS/System32/RPCRT4.dll (0x7ff8ef7e0000)
        HID.DLL => /c/WINDOWS/SYSTEM32/HID.DLL (0x7ff8eb630000)
        WINMM.dll => /c/WINDOWS/SYSTEM32/WINMM.dll (0x7ff8de3f0000)
        WINUSB.DLL => /c/WINDOWS/SYSTEM32/WINUSB.DLL (0x7ff8e46e0000)
        ucrtbase.dll => /c/WINDOWS/System32/ucrtbase.dll (0x7ff8ed040000)

$ objdump -p ./avrdude.exe | grep 'DLL Name:'
        DLL Name: SETUPAPI.dll
        DLL Name: HID.DLL
        DLL Name: WS2_32.dll
        DLL Name: KERNEL32.dll
        DLL Name: WINMM.dll
        DLL Name: WINUSB.DLL
```

2) MSYS2 mingw toolchin in CI -- pretty up-to-date MinGW GCC toolchain
Dependancies under my Windows 11 machine.
```
MINGW64 /c/work/avr/avrdude_test/avrdude_bin/github_bin/avrdude-mingw-x86_64
$ ldd avrdude.exe
        ntdll.dll => /c/WINDOWS/SYSTEM32/ntdll.dll (0x7ff8efaf0000)
        KERNEL32.DLL => /c/WINDOWS/System32/KERNEL32.DLL (0x7ff8ef260000)
        KERNELBASE.dll => /c/WINDOWS/System32/KERNELBASE.dll (0x7ff8ed1d0000)
        ADVAPI32.dll => /c/WINDOWS/System32/ADVAPI32.dll (0x7ff8ee4e0000)
        msvcrt.dll => /c/WINDOWS/System32/msvcrt.dll (0x7ff8edf80000)
        sechost.dll => /c/WINDOWS/System32/sechost.dll (0x7ff8eded0000)
        RPCRT4.dll => /c/WINDOWS/System32/RPCRT4.dll (0x7ff8ef7e0000)
        SETUPAPI.dll => /c/WINDOWS/System32/SETUPAPI.dll (0x7ff8ee030000)
        USER32.dll => /c/WINDOWS/System32/USER32.dll (0x7ff8ef0b0000)
        win32u.dll => /c/WINDOWS/System32/win32u.dll (0x7ff8ed600000)
        HID.DLL => /c/WINDOWS/SYSTEM32/HID.DLL (0x7ff8eb630000)
        GDI32.dll => /c/WINDOWS/System32/GDI32.dll (0x7ff8ee4b0000)
        gdi32full.dll => /c/WINDOWS/System32/gdi32full.dll (0x7ff8ece60000)
        msvcp_win.dll => /c/WINDOWS/System32/msvcp_win.dll (0x7ff8ed630000)
        ucrtbase.dll => /c/WINDOWS/System32/ucrtbase.dll (0x7ff8ed040000)
        WS2_32.dll => /c/WINDOWS/System32/WS2_32.dll (0x7ff8edad0000)
        cfgmgr32.DLL => /c/WINDOWS/SYSTEM32/cfgmgr32.DLL (0x7ff8ecad0000)

$ objdump -p ./avrdude.exe | grep 'DLL Name:'
        DLL Name: ADVAPI32.dll
        DLL Name: HID.DLL
        DLL Name: KERNEL32.dll
        DLL Name: msvcrt.dll
        DLL Name: SETUPAPI.dll
        DLL Name: USER32.dll
        DLL Name: WS2_32.dll
```

3) MinGW 32bit toolchain using cross-compile in CI -- this one is using a bit older MinGW GCC version

Unfortunately I do not know how to check the dependancies here for the 32bit binary.
```
MINGW64 /c/work/avr/avrdude_test/avrdude_bin/github_bin/dist/avrdude_Windows_32bit/bin
$ ldd avrdude.exe
        ntdll.dll => /c/WINDOWS/SYSTEM32/ntdll.dll (0x7ff8efaf0000)
        ntdll.dll => /c/Windows/SysWOW64/ntdll.dll (0x77c60000)
        wow64.dll => /c/WINDOWS/System32/wow64.dll (0x7ff8ef3e0000)
        wow64base.dll => /c/WINDOWS/System32/wow64base.dll (0x7ff8edde0000)
        wow64win.dll => /c/WINDOWS/System32/wow64win.dll (0x7ff8ef020000)
        wow64con.dll => /c/WINDOWS/System32/wow64con.dll (0x7ff8ef000000)

$ objdump -p avrdude.exe    | grep 'DLL Name:'
        DLL Name: hid.dll
        DLL Name: KERNEL32.dll
        DLL Name: msvcrt.dll
        DLL Name: SETUPAPI.dll
        DLL Name: USER32.dll
        DLL Name: WS2_32.dll
```

I tend to think all three toolchains will have no issues with C99, but maybe the last one can be a potential issue. Based on the last report, the last one (MinGW cross-compiler) seems to have a problem with the auto-recognition of hex file format.

The C runtime library (msvcrt.dll) in Windows 7 can potentially be a problem as well. But if it has problem, then the other toolchain should have the same problem. Let's wait for the test results of @avrfreak using the three different binaries.

mcuee, 2024-01-05:
There are two issues reported by @avrfreak here.

1) avrdude can not communicate with the chip under Windows 7. This one is rather strange. I have no idea how it happens. But if the three binaries perform differently, then the issue could be the hidapi version used.

2) avrdude can not automatically recognize the hex file format with `:a`, also under Windows 7. This is also strange but at least @stefanrueger has some ideas why it happens.

avrfreak, 2024-01-05:
OK here are some results of my testing:
I tested avrdude-mingw-i686 under windows 7 x64.
This version does not work with the snap programmer. It does work with a serialupdi cable. The hex files are accepted using ":a".

`'`avrdude: Version 7.2-20240104 (03d786e)
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is
G:\avrdude-mingw-i686\avrdude.conf

         Using port            : usb
         Using programmer      : snap_updi
avrdude: jtag3_open_updi()
avrdude: HV UPDI support: 1
avrdude: usbhid_open(): probing for max packet size
avrdude: usbhid_open(): setting max_xfer from DAP_Info response to 64
avrdude: found CMSIS-DAP compliant device, using EDBG protocol
avrdude: jtag3_getsync()

avrdude: jtag3_edbg_prepare()
avrdude: jtag3_edbg_prepare(): connection status 0x01
avrdude: sending sign-on command: 
avrdude: jtag3_edbg_send(): sending 3 bytes
avrdude: jtag3_edbg_recv():
`'`
This one is using a serialupdi cable, same computer as above:
```
avrdude: Version 7.2-20240104 (03d786e)
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is 
G:\avrdude-mingw-i686\avrdude.conf

         Using port            : COM7
         Using programmer      : serialupdi
         Setting baud rate     : 230400
avrdude: opening serial port ...
avrdude: serial_baud_lookup(): using non-standard baud rate: 230400
avrdude: sending 1 bytes [0x00]
avrdude: ser_send: . [00]
avrdude: ser_recv: . [00]
avrdude: input file flash.hex auto detected as Intel Hex
avrdude: input file eep.hex auto detected as Intel Hex
         AVR Part              : AVR16EA28
         Programming modes     : UPDI, SPM

         Memory            Size  Pg size    Offset
         -----------------------------------------
         eeprom             512        8    0x1400 
         flash            16384       64  0x800000 
         fuses               16        1    0x1050 
         fuse0/wdtcfg         1        1    0x1050 
         fuse1/bodcfg         1        1    0x1051 
         fuse2/osccfg         1        1    0x1052 
         fuse5/syscfg0        1        1    0x1055 
         fuse6/syscfg1        1        1    0x1056 
         fuse7/codesize       1        1    0x1057 
         fuse8/bootsize       1        1    0x1058 
         lock                 4        1    0x1040 
         prodsig/sigrow     128      128    0x1100 
         signature            3        1    0x1100 
         tempsense            4        1    0x1104 
         sernum              16        1    0x1110 
         userrow/usersig     64       64    0x1080 
         io                4160        1         0 
         sram              2048        1    0x7800 
         sib                 32        1         0 

[... 3868 lines cut ...]
avrdude: avr_read_mem(): skipping page 58: no interesting data
avrdude: avr_read_mem(): skipping page 59: no interesting data
avrdude: avr_read_mem(): skipping page 60: no interesting data
avrdude: avr_read_mem(): skipping page 61: no interesting data
avrdude: avr_read_mem(): skipping page 62: no interesting data
avrdude: avr_read_mem(): skipping page 63: no interesting data
avrdude: verifying ...
avrdude: 128 bytes of eeprom verified
avrdude: leaving NVM programming mode
avrdude: sending reset request
avrdude: STCS 0x59 to address 0x08
avrdude: sending 3 bytes [0x55, 0xc8, 0x59]
avrdude: ser_send: U [55] . [c8] Y [59]
avrdude: ser_recv: U [55] . [c8] Y [59]
avrdude: sending release reset request
avrdude: STCS 0x00 to address 0x08
avrdude: sending 3 bytes [0x55, 0xc8, 0x00]
avrdude: ser_send: U [55] . [c8] . [00]
avrdude: ser_recv: U [55] . [c8] . [00]
avrdude: STCS 0x0C to address 0x03
avrdude: sending 3 bytes [0x55, 0xc3, 0x0c]
avrdude: ser_send: U [55] . [c3] . [0c]
avrdude: ser_recv: U [55] . [c3] . [0c]

avrdude done.  Thank you.```

avrfreak, 2024-01-05:
Here is the output from testing avrdude-msvc-x86 in windows 7 x64.
This one works with my snap programmer. The hex files area accepted using ":a".
```
avrdude: Version 7.2-20240104 (03d786e)
         Copyright the AVRDUDE authors;
         see 

https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is G:\avrdude-msvc-

x86\avrdude.conf

         Using port            : usb
         Using programmer      : snap_updi
avrdude: jtag3_open_updi()
avrdude: HV UPDI support: 1
avrdude: usbhid_open(): probing for max packet size
avrdude: usbhid_open(): setting max_xfer from DAP_Info response 

to 64
avrdude: found CMSIS-DAP compliant device, using EDBG protocol
avrdude: jtag3_getsync()

avrdude: jtag3_edbg_prepare()
avrdude: jtag3_edbg_prepare(): connection status 0x01
avrdude: sending sign-on command: 
avrdude: jtag3_edbg_send(): sending 3 bytes
avrdude: jtag3_edbg_recv():
avrdude: jtag3_recv(): Got message seqno 0 (command_sequence == 

0)

Raw message:
01  80  00  
[general] OK
avrdude: input file flash.hex auto detected as Intel Hex
avrdude: input file eep.hex auto detected as Intel Hex
         AVR Part              : AVR16EA28
         Programming modes     : UPDI, SPM

         Memory            Size  Pg size    Offset
         -----------------------------------------
         eeprom             512        8    0x1400 
         flash            16384       64  0x800000 
         fuses               16        1    0x1050 
         fuse0/wdtcfg         1        1    0x1050 
         fuse1/bodcfg         1        1    0x1051 
         fuse2/osccfg         1        1    0x1052 
         fuse5/syscfg0        1        1    0x1055 
         fuse6/syscfg1        1        1    0x1056 
         fuse7/codesize       1        1    0x1057 
         fuse8/bootsize       1        1    0x1058 
         lock                 4        1    0x1040 
         prodsig/sigrow     128      128    0x1100 
         signature            3        1    0x1100 
         tempsense            4        1    0x1104 
         sernum              16        1    0x1110 
         userrow/usersig     64       64    0x1080 
         io                4160        1         0 
         sram              2048        1    0x7800 
         sib                 32        1         0 

         Variants   Package  F max   T range         V range    

   
         

----------------------------------------------------------
         AVR16EA28  SOIC28   20 MHz  [-40 C, 125 C]  [1.8 V, 
[... 5359 lines cut ...]
avrdude: ser_recv: . [00]
avrdude: received 1 bytes [0x00]
avrdude: NVMCMD 0 executing
avrdude: ST to 0x001000
avrdude: sending 5 bytes [0x55, 0x48, 0x00, 0x10, 0x00]
avrdude: ser_send: U [55] H [48] . [00] . [10] . [00]
avrdude: ser_recv: U [55] H [48] . [00] . [10] . [00]
avrdude: ser_recv: @ [40]
avrdude: received 1 bytes [0x40]
avrdude: sending 1 bytes [0x00]
avrdude: ser_send: . [00]
avrdude: ser_recv: . [00]
avrdude: ser_recv: @ [40]
avrdude: received 1 bytes [0x40]
###
avrdude: LD from 0x001002
avrdude: sending 5 bytes [0x55, 0x08, 0x02, 0x10, 0x00]
avrdude: ser_send: U [55] . [08] . [02] . [10] . [00]
avrdude: ser_recv: U [55] . [08] . [02] . [10] . [00]
avrdude: ser_recv: . [00]
avrdude: received 1 bytes [0x00]
avrdude: clear page buffer
avrdude: NVMCMD 15 executing
avrdude: ST to 0x001000
avrdude: sending 5 bytes [0x55, 0x48, 0x00, 0x10, 0x00]

mcuee, 2024-01-05:
@avrfreak 

Thanks a lot for the updates.

Please help to carry out the same tests under Windows 10 to see if the behavior is the same or not.

mcuee, 2024-01-05:
1) This is the version we recommend to the user under Windows.

MSVC build (using VS2022, x86, x64 and arm64 build), without libserialport, using avrdude's own libusb/libftdi, using avrdude's own readline wrapper, using hidapi 0.13.0 release
https://github.com/avrdudes/avrdude/blob/main/.github/workflows/build.yml
https://github.com/avrdudes/avrdude/wiki/Getting-Nightly-Builds-for-AVRDUDE#windows-static-linking-to-dependancies (latest run for git main)

Known Issues for this version:
* https://github.com/avrdudes/avrdude/issues/1552
* https://github.com/avrdudes/avrdude/issues/1253
* https://github.com/avrdudes/avrdude/issues/968
* https://github.com/avrdudes/avrdude/discussions/1440

2) This is the second choice for Windows users.

MSYS2 mingw32/64 (x86 and x64 build), with libserialport, with GNU Readline, with libusb-1.0.26 and libusb-compat-0.1, with libftdi1, with hidapi 0.14.0.
https://github.com/avrdudes/avrdude/blob/main/.github/workflows/build.yml
https://github.com/avrdudes/avrdude/wiki/Getting-Nightly-Builds-for-AVRDUDE#windows-static-linking-to-dependancies (latest run for git main)

Known limitation of this build
* https://github.com/avrdudes/avrdude/issues/1271 (timeout may be more likely to happen for `-c urclock` in terminal mode)

3) This is likely to be used by Arduino distributions under Windows, Linux and macOS.

using Arduino avrdude-packing script (32bit x86 build only), using a Linux container with cross compiler, without libserialport, with libusb-1.0.26 and libusb-compat-0.1.7, libftdi1-1.4, hidapi 0.12.0, GNU Readline 8.0.
https://github.com/avrdudes/avrdude/blob/main/.github/workflows/arduino_packing.yml
https://github.com/avrdudes/avrdude/wiki/Getting-Nightly-Builds-for-AVRDUDE#linux-macos-and-windows-static-linking-to-dependancies (latets run for git main)

Known limitation of this build
* https://github.com/avrdudes/avrdude/issues/1271 (timeout may be more likely to happen for `-c urclock` in terminal mode)
* https://github.com/arduino/crossbuild/issues/13 (no libserialport support)

avrfreak, 2024-01-06:
> @avrfreak
> 
> Thanks a lot for the updates.
> 
> Please help to carry out the same tests under Windows 10 to see if the behavior is the same or not.

I did all three tests of the above but in windows 10 everything is working perfectly.

So its just the avrdude-mingw-i686 under windows 7 x64.
This version does not work with the snap programmer.
Thank you

mcuee, 2024-01-06:
> I did all three tests of the above but in windows 10 everything is working perfectly.

Thanks for the updates.

> So its just the avrdude-mingw-i686 under windows 7 x64. This version does not work with the snap programmer. Thank you

Good that you can narrow down to a single version.

mcuee, 2024-01-06:
@stefanrueger 

Now we can see that under Windows 10, all three builds are good.

Under Windows 7:
1) MSVC build is all good. 
2) MSYS2 mingw build has real issues communicating with the AVR chip under Windows 7.
3) Arduino avrdude-packing build has issues with `:a` under Windows 7.

I tend to think we can live with this issue for avrdude 7.3 release. We can document the limitation in the Wiki.

mcuee, 2024-01-06:
@avrfreak 

Just want to check if this is a real regression from avrdude 7.2 or not, please test with my own build of avrdude 7.2 release using MSYS2 mingw.

Please try the following binary under your Windows 7 machine to see if it has the same issue or not. Thanks.
https://github.com/mcuee/avrdude/releases/tag/v7.2
https://github.com/mcuee/avrdude/releases/download/v7.2/avrdude-v7.2_mingw32.zip

stefanrueger, 2024-01-06:
@mcuee I still think we should get to the bottom of this. None of the maintainers has access to Windows 7. We therefore need help from people who do.

Please, @mcuee
 - Apply the following patch and create Windows 7 executables named avrdude-z-*toolchain* or similar
   ``` diff
   diff --git a/src/main.c b/src/main.c
   index 9256e197..24ca6b0b 100644
   --- a/src/main.c
   +++ b/src/main.c
   @@ -618,6 +618,9 @@ static char *concatpath(char *dst, char *dir, char *file, size_t n) {
     */
    int main(int argc, char * argv [])
    {
   +  size_t n;
   +  printf("%%zi format %s\n", 1 == sscanf("42", "%zi", &n) && n == 42? "OK": "fails");
   +
      int              rc;          /* general return code checking */
      int              exitrc;      /* exit code for main() */
      int              i;           /* general loop counter */
   ```
 - Use the different toolchains and also apply the Arduino avrdude-packaging
 - Pass on these test executables to @avrfreak 

@avrfreak please test these executables on your machine. I would like to know whether avrdude-z-*toolchain* prints `%zi format OK` or `%zi format fails` *and* whether `:a` works or not (no `-v` needed for this, just use a valid Intel `.hex` file with `:a` and tell us whether it was recognised as Intel Hex and programmed or not).

Thanks for testing this.

mcuee, 2024-01-06:
@stefanrueger 

Done here.
https://github.com/mcuee/avrdude/commit/2f479a8b9726194afd3b0e21b00604ba63d12043

@avrfreak 
Please help to test the binaries here.
https://github.com/mcuee/avrdude/actions/runs/7430832657 
Download avrdude-msvc-x86.zip and avrdude-mingw-i686.zip

https://github.com/mcuee/avrdude/actions/runs/7430832655
Download dist.zip and then extract the files for 32bit Windows

mcuee, 2024-01-06:
For now I use the following discussion to document the three types of avrdude Windows binary. 
https://github.com/avrdudes/avrdude/discussions/1607

Maybe I will convert it into a Wiki entry, or just add to the following existing page.
https://github.com/avrdudes/avrdude/wiki/Getting-Nightly-Builds-for-AVRDUDE

mcuee, 2024-01-06:
Here are another set of binaries to debug the MSYS2 mingw binary issue with communication with the AVR chip.
[avrdude_mingw64_hidapi_version_test.zip](https://github.com/avrdudes/avrdude/files/13849432/avrdude_mingw64_hidapi_version_test.zip)

They are all built with the same toolchain (MSYS2 mingw64 under Windows 11) but with different hidapi version used.

```
MINGW64 /c/work/avr/avrdude_test/avrdude_bin/avrdude_mingw64_hidapi_version_test
$ ls
avrdude.conf  avrdude_hidapi_0.12.0.exe  avrdude_hidapi_0.13.1.exe  avrdude_hidapi_0.14.0.exe
```

avrfreak, 2024-01-06:
> @stefanrueger
> 
> Done here. [mcuee@2f479a8](https://github.com/mcuee/avrdude/commit/2f479a8b9726194afd3b0e21b00604ba63d12043)
> 
> @avrfreak Please help to test the binaries here. https://github.com/mcuee/avrdude/actions/runs/7430832657 Download avrdude-msvc-x86.zip and avrdude-mingw-i686.zip
> 
> https://github.com/mcuee/avrdude/actions/runs/7430832655 Download dist.zip and then extract the files for 32bit Windows

The dist version does not allow the use of ":a" for hex files in the command line.
the build msvc-x86 does allow both ";a" and ":i".
These tests were done on a W7 X64 computer.

avrfreak, 2024-01-06:
as for the build-mingw-i686 version, 
this version does not work with the snap programmer.
here are the logs:
[logfile_a.txt](https://github.com/avrdudes/avrdude/files/13849823/logfile_a.txt)
[logfile_i.txt](https://github.com/avrdudes/avrdude/files/13849825/logfile_i.txt)

using serialupdi
[logfile_serialupdi.txt](https://github.com/avrdudes/avrdude/files/13849826/logfile_serialupdi.txt)

This version works with either ":a" or ":i" in the command line.
I should comment that I am using an RS232 serial to TTL converter with the diode and resistor mod, and it's working wonderfully. This is a **real serial port**, not a USB conversion cable. This serialupdi adapter works in all 3 versions of avrdude just tested, and in windows 7 x64.

Please let me know if there are other tests to run.
Thank you

avrfreak, 2024-01-06:
I noticed in the logfile while programming an AVR16EA28:

```avrdude: LD from 0x001002
avrdude: sending 5 bytes [0x55, 0x08, 0x02, 0x10, 0x00]
avrdude: ser_send: U [55] . [08] . [02] . [10] . [00]
avrdude: ser_recv: U [55] . [08] . [02] . [10] . [00]
avrdude: ser_recv: . [00]
avrdude: received 1 bytes [0x00]```

What this is doing is reading the NVM  status register @ 0x001002 and waiting for the  busy flags to be cleared.

If you look at the datasheet in the EA devices, **the NVM status register has been moved to 0x001006**. Therefore the above command shouldn't work, or the zero value returned is invalid.

There is no register @ 0x001002 in the AVR EA devices.
Maybe someone can verify this.

Thanks
![EA_Registers](https://github.com/avrdudes/avrdude/assets/111312821/919fffb9-06aa-4a1f-8534-b1fb6b6d8498)
![EA_Registers2](https://github.com/avrdudes/avrdude/assets/111312821/f170209e-5040-4c93-83be-a348a21b5ca9)

mcuee, 2024-01-06:
> The dist version does not allow the use of ":a" for hex files in the command line. the build msvc-x86 does allow both ";a" and ":i". These tests were done on a W7 X64 computer.

Please post the run log with `:a` which failed. Thanks.

[55 more replies]

closed 2024-01-08 by stefanrueger