avrdudes/avrdude #1595: AVR-DU series latest information

askn37, 2023-12-18:
I found that the AT pack repository has been updated and `Microchip.AVR-Dx_DFP.2.4.286.atpack` has been published.

> https://packs.download.microchip.com : From pack-status-202312141113.3240.xml

> 2.4.286 (2023-12-12) - Added AVR64DU28 and AVR64DU32.

**AVR-DU series** is now available. No new datasheets found yet. However, ATDF and header files can be used.

What can be learned from the current public information;

- The register structure of NVMCTRL version 4 (AVR-DU) appears to be similar to version 5 (AVR-EB). It may be less compatible with version 2 (AVR-DA/DB).
- SIGNATURE address and his AVR-EB changed. **This is important.**
- Added BOOTROW memory area.
- PDICFG has been added to the FUSE area.

Whatever support we provide in the future, the first thing we need to do is update `avrdude.conf.in`.

MCUdude, 2023-12-18:
@xedbg do you know when the AVR-DUs will be available?

The 7.3 release is planned for late December/early January, and I don't think I'm able to obtain any DU chips in the meantime.
NVMCTRL 4 [is documented](https://github.com/microchip-pic-avr-tools/pymcuprog/blob/main/pymcuprog/serialupdi/nvmp4.py) in the latest [pymcuprog](https://github.com/microchip-pic-avr-tools/pymcuprog) release, but has not yet been adopted to Avrdude.

> Whatever support we provide in the future, the first thing we need to do is update avrdude.conf.in.

@stefanrueger should decide whenever it makes sense to add the AVR-DUs to avrintel.c/h as a first step

mcuee, 2023-12-18:
I think we can leave AVR-DU support after official avrdude 7.3 release.

xedbg, 2023-12-18:
> @xedbg do you know when the AVR-DUs will be available?

Probably not before february...

askn37, 2023-12-19:
JTAG2UPDI (clone) has been updated. I can't test it because I don't have the actual chip or configuration file yet, but it's preparation for my next goal. https://github.com/askn37/jtag2updi

Currently, NVMCTRL version 4 assumes that the write algorithm is the same as version 2, although the control address is different. I haven't looked at the pymcuprog code at all. Assumptions are made solely from information available from the ATDF and io header files. (Completely different implementations are for proper comparison and evaluation purposes)

closed 2024-01-03 by stefanrueger