avrdudes/avrdude #1599: Minor compiler warning under NetBSD

mcuee, 2023-12-20:
Build log under NetBSD 9.3.

@stefanrueger 
Please help to check if this is a real issue or not. Thanks.

BTW, no compiler warnings under OpenBSD 7.4 (OpenBSD clang 13.0).

```
localhost$ ./build.sh
-- The C compiler identification is GNU 7.5.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Git: /usr/pkg/bin/git (found version "2.42.0")
-- Found FLEX: /usr/bin/flex (found version "2.6.4")
-- Found BISON: /usr/pkg/bin/bison (found version "3.8.2")
-- Looking for libelf.h
-- Looking for libelf.h - found
-- Looking for libelf/libelf.h
-- Looking for libelf/libelf.h - found
-- Looking for usb.h
-- Looking for usb.h - found
-- Looking for lusb0_usb.h
-- Looking for lusb0_usb.h - not found
-- Looking for libusb.h
-- Looking for libusb.h - not found
-- Looking for libusb-1.0/libusb.h
-- Looking for libusb-1.0/libusb.h - found
-- Looking for hidapi/hidapi.h
-- Looking for hidapi/hidapi.h - found
-- Looking for ftdi_tcioflush
-- Looking for ftdi_tcioflush - found
-- Configuration summary:
-- ----------------------
-- DO HAVE    libelf
-- DO HAVE    libusb
-- DO HAVE    libusb_1_0
-- DO HAVE    libhidapi
-- DO HAVE    libftdi (but prefer to use libftdi1)
-- DO HAVE    libftdi1
-- DO HAVE    libreadline
-- DON'T HAVE libserialport
-- DISABLED   doc
-- DISABLED   parport
-- DISABLED   linuxgpio
-- DISABLED   linuxspi
-- ----------------------
-- Configuring done (1.7s)
-- Generating done (0.0s)
-- Build files have been written to: /home/mcuee/build/avrdude/build_netbsd
[  1%] Generating avrdude.conf
[  1%] Built target conf
[  2%] [BISON][Parser] Building parser with bison 3.8.2
[  4%] [FLEX][Parser] Building scanner with flex 2.6.4
[  5%] Building C object src/CMakeFiles/libavrdude.dir/arduino.c.o
[  7%] Building C object src/CMakeFiles/libavrdude.dir/avr.c.o
[  8%] Building C object src/CMakeFiles/libavrdude.dir/avr910.c.o
[  9%] Building C object src/CMakeFiles/libavrdude.dir/avrcache.c.o
[ 11%] Building C object src/CMakeFiles/libavrdude.dir/avrftdi.c.o
[ 12%] Building C object src/CMakeFiles/libavrdude.dir/avrftdi_tpi.c.o
[ 14%] Building C object src/CMakeFiles/libavrdude.dir/avrintel.c.o
[ 15%] Building C object src/CMakeFiles/libavrdude.dir/avrpart.c.o
[ 16%] Building C object src/CMakeFiles/libavrdude.dir/bitbang.c.o
[ 18%] Building C object src/CMakeFiles/libavrdude.dir/buspirate.c.o
[ 19%] Building C object src/CMakeFiles/libavrdude.dir/butterfly.c.o
[ 21%] Building C object src/CMakeFiles/libavrdude.dir/ch341a.c.o
[ 22%] Building C object src/CMakeFiles/libavrdude.dir/config.c.o
[ 23%] Building C object src/CMakeFiles/libavrdude.dir/confwin.c.o
[ 25%] Building C object src/CMakeFiles/libavrdude.dir/crc16.c.o
[ 26%] Building C object src/CMakeFiles/libavrdude.dir/dfu.c.o
[ 28%] Building C object src/CMakeFiles/libavrdude.dir/dryrun.c.o
[ 29%] Building C object src/CMakeFiles/libavrdude.dir/fileio.c.o
[ 30%] Building C object src/CMakeFiles/libavrdude.dir/flip1.c.o
[ 32%] Building C object src/CMakeFiles/libavrdude.dir/flip2.c.o
[ 33%] Building C object src/CMakeFiles/libavrdude.dir/ft245r.c.o
[ 35%] Building C object src/CMakeFiles/libavrdude.dir/jtagmkI.c.o
[ 36%] Building C object src/CMakeFiles/libavrdude.dir/jtagmkII.c.o
[ 38%] Building C object src/CMakeFiles/libavrdude.dir/jtag3.c.o
[ 39%] Building C object src/CMakeFiles/libavrdude.dir/leds.c.o
[ 40%] Building C object src/CMakeFiles/libavrdude.dir/linuxgpio.c.o
[ 42%] Building C object src/CMakeFiles/libavrdude.dir/linuxspi.c.o
[ 43%] Building C object src/CMakeFiles/libavrdude.dir/lists.c.o
[ 45%] Building C object src/CMakeFiles/libavrdude.dir/micronucleus.c.o
[ 46%] Building C object src/CMakeFiles/libavrdude.dir/par.c.o
[ 47%] Building C object src/CMakeFiles/libavrdude.dir/pgm.c.o
[ 49%] Building C object src/CMakeFiles/libavrdude.dir/pgm_type.c.o
[ 50%] Building C object src/CMakeFiles/libavrdude.dir/pickit2.c.o
[ 52%] Building C object src/CMakeFiles/libavrdude.dir/pindefs.c.o
[ 53%] Building C object src/CMakeFiles/libavrdude.dir/ppi.c.o
[ 54%] Building C object src/CMakeFiles/libavrdude.dir/ppiwin.c.o
[ 56%] Building C object src/CMakeFiles/libavrdude.dir/serbb_posix.c.o
[ 57%] Building C object src/CMakeFiles/libavrdude.dir/serbb_win32.c.o
[ 59%] Building C object src/CMakeFiles/libavrdude.dir/ser_avrdoper.c.o
[ 60%] Building C object src/CMakeFiles/libavrdude.dir/ser_posix.c.o
[ 61%] Building C object src/CMakeFiles/libavrdude.dir/ser_win32.c.o
[ 63%] Building C object src/CMakeFiles/libavrdude.dir/serialadapter.c.o
[ 64%] Building C object src/CMakeFiles/libavrdude.dir/serialupdi.c.o
[ 66%] Building C object src/CMakeFiles/libavrdude.dir/stk500.c.o
[ 67%] Building C object src/CMakeFiles/libavrdude.dir/stk500v2.c.o
[ 69%] Building C object src/CMakeFiles/libavrdude.dir/stk500generic.c.o
[ 70%] Building C object src/CMakeFiles/libavrdude.dir/strutil.c.o
[ 71%] Building C object src/CMakeFiles/libavrdude.dir/teensy.c.o
[ 73%] Building C object src/CMakeFiles/libavrdude.dir/term.c.o
[ 74%] Building C object src/CMakeFiles/libavrdude.dir/updi_link.c.o
[ 76%] Building C object src/CMakeFiles/libavrdude.dir/updi_nvm.c.o
[ 77%] Building C object src/CMakeFiles/libavrdude.dir/updi_readwrite.c.o
[ 78%] Building C object src/CMakeFiles/libavrdude.dir/updi_state.c.o
[ 80%] Building C object src/CMakeFiles/libavrdude.dir/urclock.c.o
[ 81%] Building C object src/CMakeFiles/libavrdude.dir/usbasp.c.o
[ 83%] Building C object src/CMakeFiles/libavrdude.dir/usb_hidapi.c.o
[ 84%] Building C object src/CMakeFiles/libavrdude.dir/usb_libusb.c.o
[ 85%] Building C object src/CMakeFiles/libavrdude.dir/usbtiny.c.o
[ 87%] Building C object src/CMakeFiles/libavrdude.dir/update.c.o
[ 88%] Building C object src/CMakeFiles/libavrdude.dir/wiring.c.o
[ 90%] Building C object src/CMakeFiles/libavrdude.dir/xbee.c.o
[ 91%] Building C object src/CMakeFiles/libavrdude.dir/__/lexer.c.o
[ 92%] Building C object src/CMakeFiles/libavrdude.dir/__/config_gram.c.o
[ 94%] Linking C static library libavrdude.a
[ 94%] Built target libavrdude
[ 95%] Building C object src/CMakeFiles/avrdude.dir/main.c.o
/home/mcuee/build/avrdude/src/main.c: In function 'suggest_programmers':
/home/mcuee/build/avrdude/src/main.c:507:12: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     if(dst > max_distance)
            ^
[ 97%] Building C object src/CMakeFiles/avrdude.dir/developer_opts.c.o
[ 98%] Building C object src/CMakeFiles/avrdude.dir/whereami.c.o
[100%] Linking C executable avrdude
[100%] Built target avrdude

Build succeeded.

Run

sudo cmake --build build_netbsd --target install

to install.

localhost$ cc -v
Using built-in specs.
COLLECT_GCC=cc
COLLECT_LTO_WRAPPER=/usr/libexec/lto-wrapper
Target: x86_64--netbsd
Configured with: /usr/src/tools/gcc/../../external/gpl3/gcc/dist/configure --target=x86_64--netbsd --enable-long-long --enable-threads --with-bugurl=http://www.NetBSD.org/Misc/send-pr.html --with-pkgversion='NetBSD nb4 20200810' --with-system-zlib --without-isl --enable-__cxa_atexit --enable-libstdcxx-time=rt --enable-libstdcxx-threads --with-diagnostics-color=auto-if-env --with-tune=nocona --with-default-libstdcxx-abi=new --with-mpc-lib=/var/obj/mknative/amd64-x86_64/usr/src/external/lgpl3/mpc/lib/libmpc --with-mpfr-lib=/var/obj/mknative/amd64-x86_64/usr/src/external/lgpl3/mpfr/lib/libmpfr --with-gmp-lib=/var/obj/mknative/amd64-x86_64/usr/src/external/lgpl3/gmp/lib/libgmp --with-mpc-include=/usr/src/external/lgpl3/mpc/dist/src --with-mpfr-include=/usr/src/external/lgpl3/mpfr/dist/src --with-gmp-include=/usr/src/external/lgpl3/gmp/lib/libgmp/arch/x86_64 --enable-tls --disable-multilib --disable-libstdcxx-pch --build=x86_64--netbsd --host=x86_64--netbsd --with-sysroot=/var/obj/mknative/amd64-x86_64/usr/src/destdir.amd64
Thread model: posix
gcc version 7.5.0 (nb4 20200810)
```

stefanrueger, 2023-12-20:
It's an overzealous compiler warning. I have changed git main straight away to deal with this.

closed 2023-12-20 by stefanrueger

mcuee, 2023-12-21:
I have created a new label since it is too harsh to use 'bug` label.