avrdudes/avrdude #1616: Bus Pirate v3 broken from 7.0

therealdreg, 2024-01-13:
This only works on 7.0

```
avrdude -c buspirate -p m328pb -P COM8 -b 115200 -U flash:w:"C:\Users\dreg\Desktop\test\build\arduino.avr.uno\test.ino.with_bootloader.hex":i 
```

Fails on 7.2:
```
C:\Users\dreg\Downloads\avrdude-v7.2-windows-x64>avrdude -c buspirate -p m328pb -P COM8 -b 115200 -U flash:w:"C:\Users\dreg\Desktop\test\build\arduino.avr.uno\test.ino.with_bootloader.hex":i
attempting to initiate BusPirate binary mode ...
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9516 (probably m328pb)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip

avrdude: processing -U flash:w:C:\Users\dreg\Desktop\test\build\arduino.avr.uno\test.ino.with_bootloader.hex:i
avrdude: reading input file C:\Users\dreg\Desktop\test\build\arduino.avr.uno\test.ino.with_bootloader.hex for flash
         with 1426 bytes in 3 sections within [0, 0x7fff]
         using 12 pages and 110 pad bytes
avrdude: writing 1426 bytes flash ...
Writing |                                                    | 0% 0.00 s
C:\Users\dreg\Downloads\avrdude-v7.2-windows-x64>
```
I am researching the cause, after thar I will PR the fix

mcuee, 2024-01-13:
Thanks for the reports.

So this is different from the following issue which we deemed it to be a Bus Pirate FW issue (which may or may not be correct).
* https://github.com/avrdudes/avrdude/issues/459

therealdreg, 2024-01-13:
> Thanks for the reports.
> 
> So this is different from the following issue which we deemed it to be a Bus Pirate FW issue (which may or may not be correct).
> 
> * [[bug #51614] BusPirate: Paged Read command returned zero. With m1284p flash. #459](https://github.com/avrdudes/avrdude/issues/459)

avrdude: processing -U flash:w:C:\Users\dreg\Desktop\test\build\arduino.avr.uno\test.ino.with_bootloader.hex:i
avrdude: reading input file C:\Users\dreg\Desktop\test\build\arduino.avr.uno\test.ino.with_bootloader.hex for flash
         with 1426 bytes in 3 sections within [0, 0x7fff]
         using 12 pages and 110 pad bytes
avrdude: writing 1426 bytes flash ...
Writing |                                                    | 0% 0.00 s
avrdude: cannot read flash [0x0380, 0x03ff] to pad page
avrdude: cannot read flash [0x7f80, 0x7fff] to pad page

therealdreg, 2024-01-13:
```
          // Read flash contents to separate memory spc and fill in holes
          if(avr_read_page_default(pgm, p, cm, beg, spc) >= 0) {
            pmsg_notice2("padding %s [0x%04x, 0x%04x]\n", cm->desc, beg, end-1);
            for(i = beg; i < end; i++)
              if(!(cm->tags[i] & TAG_ALLOCATED)) {
                cm->tags[i] |= TAG_ALLOCATED;
                cm->buf[i] = spc[i-beg];
              }
          } else {
            pmsg_notice2("cannot read %s [0x%04x, 0x%04x] to pad page\n",
              cm->desc, beg, end-1);
          }
        }
```

therealdreg, 2024-01-13:
A lot has changed... I think it's going to take me a while

https://github.com/avrdudes/avrdude/compare/v7.0...v7.1

MCUdude, 2024-01-13:
Try looking at the commits related to the Buspirate. There are only a handful.
https://github.com/avrdudes/avrdude/commits/main/src/buspirate.c

Avrdude 7.0 was released in May 2022, so not much has happened since then.

therealdreg, 2024-01-13:
> Try looking at the commits related to the Buspirate. There are only a handful. https://github.com/avrdudes/avrdude/commits/main/src/buspirate.c
> 
> Avrdude 7.0 was released in May 2022, so not much has happened since then.
> 
> BTW which Buspirate do you have? I have Buspirate 4.0 I can test with.

BPV3 using the last firmware community with some fixes

MCUdude, 2024-01-13:
So you can confirm that it doesn't work with 7.1 as well? I'm looking through the commits and trying to narrow things down.

therealdreg, 2024-01-13:
> So you can confirm that it doesn't work with 7.1 as well? I'm looking through the commits and trying to narrow things down.

pretty Sure, only works on 7.0

MCUdude, 2024-01-13:
I'm not having any issues with the BusPirate v4.0 using the latest git main. However, when the Buspirate are powering the target, my ATmega328PB board isn't quick enough to power up before Avrdude starts to write to the board, so I had to use an external power source just to get it past the power.up

```
HiZ>i
Bus Pirate v4
Community Firmware v7.0 - goo.gl/gCzQnW 
DEVID:0x1019 REVID:0x0004 (24FJ256GB106 UNK)
http://dangerousprototypes.com
```

```
$ avrdude -cbuspirate -patmega328pb -P/dev/cu.usbmodem000000011 -Uflash:w:test_files/the_quick_brown_fox_32768B.hex:i
attempting to initiate BusPirate binary mode ...
avrdude: paged flash write enabled
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9516 (probably m328pb)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip

avrdude: processing -U flash:w:test_files/the_quick_brown_fox_32768B.hex:i
avrdude: reading input file test_files/the_quick_brown_fox_32768B.hex for flash
         with 32768 bytes in 1 section within [0, 0x7fff]
         using 256 pages and 0 pad bytes
avrdude: writing 32768 bytes flash ...
Writing | ################################################## | 100% 43.75 s 
avrdude: 32768 bytes of flash written
avrdude: verifying flash memory against test_files/the_quick_brown_fox_32768B.hex
Reading | ################################################## | 100% 39.03 s 
avrdude: 32768 bytes of flash verified

avrdude done.  Thank you.

```

therealdreg, 2024-01-13:
> I'm not having any issues with the BusPirate v4.0 using the latest git main. However, when the Buspirate are powering the target, my ATmega328PB board isn't quick enough to power up before Avrdude [...]

The problem is with v3 hardware

therealdreg, 2024-01-13:
Read works fine:
```

C:\Users\dreg\Downloads\avrdude-v7.1-windows-x64>avrdude -c buspirate -p m328pb -P COM8 -b 115200 -U flash:r:"C:\Users\dreg\Desktop\dump.hex":i

attempting to initiate BusPirate binary mode ...
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9516 (probably m328pb)
avrdude: reading flash memory ...

Reading | ################################################## | 100% 40.95 s

avrdude: writing output file C:\Users\dreg\Desktop\dump.hex
avrdude error: did not get a response to power off command

avrdude done.  Thank you.
```

Write fails:
```
C:\Users\dreg\Downloads\avrdude-v7.1-windows-x64>avrdude -c buspirate -p m328pb -P COM8 -b 115200 -U flash:w:"C:\Users\dreg\Desktop\dump.hex":i -v -v -v

avrdude: Version 7.1
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is C:\Users\dreg\Downloads\avrdude-v7.1-windows-x64\avrdude.conf

         Using Port                    : COM8
         Using Programmer              : buspirate
         Overriding Baud Rate          : 115200
         AVR Part                      : ATmega328PB
         Chip Erase delay              : 10500 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : possible i/o
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                 65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           flash                  65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           efuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
[... 103 lines cut ...]
        0x01
avrdude: buspirate_send_bin():
        0xac 0x80 0x00 0x00
avrdude: buspirate_recv_bin():
        0x00 0xac 0x80 0x00
avrdude: buspirate_send_bin():
        0x48
avrdude: buspirate_recv_bin():
        0x01
avrdude: buspirate_send_bin():
        0x13
avrdude: buspirate_recv_bin():
        0x01
avrdude: buspirate_send_bin():
        0xac 0x53 0x00 0x00
avrdude: buspirate_recv_bin():
        0x00 0xac 0x53 0x00
avrdude: reading input file C:\Users\dreg\Desktop\dump.hex for flash
         with 32768 bytes in 1 section within [0, 0x7fff]
         using 256 pages and 0 pad bytes
avrdude: writing 32768 bytes flash ...

Writing |                                                    | 0% 0.00 s
C:\Users\dreg\Downloads\avrdude-v7.1-windows-x64>
```

mcuee, 2024-01-13:
What about git main? Just want to make sure if it has the same problem as 7.2 release.

You can donwload the snapshot here.
https://github.com/avrdudes/avrdude/wiki/Getting-Nightly-Builds-for-AVRDUDE

therealdreg, 2024-01-13:
> What about git main? Just want to make sure if it has the same problem as 7.2 release.
> 
> You can donwload the snapshot here. https://github.com/avrdudes/avrdude/wiki/Getting-Nightly-Builds-for-AVRDUDE

Same problem:
```

C:\Users\dreg\Downloads\avrdude-msvc-x64>avrdude -c buspirate -p m328pb -P COM8 -b 115200 -U flash:w:"C:\Users\dreg\Desktop\dump.hex":i -v -v -v -v -v
executable_abspath = C:/Users/dreg/Downloads/avrdude-msvc-x64/avrdude.exe
executable_abspath_len = 52
executable_dirpath = C:/Users/dreg/Downloads/avrdude-msvc-x64
executable_dirpath_len = 40
sys_config = C:/Users/dreg/Downloads/avrdude-msvc-x64/avrdude.conf
sys_config_found = true

avrdude: Version 7.2-20240110 (a336e47)
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is C:\Users\dreg\Downloads\avrdude-msvc-x64\avrdude.conf

         Using port            : COM8
         Using programmer      : buspirate
         Setting baud rate     : 115200
         AVR Part              : ATmega328PB
         Programming modes     : ISP, HVPP, debugWIRE, SPM

         Memory           Size  Pg size
         ------------------------------
         eeprom           1024        4
         flash           32768      128
         efuse               1        1
         hfuse               1        1
         lfuse               1        1
         lock                1        1
         prodsig/sigrow     24        1
         signature           3        1
         calibration         1        1
         io                224        1
         sram             2048        1

         Variants         Package  F max   T range         V range
         ----------------------------------------------------------------
         ATmega328PB-AN   TQFP32   20 MHz  [-40 C, 105 C]  [1.8 V, 5.5 V]
         ATmega328PB-ANR  TQFP32   20 MHz  [-40 C, 105 C]  [1.8 V, 5.5 V]
         ATmega328PB-AU   TQFP32   20 MHz  [-40 C,  85 C]  [1.8 V, 5.5 V]
         ATmega328PB-AUR  TQFP32   20 MHz  [-40 C,  85 C]  [1.8 V, 5.5 V]
         ATmega328PB-MN   QFN32    20 MHz  [-40 C, 105 C]  [1.8 V, 5.5 V]
         ATmega328PB-MNR  QFN32    20 MHz  [-40 C,  85 C]  [1.8 V, 5.5 V]
         ATmega328PB-MU   QFN32    20 MHz  [-40 C,  85 C]  [1.8 V, 5.5 V]
         ATmega328PB-MUR  QFN32    20 MHz  [-40 C,  85 C]  [1.8 V, 5.5 V]

         Programmer Type       : BusPirate
         Description           : The Bus Pirate
attempting to initiate BusPirate binary mode ...
avrdude: buspirate_send_bin():
        0x0a 0x0a
avrdude: ser_send: . [0a] . [0a]
avrdude: buspirate_send_bin():
        0x00 0x00 0x00 0x00   0x00 0x00 0x00 0x00
        0x00 0x00 0x00 0x00   0x00 0x00 0x00 0x00
        0x00 0x00 0x00 0x00
avrdude: ser_send: . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00]
avrdude: ser_recv: B [42] B [42] I [49] O [4f] 1 [31]
avrdude: buspirate_recv_bin():
        0x42 0x42 0x49 0x4f   0x31
BusPirate binmode version: 1
avrdude: buspirate_send_bin():
        0x01
avrdude: ser_send: . [01]
[... 121 lines cut ...]
avrdude: ser_send: H [48]
avrdude: ser_recv: . [01]
avrdude: buspirate_recv_bin():
        0x01
avrdude: buspirate_send_bin():
        0x13
avrdude: ser_send: . [13]
avrdude: ser_recv: . [01]
avrdude: buspirate_recv_bin():
        0x01
avrdude: buspirate_send_bin():
        0xac 0x53 0x00 0x00
avrdude: ser_send: . [ac] S [53] . [00] . [00]
avrdude: ser_recv: . [00] . [ac] S [53] . [00]
avrdude: buspirate_recv_bin():
        0x00 0xac 0x53 0x00

avrdude: processing -U flash:w:C:\Users\dreg\Desktop\dump.hex:i
avrdude: reading input file C:\Users\dreg\Desktop\dump.hex for flash
         with 32768 bytes in 1 section within [0, 0x7fff]
         using 256 pages and 0 pad bytes
avrdude: writing 32768 bytes flash ...
Writing |                                                    | 0% 0.00 s
C:\Users\dreg\Downloads\avrdude-msvc-x64>
```

therealdreg, 2024-01-15:
@MCUdude @mcuee If you want, I can give you a Bus Pirate v3

mcuee, 2024-01-15:
> @MCUdude @mcuee If you want, I can give you a Bus Pirate v3

That will be great. I am locating in Singapore (the little Red Dot in SouthEast Asia). Shipment may be a bit expensive. :-)

mcuee, 2024-01-15:
@therealdreg 
> I don't mind the cost, can you give me the address? or I can send you the money

Thanks a lot for the offer. I have sent my mailing address to your email address listed in your github profile.

mcuee, 2024-01-15:
@therealdreg
> 
> > I don't mind the cost, can you give me the address? or I can send you the money
> 
> Thanks a lot for the offer. I have sent my mailing address to your email address listed in your github profile.

Hmm, there is an error message:
 554 5.7.1 <[dreg@fr33project.org](mailto:dreg@fr33project.org)>: Relay access denied

therealdreg, 2024-01-15:
> @therealdreg
>
>
> Hmm, there is an error message: 554 5.7.1 <[dreg@fr33project.org](mailto:dreg@fr33project.org)>: Relay access denied

dreg@rootkit.es

mcuee, 2024-01-15:
> [dreg@rootkit.es](mailto:dreg@rootkit.es)

Thanks. Email sent.

mcuee, 2024-01-15:
@therealdreg 

I will delete the post which includes your email address, just to reduce some potential spam for you.

stefanrueger, 2024-01-21:
I suspect Bus Pirate v3 gets confused by random access of pages. AVRDUDE 7.1 introduced padding of pages: when a page is only partially to be written then the page contents is first read from the device, then modified and written back. That is particularly important for bootloaders that *can* write 0xff to flash if they deploy SPM page erase before SPM page write. AVRDUDE kind of gracefully deals with read failures, but I suspect that Bus Pirtae v3 gets confused by a read page followed by a write page to the same address (that *could* happen when the firmware does auto-page increment, for example).

stefanrueger, 2024-01-21:
@therealdreg Try to write a .hex file with an integer number of flash pages to see whether this works (say, a .hex file with 512 bytes occupying addresses 0 ... 511). Then try to write two bytes less (510 bytes). If one works, and the other not, it will be that the firmware does not take kindly to AVRDUDE's padding attempt. In this case I suggest introducing a programmer variable `skip_padding` that can be set in `avrdude.conf` and then make padding conditional on `!skip_padding`.

mcuee, 2024-01-21:
Thanks to @therealdreg, I have received the Bus Pirate v3. I will carry out some tests later.

mcuee, 2024-01-21:
@therealdreg 

Maybe the following Bus Pirate issue may have some relevant info as well. It seems to me certian FW versions are causing problems with avrdude .
* https://github.com/BusPirate/Bus_Pirate/issues/122

therealdreg, 2024-01-22:
@stefanrueger do you want a bus pirate v3?

stefanrueger, 2024-01-22:
Thanks for offering, but not needed. @mcuee will be able to test my hypothesis (as could you).

MCUdude, 2024-01-22:
> Try to write a .hex file with an integer number of flash pages to see whether this works (say, a .hex file with 512 bytes occupying addresses 0 ... 511). Then try to write two bytes less (510 [...]

I'll give this a try this evening. I'll assume the following srec_cat commands would generate suitable test files:

```
srec_cat -generate 0 512 -repeat-string "THE QUICK BROWN FOX JUMPS OVER THE LAZY DOG. " -o the_quick_brown_fox_${i}B.hex -Intel # Generate test hex file with 512 bytes starting from address 0
srec_cat -generate 0 510 -repeat-string "THE QUICK BROWN FOX JUMPS OVER THE LAZY DOG. " -o the_quick_brown_fox_${i}B.hex -Intel # Generate test hex file with 510 bytes starting from address 0
```

stefanrueger, 2024-01-22:
> generate suitable test files

Well, actually, I am a bit wary of using ASCII text for flash test data. Who knows which GPIO pins are toggled when the AVR jumps into the ASCII text? It doesn't bear thinking about if someone tests a live board where I/O pins control the gate of the alligator compound :open_mouth: 

Try in a bash
``` bash
for s in 510 512; do
   srec_cat -generate 0 $s -repeat-data $(for i in {255..0}; do printf "0x%02x 0xcf " $i; done) -output rjmp-$s.hex -Intel
done
```

The generated files contain safer endless loops:
``` bash
$ avr-objdump -m avr -h -D rjmp-510.hex | tail
 1ea:	0a cf       	rjmp	.-492    	;  0x0
 1ec:	09 cf       	rjmp	.-494    	;  0x0
 1ee:	08 cf       	rjmp	.-496    	;  0x0
 1f0:	07 cf       	rjmp	.-498    	;  0x0
 1f2:	06 cf       	rjmp	.-500    	;  0x0
 1f4:	05 cf       	rjmp	.-502    	;  0x0
 1f6:	04 cf       	rjmp	.-504    	;  0x0
 1f8:	03 cf       	rjmp	.-506    	;  0x0
 1fa:	02 cf       	rjmp	.-508    	;  0x0
 1fc:	01 cf       	rjmp	.-510    	;  0x0
```

(you might consider similar techniques for your avrdude test program)

MCUdude, 2024-01-26:
@therealdreg I just received the Bus Pirate in the mail; thank you!

For me, it's working just fine. However, It's really, really, really slow:

```
$ ./avrdude -cbuspirate -patmega328p -P ft232r -Uflash:w:/Users/hans/Documents/avrdude-test-script/test_files/the_quick_brown_fox_16384B.hex -xspifreq=4
attempting to initiate BusPirate binary mode ...
avrdude: paged flash write enabled
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e950f (probably m328p)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip

avrdude: processing -U flash:w:/Users/hans/Documents/avrdude-test-script/test_files/the_quick_brown_fox_16384B.hex:i
avrdude: reading input file /Users/hans/Documents/avrdude-test-script/test_files/the_quick_brown_fox_16384B.hex for flash
         with 16384 bytes in 1 section within [0, 0x3fff]
         using 128 pages and 0 pad bytes
avrdude: writing 16384 bytes flash ...
Writing | ################################################## | 100% 529.20 s 
avrdude: 16384 bytes of flash written
avrdude: verifying flash memory against /Users/hans/Documents/avrdude-test-script/test_files/the_quick_brown_fox_16384B.hex
Reading | ################################################## | 100% 524.97 s 
avrdude: 16384 bytes of flash verified

avrdude done.  Thank you.
```

therealdreg, 2024-01-26:
> @therealdreg I just received the Bus Pirate in the mail; thank you!
>
> For me, it's working just fine. However, It's really, really, really slow:
>
> ```
> $ ./avrdude -cbuspirate -patmega328p -P ft232r [...]

what firmware are you using? (connect via COM* 115200 and write **i**)

therealdreg, 2024-01-27:
I solved the problem... 

Look the official doc about binary i/o

[image]

As you can see "If the number of bytes to read or write are out of bounds, the Bus Pirate will return 0x00 now", the last community firmware do something like this:

```
if ((bytes_to_write == 0) && (bytes_to_read == 0)) {
    REPORT_IO_FAILURE(); 
    return;
  }
```

Because for community firmware this is like the out-of-bounds case documented. 

The old bus pirate firmware accepts read == 0 && write == 0 and returns REPORT_IO_SUCCESS();

AVRDude do: 

avrdude: send: . [05] . [00] . [00] . [00] . [00] 

So this is the problem. 

Since the official documentation is not explicit about the case of read && write being zero, I believe it's best to keep the AVRdude code as it is... to ensure it continues working with older firmwares. I'm going to update the community firmware of Bus Pirate to return a 0x01 in this case, and that should solve it!

```
  if ((bytes_to_write == 0) && (bytes_to_read == 0)) {
    REPORT_IO_SUCCESS(); // needed for AVRDUDE
    return;
  }
```

Thx for your time dudes! I will update my repo firm https://github.com/therealdreg/buzzpirat

closed 2024-01-27 by therealdreg

MCUdude, 2024-01-27:
Great to hear you found the root cause @therealdreg!
But thanks for the hardware donation anyways. I'll keep this in the "test circle" when we're preparing a new release.

> what firmware are you using? (connect via COM 115200 and write 'i')

```
HiZ>i
Bus Pirate v3b
Firmware v5.10 (r559)  Bootloader v4.4
DEVID:0x0447 REVID:0x3046 (24FJ64GA002 B8)
http://dangerousprototypes.com
```

I'm also using 115200 baud. If this isn't the firmware I should be using, is there a guide I can follow to flash a newer firmware, ideally pre-compiled?