avrdudes/avrdude #1617: avrdude fails at communicating with ATMEL-ICE in ISP mode. Correct drivers installed

dzalf, 2024-01-14:
In line with an [issue](https://github.com/ZakKemble/AVRDUDESS/issues/82#issuecomment-1890970158) I raised in the avrdudess GUI repo I must report that avrdude cannot communicate with ATMEL-ICE in ISP mode despite multiple resintallations of the libusb32 drivers

However, the platformio avrdude version does work without a problem

Using the `avrdude` that comes with the GUI from avrdudess I am unable to communicate with the programmer

[image]

While when changing the avrdude target to the executable from pio the MCU gets detected without any problems:

[image]

Any ideas?

mcuee, 2024-01-16:
Please use the official avrdude 7.2 release version to see if that helps.
https://github.com/avrdudes/avrdude/releases/tag/v7.2

ZakKemble, 2024-01-17:
> Please use the official avrdude 7.2 release version to see if that helps. https://github.com/avrdudes/avrdude/releases/tag/v7.2

This is the version that avrdudess currently ships with.

mcuee, 2024-01-17:
@dzalf 

Please do not use the GUI but rather use the command line and post the debug output using `-vvvv`. Thanks.

mcuee, 2024-01-17:
@dzalf 

Yet the other debugging option -- use an alternative avrdude 7.2 binary and see if that helps. The following is my build using MSYS2 MinGW64.
https://github.com/mcuee/avrdude/releases/download/v7.2/avrdude-v7.2_mingw64.zip

The official avrdude 7.2 uses Windows MSVC64 build which may have issues with USB Compisite device.
- https://github.com/avrdudes/avrdude/issues/968 (not working with USB Composite Device programer like USBASP with the FW from https://github.com/dioannidis/usbasp)

dzalf, 2024-01-18:
@mcuee 

Thanks for your input.

I tried using different official versions from Avrdude without any success. I am simply trying to "detect" the device however I cannot communicate with the programmer.

I will try in the command line and report back.

One thing that I noticed is that pio uses an old version from back in 2016 (I was unable to determine exactly which one)

mcuee, 2024-01-19:
@dzalf 

One more try -- are you using the official driver from Microchip/Atmel? I do not have Atmel ICE so I do not know. 

It seems to me Microchip/Atmel is now using WinUSB driver more often for the tools (other than those USB HID devices). But I am not so sure about older tools like Atmel ICE.

mcuee, 2024-01-19:
@dzalf 

It seems to me that Atmel ICE should use native USB HID driver for AVR programming. Please use the default driver and not libusb-win32 driver. Older version of avrdude does not support hidapi and that was why libusb-win32 driver was used. You should NOT use that if it is an USB HID device.

Similar issues:
* https://github.com/avrdudes/avrdude/issues/1221

mcuee, 2024-01-19:
I will close this issue for now. Please reopen this issue if you still encounter the issue with the correct Windows native HID driver.

mcuee, 2024-01-19:
On the other hand, I have added this issue to #1221.
* #1221

mcuee, 2024-01-19:
Reference:
* https://github.com/ZakKemble/AVRDUDESS/issues/82
* https://github.com/ZakKemble/AVRDUDESS/issues/62

closed 2024-01-19 by mcuee

dzalf, 2024-01-20:
> @dzalf
> 
> One more try -- are you using the official driver from Microchip/Atmel? I do not have Atmel ICE so I do not know.
> 
> It seems to me Microchip/Atmel is now using WinUSB driver more often for the tools (other than those USB HID devices). But I am not so sure about older tools like Atmel ICE.

I have to report that on a new PC I could run the programmer without any issues on avrdude 7.2.

However, on the original PC where I ran my initial tests, I cannot revert back to HID or WinUSB driver. The device simply cannot be detected. I have tried MULTIPLE solutions I found online without any success. 

No matter what I do to try to uninstall the libusbK or libusb-win32 Windows 11 reverts back to those after uninstalling and replugging the programmer.

I have updated Microchip Studio and repaired it however I am unable to find the right driver.

Installing the WinUSB driver from Zadig doesn't help either.

Any ideas?

mcuee, 2024-01-20:
WInUSB driver is also not the correct driver due to the following known bug, which is of lower priority to fix since the correct driver is USB HID.
* https://github.com/avrdudes/avrdude/issues/1221

You have to revert to the default Windows USB HID driver.
* https://github.com/pbatard/libwdi/issues/290

mcuee, 2024-01-20:
BTW, I have changed the label to `bug` to be in line with Issue #1221. But then it is a `duplicate`.

dzalf, 2024-01-20:
Dear @mcuee 

I truly appreciate your continuous support on this.

After several attempts to remove drivers and not being able to find the HID drivers online, I suddenly made it work. However, I cannot pinpoint what the solution was. Right now, I can only see "Interface 1" under a libusb-win32 driver, however, the CMSIS-DAP was fortunately (and magically) configured by Windows under the HID driver:

[image]

[image]

Something that I did to make it possible to continue working on my project with VSCode+ pio while using the latest version of avrdude was to replace the stock old version that comes by default with the pio environment with the latest version available here. This way, I can use either pio, avrdude or the GUI tool at the same time.

Just to be clear, the reason why  I jump between different ways of flashing my MCUs is that for quick debugging purposes I flash a test hex into my board and the easiest way to do that is via avrdudess (simply by loading the hex file) while for regular flashing of new code, I do it directly from VSCode.

Previously, my workflow demanded using the drivers from Zadig, however, based on this issue and your kind explanations I have learnt that the way to manage the drivers with the latest revision is by NOT using the Zadig drivers anymore. I have even read that one should avoid at all costs using them since Zadig's tool does not offer the option to revert back to the HID driver (which is true).

I will try to replicate this on another PC to make sure that the issue is fully solved. In the future I will slap myself to remind me not to install any alternative drivers with new avrdude versions 😉