avrdudes/avrdude #1619: Provide a simple way to perform a chip factory reset

MCUdude, 2024-01-17:
It would be really neat if terminal mode could support some sort of factory reset option, so that `factory` or `reset` could reset the fuses to the default value and erase the EEPROM and flash memory. And if the bitclock is set too high, make sure that it's reduced to atleast 1/4 of the internal clock speed

stefanrueger, 2024-01-22:
The first part is doable within the confines of what the -c programmer can do, but how would the project attempt

> And if the bitclock is set too high, make sure that it's reduced to atleast 1/4 of the internal clock speed

That is, how would AVRDUDE know what the internal clock speed of a factory-reset part is? AVRDUDE knows ca 350 parts.

stefanrueger, 2024-01-22:
And for the time being the following would work (insert the intended real programmer for *programmer*):

`avrdude -cdryrun -qq -pATmega328P -Tc | avrdude -c`*programmer* -pATmega328P -t -T erase`

And, obviously, use the correct part name, too (twice).

Caveat: there are some parts for which AVRDUDE has no factory reset values for fuses.

stefanrueger, 2024-01-27:
Turns out the hard part is not to figure out what the initial clock speed is after factory resetting fuse values and carrying out an MCU reset, but to figure out whether a programmer can set its bit rate, and how to do this. Instead the `factory reset` command prints the new F_CPU after MCU reset.

PR #1642 (2024-01-28): Provide terminal command `factory reset`
Fixes #1619

Resets the connected part to factory state as far as possible (bootloaders, for example, cannot write fuses and may not have a means to erase EEPROM). This command may change the clock frequency F_CPU of the part after the next MCU reset when the changed fuse values come into effect. As such, this may require that future avrdude calls use a different bit clock rate (up to F_CPU/4) for the programmer next time. Note that the command `factory` can be abbreviated but the required argument `reset` needs to be spelled out in full.

```
$ cat tes [... cut]
closed 2024-01-28 by stefanrueger