avrdudes/avrdude #1620: Bad chip detection

GMagician, 2024-01-17:
New version 2.15 can't properly detect atmel328p while 2.14 works succesfully.

New version 2.15 often hang on avrdude.exe request, but the few times it ended correctly the ID was correct (1E950F) but processor was identifyed as ATA6614Q

HattonLe, 2024-01-17:
Arf... you just beat me to it... Using AVRDUDESS v2.15 and a USBasp V2.0 talking to Arduino Uno R3 (Atmega328P-PU). It identifies as ATA6614Q. Had to look that one up as never heard of it before. Tried partially dead 328P just to make sure I had everything connected up correctly as its been a while, so have now confirmed its not me LOL. (Arduino IDE correctly burns bootloader).

MCUdude, 2024-01-17:
Can you post the full Avrdude output? I know nothing about AVRDUDESS, so v2.15 tells me nothing. What matters is what kind of Avrdude binary AVRDUDESS bundles.

@stefanrueger since ATmega328P and ATA6614Q share the same device signature, how does Avrdude distinguish between them?

ZakKemble, 2024-01-17:
Heya! I'm the AVRDUDESS dev!

This isn't an avrdude problem. The ATA6614Q and 328P both have the same signatures which is confusing avrdudess when using the detect feature, see this https://github.com/ZakKemble/AVRDUDESS/issues/81

MCUdude, 2024-01-17:
> Heya! I'm the AVRDUDESS dev!
This isn't an avrdude problem. The ATA6614Q and 328P both have the same signatures which is confusing avrdudess when using the detect feature, see this https://github.com/ZakKemble/AVRDUDESS/issues/81

Hi! Very well. I'm closing the issue

closed 2024-01-17 by MCUdude