avrdudes/avrdude #1622: SerialUPDI programmer failing when performing erase/EEPROM write in single programming session against NVM V3+ chips

dbuchwald, 2024-01-17:
As reported by @askn37, there is an issue with programming all the NVM V3+ chips, which is quite easy to replicate. All you need is to perform the following operation:

```
/Users/dawid/Development/avrdude/build_darwin/src/avrdude -C /Users/dawid/Development/avrdude/build_darwin/src/avrdude.conf -c serialupdi -P /dev/cu.usbserial-1410 -p avr16eb28 -e -Ueeprom:w:test_files/0xff_512B.hex -Ueeprom:v:test_files/0xff_512B.hex
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e943f (probably avr16eb28)
avrdude: erasing chip

avrdude: processing -U eeprom:w:test_files/0xff_512B.hex:i
avrdude: reading input file test_files/0xff_512B.hex for eeprom
         with 512 bytes in 1 section within [0, 0x1ff]
         using 64 pages and 0 pad bytes
avrdude: writing 512 bytes eeprom ...
Writing | ################################################## | 100% 4.68 s
avrdude: 512 bytes of eeprom written
avrdude: verifying eeprom memory against test_files/0xff_512B.hex
Reading | ################################################## | 100% 0.75 s
avrdude warning: verification mismatch
        device 0x00 != input 0xff at addr 0x0000 (error)
avrdude error: verification mismatch

avrdude done.  Thank you.
```

Critical part here is that erase and EEPROM write operation must be executed in single programming session. Running these two commands in two separate steps does not produce the issue.

dbuchwald, 2024-01-19:
@askn37 

I implemented the fix based on your suggestion, just with one small modification to ensure we don't lose value in `status` variable. I ran some tests in my setup (avr16eb28, avr64ea28, avr128db28, t406, t416, t1626, m808, m4809, avr16dd28) and it seems to work:

```
./avrdude_test_script.sh                                                                                                                               ─╯
Prepare "-c serialupdi -P /dev/cu.usbserial-1410 -p avr16eb28" and press 'enter' or 'space' to continue. Press any other key to skip
✅ eesave fuse bit set, cleared and verified
✅ the_quick_brown_fox_16384B.hex flash -U write/verify
✅ the_quick_brown_fox_512B.hex eeprom -U write/verify
✅ lorem_ipsum_16384B.srec flash -U write/verify
✅ lorem_ipsum_512B.srec eeprom -U write/verify
✅ 0xff_512B.hex eeprom -U write/verify
✅ the_quick_brown_fox_16384B.hex:a flash -T write/verify
✅ the_quick_brown_fox_512B.hex:a eeprom -T write/verify
✅ lorem_ipsum_16384B.srec flash -T write/verify
✅ lorem_ipsum_512B.srec eeprom -T write/verify
✅ cola-vending-machine.raw flash -T/-U write/verify
✅ holes_pack_my_box_16384B.hex:a flash -U write
✅ holes_the_five_boxing_wizards_16384B.hex flash -T write
✅ holes_pack_my_box_512B.hex:a eeprom -U write
✅ holes_the_five_boxing_wizards_512B.hex eeprom -T write
✅ random_data_64B.bin usersig -T/-U write/read
Prepare "-c serialupdi -P /dev/cu.usbserial-1410 -p avr64ea28" and press 'enter' or 'space' to continue. Press any other key to skip
✅ eesave fuse bit set, cleared and verified
✅ the_quick_brown_fox_65536B.hex flash -U write/verify
✅ the_quick_brown_fox_512B.hex eeprom -U write/verify
✅ lorem_ipsum_65536B.srec flash -U write/verify
✅ lorem_ipsum_512B.srec eeprom -U write/verify
✅ 0xff_512B.hex eeprom -U write/verify
✅ the_quick_brown_fox_65536B.hex:a flash -T write/verify
✅ the_quick_brown_fox_512B.hex:a eeprom -T write/verify
✅ lorem_ipsum_65536B.srec flash -T write/verify
✅ lorem_ipsum_512B.srec eeprom -T write/verify
✅ cola-vending-machine.raw flash -T/-U write/verify
✅ holes_pack_my_box_65536B.hex:a flash -U write
✅ holes_the_five_boxing_wizards_65536B.hex flash -T write
✅ holes_pack_my_box_512B.hex:a eeprom -U write
✅ holes_the_five_boxing_wizards_512B.hex eeprom -T write
✅ random_data_64B.bin usersig -T/-U write/read
Prepare "-c serialupdi -P /dev/cu.usbserial-1410 -p avr128db28" and press 'enter' or 'space' to continue. Press any other key to skip
✅ eesave fuse bit set, cleared and verified
✅ the_quick_brown_fox_131072B.hex flash -U write/verify
✅ the_quick_brown_fox_512B.hex eeprom -U write/verify
✅ lorem_ipsum_131072B.srec flash -U write/verify
✅ lorem_ipsum_512B.srec eeprom -U write/verify
✅ 0xff_512B.hex eeprom -U write/verify
✅ the_quick_brown_fox_131072B.hex:a flash -T write/verify
✅ the_quick_brown_fox_512B.hex:a eeprom -T write/verify
✅ lorem_ipsum_131072B.srec flash -T write/verify
✅ lorem_ipsum_512B.srec eeprom -T write/verify
✅ cola-vending-machine.raw flash -T/-U write/verify
✅ holes_pack_my_box_131072B.hex:a flash -U write
✅ holes_the_five_boxing_wizards_131072B.hex flash -T write
✅ holes_pack_my_box_512B.hex:a eeprom -U write
✅ holes_the_five_boxing_wizards_512B.hex eeprom -T write
✅ random_data_32B.bin usersig -T/-U write/read
Prepare "-c serialupdi -P /dev/cu.usbserial-1410 -p t406" and press 'enter' or 'space' to continue. Press any other key to skip
✅ eesave fuse bit set, cleared and verified
✅ the_quick_brown_fox_4096B.hex flash -U write/verify
✅ the_quick_brown_fox_128B.hex eeprom -U write/verify
✅ lorem_ipsum_4096B.srec flash -U write/verify
✅ lorem_ipsum_128B.srec eeprom -U write/verify
✅ 0xff_128B.hex eeprom -U write/verify
✅ the_quick_brown_fox_4096B.hex:a flash -T write/verify
✅ the_quick_brown_fox_128B.hex:a eeprom -T write/verify
✅ lorem_ipsum_4096B.srec flash -T write/verify
✅ lorem_ipsum_128B.srec eeprom -T write/verify
✅ cola-vending-machine.raw flash -T/-U write/verify
✅ holes_pack_my_box_4096B.hex:a flash -U write
[... 67 lines cut ...]
✅ holes_pack_my_box_49152B.hex:a flash -U write
✅ holes_the_five_boxing_wizards_49152B.hex flash -T write
✅ holes_pack_my_box_256B.hex:a eeprom -U write
✅ holes_the_five_boxing_wizards_256B.hex eeprom -T write
✅ random_data_64B.bin usersig -T/-U write/read
Prepare "-c serialupdi -P /dev/cu.usbserial-1410 -p avr16dd28" and press 'enter' or 'space' to continue. Press any other key to skip
✅ eesave fuse bit set, cleared and verified
✅ the_quick_brown_fox_16384B.hex flash -U write/verify
✅ the_quick_brown_fox_256B.hex eeprom -U write/verify
✅ lorem_ipsum_16384B.srec flash -U write/verify
✅ lorem_ipsum_256B.srec eeprom -U write/verify
✅ 0xff_256B.hex eeprom -U write/verify
✅ the_quick_brown_fox_16384B.hex:a flash -T write/verify
✅ the_quick_brown_fox_256B.hex:a eeprom -T write/verify
✅ lorem_ipsum_16384B.srec flash -T write/verify
✅ lorem_ipsum_256B.srec eeprom -T write/verify
✅ cola-vending-machine.raw flash -T/-U write/verify
✅ holes_pack_my_box_16384B.hex:a flash -U write
✅ holes_the_five_boxing_wizards_16384B.hex flash -T write
✅ holes_pack_my_box_256B.hex:a eeprom -U write
✅ holes_the_five_boxing_wizards_256B.hex eeprom -T write
✅ random_data_32B.bin usersig -T/-U write/read
```

Can you pull latest `issue_1609` branch and give it a try?

askn37, 2024-01-19:
Candidates for correction

```
askn@alicia avrdude_issue1609 % ./build_darwin/src/avrdude -v -Pusb

avrdude: Version 7.2-20240119 (45b53658)
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is /Users/askn/Collaborator/avrdude_issue1609/build_darwin/src/avrdude.conf
         User configuration file is /Users/askn/.avrduderc

avrdude programmer_not_found() error: no programmer has been specified on the command line or in the
        config file(s); specify one using the -c option and try again
askn@alicia avrdude_issue1609 % 
```

run test NVM V3

```
Prepare "-c serialupdi -xrtsdtr=high -P /dev/cu.usbserial-230 -p avr64ea32" and press 'enter' or 'space' to continue. Press any other key to skip
avrdude error: programmer does not support extended parameter -x rtsdtr=high, option ignored
avrdude error: programmer does not support extended parameter -x rtsdtr=high, option ignored
avrdude error: programmer does not support extended parameter -x rtsdtr=high, option ignored
✅ eesave fuse bit set, cleared and verified
✅ the_quick_brown_fox_65536B.hex flash -U write/verify
✅ the_quick_brown_fox_512B.hex eeprom -U write/verify
✅ lorem_ipsum_65536B.srec flash -U write/verify
✅ lorem_ipsum_512B.srec eeprom -U write/verify
✅ 0xff_512B.hex eeprom -U write/verify
✅ the_quick_brown_fox_65536B.hex:a flash -T write/verify
✅ the_quick_brown_fox_512B.hex:a eeprom -T write/verify
✅ lorem_ipsum_65536B.srec flash -T write/verify
✅ lorem_ipsum_512B.srec eeprom -T write/verify
✅ cola-vending-machine.raw flash -T/-U write/verify
✅ holes_pack_my_box_65536B.hex:a flash -U write
✅ holes_the_five_boxing_wizards_65536B.hex flash -T write
✅ holes_pack_my_box_512B.hex:a eeprom -U write
✅ holes_the_five_boxing_wizards_512B.hex eeprom -T write
✅ random_data_64B.bin usersig -T/-U write/read
```

run test NVM V5

```
Prepare "-c serialupdi -xrtsdtr=high -P /dev/cu.usbserial-230 -p avr16eb32" and press 'enter' or 'space' to continue. Press any other key to skip
avrdude error: programmer does not support extended parameter -x rtsdtr=high, option ignored
avrdude error: programmer does not support extended parameter -x rtsdtr=high, option ignored
avrdude error: programmer does not support extended parameter -x rtsdtr=high, option ignored
✅ eesave fuse bit set, cleared and verified
✅ the_quick_brown_fox_16384B.hex flash -U write/verify
✅ the_quick_brown_fox_512B.hex eeprom -U write/verify
✅ lorem_ipsum_16384B.srec flash -U write/verify
✅ lorem_ipsum_512B.srec eeprom -U write/verify
✅ 0xff_512B.hex eeprom -U write/verify
✅ the_quick_brown_fox_16384B.hex:a flash -T write/verify
✅ the_quick_brown_fox_512B.hex:a eeprom -T write/verify
✅ lorem_ipsum_16384B.srec flash -T write/verify
✅ lorem_ipsum_512B.srec eeprom -T write/verify
✅ cola-vending-machine.raw flash -T/-U write/verify
✅ holes_pack_my_box_16384B.hex:a flash -U write
✅ holes_the_five_boxing_wizards_16384B.hex flash -T write
✅ holes_pack_my_box_512B.hex:a eeprom -U write
✅ holes_the_five_boxing_wizards_512B.hex eeprom -T write
✅ random_data_64B.bin usersig -T/-U write/read
```

run test NVM V0

```
[... 74 lines cut ...]
```
Prepare "-c serialupdi -xrtsdtr=high -P /dev/cu.usbserial-230 -p avr128db32" and press 'enter' or 'space' to continue. Press any other key to skip
avrdude error: programmer does not support extended parameter -x rtsdtr=high, option ignored
avrdude error: programmer does not support extended parameter -x rtsdtr=high, option ignored
avrdude error: programmer does not support extended parameter -x rtsdtr=high, option ignored
✅ eesave fuse bit set, cleared and verified
✅ the_quick_brown_fox_131072B.hex flash -U write/verify
✅ the_quick_brown_fox_512B.hex eeprom -U write/verify
✅ lorem_ipsum_131072B.srec flash -U write/verify
✅ lorem_ipsum_512B.srec eeprom -U write/verify
✅ 0xff_512B.hex eeprom -U write/verify
✅ the_quick_brown_fox_131072B.hex:a flash -T write/verify
✅ the_quick_brown_fox_512B.hex:a eeprom -T write/verify
✅ lorem_ipsum_131072B.srec flash -T write/verify
✅ lorem_ipsum_512B.srec eeprom -T write/verify
✅ cola-vending-machine.raw flash -T/-U write/verify
✅ holes_pack_my_box_131072B.hex:a flash -U write
✅ holes_the_five_boxing_wizards_131072B.hex flash -T write
✅ holes_pack_my_box_512B.hex:a eeprom -U write
✅ holes_the_five_boxing_wizards_512B.hex eeprom -T write
✅ random_data_32B.bin usersig -T/-U write/read
```

We have newly confirmed the normal operation of NVM V3 and V5.
No degradation is observed in NVM V0 and V2.

closed 2024-01-28 by stefanrueger