avrdudes/avrdude #1626: Avrdoper support is broken since version 6.4

Nibi-git, 2024-01-19:
Hello!
I'm using an old USB HID programmer Avrdoper https://www.obdev.at/products/vusb/avrdoper.html
with VID_16C0&PID_05DF&REV_0100
Typically the command line to use it looks like this:
`avrdude -c stk500v2 -F -P avrdoper -p m48p -B 20 ... and hex, fuses...`
For a long time I successfully used the version 6.3 of avrdude, but then I tried all newer versions and got an error.
In this case, the LEDs on the programmer board blink, but the operation ends with an error.
```
avrdude error: I/O Error
avrdude error: timeout
avrdude error: 
avrdude error: I/O Error
avrdude error: timeout
avrdude error: 
avrdude error: I/O Error
avrdude error: timeout
avrdude error: 
avrdude error: I/O Error
avrdude error: timeout
avrdude error: 
avrdude error: I/O Error
avrdude error: timeout
avrdude error: 
avrdude error: I/O Error
avrdude error: timeout
avrdude error: timeout communicating with programmer
avrdude error: failed to execute command 0x11
avrdude error: unable to leave programming mode

avrdude done.  Thank you.
```
[log.txt](https://github.com/avrdudes/avrdude/files/13991801/log.txt)

So for now I'm back to the old version 6.3 again.
Is it possible to fix this problem?
Best regards, Roman

mcuee, 2024-01-20:
Please provide more details of the issue as per the following Wiki FAQ entry.
https://github.com/avrdudes/avrdude/wiki/FAQ#how-to-report-an-issue-to-avrdude-project

Please post the full debug log with `-vvvv` which will help in this case.

Are you using Windows? If this is an USB HID device, please do not use libusb-win32 device driver and use Windows native HID driver to see if that works.

Reference:
https://www.obdev.at/products/vusb/avrdoper.html

Nibi-git, 2024-01-20:
Hello! 
Thanks for the answer! I updated the first post, added more detailed logs and information about the OS.
What I noticed is that the x64 version at least correctly detects the MCU:
`avrdude: device signature = 0x1e920a (probably m48p)`

mcuee, 2024-01-20:
I assume you are using the native Windows USB HID driver as the original CDC-ACM version will not work under Wndows 10.
https://www.obdev.at/products/vusb/avrdoper.html
http://www.recursion.jp/prose/avrcdc/

Looking at the debug log, latest git is complete broken for this old programmer.

But then the support may already be broken a few years back since you mention that avrdude 6.4 already does not work.
https://github.com/avrdudes/avrdude/commits/main/src/ser_avrdoper.c

mcuee, 2024-01-20:
@MCUdude and @stefanrueger 

If we compare the debug log of 7.2 release vs git main, there is a regression.

But then 7.2 does not work either. So we may be able to live with it for the upcoming avrdude 7.3 release.

dl8dtl, 2024-01-20:
> But then the support may already be broken a few years back

Since he mentioned it's broken since 6.4, the most obvious change in the 6.4 NEWS file is:

"  * Major changes compared to the previous version:

    - libhidapi support (part of patch #8717)"

So I'd assume that eventually broke AVRdoper back then.

stefanrueger, 2024-01-30:
> So I'd assume that eventually broke AVRdoper back then.

Anything that can be done? @mcuee @MCUdude

mcuee, 2024-01-30:
> > So I'd assume that eventually broke AVRdoper back then.
> 
> Anything that can be done? @mcuee @MCUdude

I do not think I can do much without the physical programmer.

MCUdude, 2024-01-31:
Sorry, I don't have any avrdoper hardware either.
However, one alternative would be to compile the firmware to run on USBasp hardware.

Nibi-git, 2024-01-31:
>  alternative would be to compile the firmware
I have ready-made firmware for the сhinese USBasp v2.0 board. I'll send it in the evening

Nibi-git, 2024-02-01:
The archive contains the compiled main.hex and sources. The firmware is configured to use the HID interface.
Atmega8, 12 Mhz
[firmware.zip](https://github.com/avrdudes/avrdude/files/14124635/firmware.zip)
[image]
[image]

mcuee, 2024-02-02:
@Nibi-git 
Your firmware build seems to work on my USBASP clone with ATmega8A.

avrdude 6.3 release works fine.
```
C:\work\avr\avrdude_test\avrdude_bin\bin_misc\avrdude-6.3-mingw32> 
.\avrdude -c stk500v2 -P avrdoper -p m328p

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.03s

avrdude.exe: Device signature = 0x1e950f (probably m328p)

avrdude.exe: safemode: Fuses OK (E:FD, H:DF, L:F7)

avrdude.exe done.  Thank you.
```

git main does not work. It is quite broken for this avrdoper programmer.

```
C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git -c stk500v2 -P avrdoper -p m328p
avrdude_git error: unable to send message: Get Input/Feature Report DeviceIoControl: (0x00000057) The parameter is incorrect.
avrdude_git error: I/O Error
avrdude_git error: unable to open port avrdoper for programmer stk500v2
Possible candidate serial ports are:
  -P COM3
  -P COM6 or -P ch340
Note that above ports might not be connected to a target board or an AVR programmer.
Also note there may be other direct serial ports not listed above.

avrdude_git done.  Thank you.
```

avrdude 7.2 does not work either, but not as broken as git main.
```
C:\work\avr\avrdude_test\avrdude_bin\bin_misc\avrdude-v7.2-windows-x64> .\avrdude -c stk500v2 -P avrdoper -p m328p
avrdude error: unable to send message: Get Input/Feature Report DeviceIoControl: (0x00000057) The parameter is incorrect.
avrdude error: I/O Error
avrdude error: unable to send message: Get Input/Feature Report DeviceIoControl: (0x00000057) The parameter is incorrect.
avrdude error: I/O Error
avrdude error: timeout
avrdude error: unable to send message: Get Input/Feature Report DeviceIoControl: (0x00000057) The parameter is incorrect.
avrdude error: I/O Error
avrdude error: timeout
avrdude error: unable to send message: Get Input/Feature Report DeviceIoControl: (0x00000057) The parameter is incorrect.
avrdude error: I/O Error
avrdude error: timeout
avrdude error: unable to send message: Get Input/Feature Report DeviceIoControl: (0x00000057) The parameter is incorrect.
avrdude error: I/O Error
avrdude error: timeout
avrdude error: unable to send message: Get Input/Feature Report DeviceIoControl: (0x00000057) The parameter is incorrect.
avrdude error: I/O Error
avrdude error: timeout
avrdude error: unable to send message: Get Input/Feature Report DeviceIoControl: (0x00000057) The parameter is incorrect.
avrdude error: I/O Error
avrdude error: timeout
avrdude error: timeout communicating with programmer
avrdude error: unable to send message: Get Input/Feature Report DeviceIoControl: (0x00000057) The parameter is incorrect.
avrdude error: I/O Error
avrdude error: unable to send message: Get Input/Feature Report DeviceIoControl: (0x00000057) The parameter is incorrect.
avrdude error: I/O Error
avrdude error: timeout
avrdude error: unable to send message: Get Input/Feature Report DeviceIoControl: (0x00000057) The parameter is incorrect.
avrdude error: I/O Error
avrdude error: timeout
avrdude error: unable to send message: Get Input/Feature Report DeviceIoControl: (0x00000057) The parameter is incorrect.
avrdude error: I/O Error
avrdude error: timeout
[... 368 lines cut ...]
avrdude.exe: stk500v2_ReceiveMessage(): timeout
Error sending message: T
avrdude.exe: avrdoperFillBuffer(): I/O Error.
avrdude.exe: stk500v2_ReceiveMessage(): timeout
Error sending message: T
avrdude.exe: avrdoperFillBuffer(): I/O Error.
avrdude.exe: stk500v2_ReceiveMessage(): timeout
Error sending message: T
avrdude.exe: avrdoperFillBuffer(): I/O Error.
avrdude.exe: stk500v2_ReceiveMessage(): timeout
Error sending message: T
avrdude.exe: avrdoperFillBuffer(): I/O Error.
avrdude.exe: stk500v2_ReceiveMessage(): timeout
Error sending message: T
avrdude.exe: avrdoperFillBuffer(): I/O Error.
avrdude.exe: stk500v2_ReceiveMessage(): timeout
Error sending message: T
avrdude.exe: avrdoperFillBuffer(): I/O Error.
avrdude.exe: stk500v2_ReceiveMessage(): timeout
avrdude.exe: stk500v2_getsync(): timeout communicating with programmer
avrdude.exe: stk500v2_command(): failed miserably to execute command 0x11
avrdude.exe: stk500v2_disable(): failed to leave programming mode

avrdude.exe done.  Thank you.
```

mcuee, 2024-02-02:
> Hello! Thanks for the answer! I updated the first post, added more detailed logs and information about the OS. What I noticed is that the 7.2 x64 version at least correctly detects the MCU: `avrdude: device signature = 0x1e920a (probably m48p)` all others - no

Hmm, I can not even get that to work with avrdude 7.2 release.
<details>

```
C:\work\avr\avrdude_test\avrdude_bin\bin_misc\avrdude-v7.2-windows-x64>
 .\avrdude -c stk500v2 -P avrdoper -p m328p -vvvv

avrdude: Version 7.2
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is C:\work\avr\avrdude_test\avrdude_bin\bin_misc\avrdude-v7.2-windows-x64\avrdude.conf

         Using Port                    : avrdoper
         Using Programmer              : stk500v2
avrdude usbGetReport() [ser_avrdoper.c:140] error: unable to send message: Get Input/Feature Report DeviceIoControl: (0x00000057) The parameter is incorrect.
avrdude avrdoperFillBuffer() [ser_avrdoper.c:289] error: I/O Error
Send: 7 bytes: 1b 01 00 01 0e 01 14  "......."
Sending 7 bytes data chunk
avrdude usbGetReport() [ser_avrdoper.c:140] error: unable to send message: Get Input/Feature Report DeviceIoControl: (0x00000057) The parameter is incorrect.
avrdude avrdoperFillBuffer() [ser_avrdoper.c:289] error: I/O Error
avrdude stk500v2_recv() [stk500v2.c:716] error: timeout
Send: 7 bytes: 1b 01 00 01 0e 01 14  "......."
Sending 7 bytes data chunk
avrdude usbGetReport() [ser_avrdoper.c:140] error: unable to send message: Get Input/Feature Report DeviceIoControl: (0x00000057) The parameter is incorrect.
avrdude avrdoperFillBuffer() [ser_avrdoper.c:289] error: I/O Error
avrdude stk500v2_recv() [stk500v2.c:716] error: timeout
Send: 7 bytes: 1b 01 00 01 0e 01 14  "......."
Sending 7 bytes data chunk
avrdude usbGetReport() [ser_avrdoper.c:140] error: unable to send message: Get Input/Feature Report DeviceIoControl: (0x00000057) The parameter is incorrect.
avrdude avrdoperFillBuffer() [ser_avrdoper.c:289] error: I/O Error
avrdude stk500v2_recv() [stk500v2.c:716] error: timeout
Send: 7 bytes: 1b 01 00 01 0e 01 14  "......."
Sending 7 bytes data chunk
avrdude usbGetReport() [ser_avrdoper.c:140] error: unable to send message: Get Input/Feature Report DeviceIoControl: (0x00000057) The parameter is incorrect.
avrdude avrdoperFillBuffer() [ser_avrdoper.c:289] error: I/O Error
avrdude stk500v2_recv() [stk500v2.c:716] error: timeout
Send: 7 bytes: 1b 01 00 01 0e 01 14  "......."
Sending 7 bytes data chunk
avrdude usbGetReport() [ser_avrdoper.c:140] error: unable to send message: Get Input/Feature Report DeviceIoControl: (0x00000057) The parameter is incorrect.
avrdude avrdoperFillBuffer() [ser_avrdoper.c:289] error: I/O Error
avrdude stk500v2_recv() [stk500v2.c:716] error: timeout
Send: 7 bytes: 1b 01 00 01 0e 01 14  "......."
Sending 7 bytes data chunk
avrdude usbGetReport() [ser_avrdoper.c:140] error: unable to send message: Get Input/Feature Report DeviceIoControl: (0x00000057) The parameter is incorrect.
avrdude avrdoperFillBuffer() [ser_avrdoper.c:289] error: I/O Error
avrdude stk500v2_recv() [stk500v2.c:716] error: timeout
avrdude stk500v2_getsync() [stk500v2.c:789] error: timeout communicating with programmer
avrdude usbGetReport() [ser_avrdoper.c:140] error: unable to send message: Get Input/Feature Report DeviceIoControl: (0x00000057) The parameter is incorrect.
avrdude avrdoperFillBuffer() [ser_avrdoper.c:289] error: I/O Error
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : possible i/o
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
[... 762 lines cut ...]
avrdude avrdoperFillBuffer() [ser_avrdoper.c:289] error: I/O Error
avrdude stk500v2_recv() [stk500v2.c:716] error: timeout
Send: 7 bytes: 1b 0f 00 01 0e 01 1a  "......."
Sending 7 bytes data chunk
avrdude usbGetReport() [ser_avrdoper.c:140] error: unable to send message: Get Input/Feature Report DeviceIoControl: (0x00000057) The parameter is incorrect.
avrdude avrdoperFillBuffer() [ser_avrdoper.c:289] error: I/O Error
avrdude stk500v2_recv() [stk500v2.c:716] error: timeout
avrdude stk500v2_getsync() [stk500v2.c:789] error: timeout communicating with programmer
Send: 9 bytes:
1b 0f 00 03 0e 11 01 01  08                        "........."
Sending 9 bytes data chunk
Received 29 bytes data chunk of total 8
Receive: 1 bytes: 1b  "."
Receive: 1 bytes: 0f  "."
Receive: 1 bytes: 00  "."
Receive: 1 bytes: 02  "."
Receive: 1 bytes: 0e  "."
Receive: 1 bytes: 11  "."
Receive: 1 bytes: 00  "."
Receive: 1 bytes: 09  "."

avrdude done.  Thank you.
```

</details>

mcuee, 2024-02-02:
Major commits related to the changes.

1) https://github.com/avrdudes/avrdude/commit/99a9a606080652a9d676df07be15e95359792a10 (libserial support)

2) https://github.com/avrdudes/avrdude/commit/59ecd4cc65359ea8924a71347518989fb28f2caf (Windows native HID api removal, only keeps hidapi)

3) https://github.com/avrdudes/avrdude/commit/845abf1d08101d1a297a4c44a3ad629eaed66e31
patch #9033: avrdoper backend uses libhidapi instead of libusb
* ser_avrdoper: Add libhidapi implementation
* stk500v2.c: Adapt #ifdefs for LIBHIDAPI

mcuee, 2024-02-02:
Unfortunately I am not a programmer and the above is what I can do now. Hopefully @MCUdude or others can chime in to help to sort out the issue.

mcuee, 2024-02-02:
I tend to think we can sort this out after avrdude 7.3 release, so we can probably target this with avrdude 7.4 release.

I will carry out some tests using hidapitester when I finished my testing of `test-avrdude` script under a few more OS over the weekend.

hidapitester
https://github.com/todbot/hidapitester

mcuee, 2024-02-02:
Wiki updated:
https://github.com/avrdudes/avrdude/wiki/Known-limitations-of-avrdude#known-issues-for-programmers

mcuee, 2024-03-17:
Remove this from avrdude 7.4 milestone since there is no active development on this topic.

alexjx, 2024-12-10:
Avrdoper was one of my first DIY programmers. As I am getting back into working on Arduino projects, I noticed this issue and decided to fix it. Please review the pull request. Thanks!

stefanrueger, 2024-12-11:
Thanks @alexjx. 

@Nibi-git @mcuee could you test PR #1937, please, and see whether it works with your HW?

mcuee, 2024-12-12:
> @mcuee could you test PR #1937, please, and see whether it works with your HW?

Yes, I can carry out the test by end of this week.

closed 2024-12-13 by stefanrueger