avrdudes/avrdude #1632: Some programmers need a delay between successive AVRDUDE calls

stefanrueger, 2024-01-22:
The new brilliant [test script](https://github.com/MCUdude/avrdude-test-script) has uncovered that some programmers [need a delay between avrdude calls](https://github.com/avrdudes/avrdude/discussions/1473#discussioncomment-8061916). We should identify a list of these programmers, determine the appropriate delay and add a `sleep(...);` into the *programmer*_close()` routine.

Please @mcuee, @MCUdude identify these instances by using `avrdude_test_script.sh -d0 ...` (no delay in the test suite), determine the needed delay and issue suitable PRs.

MCUdude, 2024-01-23:
I've now tested the following programmers, and only the Nano Every jtag2updi, and the AVR Dragon needs a delay:

| Programmer                     | Delay |
|--------------------------------|-------|
| `dragon_isp`, `dragon_jtag`    | 2s    |
| `jtag2updi` (Arduino Nano Every)  | 0.5s    |
| `jtag2updi` (Thinary Nano Every 4808)  | 0s|
| `avrispmkii` | 0s|
| `powerdebugger_isp`, `powerdebugger_updi` | 0s|
| `pickit4_isp`, `pickit4_updi` | 0s|
| `pkobn_updi` | 0s|
| `xplainedmini_updi` | 0s|
| `xplainedpro_updi` | 0s|
| `xplainedpro_pdi` | 0s|
| `serialupdi` | 0s|
| `usbtiny` | 0s|

MCUdude, 2024-01-23:
The reason why I chose 4 seconds in the script earlier was because my old, unreliable AVR Dragon required this. However, I've just received a new(er) one, and this works perfectly well with 1.5s delay.

mcuee, 2024-01-24:
Take note that the time may need to be longer when there is a fault.

For example, for the following test, some tests are expected to fail, but then it may affect the next test which is supposed to pass.

```
openbsd74$ ./test-avrdude -p "-c urclock -P /dev/ttyU0 -p m328pb"
Testing avrdude version 7.2-20240122 (2f89aece)
Prepare "-c urclock -P /dev/ttyU0 -p m328pb" and press 'enter' or 'space' to continue. Press any other key to skip
openbsd74$ ./test-avrdude -p "-c urclock -P /dev/ttyU0 -p m328pb"
Testing avrdude version 7.2-20240122 (2f89aece)
Prepare "-c urclock -P /dev/ttyU0 -p m328pb" and press 'enter' or 'space' to continue. Press any other key to skip
avrdude error: input [0x0000, 0x7fff] overlaps bootloader [0x7e80, 0x7fff]; consider -xrestore
❌   2.134 s: the_quick_brown_fox_32768B.hex flash -U write/verify (failed command below)
$ avrdude -qq -c urclock -P /dev/ttyU0 -p m328pb -Uflash:w:test_files/the_quick_brown_fox_32768B.hex
✅   6.044 s: the_quick_brown_fox_1024B.hex eeprom -U write/verify
avrdude error: input [0x0000, 0x7fff] overlaps bootloader [0x7e80, 0x7fff]; consider -xrestore
❌   2.135 s: lorem_ipsum_32768B.srec flash -U write/verify (failed command below)
$ avrdude -qq -c urclock -P /dev/ttyU0 -p m328pb -Uflash:w:test_files/lorem_ipsum_32768B.srec
avrdude warning: programmer is not responding
avrdude warning: programmer is not responding
avrdude error: unable to read byte at address 0x0034
avrdude error: read operation not supported for memory eeprom
avrdude error: unable to read all of eeprom memory, rc=-2
❌   6.504 s: lorem_ipsum_1024B.srec eeprom -U write/verify (failed command below)
$ avrdude -qq -c urclock -P /dev/ttyU0 -p m328pb -Ueeprom:w:test_files/lorem_ipsum_1024B.srec
avrdude warning: programmer is not responding
avrdude warning: programmer is not responding
avrdude error: unable to read byte at address 0x0117
avrdude error: read operation not supported for memory eeprom
avrdude error: unable to read all of eeprom memory, rc=-2
❌   7.635 s: 0xff_1024B.hex eeprom -U write/verify (failed command below)
$ avrdude -qq -c urclock -P /dev/ttyU0 -p m328pb -e -Ueeprom:w:test_files/0xff_1024B.hex
./test-avrdude: eval: line 97: conditional binary operator expected
❌   7.635 s: the_quick_brown_fox_32768B.hex:a flash -T write/verify (failed command below)
$ avrdude -qq -c urclock -P /dev/ttyU0 -p m328pb -T "write flash test_files/the_quick_brown_fox_32768B.hex:a"
./test-avrdude: eval: line 97: syntax error near unexpected token `=='
./test-avrdude: eval: line 97: `[[  ==  ]]'
❌   7.635 s: the_quick_brown_fox_1024B.hex:a eeprom -T write/verify (failed command below)
$ avrdude -qq -c urclock -P /dev/ttyU0 -p m328pb -T "write eeprom test_files/the_quick_brown_fox_1024B.hex:a"
./test-avrdude: eval: line 97: conditional binary operator expected
❌   7.635 s: lorem_ipsum_32768B.srec flash -T write/verify (failed command below)
$ avrdude -qq -c urclock -P /dev/ttyU0 -p m328pb -T "write flash test_files/lorem_ipsum_32768B.srec"
./test-avrdude: eval: line 97: syntax error near unexpected token `=='
./test-avrdude: eval: line 97: `[[  ==  ]]'
❌   7.635 s: lorem_ipsum_1024B.srec eeprom -T write/verify (failed command below)
$ avrdude -qq -c urclock -P /dev/ttyU0 -p m328pb -T "write eeprom test_files/lorem_ipsum_1024B.srec"
./test-avrdude: eval: line 97: conditional binary operator expected
❌   7.635 s: cola-vending-machine.raw flash -T/-U write/verify (failed command below)
$ avrdude -qq -c urclock -P /dev/ttyU0 -p m328pb -T 'erase flash; write flash -512 0xc0cac01a 0xcafe "secret Coca .bin recipe"' -U flash:w:test_files/cola-vending-machine.raw -T 'write flash -1024 "Hello World"'
✅   4.304 s: holes_pack_my_box_32768B.hex:a flash -U write
./test-avrdude: eval: line 97: syntax error near unexpected token `=='
./test-avrdude: eval: line 97: `[[  ==  ]]'
❌   4.304 s: holes_the_five_boxing_wizards_32768B.hex flash -T write (failed command below)
$ avrdude -qq -c urclock -P /dev/ttyU0 -p m328pb -T "write flash test_files/holes_the_five_boxing_wizards_32768B.hex"
✅   2.834 s: holes_pack_my_box_1024B.hex:a eeprom -U write
✅   2.834 s: holes_the_five_boxing_wizards_1024B.hex eeprom -T write

One or more avrdude "-c urclock -P /dev/ttyU0 -p m328pb" tests failed. Do you want to retry this particular test? (y/n):
 n

openbsd74$ avrdude -qq -c urclock -P /dev/ttyU0 -p m328pb -Ueeprom:w:test_files/lorem_ipsum_1024B.srec && echo OK
OK

ite eeprom test_files/the_quick_brown_fox_1024B.hex:a" && echo OK                                                     <
OK
```

MCUdude, 2024-01-25:
@mcuee The only programmers I know of that require a delay are the Dragon and the Nano Every.
The tests performed below are supposed to fail when writing to EEPROM. In this case, it failed the same tests regardless of whether I added a delay of 1.5 seconds or 4 seconds. This tells me that 1.5 seconds for the Dragon should be safe. I'll have to properly test the Nano Every as well, including tests where it should fail.

```
$ bash test-avrdude -d 1.5 -p "-cdragon_jtag -patmega165"
Testing avrdude version 7.2-20230720
Prepare "-cdragon_jtag -patmega165" and press 'enter' or 'space' to continue. Press any other key to skip
✅   0.000 s: set, clear and read eesave fuse bit
✅   5.265 s: the_quick_brown_fox_16384B.hex flash -U write/verify
❌   2.529 s: the_quick_brown_fox_512B.hex eeprom -U write/verify (failed command below)
$ avrdude -qqqqq -cdragon_jtag -patmega165 -Ueeprom:w:test_files/the_quick_brown_fox_512B.hex
✅   5.275 s: lorem_ipsum_16384B.srec flash -U write/verify
❌   2.524 s: lorem_ipsum_512B.srec eeprom -U write/verify (failed command below)
$ avrdude -qqqqq -cdragon_jtag -patmega165 -Ueeprom:w:test_files/lorem_ipsum_512B.srec
❌   3.164 s: 0xff_512B.hex eeprom -U write/verify (failed command below)
$ avrdude -qqqqq -cdragon_jtag -patmega165 -e -Ueeprom:w:test_files/0xff_512B.hex
✅   3.164 s: the_quick_brown_fox_16384B.hex:a flash -T write/verify
❌   3.164 s: the_quick_brown_fox_512B.hex:a eeprom -T write/verify (failed command below)
$ avrdude -qq -cdragon_jtag -patmega165 -T "write eeprom test_files/the_quick_brown_fox_512B.hex:a"
✅   3.164 s: lorem_ipsum_16384B.srec flash -T write/verify
❌   3.164 s: lorem_ipsum_512B.srec eeprom -T write/verify (failed command below)
$ avrdude -qq -cdragon_jtag -patmega165 -T "write eeprom test_files/lorem_ipsum_512B.srec"
✅   3.164 s: cola-vending-machine.raw flash -T/-U write/verify
✅   2.640 s: holes_pack_my_box_16384B.hex:a flash -U write
✅   2.640 s: holes_the_five_boxing_wizards_16384B.hex flash -T write
❌   1.882 s: holes_pack_my_box_512B.hex:a eeprom -U write (failed command below)
$ avrdude -qqqqq -cdragon_jtag -patmega165 -Ueeprom:w:test_files/holes_pack_my_box_512B.hex:a
❌   1.882 s: holes_the_five_boxing_wizards_512B.hex eeprom -T write (failed command below)
$ avrdude -qq -cdragon_jtag -patmega165 -T "write eeprom test_files/holes_the_five_boxing_wizards_512B.hex"
``

stefanrueger, 2024-01-25:
@MCUdude Brilliant research. So, could you put the necessary sleep into the respective `-c` programmer's `_close()` functions and create a PR, so that the test-avrdude script can use a zero delay?

MCUdude, 2024-01-25:
> So, could you put the necessary sleep into the respective -c programmer's _close() functions and create a PR, so that the test-avrdude script can use a zero delay?

Sure! But what should we do about the jtag2updi? It's only the Nano Every (using -r) that needs a delay. I'm pretty sure a "vanilla" jtag2updi programmer doesn't need any.

stefanrueger, 2024-01-25:
Could you compare the Nano Every `-c` string against the global variable `pgmid` and only sleep when they match?
https://github.com/avrdudes/avrdude/blob/2f89aece1e551582223440e07e07047b526a6c25/src/main.c#L217

stefanrueger, 2024-01-25:
> Take note that the time may need to be longer when there is a fault.

@mcuee Aha; I hadn't thought of that. Try to break the latest `test-avrdude` script. When you see such instances, we could generally sleep 5 seconds or so when a test fails.

MCUdude, 2024-01-25:
> Could you compare the Nano Every -c string against the global variable pgmid and only sleep when they match?

The Nano Every doesn't have a dedicated pgmid. it uses `-cjtag2updi` along with `-r`.

stefanrueger, 2024-01-25:
Add "nanoevery" to the id list of jtag2updi?

MCUdude, 2024-01-25:
> Add "nanoevery" to the id list of jtag2updi?

We could do that, yes. Alternatively, we could identify a "real" Nano Every board by using the USB VID/PID that libserialport provides

mcuee, 2024-01-25:
> > Take note that the time may need to be longer when there is a fault.
> 
> @mcuee Aha; I hadn't thought of that. Try to break the latest `test-avrdude` script. When you see such instances, we could generally sleep 5 seconds or so when a test fails.

Let me try this later this week.

mcuee, 2024-01-25:
> @mcuee The only programmers I know of that require a delay are the Dragon and the Nano Every. 

I will try as well on other programmers.

I tend to think some bootloaders may require a delay as well.

stefanrueger, 2024-01-26:
> using the USB VID/PID that libserialport provides

even better

mcuee, 2024-01-30:
@MCUdude 
`delay = 0.5 sec` still failed for my Nano Every under Windows 10.

```
$ ./test-avrdude -e "./avrdude.exe" -d 0 -p "-c jtag2updi -P COM8 -p m4809 -r"
Testing ./avrdude.exe version 7.2-20240130 (742cb9dd)
Prepare "-c jtag2updi -P COM8 -p m4809 -r" and press 'enter' or 'space' to continue. Press any other key to skip
✅   4.494 s: fuse access: clear, set and read eesave fuse bit
✅   4.407 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
✅   4.503 s: chip erase
❌  54.071 s: flash -U write/verify holes_rjmp_loops_49152B.hex (failed command below)
$ ./avrdude.exe -qq -c jtag2updi -P COM8 -p m4809 -r -l /dev/shm/test-avrdude.log.qcgCyD -Uflash:w:./test_files/holes_rjmp_loops_49152B.hex
✅  26.568 s: flash -U write/verify rjmp_loops_for_bootloaders_49152B.hex
❌  54.184 s: flash -T write/verify holes_rjmp_loops_49152B.hex (failed command below)
$ ./avrdude.exe -qq -c jtag2updi -P COM8 -p m4809 -r -l /dev/shm/test-avrdude.log.qcgCyD -T "write flash ./test_files/holes_rjmp_loops_49152B.hex:a"
✅   8.297 s: flash -T write/verify rjmp_loops_for_bootloaders_49152B.hex
❌  54.119 s: eeprom check whether programmer can flip 0s to 1s (failed command below)
$ ./avrdude.exe -qq -c jtag2updi -P COM8 -p m4809 -r -l /dev/shm/test-avrdude.log.qcgCyD -Ueeprom:w:0x55:m -Ueeprom:w:0xaa:m
# ... the next test may therefore take longer
✅   4.867 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_256B.hex
❌  54.063 s: chip erase and spot check flash is actually erased (failed command below)
$ ./avrdude.exe -qq -c jtag2updi -P COM8 -p m4809 -r -l /dev/shm/test-avrdude.log.qcgCyD -e -Uflash:v:./test_files/holes_flash_0xff_49152B.hex
❌   4.573 s: usersig -T/-U write/read random_data_64B.bin (failed command below)
$ ./avrdude.exe -qq -c jtag2updi -P COM8 -p m4809 -r -l /dev/shm/test-avrdude.log.qcgCyD -T "erase usersig; write usersig ./test_files/random_data_64B.bin" -T flush -U usersig:r:/dev/shm/usersig_dump14157_64B.bin:r -U usersig:v:/dev/shm/usersig_dump14157_64B.bin:r -T "erase usersig" -T flush -U usersig:v:./test_files/0xff_64B.hex:i

One or more AVRDUDE "-c jtag2updi -P COM8 -p m4809 -r" tests failed. Do you want to retry this particular test? (y/n): n

$ ./test-avrdude -e "./avrdude.exe" -d 0.5 -p "-c jtag2updi -P COM8 -p m4809 -r"
Testing ./avrdude.exe version 7.2-20240130 (742cb9dd)
Prepare "-c jtag2updi -P COM8 -p m4809 -r" and press 'enter' or 'space' to continue. Press any other key to skip
✅   4.463 s: fuse access: clear, set and read eesave fuse bit
✅   4.499 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
✅   4.667 s: chip erase
✅  11.995 s: flash -U write/verify holes_rjmp_loops_49152B.hex
✅   5.745 s: flash -T write/verify holes_rjmp_loops_49152B.hex
✅   4.571 s: eeprom check whether programmer can flip 0s to 1s
✅   4.631 s: eeprom -U write/verify holes_pack_my_box_256B.hex
✅   4.660 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_256B.hex
✅   5.913 s: chip erase and spot check flash is actually erased
✅   4.428 s: spot check eeprom is erased, too
❌   4.521 s: usersig -T/-U write/read random_data_64B.bin (failed command below)
$ ./avrdude.exe -qq -c jtag2updi -P COM8 -p m4809 -r -l /dev/shm/test-avrdude.log.pMXXoV -T "erase usersig; write usersig ./test_files/random_data_64B.bin" -T flush -U usersig:r:/dev/shm/usersig_dump14284_64B.bin:r -U usersig:v:/dev/shm/usersig_dump14284_64B.bin:r -T "erase usersig" -T flush -U usersig:v:./test_files/0xff_64B.hex:i

One or more AVRDUDE "-c jtag2updi -P COM8 -p m4809 -r" tests failed. Do you want to retry this particular test? (y/n): n
```

mcuee, 2024-01-30:
Test results -- in progress

| Programmer                     | Delay |
|--------------------------------|-------|
| `dragon_isp`, `dragon_jtag`    | to check   |
| `jtag2updi` (Arduino Nano Every)  | 0.5s    |
| `jtag2updi` (Thinary Nano Every 4808)  | 0s|
| `urboot urclock` | 0s|

MCUdude, 2024-01-30:
@mcuee so your Nano Every didn't need more than 0.5 seconds after all?

mcuee, 2024-01-30:
> @mcuee so your Nano Every didn't need more than 0.5 seconds after all? The delay is done within the avrdude program, so you'll have to do `$ ./test-avrdude -e "./avrdude.exe" -d 0 -p "-c jtag2updi -P COM8 -p m4809 -r"`
> 
> where the delay in the script are 0, `-d 0`

Hmm, `-d 0` failed and `-d 0.5` passed as per the above results. 
* https://github.com/avrdudes/avrdude/issues/1632#issuecomment-1916235690

MCUdude, 2024-01-30:
#1647 adds a fixed 0.5s delay if you use `-c nanoevery`. You'll have to use `-c nanoevery` to get the built-in delay.

If `-c nanoevery doesn't work` due to the built-in delay being too small, please try to increase the script delay with 100ms at the time, `-d 0.1`, `-d 0.2` etc.

mcuee, 2024-01-30:
@stefanrueger 

Your latest change does not work. MSYS2 `/dev/shm` is wrtable for log file but somehow not `bin` file. Kind of strange. 

This is with my home laptop running Windows 11.
```
MINGW64 /c/work/avr/avrdude_test/avrdude_main/tools
$ ./test-avrdude -e "./avrdude.exe" -d 0 -p "-c dryrun -p m4808" -v
Testing ./avrdude.exe version 7.2-20240130 (3d3a387d)
Prepare "-c dryrun -p m4808" and press 'enter' or 'space' to continue. Press any other key to skip
✅   0.099 s: fuse access: clear, set and read eesave fuse bit
✅   0.101 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
✅   0.102 s: chip erase
✅   0.105 s: flash -U write/verify holes_rjmp_loops_49152B.hex
✅   0.102 s: flash -T write/verify holes_rjmp_loops_49152B.hex
✅   0.100 s: eeprom check whether programmer can flip 0s to 1s
✅   0.101 s: eeprom -U write/verify holes_pack_my_box_256B.hex
✅   0.104 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_256B.hex
✅   0.113 s: chip erase and spot check flash is actually erased
✅   0.099 s: spot check eeprom is erased, too
❌   0.114 s: usersig -T/-U write/read random_data_64B.bin (failed command below)
$ ./avrdude.exe -qq -c dryrun -p m4808 -l /dev/shm/test-avrdude.log.KqMEIY -T "erase usersig; write usersig ./test_files/random_data_64B.bin" -T flush -U usersig:r:/dev/shm/usersig_dump1629_64B.bin:r -U usersig:v:/dev/shm/usersig_dump1629_64B.bin:r -T "erase usersig" -T flush -U usersig:v:./test_files/0xff_64B.hex:i
avrdude OS error: file /dev/shm/usersig_dump1629_64B.bin is not writeable: No such file or directory
avrdude OS error: file /dev/shm/usersig_dump1629_64B.bin is not readable: No such file or directory
avrdude OS error: cannot open output file /dev/shm/usersig_dump1629_64B.bin: No such file or directory
avrdude error: write to file /dev/shm/usersig_dump1629_64B.bin failed

One or more AVRDUDE "-c dryrun -p m4808" tests failed. Do you want to retry this particular test? (y/n):
```

Looks like we need to change the last one to use `/tmp`.

mcuee, 2024-01-30:
More details.

Very strange.

```
$ ./avrdude.exe -c dryrun -p m4808 -l /dev/shm/test-avrdude.log.KqMEIY -T "erase usersig; write usersig ./test_files/random_data_64B.bin" -T flush -U usersig:r:/dev/shm/usersig_dump1629_64B.bin:r -U usersig:v:/dev/shm/usersig_dump1629_64B.bin:r -T "erase usersig" -T flush -U usersig:v:./test_files/0xff_64B.hex:i

$ cat /dev/shm/test-avrdude.log.KqMEIY
avrdude OS error: file /dev/shm/usersig_dump1629_64B.bin is not writeable: No such file or directory
avrdude OS error: file /dev/shm/usersig_dump1629_64B.bin is not readable: No such file or directory
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9650 (probably m4808)

avrdude: processing -T erase usersig; write usersig ./test_files/random_data_64B.bin
Caching | ################################################## | 100% 0.00s
Caching | ################################################## | 100% 0.00s

avrdude: processing -T flush
avrdude: synching cache to device ... done

avrdude: processing -U usersig:r:/dev/shm/usersig_dump1629_64B.bin:r
avrdude: reading userrow/usersig memory ...
Reading | ################################################## | 100% 0.00s
avrdude: writing output file /dev/shm/usersig_dump1629_64B.bin
avrdude OS error: cannot open output file /dev/shm/usersig_dump1629_64B.bin: No such file or directory
avrdude error: write to file /dev/shm/usersig_dump1629_64B.bin failed

avrdude done.  Thank you.
```

mcuee, 2024-01-30:
@stefanrueger 

Not so sure how to deal with Teensy bootloader which has some limitations. I was not using the script to test Teensy before.
* https://github.com/avrdudes/avrdude/discussions/1473#discussioncomment-8140591

WIth the script, the prompt was not there and I have to rely on Windows device manager sound to press the boot button.

The failures are kind of expected -- I believe it does not support Flash/EEprom programming in terminal mode.

```
$ ./test-avrdude -e "./avrdude.exe" -d 0 -p "-c teensy -p m32u4 -x wait -V"
Testing ./avrdude.exe version 7.2-20240130 (3d3a387d)
Prepare "-c teensy -p m32u4 -x wait -V" and press 'enter' or 'space' to continue. Press any other key to skip
✅  56.875 s: chip erase
✅   0.536 s: flash -U write/verify holes_rjmp_loops_32768B.hex
❌   5.951 s: flash -T write/verify holes_rjmp_loops_32768B.hex (failed command below)
$ ./avrdude.exe -qq -c teensy -p m32u4 -x wait -V -l /dev/shm/test-avrdude.log.FPBCvP -T "write flash ./test_files/holes_rjmp_loops_32768B.hex:a"
❌   0.412 s: flash -T write/verify rjmp_loops_for_bootloaders_32768B.hex (failed command below)
$ ./avrdude.exe -qq -c teensy -p m32u4 -x wait -V -l /dev/shm/test-avrdude.log.FPBCvP -T "write flash ./test_files/rjmp_loops_for_bootloaders_32768B.hex:a"
✅   0.101 s: eeprom check whether programmer can flip 0s to 1s
✅   0.115 s: eeprom -U write/verify holes_pack_my_box_1024B.hex
❌   0.119 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_1024B.hex (failed command below)
$ ./avrdude.exe -qq -c teensy -p m32u4 -x wait -V -l /dev/shm/test-avrdude.log.FPBCvP -T "write eeprom ./test_files/holes_the_five_boxing_wizards_1024B.hex:a" -T flush -T "write eeprom ./test_files/holes_pack_my_box_1024B.hex:a"
❌   0.148 s: eeprom -T write/verify lorem_ipsum_1024B.srec (failed command below)
$ ./avrdude.exe -qq -c teensy -p m32u4 -x wait -V -l /dev/shm/test-avrdude.log.FPBCvP -T "write eeprom ./test_files/lorem_ipsum_1024B.srec:a"
❌   0.114 s: chip erase and spot check flash is actually erased (failed command below)
$ ./avrdude.exe -qq -c teensy -p m32u4 -x wait -V -l /dev/shm/test-avrdude.log.FPBCvP -e -Uflash:v:./test_files/holes_flash_0xff_32768B.hex

One or more AVRDUDE "-c teensy -p m32u4 -x wait -V" tests failed. Do you want to retry this particular test? (y/n):
```

mcuee, 2024-01-30:
Wiring bootloader is not good to be tested with the script. Anyway, `delay = 0s` is okay as `delay =4s` does not help.

```
$ ./test-avrdude -e "./avrdude.exe" -d 0 -p " -c wiring -p m2560 -P ch340"
Testing ./avrdude.exe version 7.2-20240130 (3d3a387d)
Prepare " -c wiring -p m2560 -P ch340" and press 'enter' or 'space' to continue. Press any other key to skip
❌   0.580 s: chip erase (failed command below)
$ ./avrdude.exe -qq -c wiring -p m2560 -P ch340 -l /dev/shm/test-avrdude.log.8QVBM8 -e
❌   0.595 s: flash -U write/verify holes_rjmp_loops_262144B.hex (failed command below)
$ ./avrdude.exe -qq -c wiring -p m2560 -P ch340 -l /dev/shm/test-avrdude.log.8QVBM8 -Uflash:w:./test_files/holes_rjmp_loops_262144B.hex
❌   0.586 s: flash -U write/verify rjmp_loops_for_bootloaders_262144B.hex (failed command below)
$ ./avrdude.exe -qq -c wiring -p m2560 -P ch340 -l /dev/shm/test-avrdude.log.8QVBM8 -Uflash:w:./test_files/rjmp_loops_for_bootloaders_262144B.hex
❌   5.869 s: flash -T write/verify holes_rjmp_loops_262144B.hex (failed command below)
$ ./avrdude.exe -qq -c wiring -p m2560 -P ch340 -l /dev/shm/test-avrdude.log.8QVBM8 -T "write flash ./test_files/holes_rjmp_loops_262144B.hex:a"
❌  16.161 s: flash -T write/verify rjmp_loops_for_bootloaders_262144B.hex (failed command below)
$ ./avrdude.exe -qq -c wiring -p m2560 -P ch340 -l /dev/shm/test-avrdude.log.8QVBM8 -T "write flash ./test_files/rjmp_loops_for_bootloaders_262144B.hex:a"
✅   0.652 s: eeprom check whether programmer can flip 0s to 1s
✅   5.685 s: eeprom -U write/verify holes_pack_my_box_4096B.hex
✅  11.819 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_4096B.hex
❌   0.591 s: chip erase and spot check flash is actually erased (failed command below)
$ ./avrdude.exe -qq -c wiring -p m2560 -P ch340 -l /dev/shm/test-avrdude.log.8QVBM8 -e -Uflash:v:./test_files/holes_flash_0xff_262144B.hex

One or more AVRDUDE " -c wiring -p m2560 -P ch340" tests failed. Do you want to retry this particular test? (y/n): n

$ ./test-avrdude -e "./avrdude.exe" -d 4 -p " -c wiring -p m2560 -P ch340"
Testing ./avrdude.exe version 7.2-20240130 (3d3a387d)
Prepare " -c wiring -p m2560 -P ch340" and press 'enter' or 'space' to continue. Press any other key to skip
❌   0.585 s: chip erase (failed command below)
$ ./avrdude.exe -qq -c wiring -p m2560 -P ch340 -l /dev/shm/test-avrdude.log.ssn2A4 -e
❌   0.589 s: flash -U write/verify holes_rjmp_loops_262144B.hex (failed command below)
$ ./avrdude.exe -qq -c wiring -p m2560 -P ch340 -l /dev/shm/test-avrdude.log.ssn2A4 -Uflash:w:./test_files/holes_rjmp_loops_262144B.hex
❌   0.599 s: flash -U write/verify rjmp_loops_for_bootloaders_262144B.hex (failed command below)
$ ./avrdude.exe -qq -c wiring -p m2560 -P ch340 -l /dev/shm/test-avrdude.log.ssn2A4 -Uflash:w:./test_files/rjmp_loops_for_bootloaders_262144B.hex
❌   5.886 s: flash -T write/verify holes_rjmp_loops_262144B.hex (failed command below)
$ ./avrdude.exe -qq -c wiring -p m2560 -P ch340 -l /dev/shm/test-avrdude.log.ssn2A4 -T "write flash ./test_files/holes_rjmp_loops_262144B.hex:a"
❌  16.138 s: flash -T write/verify rjmp_loops_for_bootloaders_262144B.hex (failed command below)
$ ./avrdude.exe -qq -c wiring -p m2560 -P ch340 -l /dev/shm/test-avrdude.log.ssn2A4 -T "write flash ./test_files/rjmp_loops_for_bootloaders_262144B.hex:a"
✅   0.659 s: eeprom check whether programmer can flip 0s to 1s
✅   5.693 s: eeprom -U write/verify holes_pack_my_box_4096B.hex
✅  11.838 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_4096B.hex
❌   0.637 s: chip erase and spot check flash is actually erased (failed command below)
$ ./avrdude.exe -qq -c wiring -p m2560 -P ch340 -l /dev/shm/test-avrdude.log.ssn2A4 -e -Uflash:v:./test_files/holes_flash_0xff_262144B.hex

One or more AVRDUDE " -c wiring -p m2560 -P ch340" tests failed. Do you want to retry this particular test? (y/n): n
```

Previous test results here using earlier version of the script (about two weeks ago).
https://github.com/avrdudes/avrdude/discussions/1473#discussioncomment-8165316

```
 MINGW64 /c/work/avr/avrdude_test/avrdude_bin/avrdude-test-script
$ ./avrdude_test_script_wiring_m2560.sh
Prepare "-c wiring -P COM7 -p m2560" and press 'enter' or 'space' to continue. Press any other key to skip
✅ the_quick_brown_fox_4096B.hex eeprom -U write/verify
✅ lorem_ipsum_4096B.srec eeprom -U write/verify
avrdude error: command failed
❌ 0xff_4096B.hex eeprom -U write/verify
➡️ command "./avrdude -qq -c wiring -P COM7 -p m2560 -e -Ueeprom:w:test_files/0xff_4096B.hex -Ueeprom:v:test_files/0xff_4096B.hex" failed
✅ the_quick_brown_fox_4096B.hex:a eeprom -T write/verify
✅ lorem_ipsum_4096B.srec eeprom -T write/verify
avrdude error: command failed
❌ holes_pack_my_box_262144B.hex:a flash -U write
➡️ command "./avrdude -qq -c wiring -P COM7 -p m2560 -Uflash:w:test_files/holes_pack_my_box_262144B.hex:a" failed
❌ holes_the_five_boxing_wizards_262144B.hex flash -T write
➡️ command "./avrdude -qq -c wiring -P COM7 -p m2560 -T "write flash test_files/holes_the_five_boxing_wizards_262144B.hex"" failed
✅ holes_pack_my_box_4096B.hex:a eeprom -U write
✅ holes_the_five_boxing_wizards_4096B.hex eeprom -T write

One or more avrdude "-c wiring -P COM7 -p m2560" tests failed. Do you want to retry this particular test? (y/n): n
```

mcuee, 2024-01-30:
PICKit 4 JTAG --> `delay = 0` is okay.

```
$ ./test-avrdude -e "./avrdude.exe" -d 0 -p "-c pickit4_jtag -p m32a"
Testing ./avrdude.exe version 7.2-20240130 (3d3a387d)
Prepare "-c pickit4_jtag -p m32a" and press 'enter' or 'space' to continue. Press any other key to skip
✅   0.288 s: fuse access: clear, set and read eesave fuse bit
✅   0.249 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
✅   0.258 s: chip erase
✅   2.387 s: flash -U write/verify holes_rjmp_loops_32768B.hex
✅   1.161 s: flash -T write/verify holes_rjmp_loops_32768B.hex
❌   0.296 s: eeprom check whether programmer can flip 0s to 1s (failed command below)
$ ./avrdude.exe -qq -c pickit4_jtag -p m32a -l /dev/shm/test-avrdude.log.e1j9GI -Ueeprom:w:0x55:m -Ueeprom:w:0xaa:m
# ... the next test may therefore take longer
✅  11.131 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_1024B.hex
✅   1.169 s: chip erase and spot check flash is actually erased
✅   0.609 s: spot check eeprom is erased, too

One or more AVRDUDE "-c pickit4_jtag -p m32a" tests failed. Do you want to retry this particular test? (y/n): n
```

mcuee, 2024-01-30:
Strange that I can not get Optiboot ATmega328PB to work well -- need to find the right hex file.
I tested with urboot stk500v1 bootloader first and it works well with `-c arduino`.

```
$ ./avrdude -c urclock -p m328pb -P ch340 -xshowall
avrdude: AVR device initialized and ready to accept instructions
0000ffffffff 0000-00-00 00.00  application 0 store 0 meta 1 boot 512 u7.7 wes-hprac vector 0 (RESET) ATmega328PB

$ ./test-avrdude -e "./avrdude.exe" -d 0 -v -p "-c arduino -p m328pb -P ch340"
Testing ./avrdude.exe version 7.2-20240130 (3d3a387d)
Prepare "-c arduino -p m328pb -P ch340" and press 'enter' or 'space' to continue. Press any other key to skip
✅   1.827 s: chip erase
✅   3.711 s: flash -U write/verify holes_rjmp_loops_32768B.hex
✅   2.127 s: flash -T write/verify holes_rjmp_loops_32768B.hex
✅   1.353 s: eeprom check whether programmer can flip 0s to 1s
✅   2.940 s: eeprom -U write/verify holes_pack_my_box_1024B.hex
✅   5.122 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_1024B.hex
✅   2.650 s: chip erase and spot check flash is actually erased
```

mcuee, 2024-01-30:
> Strange that I can not get Optiboot ATmega328PB to work well -- need to find the right hex file.

In the end I used @MCUdude's old Optiboot file which does not support EEPROM.
```
$ ./avrdude -c usbasp -qq -p m328pb -U optiboot_flash_atmega328pb_UART0_115200_16000000L_B5.hex && echo OK
OK

$ ./avrdude -c usbasp -p m328pb -qqt
avrdude> dump lfuse
0000  ff                                                |.               |
avrdude> dump hfuse
0000  d4                                                |.               |
avrdude> dump efuse
0000  f5                                                |.               |
avrdude> quit

$ ./test-avrdude -e "./avrdude.exe" -d 0 -s -v -p "-c arduino -p m328pb -P ch340"
Testing ./avrdude.exe version 7.2-20240130 (3d3a387d)
Prepare "-c arduino -p m328pb -P ch340" and press 'enter' or 'space' to continue. Press any other key to skip
✅   1.312 s: chip erase
✅   3.202 s: flash -U write/verify holes_rjmp_loops_32768B.hex
✅   2.118 s: flash -T write/verify holes_rjmp_loops_32768B.hex
❌   2.154 s: chip erase and spot check flash is actually erased (failed command below)
$ ./avrdude.exe -qq -c arduino -p m328pb -P ch340 -l /dev/shm/test-avrdude.log.rN5V8B -e -Uflash:v:./test_files/holes_flash_0xff_32768B.hex
avrdude warning: verification mismatch
        device 0xcf != input 0xff at addr 0x2001 (error)
avrdude error: verification mismatch

One or more AVRDUDE "-c arduino -p m328pb -P ch340" tests failed. Do you want to retry this particular test? (y/n): n

$ ./avrdude -c urclock -p m328pb -P ch340 -xshowall
avrdude: AVR device initialized and ready to accept instructions
0 0000-00-00 00.00  application 0 store 0 meta 1 boot 512 o8.0 --s-h-r-- vector 0 (RESET) ATmega328PB

```

mcuee, 2024-01-30:
@stefanrueger 

Not so sure why `-c urclock` has one more failures than `-c arduino` in the above Optiboot test. Please help to take a look.

The source code of @MCUdude's old Optiboot code is here (Minicore already migrated to urboot)
https://github.com/MCUdude/MiniCore/tree/master/avr/bootloaders/optiboot_flash

stefanrueger, 2024-01-30:
>  why -c urclock has one more failures than -c arduino in the above Optiboot test

Well, optiboot accepts a CE command and replies *all OK* but does *nothing*. This is an optiboot *bug*. `-c arduino` believes this. `-c urclock` *knows* this problem and *emulates* a CE for the next `-U` upload which (in this case) is not issued; it also tells the user about this in a warning. This is what triggers the test-avrdude script into correctly thinking that optiboot *does not* carry out the CE. So, all is OK (and `-c urclock` is better than `-c arduino` :smile:)

mcuee, 2024-01-31:
@stefanrueger 

Thanks for the explanation. To be honest I do not see any need to use Optiboot now for Classic AVR chips. That is also why I faced difficulties in finding a working Optiboot file for the ATmega328PB. Same thing I also do not see the needs of Wiring bootloader (used by Arduino Mega2560 by default). I am using them this time just for testing purpose.

urboot rules for classic AVR chips. :-)

Hopefully we will have urboot for UPDI AVR chips to replace Optiboot_x and Optiboot_dx in the future. Same for PID xmega parts. :-)

PR #1647 (2024-01-31): Add a fixed delay for AVR Dragon and Arduino Nano Every on port close
Closes #1632
closed 2024-01-31 by stefanrueger