avrdudes/avrdude #1633: Writing one-byte files to flash fails

stefanrueger, 2024-01-23:
```
$ avrdude -qqc dryrun -p m328p -U flash:w:42:m && echo OK
avrdude warning: verification mismatch
        device 0xff != input 0x2a at addr 0x0000 (error)
avrdude error: verification mismatch
```

A rather rare case, but still, if someone wants to just change/overwrite the first byte of flash it won't work. This problem is programmer and part independent. What should happen is that (at least) 2 bytes are read from flash, the first byte modified and (at least) 2 bytes written back to the device.

dl8dtl, 2024-01-23:
Well, the flash is organized in 16-bit words, so it doesn't come out as a surprise to me.

stefanrueger, 2024-01-23:
True; avrdude takes great care that files with an odd number of bytes *can* be written to flash (think data after the application), it's just that one-byte files still fail.

PR #1634 (2024-01-28): Fix upload of one-byte files to flash
Fixes #1633 

With this PR all works fine:
```
$ avrdude -qqc dryrun -p m328p -U flash:w:42:m && echo OK
OK
```
closed 2024-01-28 by stefanrueger