avrdudes/avrdude #1635: AVRDUDE warnings

weiying-chen, 2024-01-23:
I get AVRDUDE warnings like these:

>avrdude warning: ATmega329's flash writepage misses a necessary address bit a13 \[/tmp/.avrdude-OHDaQR.conf:5780\]

And these:

>avrdude warning: a4 would normally be expected to be a5 \[/tmp/.avrdude-OHDaQR.conf:11367\]

I think this means there's a discrepancy between my Arduino board and the AVRDUDE config file.

What should I do to fix this? Or maybe this won't cause any major issues since everything seems to running successfully?

    avrdude: 484 bytes of flash verified
    avrdude done.  Thank you. 

Note: I'm using [Arduino Uno R3](https://m.indiamart.com/proddetail/arduino-uno-smd-20433773355.html). And I'm running AVRDUDE using `cargo run` (Rust) on Ubuntu.

Also posted [here](https://m.indiamart.com/proddetail/arduino-uno-smd-20433773355.html).

stefanrueger, 2024-01-23:
These warnings look like the setup uses a new(er) avrdude with an old configuration file that contains errors. This is not recommended. Always use the avrdude.conf file that comes with the particular release.

If a part is not mentioned by any f these warnings (eg, the ATmega328P) then programming these parts won't be affected.

weiying-chen, 2024-01-23:
> These warnings look like the setup uses a new(er) avrdude with an old configuration file that contains errors. This is not recommended. Always use the avrdude.conf file that comes with the [...]

`avrdude -v` outputs this:

```
avrdude: Version 7.1
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is /etc/avrdude.conf
         User configuration file is /home/wei-ying-chen/.avrduderc
         User configuration file does not exist or is not a regular file, skipping
```
Sorry but where do I get the right `avrdude.conf` and where should I put it?

closed 2024-01-23 by mcuee