avrdudes/avrdude #1638: avrdude serialupdi issues under FreeBSD 14.0

mcuee, 2024-01-24:
Under both physical machine and Proxmox VM, I have the same issue.

```
$ sudo avrdude -c serialupdi -P /dev/ttyU0 -p avr64dd32
avrdude OS error: tcsetattr() failed
avrdude OS error: cannot set attributes for port /dev/ttyU0: Input/output error
avrdude error: unable to open port /dev/ttyU0 for programmer serialupdi
A possible candidate serial port is:
  -P /dev/cuaU0 or -P ch340
Note that above port might not be connected to a target board or an AVR programmer.
Also note there may be other direct serial ports not listed above.

avrdude done.  Thank you.
```

`-c urclock` works with CH340 based Uno clone using ATmega328PB.

mcuee, 2024-01-24:
Building without libserialport does not help.

```
mcuee@freebsd14:~/build/avrdude $ ./build.sh -f "-D HAVE_LIBSERIALPORT:BOOL=OFF"
...
mcuee@freebsd14:~/build/avrdude $ ./build_freebsd/src/avrdude -c serialupdi -P /dev/cuaU1 -p avr64dd32
avrdude OS error: tcsetattr() failed
avrdude OS error: cannot set attributes for port /dev/cuaU1: Input/output error
avrdude error: unable to open port /dev/cuaU1 for programmer serialupdi
avrdude error: avrdude built without libserialport support; please compile again with libserialport installed

avrdude done.  Thank you.
```

mcuee, 2024-01-24:
@dl8dtl 

Just wondering if you can help to see if you are able to reproduce the issue or not. Thanks.

mcuee, 2024-01-24:
Old reports of similar issue is for Solaris.
* https://github.com/avrdudes/avrdude/issues/123

mcuee, 2024-01-24:
No issues under NetBSD 9.3.
```
-bash-5.2$ uname -a
NetBSD localhost 9.3 NetBSD 9.3 (GENERIC) #0: Thu Aug  4 15:30:37 UTC 2022  mkrepro@mkrepro.NetBSD.org:/usr/src/sys/arch/amd64/compile/GENERIC amd64

-bash-5.2$ avrdude -c serialupdi -P /dev/ttyU0 -p avr64dd32
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e961a (probably avr64dd32)

avrdude done.  Thank you.

```

mcuee, 2024-01-24:
No issues under OpenBSD 7.4 either (Intel N4500 mini PC, physical machine).

```
openbsd74$ uname -a
OpenBSD openbsd74.my.domain 7.4 GENERIC.MP#1397 amd64

openbsd74$ ./test-avrdude -p "-c serialupdi -P /dev/ttyU0 -p avr64dd32"
Testing avrdude version 7.2-20240122 (2f89aece)
Prepare "-c serialupdi -P /dev/ttyU0 -p avr64dd32" and press 'enter' or 'space' to continue. Press any other key to skip
✅   0.000 s: set, clear and read eesave fuse bit
✅  21.795 s: the_quick_brown_fox_65536B.hex flash -U write/verify
✅  12.634 s: the_quick_brown_fox_256B.hex eeprom -U write/verify
✅  21.785 s: lorem_ipsum_65536B.srec flash -U write/verify
✅  12.634 s: lorem_ipsum_256B.srec eeprom -U write/verify
✅  12.654 s: 0xff_256B.hex eeprom -U write/verify
✅  12.654 s: the_quick_brown_fox_65536B.hex:a flash -T write/verify
✅  12.654 s: the_quick_brown_fox_256B.hex:a eeprom -T write/verify
✅  12.654 s: lorem_ipsum_65536B.srec flash -T write/verify
✅  12.654 s: lorem_ipsum_256B.srec eeprom -T write/verify
✅  12.654 s: cola-vending-machine.raw flash -T/-U write/verify
✅   7.645 s: holes_pack_my_box_65536B.hex:a flash -U write
✅   7.645 s: holes_the_five_boxing_wizards_65536B.hex flash -T write
✅   5.594 s: holes_pack_my_box_256B.hex:a eeprom -U write
✅   5.594 s: holes_the_five_boxing_wizards_256B.hex eeprom -T write
✅   3.294 s: random_data_32B.bin usersig -T/-U write/read
```

askn37, 2024-01-24:
Does the problem occur with USB serial adapters other than the CH340 series?

If you can use another serial adapter, the problem is most likely with the CH340 series driver. In some cases, reducing the speed may improve the situation.

If other serial adapters also do not work, there may be an issue with the USB passthrough feature between the physical PC and the guest operating system.

mcuee, 2024-01-24:
@askn37 

The latest tests are done with physical machine and not virtual machine.

I will see if I can test with another USB to Serial adapter.

dbuchwald, 2024-01-24:
If I can chip in: the only difference between **serialupdi** and **urclock** is the `pflag` field content. 
- serialupdi uses SERIAL_8E2
- urclock uses SERIAL_8N1

These flags are defined in here: https://github.com/avrdudes/avrdude/blob/main/src/libavrdude.h#L797-L799

Then, in turn, these are translated to OS-specific flags here: https://github.com/avrdudes/avrdude/blob/main/src/libavrdude.h#L797-L799

Difference between the two modes is SERIAL_NO_CSTOPB/SERIAL_CSTOPB and SERIAL_NO_PARITY/SERIAL_PARENB, which translates to:

```c
  if (cflags & SERIAL_CSTOPB) {
    termios.c_cflag |= CSTOPB;
  } else {
    termios.c_cflag &= ~CSTOPB;
  }
```

and

```c
  if (cflags & (SERIAL_PARENB | SERIAL_PARODD)) {
    termios.c_cflag |= PARENB;
  } else {
    termios.c_cflag &= ~PARENB;
  }

  if (cflags & SERIAL_PARODD) {
    termios.c_cflag |= PARODD;
  } else {
    termios.c_cflag &= ~PARODD;
  }
```

Maybe one of these is not recognized correctly by FreeBSD implementation of serial interface?

mcuee, 2024-01-24:
I modified the libserialport example and it seems to me `7E2` does not work and `8N1` works.

Same with the build-in example `port_config`.

I carry out the tests with another virtual machine running FreeBSD 13.2 Release this time. It was the same when I tested with the physical machine running FreeBSD 14.0 Release.

```
mcuee@freebsd132vmn100:~/build/libserialport/examples $ cc -I /usr/local/include test1.c  -L /usr/local/lib -l serialport -lusb -o test1
mcuee@freebsd132vmn100:~/build/libserialport/examples $ ./test1 /dev/cuaU0
Looking for port /dev/cuaU0.
Port name: /dev/cuaU0
Description: USB Serial
Opening port.
Setting port to 115200 8N1, no flow control.
Baudrate: 115200, data bits: 8, parity: None, stop bits: 1
mcuee@freebsd132vmn100:~/build/libserialport/examples $ ./test /dev/cuaU0
Looking for port /dev/cuaU0.
Port name: /dev/cuaU0
Description: USB Serial
Opening port.
Setting port to 115200 7E2, no flow control.
Error: Failed: Input/output error
Abort trap (core dumped)
mcuee@freebsd132vmn100:~/build/libserialport/examples $ ./port_config /dev/cuaU0
Looking for port /dev/cuaU0.
Port name: /dev/cuaU0
Description: USB Serial
Opening port.
Setting port to 115200 8N1, no flow control.
Baudrate: 115200, data bits: 8, parity: None, stop bits: 1
Creating new config for 9600 7E2, XON/XOFF flow control.
Applying new configuration.
Error: Failed: Input/output error
Abort trap (core dumped)
```

mcuee, 2024-01-24:
No issues with an FTDI USB to Serial Converter (this is not a USB to TTL converter so I can not use it).

```
mcuee@freebsd132vmn100:~/build/libserialport/examples $ ./port_config /dev/cuaU0
Looking for port /dev/cuaU0.
Port name: /dev/cuaU0
Description: (null)
Opening port.
Setting port to 115200 8N1, no flow control.
Baudrate: 115200, data bits: 8, parity: None, stop bits: 1
Creating new config for 9600 7E2, XON/XOFF flow control.
Applying new configuration.
Setting port back to previous config.

mcuee@freebsd132vmn100:~/build/libserialport/examples $ ./test1 /dev/cuaU0
Looking for port /dev/cuaU0.
Port name: /dev/cuaU0
Description: (null)
Opening port.
Setting port to 115200 8N1, no flow control.
Baudrate: 115200, data bits: 8, parity: None, stop bits: 1

mcuee@freebsd132vmn100:~/build/libserialport/examples $ ./test /dev/cuaU0
Looking for port /dev/cuaU0.
Port name: /dev/cuaU0
Description: (null)
Opening port.
Setting port to 115200 7E2, no flow control.
Baudrate: 115200, data bits: 7, parity: Even, stop bits: 2
```

mcuee, 2024-01-24:
Okay, it seems to me FreeBSD really does not like CH340.
https://cgit.freebsd.org/src/tree/sys/dev/usb/serial/uchcom.c
```
#include <sys/cdefs.h>
/*
 * Driver for WinChipHead CH341/340, the worst USB-serial chip in the
 * world.
 */
```

@dl8dtl 

Please help to check if you can set `7E2` if you happen to have a CH340 based USB to Serial converter. Thanks.

mcuee, 2024-01-24:
It seems to me only serialupdi uses `7E2`. 
`jtag2updi` does not seem to have issues. The following test is with Nano 4808 (CH340 + a USB 8051 MCU, custom FW for jtag2updi).

```
mcuee@freebsd132vmn100:~/build/avrdude-test-script $ ./test-avrdude -p "-c jtag2updi -P /dev/cuaU0 -p m4808"
Testing avrdude version 7.2-20240122 (2f89aece)
Prepare "-c jtag2updi -P /dev/cuaU0 -p m4808" and press 'enter' or 'space' to continue. Press any other key to skip
✅   0.000 s: set, clear and read eesave fuse bit
✅  23.432 s: the_quick_brown_fox_49152B.hex flash -U write/verify
✅   1.352 s: the_quick_brown_fox_256B.hex eeprom -U write/verify
✅  23.438 s: lorem_ipsum_49152B.srec flash -U write/verify
✅   1.352 s: lorem_ipsum_256B.srec eeprom -U write/verify
✅   1.502 s: 0xff_256B.hex eeprom -U write/verify
✅   1.502 s: the_quick_brown_fox_49152B.hex:a flash -T write/verify
✅   1.502 s: the_quick_brown_fox_256B.hex:a eeprom -T write/verify
✅   1.502 s: lorem_ipsum_49152B.srec flash -T write/verify
✅   1.502 s: lorem_ipsum_256B.srec eeprom -T write/verify
✅   1.502 s: cola-vending-machine.raw flash -T/-U write/verify
✅   6.885 s: holes_pack_my_box_49152B.hex:a flash -U write
✅   6.885 s: holes_the_five_boxing_wizards_49152B.hex flash -T write
✅   1.312 s: holes_pack_my_box_256B.hex:a eeprom -U write
✅   1.312 s: holes_the_five_boxing_wizards_256B.hex eeprom -T write
✅   1.242 s: random_data_64B.bin usersig -T/-U write/read
```

mcuee, 2024-01-24:
Anyway, I do not think this is an avrdude serialupdi issue. I will close this and transfer to a discussion.