avrdudes/avrdude #1643: Integrate subset of avrdude-test-script to avrdue github action with `dryrun` and `dryboot`

mcuee, 2024-01-28:
Reference:
* https://github.com/MCUdude/avrdude-test-script/pull/21#issuecomment-1913000618

@stefanrueger mentioned the following.

> this PR adds real tests that are only triggered when the -cdryrun -pm2560 test is run; these are high-level tests of some of fileio logic in AVRDUDE that are independent of any programmer and provide further sanity checking. Hence, I consider these useful to have in the overall test script.

I am thinking we can use the following subset. We may only need to test the following three build.
`linux-x86_64`, `macos-x86_64`, `msvc64` and `mingw64`.

Or we may just need to test `linux-x86_64` since the codes involved by `-c dryrun` and `-c dryboot` are mostly cross-platform.

```
    "-cdryrun -patmega328pb"
    "-cdryrun -patmega2560"
    "-cdryrun -patxmega128a1u"
    "-cdryrun -patmega4808"
    "-cdryrun -pattiny3217"
    "-cdryrun -pavr128da48"
    "-cdryrun -pavr64ea48"
    "-cdryrun -pavr16eb32"
    "-cdryboot -patmega328pb"
    "-cdryboot -patmega2560"
    "-cdryboot -patxmega128a1u"
    "-cdryboot -patmega4808"
    "-cdryboot -pattiny3217"
    "-cdryboot -pavr128da48"
    "-cdryboot -pavr64ea48"
    "-cdryboot -pavr16eb32"
```

mcuee, 2024-01-28:
The scripts may need to be adapted so that the tests will continue without manual intervention.

stefanrueger, 2024-01-28:
The following line would be one without manual interaction:
```
$ echo -e \\n | tools/test-avrdude -d0 -p"-cdryrun -pm2560" -p"-cdryrun -pavr64du28"
Testing avrdude version 7.2-20240122 (2f89aece)
Prepare "-cdryrun -pm2560" and press 'enter' or 'space' to continue. Press any other key to skip
✅   0.079 s: flash raw format -T/-U write/verify cola-vending-machine.raw
✅   0.069 s: flash extended address and hole test
✅   0.097 s: flash writing binary numbers
✅   0.115 s: flash reading and verifying binary numbers
✅   0.085 s: flash writing octal numbers
✅   0.133 s: flash reading and verifying octal numbers
✅   0.080 s: flash writing decimal numbers
✅   0.131 s: flash reading and verifying decimal numbers
✅   0.086 s: flash writing hexadecimal numbers
✅   0.105 s: flash reading and verifying hexadecimal numbers
✅   0.087 s: flash writing R numbers
✅   0.125 s: flash reading and verifying R numbers
✅   0.088 s: flash writing srec format
✅   0.075 s: flash reading and verifying srec format file
✅   0.066 s: fuse access: clear, set and read eesave fuse bit
✅   0.083 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
✅   0.066 s: chip erase
✅   0.070 s: flash -U write/verify holes_rjmp_loops_262144B.hex
✅   0.072 s: flash -T write/verify holes_rjmp_loops_262144B.hex
✅   0.068 s: eeprom check whether programmer can flip 0s to 1s
✅   0.066 s: eeprom -U write/verify holes_pack_my_box_4096B.hex
✅   0.068 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_4096B.hex
✅   0.070 s: chip erase and spot check flash is actually erased
✅   0.066 s: spot check eeprom is erased, too
Prepare "-cdryrun -pavr64du28" and press 'enter' or 'space' to continue. Press any other key to skip
✅   0.067 s: fuse access: clear, set and read eesave fuse bit
✅   0.083 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
✅   0.069 s: chip erase
✅   0.068 s: flash -U write/verify holes_rjmp_loops_65536B.hex
✅   0.068 s: flash -T write/verify holes_rjmp_loops_65536B.hex
✅   0.067 s: eeprom check whether programmer can flip 0s to 1s
✅   0.066 s: eeprom -U write/verify holes_pack_my_box_256B.hex
✅   0.067 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_256B.hex
✅   0.068 s: chip erase and spot check flash is actually erased
✅   0.067 s: spot check eeprom is erased, too
✅   0.066 s: usersig -T/-U write/read random_data_512B.bin
```

The cases `"-cdryrun -pm2560"` and `"-cdryrun -pavr64du28"` are sufficient to sanity-test AVRDUDE's high-level working
 - Can avrdude be called
 - Can it open the avrdude.conf
 - Can it parse the file
 - Can it understand the command line syntax
 - Can it read/write files from the file system and convert hex/srec/d/h/o/... formats
 - Can it call a programmer (here `dryrun`) and interact with it

All that is required (I think!) is to install the `test-avrdude` script and the test-files in the `test_files` directory in AVRDUDE's tool directory (I can do that) and to put above line into the right workflow file (I don't know how)

mcuee, 2024-01-28:
> All that is required (I think!) is to install the `test-avrdude` script and the `test_files` directory in AVRDUDE's tool directory (I can do that) 

Thanks, Please do it.

> and to put above line into the right workflow file (I don't know how)

Once you finish the above, I think this is not difficult. I can try it.

ndim, 2024-01-28:
> I am thinking we can use the following subset. We may only need to test the following three build.
> `linux-x86_64`, `macos-x86_64`, `msvc64` and `mingw64`.

Three? This looks like four to me, or what am I missing here?

mcuee, 2024-01-29:
> > I am thinking we can use the following subset. We may only need to test the following three build.
> > `linux-x86_64`, `macos-x86_64`, `msvc64` and `mingw64`.
> 
> Three? This looks like four to me, or what am I missing here?

Sorry for the typo. It is 4.

But then I think in the end we may only need one for Linux.

mcuee, 2024-01-29:
> Once you finish the above, I think this is not difficult. I can try it.

Done.
* #1646

closed 2024-01-29 by stefanrueger