avrdudes/avrdude #1644: libserialport issue under macOS 14.3 with CH340

mcuee, 2024-01-28:
Just upgraded my Mac Mini M1 2020 to latest macOS 14.3 and it seems to me avrdude can not recognize `-P ch340` now.

```
mcuee@mcuees-Mac-mini avrdude_bin % avrdude -c urclock -P ch340 -p m328pb -xshowall 
avrdude warning: -P ch340 is not connected; consider
  -P /dev/cu.Bluetooth-Incoming-Port
  -P /dev/cu.usbserial-22420
  -P /dev/cu.wchusbserial22420
  -P /dev/cu.wlan-debug
avrdude OS error: cannot open port ch340: No such file or directory
avrdude error: unable to open port ch340 for programmer urclock

avrdude done.  Thank you.

mcuee@mcuees-Mac-mini avrdude_bin % avrdude -c urclock -P /dev/cu.usbserial-22420 -p m328pb -xshowall 
avrdude: AVR device initialized and ready to accept instructions
ffffffffffff 0000-00-00 00.00  application 0 store 0 meta 1 boot 384 u7.7 weu-jPrac vector 25 (SPM_Ready) ATmega328PB

mcuee@mcuees-Mac-mini avrdude_bin % avrdude -c urclock -P /dev/cu.wchusbserial22420 -p m328pb -xshowall 
avrdude: AVR device initialized and ready to accept instructions
ffffffffffff 0000-00-00 00.00  application 0 store 0 meta 1 boot 384 u7.7 weu-jPrac vector 25 (SPM_Ready) ATmega328PB

mcuee@mcuees-Mac-mini avrdude_bin % otool -L /opt/homebrew/bin/avrdude 
/opt/homebrew/bin/avrdude:
	@rpath/libavrdude.1.dylib (compatibility version 1.0.0, current version 1.0.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1336.61.1)
	/opt/homebrew/opt/libusb-compat/lib/libusb-0.1.4.dylib (compatibility version 9.0.0, current version 9.4.0)
	/opt/homebrew/opt/libusb/lib/libusb-1.0.0.dylib (compatibility version 4.0.0, current version 4.0.0)
	/opt/homebrew/opt/hidapi/lib/libhidapi.0.dylib (compatibility version 0.0.0, current version 0.14.0)
	/opt/homebrew/opt/libftdi/lib/libftdi1.2.dylib (compatibility version 2.0.0, current version 2.5.0)
	/usr/lib/libedit.3.dylib (compatibility version 2.0.0, current version 3.0.0)
	/opt/homebrew/opt/libserialport/lib/libserialport.0.dylib (compatibility version 2.0.0, current version 2.0.0)
	/usr/lib/libncurses.5.4.dylib (compatibility version 5.4.0, current version 5.4.0)
```

mcuee, 2024-01-28:
@MCUdude and @askn37 

Please help to check if you encounter similar issue with CH340 under your Mac Thanks.

mcuee, 2024-01-28:
libserialport example output.
https://github.com/sigrokproject/libserialport/tree/master/examples
```
cuee@mcuees-Mac-mini examples % ./list_ports 
Getting port list.
Found port: /dev/cu.wlan-debug
Found port: /dev/cu.Bluetooth-Incoming-Port
Found port: /dev/cu.usbserial-22420
Found port: /dev/cu.wchusbserial22420
Found 4 ports.
Freeing port list.
```

askn37, 2024-01-28:
ch340 cannot be recognized by name.
This is a macos genuine driver.

<details>

```
% uname -a
Darwin alicia.local 23.2.0 Darwin Kernel Version 23.2.0: Wed Nov 15 21:53:34 PST 2023; root:xnu-10002.61.3~2/RELEASE_ARM64_T8103 arm64
```

```
% otool -L build_darwin/src/avrdude                                       
build_darwin/src/avrdude:
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1336.61.1)
	/opt/local/lib/libusb-0.1.4.dylib (compatibility version 9.0.0, current version 9.4.0)
	/opt/local/lib/libusb-1.0.0.dylib (compatibility version 4.0.0, current version 4.0.0)
	/opt/local/lib/libhidapi.0.dylib (compatibility version 1.0.0, current version 1.0.0)
	/opt/local/lib/libftdi.1.dylib (compatibility version 22.0.0, current version 22.0.0)
	/opt/local/lib/libftdi1.2.dylib (compatibility version 2.0.0, current version 2.5.0)
	/opt/local/lib/libreadline.8.dylib (compatibility version 8.2.0, current version 8.2.0)
	/usr/local/lib/libserialport.0.dylib (compatibility version 2.0.0, current version 2.0.0)
	/opt/local/lib/libncurses.6.dylib (compatibility version 6.0.0, current version 6.0.0)
```

```
% ./build_darwin/src/avrdude -curclock -pavr128db32 -P ch340
avrdude warning: -P ch340 is not connected; consider
  -P /dev/cu.Bluetooth-Incoming-Port
  -P /dev/cu.usbserial-110
  -P /dev/cu.usbserial-230
  -P /dev/cu.usbserial-1320
  -P /dev/cu.wlan-debug
avrdude OS error: cannot open port ch340: No such file or directory
avrdude error: unable to open port ch340 for programmer urclock

avrdude done.  Thank you.

```

```
% ./build_darwin/src/avrdude -curclock -pavr128db32 -P usb     
avrdude OS error: cannot open port usb: No such file or directory
avrdude error: unable to open port usb for programmer urclock
Possible candidate serial ports are:
  -P /dev/cu.Bluetooth-Incoming-Port
  -P /dev/cu.usbserial-110
  -P /dev/cu.usbserial-230
  -P /dev/cu.usbserial-1320
  -P /dev/cu.wlan-debug
Note that above ports might not be connected to a target board or an AVR programmer.
Also note there may be other direct serial ports not listed above.

avrdude done.  Thank you.
```

```
% system_profiler SPUSBDataType | grep -A6 "USB Serial"     
                USB Serial:

                  Product ID: 0x7523
                  Vendor ID: 0x1a86
                  Version: 2.64
                  Speed: Up to 12 Mb/s
                  Location ID: 0x00132000 / 8
--
            USB Serial:

              Product ID: 0x7523
              Vendor ID: 0x1a86
              Version: c2.33
              Speed: Up to 12 Mb/s
              Location ID: 0x00110000 / 9
--
        USB Serial:

          Product ID: 0x7523
          Vendor ID: 0x1a86
          Version: 2.64
          Speed: Up to 12 Mb/s
          Location ID: 0x02300000 / 1
```

</details>

ft231x is similarly not recognized by name.
This is a macos genuine driver.

<details>

```
% ./build_darwin/src/avrdude -curclock -pavr128db32 -P ft231x
avrdude warning: -P ft231x is not connected; consider
  -P /dev/cu.Bluetooth-Incoming-Port
  -P /dev/cu.usbserial-D306JCWB
  -P /dev/cu.wlan-debug
avrdude OS error: cannot open port ft231x: No such file or directory
avrdude error: unable to open port ft231x for programmer urclock

avrdude done.  Thank you.

```

```
% ./build_darwin/src/avrdude -curclock -pavr128db32 -P usb   
avrdude OS error: cannot open port usb: No such file or directory
avrdude error: unable to open port usb for programmer urclock
Possible candidate serial ports are:
  -P /dev/cu.Bluetooth-Incoming-Port
  -P /dev/cu.usbserial-D306JCWB
  -P /dev/cu.wlan-debug
Note that above ports might not be connected to a target board or an AVR programmer.
Also note there may be other direct serial ports not listed above.

avrdude done.  Thank you.

```

```
% system_profiler SPUSBDataType | grep -A6 "USB UART"  
        FT231X USB UART:

          Product ID: 0x6015
          Vendor ID: 0x0403  (Future Technology Devices International Limited)
          Version: 10.00
          Serial Number: D306JCWB
          Speed: Up to 12 Mb/s
```
</details>

mcuee, 2024-01-28:
@MCUdude 

Please take a look if we can improve on the current behavior. If it is not easy, then I will still close this issue but update the Wiki.

MCUdude, 2024-01-28:
Does it work if you use a different serial chip?Also, it looks like you're using the WCH driver, and not the default one that Macos provides, hence the name cu.wchusbserial

mcuee, 2024-01-28:
> Does it work if you use a different serial chip?Also, it looks like you're using the WCH driver, and not the default one that Macos provides, hence the name cu.wchusbserial

No, it is the default driver from Apple now.

MCUdude, 2024-01-28:
Ok. Ans what was the previous macos version that it worked with?

mcuee, 2024-01-28:
> Ok. Ans what was the previous macos version that it worked with?

@MCUdude 
I do not know.

But at least it works on your Mac. What is the macOS version you are using?
https://github.com/avrdudes/avrdude/pull/1498

MCUdude, 2024-01-28:
@mcuee I'm not at home right now, so can't look into it. But try other libserialport examples as well, and see if you can get it to print the USB vid/pid and SN. This is what Avrdude use to identify a port

mcuee, 2024-01-28:
> @mcuee I'm not at home right now, so can't look into it. But try other libserialport examples as well, and see if you can get it to print the USB vid/pid and SN. This is what Avrdude use to identify a port

I will carry out tests tomorrow.

mcuee, 2024-01-28:
In any case, this is not a big issue and we may not want to do anything.
`-P ch340` is just an abbreation which is nice to have but not really essential.

MCUdude, 2024-01-28:
>> Ok. And what was the previous macos version that it worked with?

>I do not know. I only have one Mac Mini M1 and I always upgrade to the latest version.

So you don't have the slightest idea what ran before you upgraded? Was it a major update, or just a minor/patch?
I haven't updated the macOS version on my mac in ages, but I'm planning a clean install sometime this spring.
I'm still on 10.14, which is ancient.

> In any case, this is not a big issue and we may not want to do anything.

Well, that means that `-Pusb:vid:pid` wouldn't work either, which mean that all features that libserialport provides are now broken. I'd consider this a big deal, as this may break `-r` as well.

@mcuee can you try to run the `port_info` libserialport example? That should print relevant information about the port attached.

This should be the output when reading a CH340 chip:

```
$ ./port_info /dev/cu.usbserial-1410 
Looking for port /dev/cu.usbserial-1410.
Port name: /dev/cu.usbserial-1410
Description: USB Serial
Type: USB
Manufacturer: (null)
Product: USB Serial
Serial: (null)
VID: 1A86 PID: 7523
Bus: 0 Address: 0
Freeing port.
```

askn37, 2024-01-29:
The macos conceptual code described here works fine in my environment. So it doesn't seem to be a driver issue.
https://github.com/avrdudes/avrdude/issues/907#issuecomment-1077408655

<details>

```
askn@alicia usb_vid_pid_test % bash -x build.sh
+ gcc -framework CoreServices -framework IOKit -o usb_vid_pid_test usb_vid_pid_test.c
usb_vid_pid_test.c:46:45: warning: 'kIOMasterPortDefault' is deprecated: first deprecated in macOS 12.0 [-Wdeprecated-declarations]
  kernResult = IOServiceGetMatchingServices(kIOMasterPortDefault, classesToMatch, matchingServices);
                                            ^~~~~~~~~~~~~~~~~~~~
                                            kIOMainPortDefault
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/IOKit.framework/Headers/IOKitLib.h:133:19: note: 'kIOMasterPortDefault' has been explicitly marked deprecated here
const mach_port_t kIOMasterPortDefault
                  ^
1 warning generated.
askn@alicia usb_vid_pid_test % ./usb_vid_pid_test
Found modem. Port: /dev/cu.usbserial-230 USB VID: 0x1A86 PID: 0x7523
Found modem. Port: /dev/cu.usbserial-D306JCWB USB VID: 0x0403 PID: 0x6015
askn@alicia usb_vid_pid_test % 
```

</details>

And it's not just the naming that doesn't work. The `-P usb:vid:pid` syntax also doesn't seem to work.

```
askn@alicia avrdude_main % ./build_darwin/src/avrdude -curclock -pavr128db32 -v -Pusb:0x1A86:0x7523

avrdude: Version 7.2-20240122 (2f89aece)
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is /Users/askn/Collaborator/avrdude_main/build_darwin/src/avrdude.conf
         User configuration file is /Users/askn/.avrduderc

avrdude setport_from_vid_pid() warning: -P usb:0x1A86:0x7523 is not connected; consider
  -P /dev/cu.Bluetooth-Incoming-Port
  -P /dev/cu.usbserial-230
  -P /dev/cu.usbserial-D306JCWB
  -P /dev/cu.wlan-debug
         Using port            : usb:0x1A86:0x7523
         Using programmer      : urclock
avrdude ser_open() OS error: cannot open port usb:0x1A86:0x7523: No such file or directory
avrdude main() error: unable to open port usb:0x1A86:0x7523 for programmer urclock

avrdude done.  Thank you.

```

mcuee, 2024-01-29:
> > In any case, this is not a big issue and we may not want to do anything.
> 
> Well, that means that `-Pusb:vid:pid` wouldn't work either, which mean that all features that libserialport provides are now broken. I'd consider this a big deal, as this may break `-r` as well.

You are right that `-Pusb:vid:pid` does not work for CH340 now. 

> @mcuee can you try to run the `port_info` libserialport example? That should print relevant information about the port attached.
>
> This should be the output when reading a CH340 chip:
>
> ```
> $ [...]

Now it is listed as a native port...

```
mcuee@mcuees-Mac-mini examples % ./list_ports 
Getting port list.
Found port: /dev/cu.wlan-debug
Found port: /dev/cu.Bluetooth-Incoming-Port
Found port: /dev/cu.usbserial-22420
Found port: /dev/cu.wchusbserial22420
Found 4 ports.
Freeing port list.
mcuee@mcuees-Mac-mini examples % ./port_info /dev/cu.usbserial-22420 
Looking for port /dev/cu.usbserial-22420.
Port name: /dev/cu.usbserial-22420
Description: USB2.0-Ser!
Type: Native
Freeing port.
mcuee@mcuees-Mac-mini examples % ./port_info /dev/cu.wchusbserial22420 
Looking for port /dev/cu.wchusbserial22420.
Port name: /dev/cu.wchusbserial22420
Description: USB2.0-Ser!
Type: Native
Freeing port.
```

mcuee, 2024-01-29:
I am not so sure if we can fix this using libserialport.

pyserial seems to work fine.
```
(py310venv) mcuee@mcuees-Mac-mini bin % ./pyserial-ports -v 
/dev/cu.Bluetooth-Incoming-Port
    desc: n/a
    hwid: n/a
/dev/cu.usbserial-22420
    desc: USB2.0-Ser!
    hwid: USB VID:PID=1A86:7523 LOCATION=2-2.4.2
/dev/cu.wchusbserial22420
    desc: USB2.0-Ser!
    hwid: USB VID:PID=1A86:7523 LOCATION=2-2.4.2
/dev/cu.wlan-debug  
    desc: n/a
    hwid: n/a
4 ports found
```

mcuee, 2024-01-29:
> So you don't have the slightest idea what ran before you upgraded? Was it a major update, or just a minor/patch?
I haven't updated the macOS version on my mac in ages, but I'm planning a clean install sometime this spring.
I'm still on 10.14, which is ancient.

@MCUdude 

Sorry I have not tested urclock under macOS for a while. I was doing a minor upgrade from macOS 14.2 to 14.3. But macOS 14 itself is a major upgrade.

My Mac Mini M1 (2020 version) came with macOS 11 Big Sur and now it is running macOS 14 Sonoma.

mcuee, 2024-01-29:
> The macos conceptual code described here works fine in my environment. So it doesn't seem to be a driver issue. [#907 (comment)](https://github.com/avrdudes/avrdude/issues/907#issuecomment-1077408655)

@askn37 

The driver (and macOS version) is not the issue per se. But I think libserialport relies on certain driver behavior to be able to work for avrdude. So it is rather a libserialport limitation.

@MCUdude 

As this is not a real avrdude issue, I will keep the label as `enhancement`. We may need some extra codes out of libserialport.

mcuee, 2024-01-29:
@askn37 
What is the macOS version you are using?

mcuee, 2024-01-29:
@dl8dtl 
Just wondering if you have the older version of macOS to check. Thanks.

mcuee, 2024-01-29:
> I'd consider this a big deal, as this may break -r as well.

@MCUdude 
Since Arduino Nano Every does not use CH340, it is not affected.

```
mcuee@mcuees-Mac-mini avr % avrdude -cjtag2updi -patmega4809 -Pusb:2341:0058 -r
avrdude: touching serial port /dev/cu.usbmodem224301 at 1200 baud
avrdude: waiting for new port... using same port /dev/cu.usbmodem224301
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9651 (probably m4809)

avrdude done.  Thank you.
```

Affected: CH340 or similar, FTDI USB to Serial may be affected (need to check).

askn37, 2024-01-29:
@mcuee 

iMac 24-inch, M1, 2021 macos Sonoma 14.2.1

```
% uname -a
Darwin alicia.local 23.2.0 Darwin Kernel Version 23.2.0: Wed Nov 15 21:53:34 PST 2023; root:xnu-10002.61.3~2/RELEASE_ARM64_T8103 arm64
```

I agree that this may be a limitation of libserialport (1.0.0). On CH340 and FT231X, `-T <name>` does not map to native port.

I think `-T <name>` is a shortcut for `-T usb:vid:pid`. Therefore, it may not generally work if a native port already exists.

mcuee, 2024-01-29:
@MCUdude 

I saw that you were involved in some discussions in the PRs for libserialport. I will try those PRs under macOS.

* https://github.com/sigrokproject/libserialport/pull/9
* https://github.com/sigrokproject/libserialport/pull/13
* https://github.com/sigrokproject/libserialport/pull/14

For Linux, the following PR may be useful.
https://github.com/sigrokproject/libserialport/pull/7

MCUdude, 2024-01-29:
Thanks for testing the port_info example. There is a compatibility issue with the latest macOS version and libserialport.

@mcuee can you try the suggested fix here [libserialport#14](https://github.com/sigrokproject/libserialport/pull/14) and try the port_info example again? Since there has been some changes to the driver, it may be that 16 character isn't enough to hold the descriptor string.

mcuee, 2024-01-29:
@MCUdude 

https://github.com/sigrokproject/libserialport/pull/14 works fine for CH340.
```
git clone https://github.com/sigrokproject/libserialport.git libserialport_test
cd libserialport_test
git fetch origin pull/14/head
git checkout -b pr14_rebase FETCH_HEAD
git rebase origin/master
cd libserialport_test
./autogen.sh
./configure --prefix=/Users/mcuee/bin
make
cd examples

mcuee@mcuees-Mac-mini examples % make -f Makefile.new 
gcc -g -Wall -I /Users/mcuee/bin/include await_events.c -L /Users/mcuee/bin/lib -lserialport -o await_events
gcc -g -Wall -I /Users/mcuee/bin/include handle_errors.c -L /Users/mcuee/bin/lib -lserialport -o handle_errors
gcc -g -Wall -I /Users/mcuee/bin/include list_ports.c -L /Users/mcuee/bin/lib -lserialport -o list_ports
gcc -g -Wall -I /Users/mcuee/bin/include port_config.c -L /Users/mcuee/bin/lib -lserialport -o port_config
gcc -g -Wall -I /Users/mcuee/bin/include port_info.c -L /Users/mcuee/bin/lib -lserialport -o port_info
gcc -g -Wall -I /Users/mcuee/bin/include send_receive.c -L /Users/mcuee/bin/lib -lserialport -o send_receive

mcuee@mcuees-Mac-mini examples % ./port_info /dev/cu.usbserial-22420 
Looking for port /dev/cu.usbserial-22420.
Port name: /dev/cu.usbserial-22420
Description: USB2.0-Ser!
Type: USB
Manufacturer: Generic
Product: USB2.0-Ser!
Serial: (null)
VID: 1A86 PID: 7523
Bus: 1 Address: 11263320
Freeing port.

mcuee@mcuees-Mac-mini examples % ./port_info /dev/cu.wchusbserial22420 
Looking for port /dev/cu.wchusbserial22420.
Port name: /dev/cu.wchusbserial22420
Description: USB2.0-Ser!
Type: USB
Manufacturer: Generic
Product: USB2.0-Ser!
Serial: (null)
VID: 1A86 PID: 7523
Bus: 1 Address: 72211800
Freeing port.
```

Original binary without your patch.
```
mcuee@mcuees-Mac-mini examples % cd bin_org 
mcuee@mcuees-Mac-mini bin_org % ./list_ports 
Getting port list.
Found port: /dev/cu.wlan-debug
Found port: /dev/cu.Bluetooth-Incoming-Port
Found port: /dev/cu.usbserial-22420
Found port: /dev/cu.wchusbserial22420
Found 4 ports.
Freeing port list.

mcuee@mcuees-Mac-mini bin_org % ./port_info /dev/cu.usbserial-22420  
Looking for port /dev/cu.usbserial-22420.
Port name: /dev/cu.usbserial-22420
Description: USB2.0-Ser!
Type: Native
Freeing port.

mcuee@mcuees-Mac-mini bin_org % ./port_info /dev/cu.wchusbserial22420
Looking for port /dev/cu.wchusbserial22420.
Port name: /dev/cu.wchusbserial22420
Description: USB2.0-Ser!
Type: Native
Freeing port.
```

mcuee, 2024-01-29:
Personally I like https://github.com/sigrokproject/libserialport/pull/9 better. It works fine with CH340.

```
git clone https://github.com/sigrokproject/libserialport.git libserialport_pr9
cd libserialport_pr9
git fetch origin pull/9/head
git checkout -b pr9_rebase FETCH_HEAD
git rebase origin/master
./autogen.sh
./configure --prefix=/Users/mcuee/bin1
make
cd examples

mcuee@mcuees-Mac-mini examples % nano Makefile.new 
mcuee@mcuees-Mac-mini examples % make -f Makefile.new 
gcc -g -Wall -I /Users/mcuee/bin1/include await_events.c -L /Users/mcuee/bin1/lib -lserialport -o await_events
gcc -g -Wall -I /Users/mcuee/bin1/include handle_errors.c -L /Users/mcuee/bin1/lib -lserialport -o handle_errors
gcc -g -Wall -I /Users/mcuee/bin1/include list_ports.c -L /Users/mcuee/bin1/lib -lserialport -o list_ports
gcc -g -Wall -I /Users/mcuee/bin1/include port_config.c -L /Users/mcuee/bin1/lib -lserialport -o port_config
gcc -g -Wall -I /Users/mcuee/bin1/include port_info.c -L /Users/mcuee/bin1/lib -lserialport -o port_info
gcc -g -Wall -I /Users/mcuee/bin1/include send_receive.c -L /Users/mcuee/bin1/lib -lserialport -o send_receive
mcuee@mcuees-Mac-mini examples % ./list_ports
Getting port list.
Found port: /dev/cu.wlan-debug
Found port: /dev/cu.Bluetooth-Incoming-Port
Found port: /dev/cu.usbserial-22420
Found port: /dev/cu.wchusbserial22420
Found 4 ports.
Freeing port list.
mcuee@mcuees-Mac-mini examples % ./port_info /dev/cu.usbserial-22420 
Looking for port /dev/cu.usbserial-22420.
Port name: /dev/cu.usbserial-22420
Description: USB2.0-Ser!
Type: USB
Manufacturer: Generic
Product: USB2.0-Ser!
Serial: (null)
VID: 1A86 PID: 7523
Bus: 1 Address: 8396120
Freeing port.
mcuee@mcuees-Mac-mini examples % ./port_info /dev/cu.wchusbserial22420 
Looking for port /dev/cu.wchusbserial22420.
Port name: /dev/cu.wchusbserial22420
Description: USB2.0-Ser!
Type: USB
Manufacturer: Generic
Product: USB2.0-Ser!
Serial: (null)
VID: 1A86 PID: 7523
Bus: 1 Address: 15785304
Freeing port.
```

MCUdude, 2024-01-29:
> https://github.com/sigrokproject/libserialport/pull/14 works fine for CH340.

Excellent! This means that the bus is a new descriptor sting that's longer than 16 bytes. A simple fix.

> Personally I like https://github.com/sigrokproject/libserialport/pull/9 better. It works fine with CH340.

Yes, https://github.com/sigrokproject/libserialport/pull/9 will resolve the issue too. But the intention for the PR was to fix the CP2102 issue that's been there for many, many years. The root cause for the issue is the buffer overflow. https://github.com/sigrokproject/libserialport/pull/9 doesn't check the descriptor string at all. I'm not qualified to tell whether this is safe or not. But https://github.com/sigrokproject/libserialport/pull/14 _just_ resolves the actual bug without changing any behavior.

mcuee, 2024-01-29:
@MCUdude 

I can also confirm that your PR works with FT232R and CP2102 where libserialport git does not work.
* https://github.com/sigrokproject/libserialport/pull/14

MCUdude, 2024-01-29:
@mcuee I think we can close this issue, now that we _know_ that it's an isolated libserialport issue, and there already exist two PRs that will fix the issue.

closed 2024-01-29 by mcuee