avrdudes/avrdude #1648: Test script usersig test does not work with `/dev/shm` under MSYS2 Windows

mcuee, 2024-01-30:
This is a strange problem. Normally we can read/write to `/dev/shm` but somehow the usersig test failed. If I use `/tmp` for the `bin` file, then it will work.

Reference:
* https://github.com/avrdudes/avrdude/issues/1632#issuecomment-1916621390
* https://github.com/avrdudes/avrdude/issues/1632#issuecomment-1916629840

```
MINGW64 /c/work/avr/avrdude_test/avrdude_main/tools
$ ls -la /dev/shm/
total 28
drwxr-xr-x 1 xiaof xiaof     0 Jan 30 19:14 .
drwxr-xr-x 1 xiaof xiaof     0 Oct  3  2021 ..
-rw-r--r-- 1 xiaof xiaof  3516 Oct 25 18:09 errorv.log
-rw-r--r-- 1 xiaof xiaof  4106 Oct 25 18:10 errorvv.log
-rw-r--r-- 1 xiaof xiaof 11673 Oct 25 18:05 errorvvvv.log

$ cp ./test_files/random_data_64B.bin /dev/shm/

$ ls -la /dev/shm/
total 29
drwxr-xr-x 1 xiaof xiaof     0 Jan 30 19:18 .
drwxr-xr-x 1 xiaof xiaof     0 Oct  3  2021 ..
-rw-r--r-- 1 xiaof xiaof  3516 Oct 25 18:09 errorv.log
-rw-r--r-- 1 xiaof xiaof  4106 Oct 25 18:10 errorvv.log
-rw-r--r-- 1 xiaof xiaof 11673 Oct 25 18:05 errorvvvv.log
-rw-r--r-- 1 xiaof xiaof    64 Jan 30 19:18 random_data_64B.bin

$ ./test-avrdude -e "./avrdude.exe" -d 0 -p "-c dryrun -p m4808" -v
Testing ./avrdude.exe version 7.2-20240130 (3d3a387d)
Prepare "-c dryrun -p m4808" and press 'enter' or 'space' to continue. Press any other key to skip
✅   0.099 s: fuse access: clear, set and read eesave fuse bit
✅   0.101 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
✅   0.102 s: chip erase
✅   0.105 s: flash -U write/verify holes_rjmp_loops_49152B.hex
✅   0.102 s: flash -T write/verify holes_rjmp_loops_49152B.hex
✅   0.100 s: eeprom check whether programmer can flip 0s to 1s
✅   0.101 s: eeprom -U write/verify holes_pack_my_box_256B.hex
✅   0.104 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_256B.hex
✅   0.113 s: chip erase and spot check flash is actually erased
✅   0.099 s: spot check eeprom is erased, too
❌   0.114 s: usersig -T/-U write/read random_data_64B.bin (failed command below)
$ ./avrdude.exe -qq -c dryrun -p m4808 -l /dev/shm/test-avrdude.log.KqMEIY -T "erase usersig; write usersig ./test_files/random_data_64B.bin" -T flush -U usersig:r:/dev/shm/usersig_dump1629_64B.bin:r -U usersig:v:/dev/shm/usersig_dump1629_64B.bin:r -T "erase usersig" -T flush -U usersig:v:./test_files/0xff_64B.hex:i
avrdude OS error: file /dev/shm/usersig_dump1629_64B.bin is not writeable: No such file or directory
avrdude OS error: file /dev/shm/usersig_dump1629_64B.bin is not readable: No such file or directory
avrdude OS error: cannot open output file /dev/shm/usersig_dump1629_64B.bin: No such file or directory
avrdude error: write to file /dev/shm/usersig_dump1629_64B.bin failed

One or more AVRDUDE "-c dryrun -p m4808" tests failed. Do you want to retry this particular test? (y/n): n

$ ./avrdude.exe -c dryrun -p m4808 -l /dev/shm/test-avrdude.log.KqMEIY -T "erase usersig; write usersig ./test_files/random_data_64B.bin" -T flush -U usersig:r:/dev/shm/usersig_dump1629_64B.bin:r -U usersig:v:/dev/shm/usersig_dump1629_64B.bin:r -T "erase usersig" -T flush -U usersig:v:./test_files/0xff_64B.hex:i

$ cat /dev/shm/test-avrdude.log.KqMEIY
avrdude OS error: file /dev/shm/usersig_dump1629_64B.bin is not writeable: No such file or directory
avrdude OS error: file /dev/shm/usersig_dump1629_64B.bin is not readable: No such file or directory
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9650 (probably m4808)

avrdude: processing -T erase usersig; write usersig ./test_files/random_data_64B.bin
Caching | ################################################## | 100% 0.00s
Caching | ################################################## | 100% 0.00s

avrdude: processing -T flush
avrdude: synching cache to device ... done

avrdude: processing -U usersig:r:/dev/shm/usersig_dump1629_64B.bin:r
avrdude: reading userrow/usersig memory ...
Reading | ################################################## | 100% 0.00s
avrdude: writing output file /dev/shm/usersig_dump1629_64B.bin
avrdude OS error: cannot open output file /dev/shm/usersig_dump1629_64B.bin: No such file or directory
avrdude error: write to file /dev/shm/usersig_dump1629_64B.bin failed

avrdude done.  Thank you.

$ ls -la /dev/shm/
total 33
drwxr-xr-x 1 xiaof xiaof     0 Jan 30 19:20 .
drwxr-xr-x 1 xiaof xiaof     0 Oct  3  2021 ..
-rw-r--r-- 1 xiaof xiaof  3516 Oct 25 18:09 errorv.log
-rw-r--r-- 1 xiaof xiaof  4106 Oct 25 18:10 errorvv.log
-rw-r--r-- 1 xiaof xiaof 11673 Oct 25 18:05 errorvvvv.log
-rw-r--r-- 1 xiaof xiaof    64 Jan 30 19:18 random_data_64B.bin
-rw-r--r-- 1 xiaof xiaof  1100 Jan 30 19:20 test-avrdude.log.KqMEIY

```

stefanrueger, 2024-01-30:
It's not so clear to me where the bug lies. OS? Script? /dev/shm works under Linux as I expect it, but perhaps it's not well emulated under Windows? Maybe the best move is to add an option for the directory with temporary files.

stefanrueger, 2024-01-30:
Maybe `/tmp/shm` is *full*? What does `df -h /dev/shm` say?

mcuee, 2024-01-31:
> It's not so clear to me where the bug lies. OS? Script? /dev/shm works under Linux as I expect it, but perhaps it's not well emulated under Windows? 

I really have no idea why it does not work. This is MSYS2 which is forked off a version of Cygwin and probably it is not good.

> Maybe the best move is to add an option for the directory with temporary files.

I agree.

BTW, `/dev/shm` does not even exist under macOS.

```
mcuee@mcuees-Mac-mini tools % sw_vers 
ProductName:		macOS
ProductVersion:		14.3
BuildVersion:		23D56
mcuee@mcuees-Mac-mini tools % ls -la /dev/shm
ls: /dev/shm: No such file or directory
```

Reference:
https://www.deepanseeralan.com/tech/playing-with-shared-memory/

> macOS
> Support for POSIX shared memory is somewhat limited. Though the functionality is implemented, there is no file system interface like /dev/shm to manage the shared memory segments.

mcuee, 2024-01-31:
> Maybe `/tmp/shm` is _full_? What does `df -h /dev/shm` say?

No.

```
$ df -h /dev/shm
Filesystem      Size  Used Avail Use% Mounted on
C:/msys64       411G  370G   42G  90% /
```

stefanrueger, 2024-01-31:
> permission

What about `ls -ld /dev/shm`?

mcuee, 2024-01-31:
> > permission
> 
> What about `ls -ld /dev/shm`?

This is with work laptop running Windows 10 and MSYS2.

```
$ ls -ld /dev/shm/
drwxr-xr-x 1 XFCHEN Domain Users 0 Jan 30 17:18 /dev/shm/

$ ls -la /dev/shm/
total 15
drwxr-xr-x 1 XFCHEN Domain Users   0 Jan 30 17:18 .
drwxr-xr-x 1 XFCHEN Domain Users   0 Mar 22  2021 ..
-rw-r--r-- 1 XFCHEN Domain Users  12 Apr  8  2021 fish_shmem_1189162
-rw-r--r-- 1 XFCHEN Domain Users 382 Jan 30 15:33 test-avrdude.log.pMXXoV
-rw-r--r-- 1 XFCHEN Domain Users 382 Jan 30 15:47 test-avrdude.log.Rv8F1B
-rw-r--r-- 1 XFCHEN Domain Users 846 Jan 30 15:11 test-avrdude_mod.log.9Ays8m
-rw-r--r-- 1 XFCHEN Domain Users 846 Jan 30 15:13 test-avrdude_mod.log.x0wLUy
```

stefanrueger, 2024-01-31:
OK, PR #1650 adds the option `-t` so you can use `-t /tmp` but it is still a riddle why `/dev/shm` does not work. Oh well.

mcuee, 2024-01-31:
> OK, PR #1650 adds the option `-t` so you can use `-t /tmp` but it is still a riddle why `/dev/shm` does not work. Oh well.

`-t tmp` is supposed to work but it also fails. Something is very strange now.

The command is supposed to be successful but somehow the script still returns a failure.

```
$ ./test-avrdude -e "./avrdude.exe" -d 0 -t /tmp -v -p "-c dryrun -p m4808"
Testing ./avrdude.exe version 7.2-20240130 (3d3a387d)
Prepare "-c dryrun -p m4808" and press 'enter' or 'space' to continue. Press any other key to skip
✅   0.199 s: fuse access: clear, set and read eesave fuse bit
config eesave=eex_preserved # 1
✅   0.232 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
✅   0.209 s: chip erase
✅   0.235 s: flash -U write/verify holes_rjmp_loops_49152B.hex
✅   0.193 s: flash -T write/verify holes_rjmp_loops_49152B.hex
✅   0.224 s: eeprom check whether programmer can flip 0s to 1s
✅   0.197 s: eeprom -U write/verify holes_pack_my_box_256B.hex
✅   0.216 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_256B.hex
✅   0.222 s: chip erase and spot check flash is actually erased
✅   0.206 s: spot check eeprom is erased, too
❌   0.232 s: usersig -T/-U write/read random_data_64B.bin (failed command below)
$ ./avrdude.exe -qq -c dryrun -p m4808 -l /tmp/test-avrdude.log.4nEfQF -T "erase usersig; write usersig ./test_files/random_data_64B.bin" -T flush -U usersig:r:/tmp/usersig_dump5750_64B.bin:r -U usersig:v:/tmp/usersig_dump5750_64B.bin:r -T "erase usersig" -T flush -U usersig:v:./test_files/0xff_64B.hex:i

One or more AVRDUDE "-c dryrun -p m4808" tests failed. Do you want to retry this particular test? (y/n): n

$ ./avrdude.exe -c dryrun -p m4808 -T "erase usersig; write usersig ./test_files/random_data_64B.bin" -T flush -U usersig:r:/tmp/usersig_dump5750_64B.bin:
r -U usersig:v:/tmp/usersig_dump5750_64B.bin:r -T "erase usersig" -T flush -U usersig:v:./test_files/0xff_64B.hex:i
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9650 (probably m4808)

avrdude: processing -T erase usersig; write usersig ./test_files/random_data_64B.bin
Caching | ################################################## | 100% 0.21 s
Caching | ################################################## | 100% 0.20 s

avrdude: processing -T flush
avrdude: synching cache to device ... done

avrdude: processing -U usersig:r:/tmp/usersig_dump5750_64B.bin:r
avrdude: reading userrow/usersig memory ...
Reading | ################################################## | 100% 0.00 s
avrdude: writing output file /tmp/usersig_dump5750_64B.bin

avrdude: processing -U usersig:v:/tmp/usersig_dump5750_64B.bin:r
avrdude: verifying userrow/usersig memory against /tmp/usersig_dump5750_64B.bin
Reading | ################################################## | 100% 0.00 s
avrdude: 64 bytes of userrow/usersig verified

avrdude: processing -T erase usersig
Caching | ################################################## | 100% 0.22 s

avrdude: processing -T flush
avrdude: synching cache to device ... done

avrdude: processing -U usersig:v:./test_files/0xff_64B.hex:i
avrdude: verifying userrow/usersig memory against ./test_files/0xff_64B.hex
Reading | ################################################## | 100% 0.01 s
avrdude: 64 bytes of userrow/usersig verified

avrdude done.  Thank you.
```

stefanrueger, 2024-01-31:
Try `-t /tmp`  (slash tmp)

stefanrueger, 2024-01-31:
Try this 
``` diff
diff --git a/tools/test-avrdude b/tools/test-avrdude
index b63432da..ddf7caef 100755
--- a/tools/test-avrdude
+++ b/tools/test-avrdude
@@ -387,7 +387,7 @@ for (( p=0; p<$arraylength; p++ )); do
         -U usersig:v:$tfiles/0xff_${USERSIG_SIZE}B.hex:i)
       execute "${command[@]}" >$outfile
       touch "$tmpfile"
-      result [[ ! -s $outfile '&&' ! -s $logfile ]] '&&' cmp -s "$tfiles/random_data_${USERSIG_SIZE}B.bin" "$tmpfile"
+      result [[ -s $logfile ]] '&&' cmp -s "$tfiles/random_data_${USERSIG_SIZE}B.bin" "$tmpfile"
       rm -f "$tmpfile"
     fi
 
```

mcuee, 2024-01-31:
@stefanrueger 

It still does not work.
```
$ git diff
diff --git a/tools/test-avrdude b/tools/test-avrdude
index b63432da..d484eca6 100755
--- a/tools/test-avrdude
+++ b/tools/test-avrdude
@@ -387,8 +387,11 @@ for (( p=0; p<$arraylength; p++ )); do
         -U usersig:v:$tfiles/0xff_${USERSIG_SIZE}B.hex:i)
       execute "${command[@]}" >$outfile
       touch "$tmpfile"
-      result [[ ! -s $outfile '&&' ! -s $logfile ]] '&&' cmp -s "$tfiles/random_data_${USERSIG_SIZE}B.bin" "$tmpfile"
-      rm -f "$tmpfile"
+         result [[ -s $logfile ]] '&&' cmp -s "$tfiles/random_data_${USERSIG_SIZE}B.bin" "$tmpfile"
+      cat $tmpfile
+         cat $outfile
+         cat $logfile
+         rm -f "$tmpfile"
     fi

     if [ $FAIL == true ]; then

XFCHEN@1TV4433 MINGW64 /c/work/avr/avrdude_test/avrdude_main/tools
$ ./test-avrdude -e "./avrdude.exe" -d 0 -t /tmp -v -p "-c dryrun -p m4808"
Testing ./avrdude.exe version 7.2-20240131 (5684b85f)
Prepare "-c dryrun -p m4808" and press 'enter' or 'space' to continue. Press any other key to skip
✅   0.225 s: fuse access: clear, set and read eesave fuse bit
config eesave=eex_preserved # 1
✅   0.222 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
✅   0.202 s: chip erase
✅   0.222 s: flash -U write/verify holes_rjmp_loops_49152B.hex
✅   0.195 s: flash -T write/verify holes_rjmp_loops_49152B.hex
✅   0.230 s: eeprom check whether programmer can flip 0s to 1s
✅   0.203 s: eeprom -U write/verify holes_pack_my_box_256B.hex
✅   0.228 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_256B.hex
✅   0.209 s: chip erase and spot check flash is actually erased
✅   0.235 s: spot check eeprom is erased, too
❌   0.209 s: usersig -T/-U write/read random_data_64B.bin (failed command below)
$ ./avrdude.exe -qq -c dryrun -p m4808 -l /tmp/test-avrdude.log.u2qq69 -T "erase usersig; write usersig ./test_files/random_data_64B.bin" -T flush -U usersig:r:/tmp/usersig_dump6886_64B.bin:r -U usersig:v:/tmp/usersig_dump6886_64B.bin:r -T "erase usersig" -T flush -U usersig:v:./test_files/0xff_64B.hex:i
cat: /tmp/test-avrdude.out.je8ppU: No such file or directory
cat: /tmp/test-avrdude.log.u2qq69: No such file or directory

One or more AVRDUDE "-c dryrun -p m4808" tests failed. Do you want to retry this particular test? (y/n): n
```

mcuee, 2024-01-31:
Super strange.

```
$ git diff
diff --git a/tools/test-avrdude b/tools/test-avrdude
index b63432da..a5c8c99d 100755
--- a/tools/test-avrdude
+++ b/tools/test-avrdude
@@ -386,9 +386,14 @@ for (( p=0; p<$arraylength; p++ )); do
         -T flush
         -U usersig:v:$tfiles/0xff_${USERSIG_SIZE}B.hex:i)
       execute "${command[@]}" >$outfile
-      touch "$tmpfile"
-      result [[ ! -s $outfile '&&' ! -s $logfile ]] '&&' cmp -s "$tfiles/random_data_${USERSIG_SIZE}B.bin" "$tmpfile"
-      rm -f "$tmpfile"
+         hexdump $tfiles/random_data_${USERSIG_SIZE}B.bin
+      hexdump "$tmpfile"
+         result [[ -s $logfile ]] '&&' cmp -s "$tfiles/random_data_${USERSIG_SIZE}B.bin" "$tmpfile"
+         touch "$tmpfile"
+         result [[ -s $logfile ]] '&&' cmp -s "$tfiles/random_data_${USERSIG_SIZE}B.bin" "$tmpfile"
+         hexdump $tfiles/random_data_${USERSIG_SIZE}B.bin
+         hexdump "$tmpfile"
+         rm -f "$tmpfile"
     fi

     if [ $FAIL == true ]; then

$ ./test-avrdude -e "./avrdude.exe" -t /tmp -d 0 -v -p "-c dryrun -p m4808"
Testing ./avrdude.exe version 7.2-20240131 (5684b85f)
Prepare "-c dryrun -p m4808" and press 'enter' or 'space' to continue. Press any other key to skip
✅   0.220 s: fuse access: clear, set and read eesave fuse bit
config eesave=eex_preserved # 1
✅   0.219 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
✅   0.200 s: chip erase
✅   0.235 s: flash -U write/verify holes_rjmp_loops_49152B.hex
✅   0.199 s: flash -T write/verify holes_rjmp_loops_49152B.hex
✅   0.217 s: eeprom check whether programmer can flip 0s to 1s
✅   0.210 s: eeprom -U write/verify holes_pack_my_box_256B.hex
✅   0.214 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_256B.hex
✅   0.211 s: chip erase and spot check flash is actually erased
✅   0.214 s: spot check eeprom is erased, too
0000000 aac2 fbd7 519a 9513 2d47 b3ea 4329 6684
0000010 704f 0fad 7491 8ee0 863e ffd1 4df5 5cb3
0000020 0def e340 d730 eb91 8155 9332 21c2 7ba8
0000030 7621 ace2 5aff d4cb d518 6359 bc33 d84c
0000040
hexdump: /tmp/usersig_dump2992_64B.bin: No such file or directory
hexdump: all input file arguments failed
❌   0.208 s: usersig -T/-U write/read random_data_64B.bin (failed command below)
$ ./avrdude.exe -qq -c dryrun -p m4808 -l /tmp/test-avrdude.log.pSIduP -T "erase usersig; write usersig ./test_files/random_data_64B.bin" -T flush -U usersig:r:/tmp/usersig_dump2992_64B.bin:r -U usersig:v:/tmp/usersig_dump2992_64B.bin:r -T "erase usersig" -T flush -U usersig:v:./test_files/0xff_64B.hex:i
❌ -999.000 s: unknown (failed command below)
$ sleep 0.1
0000000 aac2 fbd7 519a 9513 2d47 b3ea 4329 6684
0000010 704f 0fad 7491 8ee0 863e ffd1 4df5 5cb3
0000020 0def e340 d730 eb91 8155 9332 21c2 7ba8
0000030 7621 ace2 5aff d4cb d518 6359 bc33 d84c
0000040

One or more AVRDUDE "-c dryrun -p m4808" tests failed. Do you want to retry this particular test? (y/n): n

$ ./test-avrdude -e "./avrdude.exe" -t . -d 0 -v -p "-c dryrun -p m4808"
Testing ./avrdude.exe version 7.2-20240131 (5684b85f)
Prepare "-c dryrun -p m4808" and press 'enter' or 'space' to continue. Press any other key to skip
✅   0.206 s: fuse access: clear, set and read eesave fuse bit
config eesave=eex_preserved # 1
✅   0.219 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
✅   0.198 s: chip erase
✅   0.231 s: flash -U write/verify holes_rjmp_loops_49152B.hex
✅   0.206 s: flash -T write/verify holes_rjmp_loops_49152B.hex
✅   0.225 s: eeprom check whether programmer can flip 0s to 1s
✅   0.206 s: eeprom -U write/verify holes_pack_my_box_256B.hex
✅   0.209 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_256B.hex
✅   0.201 s: chip erase and spot check flash is actually erased
✅   0.224 s: spot check eeprom is erased, too
0000000 aac2 fbd7 519a 9513 2d47 b3ea 4329 6684
0000010 704f 0fad 7491 8ee0 863e ffd1 4df5 5cb3
0000020 0def e340 d730 eb91 8155 9332 21c2 7ba8
0000030 7621 ace2 5aff d4cb d518 6359 bc33 d84c
0000040
0000000 aac2 fbd7 519a 9513 2d47 b3ea 4329 6684
0000010 704f 0fad 7491 8ee0 863e ffd1 4df5 5cb3
0000020 0def e340 d730 eb91 8155 9332 21c2 7ba8
0000030 7621 ace2 5aff d4cb d518 6359 bc33 d84c
0000040
❌   0.211 s: usersig -T/-U write/read random_data_64B.bin (failed command below)
$ ./avrdude.exe -qq -c dryrun -p m4808 -l ./test-avrdude.log.HMudNi -T "erase usersig; write usersig ./test_files/random_data_64B.bin" -T flush -U usersig:r:./usersig_dump3129_64B.bin:r -U usersig:v:./usersig_dump3129_64B.bin:r -T "erase usersig" -T flush -U usersig:v:./test_files/0xff_64B.hex:i
❌ -999.000 s: unknown (failed command below)
$ sleep 0.1
0000000 aac2 fbd7 519a 9513 2d47 b3ea 4329 6684
0000010 704f 0fad 7491 8ee0 863e ffd1 4df5 5cb3
0000020 0def e340 d730 eb91 8155 9332 21c2 7ba8
0000030 7621 ace2 5aff d4cb d518 6359 bc33 d84c
0000040
0000000 aac2 fbd7 519a 9513 2d47 b3ea 4329 6684
0000010 704f 0fad 7491 8ee0 863e ffd1 4df5 5cb3
0000020 0def e340 d730 eb91 8155 9332 21c2 7ba8
0000030 7621 ace2 5aff d4cb d518 6359 bc33 d84c
0000040

One or more AVRDUDE "-c dryrun -p m4808" tests failed. Do you want to retry this particular test? (y/n): n
```

mcuee, 2024-01-31:
Carry out the git reset and the last one now works. `-t /tmp` does not work, `-t .` works.

```
C:\work\avr\avrdude_test\avrdude_main [main ≡ +8 ~1 -0 !]> git reset --hard
Updating files: 100% (582/582), done.
HEAD is now at 5684b85f Update NEWS

$ ./test-avrdude -e "./avrdude.exe" -d 0 -t /tmp -v -p "-c dryrun -p m4808"
Testing ./avrdude.exe version 7.2-20240131 (5684b85f)
Prepare "-c dryrun -p m4808" and press 'enter' or 'space' to continue. Press any other key to skip
✅   0.225 s: fuse access: clear, set and read eesave fuse bit
config eesave=eex_preserved # 1
✅   0.203 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
✅   0.196 s: chip erase
✅   0.239 s: flash -U write/verify holes_rjmp_loops_49152B.hex
✅   0.204 s: flash -T write/verify holes_rjmp_loops_49152B.hex
✅   0.222 s: eeprom check whether programmer can flip 0s to 1s
✅   0.205 s: eeprom -U write/verify holes_pack_my_box_256B.hex
✅   0.220 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_256B.hex
✅   0.203 s: chip erase and spot check flash is actually erased
✅   0.223 s: spot check eeprom is erased, too
❌   0.209 s: usersig -T/-U write/read random_data_64B.bin (failed command below)
$ ./avrdude.exe -qq -c dryrun -p m4808 -l /tmp/test-avrdude.log.lWuvEV -T "erase usersig; write usersig ./test_files/random_data_64B.bin" -T flush -U usersig:r:/tmp/usersig_dump2407_64B.bin:r -U usersig:v:/tmp/usersig_dump2407_64B.bin:r -T "erase usersig" -T flush -U usersig:v:./test_files/0xff_64B.hex:i

One or more AVRDUDE "-c dryrun -p m4808" tests failed. Do you want to retry this particular test? (y/n): n

$ ./test-avrdude -e "./avrdude.exe" -d 0 -t . -v -p "-c dryrun -p m4808"
Testing ./avrdude.exe version 7.2-20240131 (5684b85f)
Prepare "-c dryrun -p m4808" and press 'enter' or 'space' to continue. Press any other key to skip
✅   0.216 s: fuse access: clear, set and read eesave fuse bit
config eesave=eex_preserved # 1
✅   0.222 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
✅   0.196 s: chip erase
✅   0.229 s: flash -U write/verify holes_rjmp_loops_49152B.hex
✅   0.195 s: flash -T write/verify holes_rjmp_loops_49152B.hex
✅   0.220 s: eeprom check whether programmer can flip 0s to 1s
✅   0.203 s: eeprom -U write/verify holes_pack_my_box_256B.hex
✅   0.207 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_256B.hex
✅   0.200 s: chip erase and spot check flash is actually erased
✅   0.218 s: spot check eeprom is erased, too
✅   0.205 s: usersig -T/-U write/read random_data_64B.bin
```

mcuee, 2024-01-31:
@stefanrueger 

It seems to be very specific to MSYS2 here. I think we can just add a note to say to use `-t .` when using MSYS2 under Windows.

mcuee, 2024-01-31:
I tried git bash (git for Windows) and it has the same behavior. 

```
$ ./test-avrdude -e "./avrdude.exe" -d 0 -t /tmp -p "-c dryrun -p m4808"
Testing ./avrdude.exe version 7.2-20240131 (5684b85f)
Prepare "-c dryrun -p m4808" and press 'enter' or 'space' to continue. Press any other key to skip
✅   0.239 s: fuse access: clear, set and read eesave fuse bit
✅   0.222 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
✅   0.235 s: chip erase
✅   0.229 s: flash -U write/verify holes_rjmp_loops_49152B.hex
✅   0.234 s: flash -T write/verify holes_rjmp_loops_49152B.hex
✅   0.219 s: eeprom check whether programmer can flip 0s to 1s
✅   0.239 s: eeprom -U write/verify holes_pack_my_box_256B.hex
✅   0.221 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_256B.hex
✅   0.223 s: chip erase and spot check flash is actually erased
✅   0.249 s: spot check eeprom is erased, too
❌   0.224 s: usersig -T/-U write/read random_data_64B.bin (failed command below)
$ ./avrdude.exe -qq -c dryrun -p m4808 -l /tmp/test-avrdude.log.UWGDM3 -T "erase usersig; write usersig ./test_files/random_data_64B.bin" -T flush -U usersig:r:/tmp/usersig_dump2167_64B.bin:r -U usersig:v:/tmp/usersig_dump2167_64B.bin:r -T "erase usersig" -T flush -U usersig:v:./test_files/0xff_64B.hex:i

One or more AVRDUDE "-c dryrun -p m4808" tests failed. Do you want to retry this particular test? (y/n): n

$ ./test-avrdude -e "./avrdude.exe" -d 0 -t . -p "-c dryrun -p m4808"
Testing ./avrdude.exe version 7.2-20240131 (5684b85f)
Prepare "-c dryrun -p m4808" and press 'enter' or 'space' to continue. Press any other key to skip
✅   0.235 s: fuse access: clear, set and read eesave fuse bit
✅   0.217 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
✅   0.229 s: chip erase
✅   0.223 s: flash -U write/verify holes_rjmp_loops_49152B.hex
✅   0.246 s: flash -T write/verify holes_rjmp_loops_49152B.hex
✅   0.215 s: eeprom check whether programmer can flip 0s to 1s
✅   0.224 s: eeprom -U write/verify holes_pack_my_box_256B.hex
✅   0.230 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_256B.hex
✅   0.220 s: chip erase and spot check flash is actually erased
✅   0.245 s: spot check eeprom is erased, too
✅   0.229 s: usersig -T/-U write/read random_data_64B.bin

```

stefanrueger, 2024-01-31:
One last try before I give up: how about
``` diff
diff --git a/tools/test-avrdude b/tools/test-avrdude
index b63432da..7bc0ab32 100755
--- a/tools/test-avrdude
+++ b/tools/test-avrdude
@@ -387,7 +387,7 @@ for (( p=0; p<$arraylength; p++ )); do
         -U usersig:v:$tfiles/0xff_${USERSIG_SIZE}B.hex:i)
       execute "${command[@]}" >$outfile
       touch "$tmpfile"
-      result [[ ! -s $outfile '&&' ! -s $logfile ]] '&&' cmp -s "$tfiles/random_data_${USERSIG_SIZE}B.bin" "$tmpfile"
+      result cmp -s "$tfiles/random_data_${USERSIG_SIZE}B.bin" "$tmpfile"
       rm -f "$tmpfile"
     fi
```

mcuee, 2024-01-31:
> One last try before I give up: how about
>
> ```diff
> diff --git a/tools/test-avrdude b/tools/test-avrdude
> index b63432da..7bc0ab32 100755
> --- a/tools/test-avrdude
> +++ b/tools/test-avrdude
> @@ -387,7 [...]

Unfortunately it does not work with `-t /tmp`.

```
$ git diff
diff --git a/tools/test-avrdude b/tools/test-avrdude
index b63432da..fe7eaa10 100755
--- a/tools/test-avrdude
+++ b/tools/test-avrdude
@@ -387,7 +387,8 @@ for (( p=0; p<$arraylength; p++ )); do
         -U usersig:v:$tfiles/0xff_${USERSIG_SIZE}B.hex:i)
       execute "${command[@]}" >$outfile
       touch "$tmpfile"
-      result [[ ! -s $outfile '&&' ! -s $logfile ]] '&&' cmp -s "$tfiles/random_data_${USERSIG_SIZE}B.bin" "$tmpfile"
+      #result [[ ! -s $outfile '&&' ! -s $logfile ]] '&&' cmp -s "$tfiles/random_data_${USERSIG_SIZE}B.bin" "$tmpfile"
+         result cmp -s "$tfiles/random_data_${USERSIG_SIZE}B.bin" "$tmpfile"
       rm -f "$tmpfile"
     fi

$ ./test-avrdude -e "./avrdude.exe" -d 0 -t /tmp -v  -p "-c jtag2updi -p m4808 -P ch340"
Testing ./avrdude.exe version 7.2-20240131 (5684b85f)
Prepare "-c jtag2updi -p m4808 -P ch340" and press 'enter' or 'space' to continue. Press any other key to skip
✅   1.029 s: fuse access: clear, set and read eesave fuse bit
config eesave=eex_preserved # 1
✅   1.014 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
✅   1.148 s: chip erase
✅   4.889 s: flash -U write/verify holes_rjmp_loops_49152B.hex
✅   2.456 s: flash -T write/verify holes_rjmp_loops_49152B.hex
✅   1.091 s: eeprom check whether programmer can flip 0s to 1s
✅   1.089 s: eeprom -U write/verify holes_pack_my_box_256B.hex
✅   1.152 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_256B.hex
✅   2.631 s: chip erase and spot check flash is actually erased
✅   1.026 s: spot check eeprom is erased, too
❌   1.134 s: usersig -T/-U write/read random_data_64B.bin (failed command below)
$ ./avrdude.exe -qq -c jtag2updi -p m4808 -P ch340 -l /tmp/test-avrdude.log.4JnXel -T "erase usersig; write usersig ./test_files/random_data_64B.bin" -T flush -U usersig:r:/tmp/usersig_dump1342_64B.bin:r -U usersig:v:/tmp/usersig_dump1342_64B.bin:r -T "erase usersig" -T flush -U usersig:v:./test_files/0xff_64B.hex:i

One or more AVRDUDE "-c jtag2updi -p m4808 -P ch340" tests failed. Do you want to retry this particular test? (y/n): n
```

stefanrueger, 2024-01-31:
Well, I am running out of ideas why one directory works but another doesn't. Maybe the length of the filename? Could remove `_dump` or more from filename and try that. Also could remove the `-s` after `cmp` to see the kind of error `cmp` throws. Or `cmp` isn't well programmed and doesn't return proper exit values? But, yes, scratching the bottom of the barrel here. Maybe we should just be happy you found a workaround with `-t .` and close the issue?

mcuee, 2024-01-31:
> Well, I am running out of ideas why one directory works but another doesn't. Maybe the length of the filename? Could remove `_dump` or more from filename and try that. Also could remove the `-s` [...]

Give me until end of this week to carry out the investigation and testing on other platforms.
 
If we can not sort out this minor issue in the end, I think we can just add a note on the top of the script to ask the user to use `-t . ` as a work-around.

mcuee, 2024-01-31:
The following work-around also works. Looks like Unix style directory name does not work whereas modified Windows style (or DOS style) directory name works.

1) `c:/tmp` works but not `c:\tmp`.
2) `/tmp` and `/c/tmp` do not work

```
$ ./test-avrdude -e "./avrdude.exe" -d 0 -v -t c:/temp -p "-c dryrun -p m4808"
Testing ./avrdude.exe version 7.2-20240131 (5684b85f)
Prepare "-c dryrun -p m4808" and press 'enter' or 'space' to continue. Press any other key to skip
✅   0.095 s: fuse access: clear, set and read eesave fuse bit
config eesave=eex_preserved # 1
✅   0.104 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
✅   0.113 s: chip erase
✅   0.098 s: flash -U write/verify holes_rjmp_loops_49152B.hex
✅   0.103 s: flash -T write/verify holes_rjmp_loops_49152B.hex
✅   0.100 s: eeprom check whether programmer can flip 0s to 1s
✅   0.106 s: eeprom -U write/verify holes_pack_my_box_256B.hex
✅   0.104 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_256B.hex
✅   0.098 s: chip erase and spot check flash is actually erased
✅   0.102 s: spot check eeprom is erased, too
✅   0.106 s: usersig -T/-U write/read random_data_64B.bin
```

Interestingly `-t /c/tmp` does not work.

```
$ ./test-avrdude -e "./avrdude.exe" -d 0 -v -t /c/tmp -p "-c dryrun -p m4808"
Testing ./avrdude.exe version 7.2-20240131 (5684b85f)
Prepare "-c dryrun -p m4808" and press 'enter' or 'space' to continue. Press any other key to skip
✅   0.105 s: fuse access: clear, set and read eesave fuse bit
config eesave=eex_preserved # 1
✅   0.099 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
✅   0.106 s: chip erase
✅   0.113 s: flash -U write/verify holes_rjmp_loops_49152B.hex
✅   0.097 s: flash -T write/verify holes_rjmp_loops_49152B.hex
✅   0.099 s: eeprom check whether programmer can flip 0s to 1s
✅   0.107 s: eeprom -U write/verify holes_pack_my_box_256B.hex
✅   0.107 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_256B.hex
✅   0.120 s: chip erase and spot check flash is actually erased
✅   0.108 s: spot check eeprom is erased, too
❌   0.120 s: usersig -T/-U write/read random_data_64B.bin (failed command below)
$ ./avrdude.exe -qq -c dryrun -p m4808 -l /c/tmp/test-avrdude.log.KRoXqh -T "erase usersig; write usersig ./test_files/random_data_64B.bin" -T flush -U usersig:r:/c/tmp/usersig_dump477_64B.bin:r -U usersig:v:/c/tmp/usersig_dump477_64B.bin:r -T "erase usersig" -T flush -U usersig:v:./test_files/0xff_64B.hex:i
avrdude OS error: file /c/tmp/usersig_dump477_64B.bin is not writeable: No such file or directory
avrdude OS error: file /c/tmp/usersig_dump477_64B.bin is not readable: No such file or directory
avrdude OS error: cannot open output file /c/tmp/usersig_dump477_64B.bin: No such file or directory
avrdude error: write to file /c/tmp/usersig_dump477_64B.bin failed

One or more AVRDUDE "-c dryrun -p m4808" tests failed. Do you want to retry this particular test? (y/n): n
```

mcuee, 2024-01-31:
Looks like the file creation just failed.

```
$ git diff
diff --git a/tools/test-avrdude b/tools/test-avrdude
index b63432da..637781ba 100755
--- a/tools/test-avrdude
+++ b/tools/test-avrdude
@@ -386,8 +386,12 @@ for (( p=0; p<$arraylength; p++ )); do
         -T flush
         -U usersig:v:$tfiles/0xff_${USERSIG_SIZE}B.hex:i)
       execute "${command[@]}" >$outfile
+         cat "$outfile"
+         hexdump "$tmpfile"
+         hexdump "$tfiles/random_data_${USERSIG_SIZE}B.bin"
       touch "$tmpfile"
-      result [[ ! -s $outfile '&&' ! -s $logfile ]] '&&' cmp -s "$tfiles/random_data_${USERSIG_SIZE}B.bin" "$tmpfile"
+      #result [[ ! -s $outfile '&&' ! -s $logfile ]] '&&' cmp "$tfiles/random_data_${USERSIG_SIZE}B.bin" "$tmpfile"
+         result cmp -l "$tfiles/random_data_${USERSIG_SIZE}B.bin" "$tmpfile"
       rm -f "$tmpfile"
     fi

xiaof@LAPTOP-913SIMUD MINGW64 /c/work/avr/avrdude_test/avrdude_main/tools
$ ./test-avrdude -e "./avrdude.exe" -d 0 -v -t /tmp -p "-c dryrun -p m4808"
Testing ./avrdude.exe version 7.2-20240131 (5684b85f)
Prepare "-c dryrun -p m4808" and press 'enter' or 'space' to continue. Press any other key to skip
✅   0.093 s: fuse access: clear, set and read eesave fuse bit
config eesave=eex_preserved # 1
✅   0.096 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
✅   0.099 s: chip erase
✅   0.112 s: flash -U write/verify holes_rjmp_loops_49152B.hex
✅   0.108 s: flash -T write/verify holes_rjmp_loops_49152B.hex
✅   0.098 s: eeprom check whether programmer can flip 0s to 1s
✅   0.099 s: eeprom -U write/verify holes_pack_my_box_256B.hex
✅   0.101 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_256B.hex
✅   0.109 s: chip erase and spot check flash is actually erased
✅   0.100 s: spot check eeprom is erased, too
hexdump: /tmp/usersig_dump2057_64B.bin: No such file or directory
hexdump: all input file arguments failed
0000000 aac2 fbd7 519a 9513 2d47 b3ea 4329 6684
0000010 704f 0fad 7491 8ee0 863e ffd1 4df5 5cb3
0000020 0def e340 d730 eb91 8155 9332 21c2 7ba8
0000030 7621 ace2 5aff d4cb d518 6359 bc33 d84c
0000040
cmp: EOF on /tmp/usersig_dump2057_64B.bin which is empty
❌   0.102 s: usersig -T/-U write/read random_data_64B.bin (failed command below)
$ ./avrdude.exe -qq -c dryrun -p m4808 -l /tmp/test-avrdude.log.6YJoMt -T "erase usersig; write usersig ./test_files/random_data_64B.bin" -T flush -U usersig:r:/tmp/usersig_dump2057_64B.bin:r -U usersig:v:/tmp/usersig_dump2057_64B.bin:r -T "erase usersig" -T flush -U usersig:v:./test_files/0xff_64B.hex:i

One or more AVRDUDE "-c dryrun -p m4808" tests failed. Do you want to retry this particular test? (y/n): n
```

mcuee, 2024-01-31:
>> Maybe we should just be happy you found a workaround with -t . and close the issue?
>
> If we can not sort out this minor issue in the end, I think we can just add a note on the top of the script to ask the user to use -t .  as a work-around.

I think it is not worth the time to scratch the head here. Adding a note will do. Or we can use the Wiki.

mcuee, 2024-02-01:
> I think it is not worth the time to scratch the head here. Adding a note will do. Or we can use the Wiki.

I will close this issue and update the Wiki.

closed 2024-02-01 by mcuee

mcuee, 2024-02-01:
> > I think it is not worth the time to scratch the head here. Adding a note will do. Or we can use the Wiki.
> 
> I will close this issue and update the Wiki.

Done.
https://github.com/avrdudes/avrdude/wiki/Known-limitations-of-avrdude#known-issues-for-test-avrdude-tool

stefanrueger, 2024-02-01:
> Looks like the file creation just failed

Pretty sure avrdude could create and use the file (otherwise avrdude would have issued a ton of error messages). Maybe the OS has some sort of mechanism to delete a file in the /tmp folder if the program that created it has finished? I have now shifted the creation of temporary file from the avrdude invocation to the calling bash script in PR #1651

mcuee, 2024-02-01:
> > Looks like the file creation just failed
> 
> Pretty sure avrdude could create and use the file (otherwise avrdude would have issued a ton of error messages). Maybe the OS has some sort of mechanism to delete a file in the /tmp folder if the program that created it has finished? I have now shifted the creation of temporary files from the avrdude invocation to the calling bash script in PR #1651

Unfortunately PR #1651 still does not help.

```
$ ./test-avrdude -e "./avrdude.exe" -v -p "-c dryrun -p avr16ea28"
Testing ./avrdude.exe version 7.2-20240201 (acd5584c)
Prepare "-c dryrun -p avr16ea28" and press 'enter' or 'space' to continue. Press any other key to skip
✅   0.209 s: fuse access: clear, set and read eesave fuse bit
config eesave=eex_preserved # 1
✅   0.220 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
✅   0.216 s: chip erase
✅   0.218 s: flash -U write/verify holes_rjmp_loops_16384B.hex
✅   0.208 s: flash -T write/verify holes_rjmp_loops_16384B.hex
✅   0.195 s: eeprom check whether programmer can flip 0s to 1s
✅   0.209 s: eeprom -U write/verify holes_pack_my_box_512B.hex
✅   0.212 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_512B.hex
✅   0.211 s: chip erase and spot check flash is actually erased
✅   0.198 s: spot check eeprom is erased, too
❌   0.203 s: usersig -T/-U write/read random_data_64B.bin (failed command below)
$ ./avrdude.exe -qq -c dryrun -p avr16ea28 -l /dev/shm/test-avrdude.log.u3OOhk -T "erase usersig; write usersig ./test_files/random_data_64B.bin" -T flush -U usersig:r:/dev/shm/test-avrdude.tmp.1sd2TK:r -U usersig:v:/dev/shm/test-avrdude.tmp.1sd2TK:r -T "erase usersig" -T flush -U usersig:v:./test_files/0xff_64B.hex:i
avrdude OS error: file /dev/shm/test-avrdude.tmp.1sd2TK is not writeable: No such file or directory
avrdude OS error: file /dev/shm/test-avrdude.tmp.1sd2TK is not readable: No such file or directory
avrdude OS error: cannot open output file /dev/shm/test-avrdude.tmp.1sd2TK: No such file or directory
avrdude error: write to file /dev/shm/test-avrdude.tmp.1sd2TK failed

One or more AVRDUDE "-c dryrun -p avr16ea28" tests failed. Do you want to retry this particular test? (y/n):

$ ./test-avrdude -e "./avrdude.exe" -v -t /tmp -p "-c dryrun -p avr16ea28"
Testing ./avrdude.exe version 7.2-20240201 (acd5584c)
Prepare "-c dryrun -p avr16ea28" and press 'enter' or 'space' to continue. Press any other key to skip
✅   0.239 s: fuse access: clear, set and read eesave fuse bit
config eesave=eex_preserved # 1
✅   0.210 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
✅   0.209 s: chip erase
✅   0.220 s: flash -U write/verify holes_rjmp_loops_16384B.hex
✅   0.195 s: flash -T write/verify holes_rjmp_loops_16384B.hex
✅   0.228 s: eeprom check whether programmer can flip 0s to 1s
✅   0.196 s: eeprom -U write/verify holes_pack_my_box_512B.hex
✅   0.240 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_512B.hex
✅   0.195 s: chip erase and spot check flash is actually erased
✅   0.221 s: spot check eeprom is erased, too
❌   0.210 s: usersig -T/-U write/read random_data_64B.bin (failed command below)
$ ./avrdude.exe -qq -c dryrun -p avr16ea28 -l /tmp/test-avrdude.log.QxDenI -T "erase usersig; write usersig ./test_files/random_data_64B.bin" -T flush -U usersig:r:/tmp/test-avrdude.tmp.HtSiFd:r -U usersig:v:/tmp/test-avrdude.tmp.HtSiFd:r -T "erase usersig" -T flush -U usersig:v:./test_files/0xff_64B.hex:i

One or more AVRDUDE "-c dryrun -p avr16ea28" tests failed. Do you want to retry this particular test? (y/n): n

$ ./test-avrdude -e "./avrdude.exe" -v -t . -p "-c dryrun -p avr16ea28"
Testing ./avrdude.exe version 7.2-20240201 (acd5584c)
Prepare "-c dryrun -p avr16ea28" and press 'enter' or 'space' to continue. Press any other key to skip
✅   0.208 s: fuse access: clear, set and read eesave fuse bit
config eesave=eex_preserved # 1
✅   0.210 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
✅   0.267 s: chip erase
✅   0.206 s: flash -U write/verify holes_rjmp_loops_16384B.hex
✅   0.229 s: flash -T write/verify holes_rjmp_loops_16384B.hex
✅   0.204 s: eeprom check whether programmer can flip 0s to 1s
✅   0.217 s: eeprom -U write/verify holes_pack_my_box_512B.hex
✅   0.203 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_512B.hex
✅   0.224 s: chip erase and spot check flash is actually erased
✅   0.209 s: spot check eeprom is erased, too
✅   0.219 s: usersig -T/-U write/read random_data_64B.bin
```

PR #1650 (2024-01-31): Update test-avrdude script
 - Avoid sed -i in test-avrdude script
 - Add option test-avrdude -t <dir> to set temporary directory

Fixes #1648 and #1649 (fingers crossed)