avrdudes/avrdude #1649: Test script issues

MCUdude, 2024-01-30:
While testing #1647, I stumbled across this issue. I'm getting a sed rename() error, and the eesave fuse bit is still set to 1, which means that the EEPROM is preserved, even though it shouldn't be.

```
$ ./test-avrdude -d 0 -p "-c nanoevery -p atmega4809 -r -P /dev/cu.usbmodem14101" -v
Testing avrdude version 7.2-20230720
Prepare "-c nanoevery -p atmega4809 -r -P /dev/cu.usbmodem14101" and press 'enter' or 'space' to continue. Press any other key to skip
sed: rename(): No such file or directory
❌   3.452 s: fuse access: clear, set and read eesave fuse bit (failed command below)
$ avrdude -qq -c nanoevery -p atmega4809 -r -P /dev/cu.usbmodem14101 -l /tmp/test-avrdude.log.xcLt9q -T "config eesave=0; config eesave=1; config eesave"
config eesave=eex_preserved # 1
sed: rename(): No such file or directory
✅   3.431 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
✅   3.551 s: chip erase
✅  10.358 s: flash -U write/verify holes_rjmp_loops_49152B.hex
✅   4.256 s: flash -T write/verify holes_rjmp_loops_49152B.hex
✅   3.523 s: eeprom check whether programmer can flip 0s to 1s
✅   3.541 s: eeprom -U write/verify holes_pack_my_box_256B.hex
✅   3.649 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_256B.hex
✅   4.398 s: chip erase and spot check flash is actually erased
❌   3.417 s: spot check eeprom is erased, too (failed command below)
$ avrdude -qq -c nanoevery -p atmega4809 -r -P /dev/cu.usbmodem14101 -l /tmp/test-avrdude.log.xcLt9q -Ueeprom:v:./test_files/holes_eeprom_0xff_256B.hex
avrdude warning: verification mismatch
        device 0x50 != input 0xff at addr 0x0022 (error)
avrdude error: verification mismatch
✅   3.601 s: usersig -T/-U write/read random_data_64B.bin

One or more AVRDUDE "-c nanoevery -p atmega4809 -r -P /dev/cu.usbmodem14101" tests failed. Do you want to retry this particular test? (y/n): n

$ avrdude -c nanoevery -p atmega4809 -r -P /dev/cu.usbmodem14101 -T "config eesave"
avrdude: touching serial port /dev/cu.usbmodem14101 at 1200 baud
avrdude: waiting for new port... using same port /dev/cu.usbmodem14101
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9651 (probably m4809)

avrdude: processing -T config eesave
config eesave=eex_preserved # 1

avrdude done.  Thank you.
```

stefanrueger, 2024-01-30:
For the `sed` problem you could try to see whether it is a problem with `dryrun`, too.

mcuee, 2024-01-30:
I can reproduce the issue under macOS 14.3 with my Mac Mini M1 2020.
```
mcuee@mcuees-Mac-mini tools % ./test-avrdude -p "-c dryrun -p m4809"
Testing avrdude version 7.2-20240130 (3d3a387d)
Prepare "-c dryrun -p m4809" and press 'enter' or 'space' to continue. Press any other key to skip
sed: rename(): No such file or directory
❌   0.090 s: fuse access: clear, set and read eesave fuse bit (failed command below)
$ avrdude -qq -c dryrun -p m4809 -l /tmp/test-avrdude.log.vlbEnL -T "config eesave=0; config eesave=1; config eesave"
sed: rename(): No such file or directory
✅   0.090 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
✅   0.076 s: chip erase
✅   0.090 s: flash -U write/verify holes_rjmp_loops_49152B.hex
✅   0.092 s: flash -T write/verify holes_rjmp_loops_49152B.hex
✅   0.091 s: eeprom check whether programmer can flip 0s to 1s
✅   0.100 s: eeprom -U write/verify holes_pack_my_box_256B.hex
✅   0.056 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_256B.hex
✅   0.058 s: chip erase and spot check flash is actually erased
✅   0.059 s: spot check eeprom is erased, too
✅   0.061 s: usersig -T/-U write/read random_data_64B.bin

One or more AVRDUDE "-c dryrun -p m4809" tests failed. Do you want to retry this particular test? (y/n): n
```

mcuee, 2024-01-31:
@stefanrueger and @MCUdude 

BTW, I think avrdude git main is pretty good now. The issue found are mainly with the scripts. I think it is worth the time and effort to improve the script.

stefanrueger, 2024-01-31:
> worth the time and effort to improve the script

As so often it's about programming so that *old* bash versions work and other OS as well. Good job you have such an impressive array of OS.

If you have an atmega4808 (or any other UPDI part) could you try to reproduce @MCUdude's observation that after the test script runs `-T config eesave` shows `config eesave=eex_preserved # 1`. I don't get it how that could be

stefanrueger, 2024-01-31:
I'll merge the updated script, so this issue closes. I am still concerned about the `ee`save=eex_preserved` business which I cannot explain and which may need separate examination outside the `test-avrdude` script and, if warranted, a separate issue. It would be great if @MCUdude or @mcuee could play with
```
$ avrdude -qq -c nanoevery -p atmega4809 -r -P /dev/cu.usbmodem14101 -T "config eesave=ee*erased"
```
and see what the eeave bit is afterwards and whether or not EEPROM *is* erased after `avrdude -e`.

mcuee, 2024-01-31:
@stefanrueger and @MCUdude 

Here is what I have with latest git main under Windows 10.

```
C:\work\avr\avrdude_test\avrdude_main\tools [main ≡ +9 ~0 -0 !]> 
.\avrdude -c jtag2updi -p atmega4808 -P ch340 -T "config eesave=ee*erased" -T "config eesave"
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9650 (probably m4808)

avrdude: processing -T config eesave=ee*erased

avrdude: processing -T config eesave
config eesave=eex_erased # 0

avrdude done.  Thank you.

C:\work\avr\avrdude_test\avrdude_main\tools [main ≡ +9 ~0 -0 !]> 
.\avrdude -c nanoevery -p atmega4809 -r -P COM8 -T "config eesave=ee*erased" -T "config eesave"
avrdude: touching serial port COM8 at 1200 baud
avrdude: waiting for new port... using same port COM8
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9651 (probably m4809)

avrdude: processing -T config eesave=ee*erased

avrdude: processing -T config eesave
config eesave=eex_erased # 0

avrdude done.  Thank you.
```

stefanrueger, 2024-01-31:
@mcuee Thanks for testing. Brilliant. So,`test-avrdude -p "-c nanoevery -p m4808 -P ch340 -r"` should work with the right `-t` option. Good to know

mcuee, 2024-01-31:
Test results under FreeBSD 14.0 Release VM (Proxmox PVE 8.0 VM, using an Intel N100 CPU based mini PC).

The script runs well once I installed bash.

For Arduino Nano Every, extra delay `-d 0`, `-d 1` and `-d 2` results are the same, the second tests will fail.

```
mcuee@freebsd14vmn100:~/build/avrdude/tools $ sudo ./test-avrdude -d 0 -v -p "-c nanoevery -p m4809 -P /dev/ttyU0 -r"
Testing avrdude version 7.2-20240131 (5684b85f)
Prepare "-c nanoevery -p m4809 -P /dev/ttyU0 -r" and press 'enter' or 'space' to continue. Press any other key to skip
✅   4.388 s: fuse access: clear, set and read eesave fuse bit
config eesave=eex_preserved # 1
❌   9.412 s: fuse access: set eesave fusebit to delete EEPROM on chip erase (failed command below)
$ avrdude -qq -c nanoevery -p m4809 -P /dev/ttyU0 -r -l /tmp/test-avrdude.log.3UJFNB -T "config eesave=ee*erased"
avrdude warning: attempt 1 of 10: sign-on command: status -1
✅  14.648 s: chip erase
avrdude warning: attempt 1 of 10: sign-on command: status -1
avrdude warning: attempt 2 of 10: sign-on command: status -1
✅  22.086 s: flash -U write/verify holes_rjmp_loops_49152B.hex
avrdude warning: attempt 1 of 10: sign-on command: status -1
avrdude warning: attempt 2 of 10: sign-on command: status -1
✅   5.801 s: flash -T write/verify holes_rjmp_loops_49152B.hex
✅   4.531 s: eeprom check whether programmer can flip 0s to 1s
✅   4.539 s: eeprom -U write/verify holes_pack_my_box_256B.hex
✅   4.653 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_256B.hex
✅   5.953 s: chip erase and spot check flash is actually erased
✅  14.576 s: spot check eeprom is erased, too
avrdude warning: attempt 1 of 10: sign-on command: status -1
avrdude warning: attempt 2 of 10: sign-on command: status -1
✅   4.564 s: usersig -T/-U write/read random_data_64B.bin

One or more AVRDUDE "-c nanoevery -p m4809 -P /dev/ttyU0 -r" tests failed. Do you want to retry this particular test? (y/n): n

mcuee@freebsd14vmn100:~/build/avrdude/tools $ sudo ./test-avrdude -d 1 -v -p "-c nanoevery -p m4809 -P /dev/ttyU0 -r"
Testing avrdude version 7.2-20240131 (5684b85f)
Prepare "-c nanoevery -p m4809 -P /dev/ttyU0 -r" and press 'enter' or 'space' to continue. Press any other key to skip
✅   4.395 s: fuse access: clear, set and read eesave fuse bit
config eesave=eex_preserved # 1
❌   9.440 s: fuse access: set eesave fusebit to delete EEPROM on chip erase (failed command below)
$ avrdude -qq -c nanoevery -p m4809 -P /dev/ttyU0 -r -l /tmp/test-avrdude.log.Ckafeb -T "config eesave=ee*erased"
avrdude warning: attempt 1 of 10: sign-on command: status -1
✅  14.568 s: chip erase
avrdude warning: attempt 1 of 10: sign-on command: status -1
avrdude warning: attempt 2 of 10: sign-on command: status -1
✅  22.050 s: flash -U write/verify holes_rjmp_loops_49152B.hex
avrdude warning: attempt 1 of 10: sign-on command: status -1
avrdude warning: attempt 2 of 10: sign-on command: status -1
✅   5.786 s: flash -T write/verify holes_rjmp_loops_49152B.hex
✅   4.530 s: eeprom check whether programmer can flip 0s to 1s
✅   4.542 s: eeprom -U write/verify holes_pack_my_box_256B.hex
✅   4.664 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_256B.hex
✅   5.936 s: chip erase and spot check flash is actually erased
✅  14.436 s: spot check eeprom is erased, too
avrdude warning: attempt 1 of 10: sign-on command: status -1
avrdude warning: attempt 2 of 10: sign-on command: status -1
✅   4.572 s: usersig -T/-U write/read random_data_64B.bin

One or more AVRDUDE "-c nanoevery -p m4809 -P /dev/ttyU0 -r" tests failed. Do you want to retry this particular test? (y/n): n

mcuee@freebsd14vmn100:~/build/avrdude/tools $ sudo ./test-avrdude -d 2 -v -p "-c nanoevery -p m4809 -P /dev/ttyU0 -r"
Testing avrdude version 7.2-20240131 (5684b85f)
Prepare "-c nanoevery -p m4809 -P /dev/ttyU0 -r" and press 'enter' or 'space' to continue. Press any other key to skip
✅   4.385 s: fuse access: clear, set and read eesave fuse bit
config eesave=eex_preserved # 1
❌   9.442 s: fuse access: set eesave fusebit to delete EEPROM on chip erase (failed command below)
$ avrdude -qq -c nanoevery -p m4809 -P /dev/ttyU0 -r -l /tmp/test-avrdude.log.G9RA99 -T "config eesave=ee*erased"
avrdude warning: attempt 1 of 10: sign-on command: status -1
✅  14.571 s: chip erase
avrdude warning: attempt 1 of 10: sign-on command: status -1
avrdude warning: attempt 2 of 10: sign-on command: status -1
✅  22.149 s: flash -U write/verify holes_rjmp_loops_49152B.hex
avrdude warning: attempt 1 of 10: sign-on command: status -1
avrdude warning: attempt 2 of 10: sign-on command: status -1
✅   5.787 s: flash -T write/verify holes_rjmp_loops_49152B.hex
✅   4.524 s: eeprom check whether programmer can flip 0s to 1s
✅   4.541 s: eeprom -U write/verify holes_pack_my_box_256B.hex
✅   4.646 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_256B.hex
✅   5.972 s: chip erase and spot check flash is actually erased
✅  14.542 s: spot check eeprom is erased, too
avrdude warning: attempt 1 of 10: sign-on command: status -1
avrdude warning: attempt 2 of 10: sign-on command: status -1
✅   4.571 s: usersig -T/-U write/read random_data_64B.bin

One or more AVRDUDE "-c nanoevery -p m4809 -P /dev/ttyU0 -r" tests failed. Do you want to retry this particular test? (y/n): n
```

MCUdude, 2024-01-31:
All good!

```
$ ./test-avrdude -d 0 -p "-c nanoevery -patmega4809 -r -P /dev/cu.usbmodem14101"
Testing avrdude version 7.2-20230720
Prepare "-c nanoevery -patmega4809 -r -P /dev/cu.usbmodem14101" and press 'enter' or 'space' to continue. Press any other key to skip
✅   3.433 s: fuse access: clear, set and read eesave fuse bit
✅   3.431 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
✅   3.657 s: chip erase
✅  10.398 s: flash -U write/verify holes_rjmp_loops_49152B.hex
✅   4.294 s: flash -T write/verify holes_rjmp_loops_49152B.hex
✅   3.613 s: eeprom check whether programmer can flip 0s to 1s
✅   3.872 s: eeprom -U write/verify holes_pack_my_box_256B.hex
✅   3.671 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_256B.hex
✅   4.507 s: chip erase and spot check flash is actually erased
✅   3.438 s: spot check eeprom is erased, too
✅   3.576 s: usersig -T/-U write/read random_data_64B.bin
```

mcuee, 2024-02-03:
@MCUdude 

It seems to me you are tesing an older version of avrdude (`avrdude version 7.2-20230720`).

I am having some issues here with latest git main under macOS. Please check on your side as well. Thanks.

* https://github.com/avrdudes/avrdude/discussions/1473#discussioncomment-8352833

```
mcuee@mcuees-Mac-mini tools % ./test-avrdude -v -p "-c nanoevery -p m4809 -P usb:2341:0058 -r" 
Testing avrdude version 7.2-20240201 (571a8063)
Prepare "-c nanoevery -p m4809 -P usb:2341:0058 -r" and press 'enter' or 'space' to continue. Press any other key to skip
✅   3.478 s: fuse access: clear, set and read eesave fuse bit
config eesave=eex_preserved # 1
✅   3.500 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
✅   3.639 s: chip erase
✅  10.579 s: flash -U write/verify holes_rjmp_loops_49152B.hex
✅   4.445 s: flash -T write/verify holes_rjmp_loops_49152B.hex
✅   3.628 s: eeprom check whether programmer can flip 0s to 1s
✅   3.636 s: eeprom -U write/verify holes_pack_my_box_256B.hex
✅   3.736 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_256B.hex
✅   4.604 s: chip erase and spot check flash is actually erased
❌   3.521 s: spot check eeprom is erased, too (failed command below)
$ avrdude -qq -c nanoevery -p m4809 -P usb:2341:0058 -r -l /tmp/test-avrdude.log.TmPKu4 -Ueeprom:v:./test_files/holes_eeprom_0xff_256B.hex
avrdude warning: verification mismatch
        device 0x50 != input 0xff at addr 0x0022 (error)
avrdude error: verification mismatch
✅   3.666 s: usersig -T/-U write/read random_data_64B.bin

One or more AVRDUDE "-c nanoevery -p m4809 -P usb:2341:0058 -r" tests failed. Do you want to retry this particular test? (y/n): n
```

On the other hand, the result seems to be expected because of the first step.

```
mcuee@mcuees-Mac-mini tools % avrdude -c nanoevery -p m4809 -P usb:2341:0058 -r -t
avrdude: touching serial port /dev/cu.usbmodem224401 at 1200 baud
avrdude: waiting for new port... using same port /dev/cu.usbmodem224401
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9651 (probably m4809)

avrdude: processing -t interactive terminal
avrdude> config eesave
config eesave=eex_preserved # 1
avrdude> dump eeprom
Reading | ################################################## | 100% 0.04 s 
0000  aa ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0020  ff ff 50 41 43 4b 20 4d  59 20 42 4f 58 20 57 49  |..PACK MY BOX WI|
0030  54 48 20 46 49 56 45 20  44 4f 5a 45 4e 20 4c 49  |TH FIVE DOZEN LI|
0040  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0050  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0060  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0070  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0080  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0090  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00a0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00b0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00c0  43 4b 20 4d 59 20 42 4f  58 20 57 49 54 48 20 46  |CK MY BOX WITH F|
00d0  49 56 45 20 44 4f 5a 45  4e 20 4c 49 51 55 ff ff  |IVE DOZEN LIQU..|
00e0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00f0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
avrdude> dump fuses
Reading | ################################################## | 100% 0.03 s 
0000  00 54 01 ff 00 ff 06 00  02 ff                    |.T........      |
avrdude> quit

avrdude done.  Thank you.
```

mcuee, 2024-02-03:
I get inconsistent results.

The following patch to the script seems to work once.

```
mcuee@mcuees-Mac-mini tools % git diff                                       
diff --git a/tools/test-avrdude b/tools/test-avrdude
index 85059976..85fcbbaa 100755
--- a/tools/test-avrdude
+++ b/tools/test-avrdude
@@ -410,6 +410,9 @@ for (( p=0; p<$arraylength; p++ )); do
     ######
     # Chip erase and verify
     #
+    specify="fuse access: read eesave fuse bit"
+    command=(${avrdude[@]} -T '"config eesave"')
+    execute "${command[@]}"
     specify="chip erase and spot check flash is actually erased"
     command=(${avrdude[@]} -e -Uflash:v:$tfiles/holes_flash_0xff_${FLASH_SIZE}B.hex)
     execute "${command[@]}"
mcuee@mcuees-Mac-mini tools % ./test-avrdude -v -p "-c nanoevery -p m4809 -P usb:2341:0058 -r" 
Testing avrdude version 7.2-20240201 (571a8063)
Prepare "-c nanoevery -p m4809 -P usb:2341:0058 -r" and press 'enter' or 'space' to continue. Press any other key to skip
✅   3.433 s: fuse access: clear, set and read eesave fuse bit
config eesave=eex_preserved # 1
✅   3.475 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
✅   3.612 s: chip erase
✅  10.559 s: flash -U write/verify holes_rjmp_loops_49152B.hex
✅   4.426 s: flash -T write/verify holes_rjmp_loops_49152B.hex
✅   3.607 s: eeprom check whether programmer can flip 0s to 1s
✅   3.608 s: eeprom -U write/verify holes_pack_my_box_256B.hex
✅   3.725 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_256B.hex
config eesave=eex_erased # 0
./test-avrdude: line 168: [[: 3.470: syntax error: invalid arithmetic operator (error token is ".470")
✅   4.567 s: chip erase and spot check flash is actually erased
✅   3.478 s: spot check eeprom is erased, too
✅   3.625 s: usersig -T/-U write/read random_data_64B.bin
```

After that it does not seem to work even with `-d 2`.
`-d 0` --> fuse bit is not correct
`-d 1` and `-d 2` -->fuse bit seems to be correct but no idea why it still failed.

```
bash-5.2$ git diff
diff --git a/tools/test-avrdude b/tools/test-avrdude
index 85059976..85fcbbaa 100755
--- a/tools/test-avrdude
+++ b/tools/test-avrdude
@@ -410,6 +410,9 @@ for (( p=0; p<$arraylength; p++ )); do
     ######
     # Chip erase and verify
     #
+    specify="fuse access: read eesave fuse bit"
+    command=(${avrdude[@]} -T '"config eesave"')
+    execute "${command[@]}"
     specify="chip erase and spot check flash is actually erased"
     command=(${avrdude[@]} -e -Uflash:v:$tfiles/holes_flash_0xff_${FLASH_SIZE}B.hex)
     execute "${command[@]}"
bash-5.2$ ./test-avrdude -d 0 -v -p "-c nanoevery -p m4809 -P usb:2341:0058 -r"
Testing avrdude version 7.2-20240201 (571a8063)
Prepare "-c nanoevery -p m4809 -P usb:2341:0058 -r" and press 'enter' or 'space' to continue. Press any other key to skip
✅   3.447 s: fuse access: clear, set and read eesave fuse bit
config eesave=eex_preserved # 1
✅   3.458 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
✅   3.604 s: chip erase
✅  10.540 s: flash -U write/verify holes_rjmp_loops_49152B.hex
✅   4.394 s: flash -T write/verify holes_rjmp_loops_49152B.hex
✅   3.585 s: eeprom check whether programmer can flip 0s to 1s
✅   3.575 s: eeprom -U write/verify holes_pack_my_box_256B.hex
[... 36 lines cut ...]

bash-5.2$ ./test-avrdude -d 2 -v -p "-c nanoevery -p m4809 -P usb:2341:0058 -r"
Testing avrdude version 7.2-20240201 (571a8063)
Prepare "-c nanoevery -p m4809 -P usb:2341:0058 -r" and press 'enter' or 'space' to continue. Press any other key to skip
✅   3.438 s: fuse access: clear, set and read eesave fuse bit
config eesave=eex_preserved # 1
✅   3.456 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
✅   3.612 s: chip erase
✅  10.548 s: flash -U write/verify holes_rjmp_loops_49152B.hex
✅   4.416 s: flash -T write/verify holes_rjmp_loops_49152B.hex
✅   3.603 s: eeprom check whether programmer can flip 0s to 1s
✅   3.595 s: eeprom -U write/verify holes_pack_my_box_256B.hex
✅   3.723 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_256B.hex
config eesave=eex_erased # 0
./test-avrdude: line 168: [[: 3.469: syntax error: invalid arithmetic operator (error token is ".469")
✅   4.568 s: chip erase and spot check flash is actually erased
❌   3.480 s: spot check eeprom is erased, too (failed command below)
$ avrdude -qq -c nanoevery -p m4809 -P usb:2341:0058 -r -l /tmp/test-avrdude.log.wfUidK -Ueeprom:v:./test_files/holes_eeprom_0xff_256B.hex
avrdude warning: verification mismatch
        device 0x50 != input 0xff at addr 0x0022 (error)
avrdude error: verification mismatch
✅   3.616 s: usersig -T/-U write/read random_data_64B.bin

One or more AVRDUDE "-c nanoevery -p m4809 -P usb:2341:0058 -r" tests failed. Do you want to retry this particular test? (y/n): n
```

mcuee, 2024-02-03:
@MCUdude 
Please help to carry out the test under your macOS system with latest git main. Thanks.

@askn37 and @stefanrueger 
You two seem to have some insights here. Please help to comment as well. Thanks.

askn37, 2024-02-03:
I will write down my environment and information about nanoevery.

```
askn@alicia avrdude_main % uname -a
Darwin alicia.local 23.2.0 Darwin Kernel Version 23.2.0: Wed Nov 15 21:53:34 PST 2023; root:xnu-10002.61.3~2/RELEASE_ARM64_T8103 arm64
askn@alicia avrdude_main % ./build_darwin/src/avrdude -cnanoevery -r -P /dev/cu.usbmodem2301 -pm4809 -v

avrdude: Version 7.2-20240201 (571a8063)
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is /Users/askn/Collaborator/avrdude_main/build_darwin/src/avrdude.conf
         User configuration file is /Users/askn/.avrduderc

avrdude: touching serial port /dev/cu.usbmodem2301 at 1200 baud
avrdude: waiting for new port... using same port /dev/cu.usbmodem2301
         Using port            : /dev/cu.usbmodem2301
         Using programmer      : nanoevery
         Programmer baud rate  : 115200
         AVR Part              : ATmega4809
         Programming modes     : UPDI, SPM
         Programmer Type       : JTAGMKII_UPDI
         Description           : JTAGv2 to UPDI bridge
         Main MCU HW version   : 1
         Main MCU FW version   : 1.07
         Sec. MCU HW version   : 1
         Sec. MCU FW version   : 6.07
         Serial number         : 00:00:00:00:00:00
avrdude: silicon revision: 0.1
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9651 (probably m4809)

avrdude done.  Thank you.

```

In this situation, no fault is found at this time. You will notice that each test portion takes an extra 0.3 to 0.8 seconds.

```
+ time ./test-avrdude -d 0 -e ../build_darwin/src/avrdude -p "-cnanoevery -r -P/dev/cu.usbmodem2301 -pm4809"
Testing ../build_darwin/src/avrdude version 7.2-20240201 (571a8063)
Prepare "-cnanoevery -r -P/dev/cu.usbmodem2301 -pm4809" and press 'enter' or 'space' to continue. Press any other key to skip
✅   3.779 s: fuse access: clear, set and read eesave fuse bit
✅   3.788 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
✅   4.010 s: chip erase
✅  11.320 s: flash -U write/verify holes_rjmp_loops_49152B.hex
✅   5.042 s: flash -T write/verify holes_rjmp_loops_49152B.hex
✅   3.925 s: eeprom check whether programmer can flip 0s to 1s
✅   3.924 s: eeprom -U write/verify holes_pack_my_box_256B.hex
✅   4.055 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_256B.hex
✅   5.271 s: chip erase and spot check flash is actually erased
✅   3.775 s: spot check eeprom is erased, too
✅   3.991 s: usersig -T/-U write/read random_data_64B.
```

mcuee, 2024-02-03:
Here is my setup, Mac Mini M1 2020 model.

Mac Mini M1 -- 7-port powered USB 3.0 hub -- 13-port powered USB 2.0 hub -- Arduino Nano Every (official version from Arduino)

```
bash-5.2$ pwd
/Users/mcuee/build/avr/avrdude_test/avrdude_main/tools
bash-5.2$ uname -a
Darwin mcuees-Mac-mini.lan 23.3.0 Darwin Kernel Version 23.3.0: Wed Dec 20 21:30:27 PST 2023; root:xnu-10002.81.5~7/RELEASE_ARM64_T8103 arm64
bash-5.2$ avrdude
Usage: avrdude [options]
Options:
  -p <partno>            Specify AVR device; -p ? lists all known parts
  -p <wildcard>/<flags>  Run developer options for matched AVR devices,
                         e.g., -p ATmega328P/s or /S for part definition
  -b <baudrate>          Override RS-232 baud rate
  -B <bitclock>          Specify bit clock period (us)
  -C <config-file>       Specify location of configuration file
  -C +<config-file>      Specify additional config file, can be repeated
  -N                     Do not load config file 
  -c <programmer>        Specify programmer; -c ? and -c ?type list all
  -c <wildcard>/<flags>  Run developer options for matched programmers,
                         e.g., -c 'ur*'/s for programmer info/definition
  -A                     Disable trailing-0xff removal for file/AVR read
  -D                     Disable auto erase for flash memory; implies -A
  -i <delay>             ISP Clock Delay [in microseconds]
  -P <port>              Connection; -P ?s or -P ?sa lists serial ones
  -r                     Reconnect to -P port after "touching" it; wait
                         400 ms for each -r; needed for some USB boards
  -F                     Override invalid signature or initial checks
  -e                     Perform a chip erase
  -O                     Perform RC oscillator calibration (see AVR053)
  -t                     Run an interactive terminal when it is its turn
  -T <terminal cmd line> Run terminal line when it is its turn
  -U <memstr>:r|w|v:<filename>[:format]
                         Carry out memory operation when it is its turn
                         Multiple -t, -T and -U options can be specified
  -n                     Do not write to the device whilst processing -U
  -V                     Do not automatically verify during -U
  -E <exitsp>[,<exitsp>] List programmer exit specifications
  -x <extended_param>    Pass <extended_param> to programmer, see -xhelp
  -v                     Verbose output; -v -v for more
  -q                     Quell progress output; -q -q for less
  -l logfile             Use logfile rather than stderr for diagnostics
  -?                     Display this usage

avrdude version 7.2-20240201 (571a8063), https://github.com/avrdudes/avrdude
bash-5.2$ ./test-avrdude -v -d 2 -p "-c nanoevery -p m4809 -P /dev/cu.usbmodem224401 -r" 
Testing avrdude version 7.2-20240201 (571a8063)
Prepare "-c nanoevery -p m4809 -P /dev/cu.usbmodem224401 -r" and press 'enter' or 'space' to continue. Press any other key to skip
✅   3.433 s: fuse access: clear, set and read eesave fuse bit
config eesave=eex_preserved # 1
✅   3.478 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
✅   3.618 s: chip erase
✅  10.542 s: flash -U write/verify holes_rjmp_loops_49152B.hex
✅   4.409 s: flash -T write/verify holes_rjmp_loops_49152B.hex
✅   3.597 s: eeprom check whether programmer can flip 0s to 1s
✅   3.610 s: eeprom -U write/verify holes_pack_my_box_256B.hex
✅   3.712 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_256B.hex
✅   4.570 s: chip erase and spot check flash is actually erased
❌   3.484 s: spot check eeprom is erased, too (failed command below)
$ avrdude -qq -c nanoevery -p m4809 -P /dev/cu.usbmodem224401 -r -l /tmp/test-avrdude.log.Ib3vUh -Ueeprom:v:./test_files/holes_eeprom_0xff_256B.hex
avrdude warning: verification mismatch
        device 0x50 != input 0xff at addr 0x0022 (error)
avrdude error: verification mismatch
✅   3.616 s: usersig -T/-U write/read random_data_64B.bin

One or more AVRDUDE "-c nanoevery -p m4809 -P /dev/cu.usbmodem224401 -r" tests failed. Do you want to retry this particular test? (y/n): n
```

askn37, 2024-02-03:
Oh, in my case it's not via a USB hub. Only an ammeter and an extension cable with a power switch are placed between it and the iMac. We have also found that the USB-to-serial converter becomes significantly less stable when running VMware Fusion 13.5 at the same time. When testing AVRDUDE, always stop the virtual environment.

mcuee, 2024-02-03:
I do not have virtual environment running under macOS.

Here is the test result when directly attach the Nano Every to the USB port of Mac Mini and it has the same issue. Strange.

```
bash-5.2$ avrdude -c nanoevery -p m4809 -P /dev/cu.usbmodem2101 -r -v

avrdude: Version 7.2-20240201 (571a8063)
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is /opt/homebrew/etc/avrdude.conf
         User configuration file is /Users/mcuee/.avrduderc
         User configuration file does not exist or is not a regular file, skipping

avrdude: touching serial port /dev/cu.usbmodem2101 at 1200 baud
avrdude: waiting for new port... using same port /dev/cu.usbmodem2101
         Using port            : /dev/cu.usbmodem2101
         Using programmer      : nanoevery
         Programmer baud rate  : 115200
         AVR Part              : ATmega4809
         Programming modes     : UPDI, SPM
         Programmer Type       : JTAGMKII_UPDI
         Description           : JTAGv2 to UPDI bridge
         Main MCU HW version   : 1
         Main MCU FW version   : 1.07
         Sec. MCU HW version   : 1
         Sec. MCU FW version   : 6.07
         Serial number         : 00:00:00:00:00:00
avrdude: silicon revision: 0.1
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9651 (probably m4809)

avrdude done.  Thank you.

bash-5.2$ ./test-avrdude -v -d 2 -p "-c nanoevery -p m4809 -P /dev/cu.usbmodem2101 -r" 
Testing avrdude version 7.2-20240201 (571a8063)
Prepare "-c nanoevery -p m4809 -P /dev/cu.usbmodem2101 -r" and press 'enter' or 'space' to continue. Press any other key to skip
✅   3.428 s: fuse access: clear, set and read eesave fuse bit
config eesave=eex_preserved # 1
✅   3.464 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
✅   3.606 s: chip erase
✅  10.490 s: flash -U write/verify holes_rjmp_loops_49152B.hex
✅   4.355 s: flash -T write/verify holes_rjmp_loops_49152B.hex
✅   3.566 s: eeprom check whether programmer can flip 0s to 1s
✅   3.582 s: eeprom -U write/verify holes_pack_my_box_256B.hex
✅   3.710 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_256B.hex
✅   4.498 s: chip erase and spot check flash is actually erased
❌   3.466 s: spot check eeprom is erased, too (failed command below)
$ avrdude -qq -c nanoevery -p m4809 -P /dev/cu.usbmodem2101 -r -l /tmp/test-avrdude.log.gDongC -Ueeprom:v:./test_files/holes_eeprom_0xff_256B.hex
avrdude warning: verification mismatch
        device 0x50 != input 0xff at addr 0x0022 (error)
avrdude error: verification mismatch
✅   3.606 s: usersig -T/-U write/read random_data_64B.bin

One or more AVRDUDE "-c nanoevery -p m4809 -P /dev/cu.usbmodem2101 -r" tests failed. Do you want to retry this particular test? (y/n): n
```

mcuee, 2024-02-03:
Just want to make sure my Nano Every is sitll working and here are the results under Windows 11 (no issues).

```
$ ./test-avrdude -e "./avrdude.exe" -d 0 -t . -v -p "-c nanoevery -p m4809 -P usb:2341:0058 -r"
Testing ./avrdude.exe version 7.2-20240201 (571a8063)
Prepare "-c nanoevery -p m4809 -P usb:2341:0058 -r" and press 'enter' or 'space' to continue. Press any other key to skip
✅   3.798 s: fuse access: clear, set and read eesave fuse bit
config eesave=eex_preserved # 1
✅   3.765 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
✅   3.884 s: chip erase
✅  11.285 s: flash -U write/verify holes_rjmp_loops_49152B.hex
✅   5.057 s: flash -T write/verify holes_rjmp_loops_49152B.hex
✅   3.962 s: eeprom check whether programmer can flip 0s to 1s
✅   3.930 s: eeprom -U write/verify holes_pack_my_box_256B.hex
✅   4.061 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_256B.hex
✅   5.171 s: chip erase and spot check flash is actually erased
✅   3.782 s: spot check eeprom is erased, too
✅   3.929 s: usersig -T/-U write/read random_data_64B.bin
```

stefanrueger, 2024-02-03:
I am not sure this is an issue with the test script. It looks more like the test script has uncovered a difficult to understand issue in the way the programmer writes fuses and/or waits insufficiently long after a CE. The next step would be to try to reproduce the issue with a single plain avrdude command. You can use `test-avrdude -l` to list the avrdude command lines that the test program issues; then remove the `-l logfile` parts of the avrdude commands and reduce the set of commands as much as you can while still observing the problem.

mcuee, 2024-02-03:
> I am not sure this is an issue with the test script. It looks more like the test script has uncovered a difficult to understand issue in the way the programmer writes fuses and/or waits [...]

Okay, this is a possibility as well. I'll see if I can reproduce the issue with a single plain avrdude command.

mcuee, 2024-02-04:
@stefanrueger 

BTW, the error only happens after the patch. But I really do not understand why that happens.

This bash version is from Homebrew.

```
 ./test-avrdude: line 168: [[: 3.451: syntax error: invalid arithmetic operator (error token is ".451")
``` 

```
bash-5.2$ git diff
diff --git a/tools/test-avrdude b/tools/test-avrdude
index 85059976..85fcbbaa 100755
--- a/tools/test-avrdude
+++ b/tools/test-avrdude
@@ -410,6 +410,9 @@ for (( p=0; p<$arraylength; p++ )); do
     ######
     # Chip erase and verify
     #
+    specify="fuse access: read eesave fuse bit"
+    command=(${avrdude[@]} -T '"config eesave"')
+    execute "${command[@]}"
     specify="chip erase and spot check flash is actually erased"
     command=(${avrdude[@]} -e -Uflash:v:$tfiles/holes_flash_0xff_${FLASH_SIZE}B.hex)
     execute "${command[@]}"
bash-5.2$ ./test-avrdude -d 0 -v -p "-c nanoevery -p m4809 -P usb:2341:0058 -r"
Testing avrdude version 7.2-20240201 (571a8063)
Prepare "-c nanoevery -p m4809 -P usb:2341:0058 -r" and press 'enter' or 'space' to continue. Press any other key to skip
✅   3.447 s: fuse access: clear, set and read eesave fuse bit
config eesave=eex_preserved # 1
✅   3.458 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
✅   3.604 s: chip erase
✅  10.540 s: flash -U write/verify holes_rjmp_loops_49152B.hex
✅   4.394 s: flash -T write/verify holes_rjmp_loops_49152B.hex
✅   3.585 s: eeprom check whether programmer can flip 0s to 1s
✅   3.575 s: eeprom -U write/verify holes_pack_my_box_256B.hex
✅   3.709 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_256B.hex
config eesave=eex_preserved # 1
./test-avrdude: line 168: [[: 3.451: syntax error: invalid arithmetic operator (error token is ".451")
✅   4.562 s: chip erase and spot check flash is actually erased
❌   3.465 s: spot check eeprom is erased, too (failed command below)
$ avrdude -qq -c nanoevery -p m4809 -P usb:2341:0058 -r -l /tmp/test-avrdude.log.yNTKYp -Ueeprom:v:./test_files/holes_eeprom_0xff_256B.hex
avrdude warning: verification mismatch
        device 0x50 != input 0xff at addr 0x0022 (error)
avrdude error: verification mismatch
✅   3.614 s: usersig -T/-U write/read random_data_64B.bin

One or more AVRDUDE "-c nanoevery -p m4809 -P usb:2341:0058 -r" tests failed. Do you want to retry this particular test?
```

askn37, 2024-02-04:
I updated the kernel, so please check again.

```
askn@alicia avrdude-test % uname -a
Darwin alicia.local 23.3.0 Darwin Kernel Version 23.3.0: Wed Dec 20 21:30:27 PST 2023; root:xnu-10002.81.5~7/RELEASE_ARM64_T8103 arm64

askn@alicia avrdude-test % bash --version
GNU bash, version 3.2.57(1)-release (arm64-apple-darwin23)
Copyright (C) 2007 Free Software Foundation, Inc.

askn@alicia avrdude-test % ./nanoevery.sh
/Users/askn/Collaborator/avrdude_main/tools
+ time ./test-avrdude -d 0 -e ../build_darwin/src/avrdude -p "-cnanoevery -r -P/dev/cu.usbmodem2301 -pm4809"
Testing ../build_darwin/src/avrdude version 7.2-20240201 (571a8063)
Prepare "-cnanoevery -r -P/dev/cu.usbmodem2301 -pm4809" and press 'enter' or 'space' to continue. Press any other key to skip
✅   3.783 s: fuse access: clear, set and read eesave fuse bit
✅   3.806 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
✅   4.011 s: chip erase
✅  11.326 s: flash -U write/verify holes_rjmp_loops_49152B.hex
✅   5.034 s: flash -T write/verify holes_rjmp_loops_49152B.hex
✅   3.947 s: eeprom check whether programmer can flip 0s to 1s
✅   3.948 s: eeprom -U write/verify holes_pack_my_box_256B.hex
✅   4.077 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_256B.hex
✅   5.309 s: chip erase and spot check flash is actually erased
✅   3.827 s: spot check eeprom is erased, too
✅   3.969 s: usersig -T/-U write/read random_data_64B.bin
       53.76 real         1.06 user         0.99 sys
```

There doesn't seem to be any difference.

mcuee, 2024-02-04:
> @stefanrueger
> 
> BTW, the following script error only happens after the patch. But I really do not understand why that happens. This bash version is from Homebrew.
> 
> ```
>  ./test-avrdude: line 168: [[: 3.451: syntax error: invalid arithmetic operator (error token is ".451")
> ```

Same thing happens under Ubuntu Linux 20.04.

```
mcuee@UbuntuSwift3 ~/build/avr/avrdude_main/tools (main)$ git diff
diff --git a/tools/test-avrdude b/tools/test-avrdude
index 85059976..ef397a6a 100755
--- a/tools/test-avrdude
+++ b/tools/test-avrdude
@@ -410,6 +410,10 @@ for (( p=0; p<$arraylength; p++ )); do
     ######
     # Chip erase and verify
     #
+    specify="fuse access: read eesave fuse bit"
+    command=(${avrdude[@]} -T '"config eesave"')
+    execute "${command[@]}"
+
     specify="chip erase and spot check flash is actually erased"
     command=(${avrdude[@]} -e -Uflash:v:$tfiles/holes_flash_0xff_${FLASH_SIZE}B.hex)
     execute "${command[@]}"
mcuee@UbuntuSwift3 ~/build/avr/avrdude_main/tools (main)$ ./test-avrdude -d 0 -v -p "-c nanoevery -p m4809 -P usb:2341:0058 -r"
Testing avrdude version 7.2-20240201 (571a8063)
Prepare "-c nanoevery -p m4809 -P usb:2341:0058 -r" and press 'enter' or 'space' to continue. Press any other key to skip
✅   3.266 s: fuse access: clear, set and read eesave fuse bit
config eesave=eex_preserved # 1
✅   3.263 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
✅   3.412 s: chip erase
✅  10.276 s: flash -U write/verify holes_rjmp_loops_49152B.hex
✅   4.127 s: flash -T write/verify holes_rjmp_loops_49152B.hex
✅   3.392 s: eeprom check whether programmer can flip 0s to 1s
✅   3.390 s: eeprom -U write/verify holes_pack_my_box_256B.hex
✅   3.512 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_256B.hex
config eesave=eex_erased # 0
./test-avrdude: line 168: [[: 3.256: syntax error: invalid arithmetic operator (error token is ".256")
✅   4.264 s: chip erase and spot check flash is actually erased
✅   3.271 s: spot check eeprom is erased, too
✅   3.408 s: usersig -T/-U write/read random_data_64B.bin
mcuee@UbuntuSwift3 ~/build/avr/avrdude_main/tools (main)$ bash --version
GNU bash, version 5.0.17(1)-release (x86_64-pc-linux-gnu)
Copyright (C) 2019 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>

This is free software; you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```

mcuee, 2024-02-04:
> I am not sure this is an issue with the test script. It looks more like the test script has uncovered a difficult to understand issue in the way the programmer writes fuses and/or waits [...]

@stefanrueger 

This is a one-line test and there is no issue under my Mac Mini M1 running macOS 14.3.

```
mcuee@mcuees-Mac-mini tools % 
./avrdude -c nanoevery -p m4809 -P usb:2341:0058 -r \
-T "config eesave=0; config eesave=1; config eesave" \
-T "config eesave=ee*erased; config eesave" \
-T "erase" -Ueeprom:v:./test_files/holes_eeprom_0xff_256B.hex
avrdude: touching serial port /dev/cu.usbmodem224401 at 1200 baud
avrdude: waiting for new port... using same port /dev/cu.usbmodem224401
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9651 (probably m4809)

avrdude: processing -T config eesave=0; config eesave=1; config eesave
config eesave=eex_preserved # 1

avrdude: processing -T config eesave=ee*erased; config eesave
config eesave=eex_erased # 0

avrdude: processing -T erase
erasing chip ...

avrdude: processing -U eeprom:v:./test_files/holes_eeprom_0xff_256B.hex:i
avrdude: verifying eeprom memory against ./test_files/holes_eeprom_0xff_256B.hex
Reading | ################################################## | 100% 0.01 s 
avrdude: 60 bytes of eeprom verified

avrdude done.  Thank you.
```

mcuee, 2024-02-04:
Even if I combined all of the commands in the same `-T` command, it is still okay.

```
mcuee@mcuees-Mac-mini tools % ./avrdude -c nanoevery -p m4809 -P usb:2341:0058 -r \
-T "config eesave=0; config eesave=1; config eesave; config eesave=ee*erased; erase" \
-Ueeprom:v:./test_files/holes_eeprom_0xff_256B.hex
avrdude: touching serial port /dev/cu.usbmodem224401 at 1200 baud
avrdude: waiting for new port... using same port /dev/cu.usbmodem224401
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9651 (probably m4809)

avrdude: processing -T config eesave=0; config eesave=1; config eesave; config eesave=ee*erased; erase
config eesave=eex_preserved # 1
erasing chip ...

avrdude: processing -U eeprom:v:./test_files/holes_eeprom_0xff_256B.hex:i
avrdude: verifying eeprom memory against ./test_files/holes_eeprom_0xff_256B.hex
Reading | ################################################## | 100% 0.01 s 
avrdude: 60 bytes of eeprom verified

avrdude done.  Thank you.
```

stefanrueger, 2024-02-04:
> Same thing happens under Ubuntu Linux 20.04.

Not in mine. I had to use the `dryrun` programmer though.

``` diff
diff --git a/tools/test-avrdude b/tools/test-avrdude
index 85059976..ef397a6a 100755
--- a/tools/test-avrdude
+++ b/tools/test-avrdude
@@ -410,6 +410,10 @@ for (( p=0; p<$arraylength; p++ )); do
     ######
     # Chip erase and verify
     #
+    specify="fuse access: read eesave fuse bit"
+    command=(${avrdude[@]} -T '"config eesave"')
+    execute "${command[@]}"
+
     specify="chip erase and spot check flash is actually erased"
     command=(${avrdude[@]} -e -Uflash:v:$tfiles/holes_flash_0xff_${FLASH_SIZE}B.hex)
     execute "${command[@]}"
```

Then
```
$ tools/test-avrdude -d 0 -v -p "-c dryrun -p m4809 -P usb:2341:0058 -r"
Testing avrdude version 7.2-20240201 (571a8063)
Prepare "-c dryrun -p m4809 -P usb:2341:0058 -r" and press 'enter' or 'space' to continue. Press any other key to skip
✅   0.064 s: fuse access: clear, set and read eesave fuse bit
config eesave=eex_preserved # 1
✅   0.065 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
✅   0.065 s: chip erase
✅   0.065 s: flash -U write/verify holes_rjmp_loops_49152B.hex
✅   0.066 s: flash -T write/verify holes_rjmp_loops_49152B.hex
✅   0.064 s: eeprom check whether programmer can flip 0s to 1s
✅   0.065 s: eeprom -U write/verify holes_pack_my_box_256B.hex
✅   0.065 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_256B.hex
config eesave=eex_erased # 0
✅   0.065 s: chip erase and spot check flash is actually erased
✅   0.064 s: spot check eeprom is erased, too
✅   0.065 s: usersig -T/-U write/read random_data_64B.bin
```

You can always try and quote the strings in line 168; and echo them after the intervention like so
``` diff
diff --git a/tools/test-avrdude b/tools/test-avrdude
index 85059976..afdb4217 100755
--- a/tools/test-avrdude
+++ b/tools/test-avrdude
@@ -165,7 +165,7 @@ execute () {
     echo "\$ ${command[@]}" | tr -s " "
     return 0;
   fi
-  [[ $programmer != -cdry* && $elapsed -ne -1 ]] && sleep "$delay"
+  [[ "$programmer" != -cdry* && "$elapsed" -ne -1 ]] && sleep "$delay"
   # These shenanigans keep stdout, stderr and the exit code of the command
   { read elapsed < <({ time { eval "$@"; echo $? >"$status"; } 2>&4 4>&-; } 4>&2 2>&1 >&3 3>&-); } 3>&1
   return $(cat "$status")
@@ -410,6 +410,11 @@ for (( p=0; p<$arraylength; p++ )); do
     ######
     # Chip erase and verify
     #
+    specify="fuse access: read eesave fuse bit"
+    command=(${avrdude[@]} -T '"config eesave"')
+    execute "${command[@]}"
+    echo $elapsed $programmer
+
     specify="chip erase and spot check flash is actually erased"
     command=(${avrdude[@]} -e -Uflash:v:$tfiles/holes_flash_0xff_${FLASH_SIZE}B.hex)
     execute "${command[@]}"
```

askn37, 2024-02-04:
Insert exit at the position before the test fails. That's `chip erase and spot check flash is actually erased`. Next, try dumping her EEPROM of the device. Next, try running the code from the failing test.

My guess is that the memory is not configured correctly for the next test, even if the previous test finished correctly.

mcuee, 2024-02-04:
@stefanrueger 

I think I have found the issue. It is due to the script -- the extra delay does not work for any of the real programmers. Once I change that line, it works fine.

```
mcuee@mcuees-Mac-mini tools % git diff
diff --git a/tools/test-avrdude b/tools/test-avrdude
index 85059976..86368904 100755
--- a/tools/test-avrdude
+++ b/tools/test-avrdude
@@ -165,7 +165,7 @@ execute () {
     echo "\$ ${command[@]}" | tr -s " "
     return 0;
   fi
-  [[ $programmer != -cdry* && $elapsed -ne -1 ]] && sleep "$delay"
+  sleep "$delay"
   # These shenanigans keep stdout, stderr and the exit code of the command
   { read elapsed < <({ time { eval "$@"; echo $? >"$status"; } 2>&4 4>&-; } 4>&2 2>&1 >&3 3>&-); } 3>&1
   return $(cat "$status")
@@ -410,6 +410,10 @@ for (( p=0; p<$arraylength; p++ )); do
     ######
     # Chip erase and verify
     #
+    specify="fuse access: read eesave fuse bit"
+    command=(${avrdude[@]} -T '"config eesave"')
+    execute "${command[@]}"
+    echo $elapsed $programmer
     specify="chip erase and spot check flash is actually erased"
     command=(${avrdude[@]} -e -Uflash:v:$tfiles/holes_flash_0xff_${FLASH_SIZE}B.hex)
     execute "${command[@]}"

mcuee@mcuees-Mac-mini tools % ./test-avrdude -e "./avrdude" -v -d 2 -p "-c nanoevery -p m4809 -P usb:2341:0058 -r" 
Testing ./avrdude version 7.2-20240201 (571a8063)
Prepare "-c nanoevery -p m4809 -P usb:2341:0058 -r" and press 'enter' or 'space' to continue. Press any other key to skip
✅   3.480 s: fuse access: clear, set and read eesave fuse bit
config eesave=eex_preserved # 1
✅   3.475 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
✅   3.607 s: chip erase
✅  10.543 s: flash -U write/verify holes_rjmp_loops_49152B.hex
✅   4.421 s: flash -T write/verify holes_rjmp_loops_49152B.hex
✅   3.581 s: eeprom check whether programmer can flip 0s to 1s
✅   3.598 s: eeprom -U write/verify holes_pack_my_box_256B.hex
✅   3.716 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_256B.hex
config eesave=eex_erased # 0
3.465 -cnanoevery
✅   4.570 s: chip erase and spot check flash is actually erased
✅   3.476 s: spot check eeprom is erased, too
✅   3.604 s: usersig -T/-U write/read random_data_64B.bin
```

stefanrueger, 2024-02-04:
What about if you use
```
[[ $programmer != -cdryrun && $programmer != -dryboot && $elapsed -ne -1 ]] && sleep "$delay"
```

mcuee, 2024-02-04:
> What about if you use
> 
> ```
> [[ $programmer != -cdryrun && $programmer != -dryboot && $elapsed -ne -1 ]] && sleep "$delay"
> ```

The script error is still there but then it seems to work. Strange.

Another thing, I also found that the tests may still fail after one or two runs. You can see that the 3rd run failed in this case.

I have to unplug and plug and then the first run is always okay (even if I use the previous patch without script error).

```
mcuee@mcuees-Mac-mini tools % git diff
diff --git a/tools/test-avrdude b/tools/test-avrdude
index 85059976..dd8838e2 100755
--- a/tools/test-avrdude
+++ b/tools/test-avrdude
@@ -165,7 +165,7 @@ execute () {
     echo "\$ ${command[@]}" | tr -s " "
     return 0;
   fi
-  [[ $programmer != -cdry* && $elapsed -ne -1 ]] && sleep "$delay"
+  [[ $programmer != -cdryrun && $programmer != -dryboot && $elapsed -ne -1 ]] && sleep "$delay"
   # These shenanigans keep stdout, stderr and the exit code of the command
   { read elapsed < <({ time { eval "$@"; echo $? >"$status"; } 2>&4 4>&-; } 4>&2 2>&1 >&3 3>&-); } 3>&1
   return $(cat "$status")
@@ -410,6 +410,10 @@ for (( p=0; p<$arraylength; p++ )); do
     ######
     # Chip erase and verify
     #
+    specify="fuse access: read eesave fuse bit"
+    command=(${avrdude[@]} -T '"config eesave"')
+    execute "${command[@]}"
+    echo $elapsed $programmer
     specify="chip erase and spot check flash is actually erased"
     command=(${avrdude[@]} -e -Uflash:v:$tfiles/holes_flash_0xff_${FLASH_SIZE}B.hex)
     execute "${command[@]}"

mcuee@mcuees-Mac-mini tools % ./test-avrdude -e "./avrdude" -v -d 2 -p "-c nanoevery -p m4809 -P usb:2341:0058 -r" 
Testing ./avrdude version 7.2-20240201 (571a8063)
Prepare "-c nanoevery -p m4809 -P usb:2341:0058 -r" and press 'enter' or 'space' to continue. Press any other key to skip
✅   3.441 s: fuse access: clear, set and read eesave fuse bit
config eesave=eex_preserved # 1
✅   3.471 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
✅   3.618 s: chip erase
✅  10.553 s: flash -U write/verify holes_rjmp_loops_49152B.hex
✅   4.395 s: flash -T write/verify holes_rjmp_loops_49152B.hex
✅   3.604 s: eeprom check whether programmer can flip 0s to 1s
✅   3.608 s: eeprom -U write/verify holes_pack_my_box_256B.hex
✅   3.691 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_256B.hex
config eesave=eex_erased # 0
3.452 -cnanoevery
./test-avrdude: line 168: [[: 3.452: syntax error: invalid arithmetic operator (error token is ".452")
✅   4.559 s: chip erase and spot check flash is actually erased
✅   3.465 s: spot check eeprom is erased, too
✅   3.617 s: usersig -T/-U write/read random_data_64B.bin

mcuee@mcuees-Mac-mini tools % ./test-avrdude -e "./avrdude" -v -d 2 -p "-c nanoevery -p m4809 -P usb:2341:0058 -r" 
Testing ./avrdude version 7.2-20240201 (571a8063)
Prepare "-c nanoevery -p m4809 -P usb:2341:0058 -r" and press 'enter' or 'space' to continue. Press any other key to skip
✅   3.439 s: fuse access: clear, set and read eesave fuse bit
config eesave=eex_preserved # 1
✅   3.477 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
✅   3.625 s: chip erase
✅  10.546 s: flash -U write/verify holes_rjmp_loops_49152B.hex
✅   4.414 s: flash -T write/verify holes_rjmp_loops_49152B.hex
✅   3.593 s: eeprom check whether programmer can flip 0s to 1s
✅   3.596 s: eeprom -U write/verify holes_pack_my_box_256B.hex
✅   3.715 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_256B.hex
config eesave=eex_erased # 0
3.463 -cnanoevery
./test-avrdude: line 168: [[: 3.463: syntax error: invalid arithmetic operator (error token is ".463")
✅   4.563 s: chip erase and spot check flash is actually erased
✅   3.485 s: spot check eeprom is erased, too
✅   3.612 s: usersig -T/-U write/read random_data_64B.bin

mcuee@mcuees-Mac-mini tools % ./test-avrdude -e "./avrdude" -v -d 2 -p "-c nanoevery -p m4809 -P usb:2341:0058 -r" 
Testing ./avrdude version 7.2-20240201 (571a8063)
Prepare "-c nanoevery -p m4809 -P usb:2341:0058 -r" and press 'enter' or 'space' to continue. Press any other key to skip
✅   3.436 s: fuse access: clear, set and read eesave fuse bit
config eesave=eex_preserved # 1
✅   3.463 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
✅   3.618 s: chip erase
✅  10.554 s: flash -U write/verify holes_rjmp_loops_49152B.hex
✅   4.428 s: flash -T write/verify holes_rjmp_loops_49152B.hex
✅   3.598 s: eeprom check whether programmer can flip 0s to 1s
✅   3.602 s: eeprom -U write/verify holes_pack_my_box_256B.hex
✅   3.710 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_256B.hex
config eesave=eex_preserved # 1
3.470 -cnanoevery
./test-avrdude: line 168: [[: 3.470: syntax error: invalid arithmetic operator (error token is ".470")
✅   4.564 s: chip erase and spot check flash is actually erased
❌   3.472 s: spot check eeprom is erased, too (failed command below)
$ ./avrdude -qq -c nanoevery -p m4809 -P usb:2341:0058 -r -l /tmp/test-avrdude.log.tjhgng -Ueeprom:v:./test_files/holes_eeprom_0xff_256B.hex
avrdude warning: verification mismatch
        device 0x50 != input 0xff at addr 0x0022 (error)
avrdude error: verification mismatch
✅   3.620 s: usersig -T/-U write/read random_data_64B.bin

One or more AVRDUDE "-c nanoevery -p m4809 -P usb:2341:0058 -r" tests failed. Do you want to retry this particular test? (y/n): n
```

mcuee, 2024-02-04:
The following is without script error, first run is okay and second run fails.

```
mcuee@mcuees-Mac-mini tools % 
./test-avrdude -e "./avrdude" -v -d 2 -p "-c nanoevery -p m4809 -P usb:2341:0058 -r" 
Testing ./avrdude version 7.2-20240201 (571a8063)
Prepare "-c nanoevery -p m4809 -P usb:2341:0058 -r" and press 'enter' or 'space' to continue. Press any other key to skip
✅   3.481 s: fuse access: clear, set and read eesave fuse bit
config eesave=eex_preserved # 1
✅   3.466 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
✅   3.623 s: chip erase
✅  10.560 s: flash -U write/verify holes_rjmp_loops_49152B.hex
✅   4.417 s: flash -T write/verify holes_rjmp_loops_49152B.hex
✅   3.597 s: eeprom check whether programmer can flip 0s to 1s
✅   3.587 s: eeprom -U write/verify holes_pack_my_box_256B.hex
✅   3.707 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_256B.hex
config eesave=eex_erased # 0
3.468 -cnanoevery
✅   4.562 s: chip erase and spot check flash is actually erased
✅   3.482 s: spot check eeprom is erased, too
✅   3.617 s: usersig -T/-U write/read random_data_64B.bin

mcuee@mcuees-Mac-mini tools % 
./test-avrdude -e "./avrdude" -v -d 2 -p "-c nanoevery -p m4809 -P usb:2341:0058 -r" 
Testing ./avrdude version 7.2-20240201 (571a8063)
Prepare "-c nanoevery -p m4809 -P usb:2341:0058 -r" and press 'enter' or 'space' to continue. Press any other key to skip
✅   3.469 s: fuse access: clear, set and read eesave fuse bit
config eesave=eex_preserved # 1
✅   3.458 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
✅   3.617 s: chip erase
✅  10.556 s: flash -U write/verify holes_rjmp_loops_49152B.hex
✅   4.402 s: flash -T write/verify holes_rjmp_loops_49152B.hex
✅   3.602 s: eeprom check whether programmer can flip 0s to 1s
✅   3.598 s: eeprom -U write/verify holes_pack_my_box_256B.hex
✅   3.710 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_256B.hex
config eesave=eex_preserved # 1
3.456 -cnanoevery
✅   4.571 s: chip erase and spot check flash is actually erased
❌   3.469 s: spot check eeprom is erased, too (failed command below)
$ ./avrdude -qq -c nanoevery -p m4809 -P usb:2341:0058 -r -l /tmp/test-avrdude.log.tnRQtg -Ueeprom:v:./test_files/holes_eeprom_0xff_256B.hex
avrdude warning: verification mismatch
        device 0x50 != input 0xff at addr 0x0022 (error)
avrdude error: verification mismatch
✅   3.614 s: usersig -T/-U write/read random_data_64B.bin

One or more AVRDUDE "-c nanoevery -p m4809 -P usb:2341:0058 -r" tests failed. Do you want to retry this particular test? (y/n): n
```

askn37, 2024-02-04:
nanoevery (i.e. variants of JTAG2UPDI) poses a potential problem with the following tests:

```
-cnanoevery -r -pm4809 -T 'config eesave=ee*erased
```

Running this with `-vvvv` gives the following result:

```
avrdude: jtagmkII_write_byte(.., fuse5, 0x0, ...)
avrdude: jtagmkII_write_byte(): sending write memory command: 
avrdude: jtagmkII_send(): sending 11 bytes
avrdude: ser_send: . [1b] . [0f] . [00] . [0b] . [00] . [00] . [00] . [0e] . [04] . [b2] . [01] . [00] . [00] . [00] . [85] . [12] . [00] . [00] . [c8] { [7b] . [b4]
avrdude: jtagmkII_recv():
avrdude: ser_recv: . [1b]
avrdude: ser_recv: . [0f]
avrdude: ser_recv: . [00]
avrdude: ser_recv: . [01]
avrdude: ser_recv: . [00]
avrdude: ser_recv: . [00]
avrdude: ser_recv: . [00]
avrdude: ser_recv: . [0e]
avrdude: ser_recv: . [80]
avrdude: ser_recv: . [c0]
avrdude: ser_recv: . [b3]

avrdude: jtagmkII_recv(): got message seqno 15 (command_sequence == 15)
avrdude: jtagmkII_recv: . [80]

Raw message:
0x80 
OK

avrdude: jtagmkII_program_disable(): Sending leave progmode command: 
avrdude: jtagmkII_send(): sending 1 bytes
avrdude: ser_send: . [1b] . [10] . [00] . [01] . [00] . [00] . [00] . [0e] . [15] . [2e] . [9a]
avrdude: jtagmkII_recv():
avrdude: ser_recv: . [1b]
avrdude: ser_recv: . [10]
avrdude: ser_recv: . [00]
avrdude: ser_recv: . [01]
avrdude: ser_recv: . [00]
avrdude: ser_recv: . [00]
avrdude: ser_recv: . [00]
avrdude: ser_recv: . [0e]
avrdude: ser_recv: . [80]
avrdude: ser_recv: . [0a]
avrdude: ser_recv: Y [59]

avrdude: jtagmkII_recv(): got message seqno 16 (command_sequence == 16)
avrdude: jtagmkII_recv: . [80]

Raw message:
0x80 
OK
```

I write the FUSE and immediately LEAVE it, but the unmodified JTAG2UPDI implementation requires a real delay during this time.

Writes a FUSE and LEAVEs immediately, but the unmodified JTAG2UPDI implementation requires an actual delay during this time. If you can at least insert verification here, the problem is likely to disappear. (AVR will block reads until the FUSE is truly rewritten.)

mcuee, 2024-02-04:
@askn37 

Please help to provide a patch for me to test. Thanks.

askn37, 2024-02-04:
How about this?

```diff
diff --git a/tools/test-avrdude b/tools/test-avrdude
index 85059976..1ad5d23b 100755
--- a/tools/test-avrdude
+++ b/tools/test-avrdude
@@ -326,7 +326,7 @@ for (( p=0; p<$arraylength; p++ )); do
 
       if [ -n "$EE_SIZE" ]; then
         specify="fuse access: set eesave fusebit to delete EEPROM on chip erase"
-        command=(${avrdude[@]} -T '"config eesave=ee*erased"')
+        command=(${avrdude[@]} -T '"config eesave=ee*erased; config eesave"')
         execute "${command[@]}" > $outfile
         sed -e/^config/d -e/"> "/d $outfile > ${outfile}-2
         result [[ ! -s ${outfile}-2  '&&' ! -s $logfile ]]
```

mcuee, 2024-02-04:
> How about this?
>
> ```diff
> diff --git a/tools/test-avrdude b/tools/test-avrdude
> index 85059976..1ad5d23b 100755
> --- a/tools/test-avrdude
> +++ b/tools/test-avrdude
> @@ -326,7 +326,7 @@ for (( p=0; [...]

Great. Now I get consistent passing results, even with multiple run.

```
mcuee@mcuees-Mac-mini tools % git diff
diff --git a/tools/test-avrdude b/tools/test-avrdude
index 85059976..0e7d481f 100755
--- a/tools/test-avrdude
+++ b/tools/test-avrdude
@@ -165,7 +165,8 @@ execute () {
     echo "\$ ${command[@]}" | tr -s " "
     return 0;
   fi
-  [[ $programmer != -cdry* && $elapsed -ne -1 ]] && sleep "$delay"
+  #[[ $programmer != -cdryrun && $programmer != -dryboot && $elapsed -ne -1 ]] && sleep "$delay"
+  sleep "$delay"
   # These shenanigans keep stdout, stderr and the exit code of the command
   { read elapsed < <({ time { eval "$@"; echo $? >"$status"; } 2>&4 4>&-; } 4>&2 2>&1 >&3 3>&-); } 3>&1
   return $(cat "$status")
@@ -326,7 +327,7 @@ for (( p=0; p<$arraylength; p++ )); do
 
       if [ -n "$EE_SIZE" ]; then
         specify="fuse access: set eesave fusebit to delete EEPROM on chip erase"
-        command=(${avrdude[@]} -T '"config eesave=ee*erased"')
+        command=(${avrdude[@]} -T '"config eesave=ee*erased; config eesave"')
         execute "${command[@]}" > $outfile
         sed -e/^config/d -e/"> "/d $outfile > ${outfile}-2
         result [[ ! -s ${outfile}-2  '&&' ! -s $logfile ]]
@@ -410,6 +411,10 @@ for (( p=0; p<$arraylength; p++ )); do
     ######
     # Chip erase and verify
     #
+    specify="fuse access: read eesave fuse bit"
+    command=(${avrdude[@]} -T '"config eesave"')
+    execute "${command[@]}"
+    echo $elapsed $programmer
     specify="chip erase and spot check flash is actually erased"
     command=(${avrdude[@]} -e -Uflash:v:$tfiles/holes_flash_0xff_${FLASH_SIZE}B.hex)
     execute "${command[@]}"
mcuee@mcuees-Mac-mini tools % ./test-avrdude -e "./avrdude" -v -d 4 -p "-c jtag2updi -p m4809 -P usb:2341:0058 -r"  
Testing ./avrdude version 7.2-20240201 (571a8063)
Prepare "-c jtag2updi -p m4809 -P usb:2341:0058 -r" and press 'enter' or 'space' to continue. Press any other key to skip
✅   2.969 s: fuse access: clear, set and read eesave fuse bit
config eesave=eex_preserved # 1
✅   2.958 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
config eesave=eex_erased # 0
✅   3.114 s: chip erase
✅  10.042 s: flash -U write/verify holes_rjmp_loops_49152B.hex
✅   3.915 s: flash -T write/verify holes_rjmp_loops_49152B.hex
✅   3.083 s: eeprom check whether programmer can flip 0s to 1s
✅   3.100 s: eeprom -U write/verify holes_pack_my_box_256B.hex
✅   3.200 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_256B.hex
config eesave=eex_erased # 0
2.948 -cjtag2updi
✅   4.062 s: chip erase and spot check flash is actually erased
✅   2.969 s: spot check eeprom is erased, too
✅   3.103 s: usersig -T/-U write/read random_data_64B.bin
mcuee@mcuees-Mac-mini tools % ./test-avrdude -e "./avrdude" -v -d 1 -p "-c nanoevery -p m4809 -P usb:2341:0058 -r" 
Testing ./avrdude version 7.2-20240201 (571a8063)
Prepare "-c nanoevery -p m4809 -P usb:2341:0058 -r" and press 'enter' or 'space' to continue. Press any other key to skip
✅   3.478 s: fuse access: clear, set and read eesave fuse bit
config eesave=eex_preserved # 1
✅   3.456 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
config eesave=eex_erased # 0
✅   3.613 s: chip erase
✅  10.557 s: flash -U write/verify holes_rjmp_loops_49152B.hex
✅   4.416 s: flash -T write/verify holes_rjmp_loops_49152B.hex
✅   3.606 s: eeprom check whether programmer can flip 0s to 1s
✅   3.608 s: eeprom -U write/verify holes_pack_my_box_256B.hex
✅   3.724 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_256B.hex
config eesave=eex_erased # 0
3.458 -cnanoevery
✅   4.566 s: chip erase and spot check flash is actually erased
✅   3.470 s: spot check eeprom is erased, too
✅   3.625 s: usersig -T/-U write/read random_data_64B.bin
mcuee@mcuees-Mac-mini tools % ./test-avrdude -e "./avrdude" -v -d 0 -p "-c nanoevery -p m4809 -P usb:2341:0058 -r" 
Testing ./avrdude version 7.2-20240201 (571a8063)
Prepare "-c nanoevery -p m4809 -P usb:2341:0058 -r" and press 'enter' or 'space' to continue. Press any other key to skip
✅   3.433 s: fuse access: clear, set and read eesave fuse bit
config eesave=eex_preserved # 1
✅   3.457 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
config eesave=eex_erased # 0
✅   3.605 s: chip erase
✅  10.535 s: flash -U write/verify holes_rjmp_loops_49152B.hex
✅   4.386 s: flash -T write/verify holes_rjmp_loops_49152B.hex
✅   3.580 s: eeprom check whether programmer can flip 0s to 1s
✅   3.589 s: eeprom -U write/verify holes_pack_my_box_256B.hex
✅   3.706 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_256B.hex
config eesave=eex_erased # 0
3.446 -cnanoevery
✅   4.565 s: chip erase and spot check flash is actually erased
✅   3.463 s: spot check eeprom is erased, too
✅   3.595 s: usersig -T/-U write/read random_data_64B.bin
```

stefanrueger, 2024-02-04:
> invalid arithmetic operator

Try `[[ $programmer != -cdryrun && $programmer != -dryboot && $elapsed != -1 ]] && sleep "$delay"`

I get it now. Your intervention does not call the `result` sub-routine, which resets `$elapsed` to -999, an integer. The `execute` subroutine sets elapsed to a floating point, which is what some `bash` complain about and some not when used with the integer `-ne` operator. Some older `bash` scripts are unable to do wildcard string match, so ` != -cdry*` should be avoided to allow older bash versions use the test script.

stefanrueger, 2024-02-04:
> Now I get consistent passing results, even with multiple run.

Wonderful. So we have an *explanation* for the behaviour with nanoevery. @mcuee @askn37 What's to do? Is the consensus that this is an error in the firmware? Should AVRDUDE change its implementation work around this FW bug? If so, how? I suggest *not* changing the test script. This has, after all, indicated that *something was wrong* even though it took the team a while to figure out what.

mcuee, 2024-02-04:
I do not know how to fix avrdude. 

With the following patch, I have tested the commands x 3 times and it works fine. I also tested with `-d 1` and `-d 2`, no more script error. And all 5 runs work fine.

I will reduce the patch to the minimum to see how it goes.

```
mcuee@mcuees-Mac-mini tools % git diff
diff --git a/tools/test-avrdude b/tools/test-avrdude
index 85059976..9207044d 100755
--- a/tools/test-avrdude
+++ b/tools/test-avrdude
@@ -165,7 +165,7 @@ execute () {
     echo "\$ ${command[@]}" | tr -s " "
     return 0;
   fi
-  [[ $programmer != -cdry* && $elapsed -ne -1 ]] && sleep "$delay"
+  [[ $programmer != -cdryrun && $programmer != -dryboot && $elapsed != -1 ]] && sleep "$delay"
   # These shenanigans keep stdout, stderr and the exit code of the command
   { read elapsed < <({ time { eval "$@"; echo $? >"$status"; } 2>&4 4>&-; } 4>&2 2>&1 >&3 3>&-); } 3>&1
   return $(cat "$status")
@@ -326,7 +326,7 @@ for (( p=0; p<$arraylength; p++ )); do
 
       if [ -n "$EE_SIZE" ]; then
         specify="fuse access: set eesave fusebit to delete EEPROM on chip erase"
-        command=(${avrdude[@]} -T '"config eesave=ee*erased"')
+        command=(${avrdude[@]} -T '"config eesave=ee*erased; config eesave"')
         execute "${command[@]}" > $outfile
         sed -e/^config/d -e/"> "/d $outfile > ${outfile}-2
         result [[ ! -s ${outfile}-2  '&&' ! -s $logfile ]]
@@ -410,6 +410,11 @@ for (( p=0; p<$arraylength; p++ )); do
     ######
     # Chip erase and verify
     #
+    specify="fuse access: read eesave fuse bit"
+    command=(${avrdude[@]} -T '"config eesave"')
+    execute "${command[@]}"
+    result [ $? == 0 ]
+    echo $elapsed $programmer
     specify="chip erase and spot check flash is actually erased"
     command=(${avrdude[@]} -e -Uflash:v:$tfiles/holes_flash_0xff_${FLASH_SIZE}B.hex)
     execute "${command[@]}"

mcuee@mcuees-Mac-mini tools % ./test-avrdude -e "./avrdude" -v -d 0 -p "-c nanoevery -p m4809 -P usb:2341:0058 -r" 
Testing ./avrdude version 7.2-20240201 (571a8063)
Prepare "-c nanoevery -p m4809 -P usb:2341:0058 -r" and press 'enter' or 'space' to continue. Press any other key to skip
✅   3.439 s: fuse access: clear, set and read eesave fuse bit
config eesave=eex_preserved # 1
✅   3.441 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
config eesave=eex_erased # 0
✅   3.590 s: chip erase
✅  10.531 s: flash -U write/verify holes_rjmp_loops_49152B.hex
✅   4.403 s: flash -T write/verify holes_rjmp_loops_49152B.hex
✅   3.571 s: eeprom check whether programmer can flip 0s to 1s
✅   3.579 s: eeprom -U write/verify holes_pack_my_box_256B.hex
✅   3.703 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_256B.hex
config eesave=eex_erased # 0
✅   3.459 s: fuse access: read eesave fuse bit
-999 -cnanoevery
✅   4.551 s: chip erase and spot check flash is actually erased
✅   3.479 s: spot check eeprom is erased, too
✅   3.594 s: usersig -T/-U write/read random_data_64B.bin
```

stefanrueger, 2024-02-04:
When you use ` [[ $programmer != -cdryrun && $programmer != -dryboot && $elapsed != -1 ]] && sleep "$delay"` there is then no need to call `result` as the sometimes troublesome `$elapsed -ne -1` integer test has been replaced with the always correct `$elapsed != -1` test.

mcuee, 2024-02-04:
Here is the minimum patch which works fine.

```
mcuee@mcuees-Mac-mini tools % git diff
diff --git a/tools/test-avrdude b/tools/test-avrdude
index 85059976..65247803 100755
--- a/tools/test-avrdude
+++ b/tools/test-avrdude
@@ -165,7 +165,7 @@ execute () {
     echo "\$ ${command[@]}" | tr -s " "
     return 0;
   fi
-  [[ $programmer != -cdry* && $elapsed -ne -1 ]] && sleep "$delay"
+  [[ $programmer != -cdryrun && $programmer != -dryboot && $elapsed != -1 ]] && sleep "$delay"
   # These shenanigans keep stdout, stderr and the exit code of the command
   { read elapsed < <({ time { eval "$@"; echo $? >"$status"; } 2>&4 4>&-; } 4>&2 2>&1 >&3 3>&-); } 3>&1
   return $(cat "$status")
@@ -326,7 +326,7 @@ for (( p=0; p<$arraylength; p++ )); do
 
       if [ -n "$EE_SIZE" ]; then
         specify="fuse access: set eesave fusebit to delete EEPROM on chip erase"
-        command=(${avrdude[@]} -T '"config eesave=ee*erased"')
+        command=(${avrdude[@]} -T '"config eesave=ee*erased; config eesave"')
         execute "${command[@]}" > $outfile
         sed -e/^config/d -e/"> "/d $outfile > ${outfile}-2
         result [[ ! -s ${outfile}-2  '&&' ! -s $logfile ]]

mcuee@mcuees-Mac-mini tools % ./test-avrdude -e "./avrdude" -v -d 1 -p "-c nanoevery -p m4809 -P usb:2341:0058 -r" 
Testing ./avrdude version 7.2-20240201 (571a8063)
Prepare "-c nanoevery -p m4809 -P usb:2341:0058 -r" and press 'enter' or 'space' to continue. Press any other key to skip
✅   3.442 s: fuse access: clear, set and read eesave fuse bit
config eesave=eex_preserved # 1
✅   3.464 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
config eesave=eex_erased # 0
✅   3.629 s: chip erase
✅  10.560 s: flash -U write/verify holes_rjmp_loops_49152B.hex
✅   4.405 s: flash -T write/verify holes_rjmp_loops_49152B.hex
✅   3.599 s: eeprom check whether programmer can flip 0s to 1s
✅   3.604 s: eeprom -U write/verify holes_pack_my_box_256B.hex
✅   3.723 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_256B.hex
✅   4.566 s: chip erase and spot check flash is actually erased
✅   3.469 s: spot check eeprom is erased, too
✅   3.617 s: usersig -T/-U write/read random_data_64B.bin

mcuee@mcuees-Mac-mini tools % ./test-avrdude -e "./avrdude" -v -d 0 -p "-c nanoevery -p m4809 -P usb:2341:0058 -r" 
Testing ./avrdude version 7.2-20240201 (571a8063)
Prepare "-c nanoevery -p m4809 -P usb:2341:0058 -r" and press 'enter' or 'space' to continue. Press any other key to skip
✅   3.422 s: fuse access: clear, set and read eesave fuse bit
config eesave=eex_preserved # 1
✅   3.450 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
config eesave=eex_erased # 0
✅   3.602 s: chip erase
✅  10.546 s: flash -U write/verify holes_rjmp_loops_49152B.hex
✅   4.392 s: flash -T write/verify holes_rjmp_loops_49152B.hex
✅   3.575 s: eeprom check whether programmer can flip 0s to 1s
✅   3.585 s: eeprom -U write/verify holes_pack_my_box_256B.hex
✅   3.699 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_256B.hex
✅   4.551 s: chip erase and spot check flash is actually erased
✅   3.471 s: spot check eeprom is erased, too
✅   3.601 s: usersig -T/-U write/read random_data_64B.bin

mcuee@mcuees-Mac-mini tools % ./test-avrdude -e "./avrdude" -v -d 0 -p "-c nanoevery -p m4809 -P usb:2341:0058 -r" 
Testing ./avrdude version 7.2-20240201 (571a8063)
Prepare "-c nanoevery -p m4809 -P usb:2341:0058 -r" and press 'enter' or 'space' to continue. Press any other key to skip
✅   3.443 s: fuse access: clear, set and read eesave fuse bit
config eesave=eex_preserved # 1
✅   3.453 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
config eesave=eex_erased # 0
✅   3.598 s: chip erase
✅  10.538 s: flash -U write/verify holes_rjmp_loops_49152B.hex
✅   4.406 s: flash -T write/verify holes_rjmp_loops_49152B.hex
✅   3.593 s: eeprom check whether programmer can flip 0s to 1s
✅   3.574 s: eeprom -U write/verify holes_pack_my_box_256B.hex
✅   3.713 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_256B.hex
✅   4.536 s: chip erase and spot check flash is actually erased
✅   3.446 s: spot check eeprom is erased, too
✅   3.605 s: usersig -T/-U write/read random_data_64B.bin

mcuee@mcuees-Mac-mini tools % ./test-avrdude -e "./avrdude" -v -d 0 -p "-c nanoevery -p m4809 -P usb:2341:0058 -r" 
Testing ./avrdude version 7.2-20240201 (571a8063)
Prepare "-c nanoevery -p m4809 -P usb:2341:0058 -r" and press 'enter' or 'space' to continue. Press any other key to skip
✅   3.427 s: fuse access: clear, set and read eesave fuse bit
config eesave=eex_preserved # 1
✅   3.446 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
config eesave=eex_erased # 0
✅   3.600 s: chip erase
✅  10.531 s: flash -U write/verify holes_rjmp_loops_49152B.hex
✅   4.390 s: flash -T write/verify holes_rjmp_loops_49152B.hex
✅   3.583 s: eeprom check whether programmer can flip 0s to 1s
✅   3.582 s: eeprom -U write/verify holes_pack_my_box_256B.hex
✅   3.700 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_256B.hex
✅   4.550 s: chip erase and spot check flash is actually erased
✅   3.469 s: spot check eeprom is erased, too
✅   3.589 s: usersig -T/-U write/read random_data_64B.bin
```

stefanrueger, 2024-02-04:
I think we definitely should *not* make that change in the test script
``` diff
-        command=(${avrdude[@]} -T '"config eesave=ee*erased"')
+        command=(${avrdude[@]} -T '"config eesave=ee*erased; config eesave"')
```
as the original version has *uncovered* a problem in the nanoevery FW and/or avrdude. Uncovering problems is *good*.

stefanrueger, 2024-02-04:
Making that change has proven @askn37 's hypothesis correct. @askn37 The question remains what should we do about avrdude? Can it provide a workaround for the problem that does not unnecessarily slow down programming?

mcuee, 2024-02-04:
> I think we definitely should _not_ make that change in the test script
>
> ```diff
> -        command=(${avrdude[@]} -T '"config eesave=ee*erased"')
> +        command=(${avrdude[@]} -T '"config [...]

@stefanrueger 

I see. In that case, we may want to remove the similar thing (`config eesave` and `config wdton`) in the following lines.

```
    #####
    # Fuse test (bootloaders usually cannot set fuses)
    #
    if [[ $is_bootloader -ne 1 && $fusetest -eq 1 ]]; then
      if [ -n "$EE_SIZE" ]; then
        specify="fuse access: clear, set and read eesave fuse bit"
        command=(${avrdude[@]} -T '"config eesave=0; config eesave=1; config eesave"')
      else
        specify="fuse access: clear, set and read wdton fuse bit"
        command=(${avrdude[@]} -T '"config wdton=0; config wdton=1; config wdton"')
      fi
```

askn37, 2024-02-04:
In my opinion, the original JTAG2UPDI has the following limitations:

- The last operation performed before ending the session must not be a write or erase. That operation may be lost. Many users may not be aware of this, as this is typically a write-verify pair. Adding a delay after the session ends does not solve the problem. This also applies to Nano Every.
- Any process that manipulates the status LED after a series of sessions has completed conflicts with starting a new session. The built-in timer malfunctions and times out, making it impossible to continue. To avoid this, a 1-2 second delay is required after the previous session ends. This is not the case with Nano Every, but it has a similar timer clearing process, so adding a short delay is recommended.

mcuee, 2024-02-04:
> I write the FUSE and immediately LEAVE it, but the unmodified JTAG2UPDI implementation requires a real delay during this time.
> 
> Writes a FUSE and LEAVEs immediately, but the unmodified JTAG2UPDI implementation requires an actual delay during this time.

@askn37 and @stefanrueger 

Would you please provide a patch to addd a delay when writing the fuse using jtag2updi? Maybe that is the right approach for avrdude 7.3 release.

Later @askn37 can probably help to provide a way to identify patched jtag2updi version which does not suffer this FW issue. Then we can either have delay (unpatched version) and no delay (patched version).

@stefanrueger 
I think this will not really slow down much of the programming since fuse is seldom touched.

mcuee, 2024-02-04:
> In my opinion, the original JTAG2UPDI has the following limitations:
>
> * The last operation performed before ending the session must not be a write or erase. That operation may be lost. Many users [...]

@askn37 

Wow, you really know the ins-and-outs of jtag2updi. Just wondering if you can provide a PR for avrdude to fix the issues with the various jtag2updi implementations out there.
1) official jtag2updi implementation
2) Arduino Nano Every
3) Nano 4808, usually using CH340 and an 8051 MCU (this one seems to be pretty okay)

askn37, 2024-02-04:
Maybe AVR Dragon has similar characteristics?

For the first issue, it seems best to patch stk500v2.c. If the last command executed before ser_close was a write/erase, further reading the corresponding last byte of memory or EEPROM should provide a sufficient delay of tens of milliseconds.

The second problem requires a lazy insert after the ser_close. How much is needed can only be determined empirically. I'm wondering if I need this just to get my test script to complete.

stefanrueger, 2024-02-04:
@askn37 AVR Dragon and nanoevery alread have a recently inserted sleep in _close() of jtagmkII.c:
https://github.com/avrdudes/avrdude/blob/571a8063b9258cf6d361a02abd4387e55dc28af8/src/jtagmkII.c#L1805-L1808

Is that sufficient?

>  If the last command executed before ser_close was a write/erase, further reading the corresponding last byte

@askn37 Could you please propose an issue and a PR? I don't have the programmers or parts to test.

askn37, 2024-02-04:
> AVR Dragon and nanoevery alread have a recently inserted sleep in _close() of jtagmkII.c:

On NanoEvery, that delay bypasses the next check in the firmware.
In other words, UPDI mode cannot be exited until 500ms has passed after passing through CMND_LEAVE_PROGMODE.

https://github.com/arduino/ArduinoCore-megaavr/blob/5e639ee40afa693354d3d056ba7fb795a8948c11/firmwares/MuxTO/MuxTO.ino#L168-L173

```
    // updi_mode cannot last more than 1 minute; in that case, break forcibly
    if ((updi_mode_end != 0 && (millis() - updi_mode_end) > 500) || ((millis() - updi_mode_start) > 60000)) {
      updi_mode = false;
      baudrate = -1;
      return;
    }
```

https://github.com/arduino/ArduinoCore-megaavr/blob/5e639ee40afa693354d3d056ba7fb795a8948c11/firmwares/MuxTO/MuxTO.ino#L199-L202

```
      case JTAG2::CMND_LEAVE_PROGMODE:
        JTAG2::leave_progmode();
        updi_mode_end = millis();
        break;
```

A delay is required before JTAG2::leave_progmode() to avoid FUSE write loss. Enough time for UPDI to read one byte of memory.

mcuee, 2024-02-04:
https://github.com/avrdudes/avrdude/issues/1649 discovered two issues. PR #1657 by @stefanrueger fixes the script error issue. 
* #1657 

Hopefully @askn37 can help to fix the fuse writing issue in another PR.

mcuee, 2024-02-04:
> https://github.com/arduino/ArduinoCore-megaavr/blob/5e639ee40afa693354d3d056ba7fb795a8948c11/firmwares/MuxTO/MuxTO.ino#L199-L202
>
> ```
>       case JTAG2::CMND_LEAVE_PROGMODE: [...]

So this one will be a simple PR, right?

I think it may be good to have two PRs to address the two different issues.

mcuee, 2024-02-04:
> On NanoEvery, that delay bypasses the next check in the firmware. In other words, UPDI mode cannot be exited until 500ms has passed after passing through CMND_LEAVE_PROGMODE. [...]

I am sorry but I do not quite understand this issue. Please help to elaborate. Thanks.

Or maybe @stefanrueger can chime in as well. Thanks.

mcuee, 2024-02-04:
> As far as using the test script is concerned, the problem only occurs in one place: writing FUSE (and disconnecting without verifying), so would it be enough to implement just one taint flag?

I think the idea is not just to deal with the current test script, but rather the underlying issues of avrdude when dealing with different jtag2updi implementations. But if the changes are too complicated, maybe we can deal with the easier one first for avrdude 7.3 release.

askn37, 2024-02-04:
A very simple implementation of the taint_write flag.

```diff
diff --git a/src/jtagmkII.c b/src/jtagmkII.c
index 0aecb7ce..7ad73c84 100644
--- a/src/jtagmkII.c
+++ b/src/jtagmkII.c
@@ -87,6 +87,9 @@ struct pdata
   /* Major firmware version (needed for Xmega programming) */
   unsigned int fwver;
 
+  /* Remember if the last execution was a memory write */
+  int taint_write;
+
 #define FLAGS32_INIT_SMC      1 // Part will undergo chip erase
 #define FLAGS32_WRITE         2 // At least one write operation specified
   // Couple of flag bits for AVR32 programming
@@ -142,6 +145,7 @@ static int jtagmkII_paged_write(const PROGRAMMER *pgm, const AVRPART *p, const A
                                 unsigned int addr, unsigned int n_bytes);
 static unsigned char jtagmkII_mtype(const PROGRAMMER *pgm, const AVRPART *p, unsigned long addr);
 static unsigned int jtagmkII_memaddr(const PROGRAMMER *pgm, const AVRPART *p, const AVRMEM *m, unsigned long addr);
+static int jtagmkII_updi_term_keep_alive(const PROGRAMMER *pgm, const AVRPART *p_unused);
 
 // AVR32
 #define ERROR_SAB 0xFFFFFFFF
@@ -850,6 +854,7 @@ static int jtagmkII_chip_erase(const PROGRAMMER *pgm, const AVRPART *p) {
   if (!(p->prog_modes & (PM_PDI | PM_UPDI)))
       pgm->initialize(pgm, p);
 
+  PDATA(pgm)->taint_write = 1;
   return 0;
 }
 
@@ -1105,6 +1110,13 @@ static int jtagmkII_program_disable(const PROGRAMMER *pgm) {
   if (!PDATA(pgm)->prog_enabled)
     return 0;
 
+  // Delay early termination to ensure memory rewrite operations.
+  if (PDATA(pgm)->taint_write) {
+    pmsg_notice2("jtagmkII_program_disable(): Delay after writing: ");
+    jtagmkII_updi_term_keep_alive(pgm, /* unused */ (AVRPART*)NULL);
+    jtagmkII_updi_term_keep_alive(pgm, /* unused */ (AVRPART*)NULL);
+  }
+
   buf[0] = CMND_LEAVE_PROGMODE;
   pmsg_notice2("jtagmkII_program_disable(): "
     "Sending leave progmode command: ");
@@ -1128,6 +1140,7 @@ static int jtagmkII_program_disable(const PROGRAMMER *pgm) {
     return -1;
   }
 
+  PDATA(pgm)->taint_write  = 0;
   PDATA(pgm)->prog_enabled = 0;
   (void)jtagmkII_reset(pgm, 0x01);
 
@@ -2012,6 +2025,7 @@ static int jtagmkII_paged_write(const PROGRAMMER *pgm, const AVRPART *p, const A
   free(cmd);
   serial_recv_timeout = otimeout;
 
+  PDATA(pgm)->taint_write = 1;
   return n_bytes;
 }
 
@@ -2102,6 +2116,7 @@ static int jtagmkII_paged_load(const PROGRAMMER *pgm, const AVRPART *p, const AV
   }
   serial_recv_timeout = otimeout;
 
+  PDATA(pgm)->taint_write = 0;
   return n_bytes;
 }
 
@@ -2290,6 +2305,7 @@ retry:
     *value = resp[1];
 
   free(resp);
+  PDATA(pgm)->taint_write = 0;
   return 0;
 
 fail:
@@ -2400,6 +2416,7 @@ retry:
   }
 
   free(resp);
+  PDATA(pgm)->taint_write = 1;
   return 0;
 
 fail:
```

This is still very experimental. I can't reproduce the problem with my nanoevery, but I can assure you it was harmless.

To create a delay, reuse jtagmkII_updi_term_keep_alive and issue a GET_SYNC. I still don't know if once is enough or twice is not enough. You can replace this with usleep if you know the delay time you want.

For now, I'm avoiding using commands that read memory, as I need to store many parameters in cookies when writing to memory.

You may also need to update the taint_write flag elsewhere. Currently, it is only inserted where it is deemed necessary.

mcuee, 2024-02-04:
@askn37 

Thanks a lot. Your patch works well. I apply your patch on top of PR #1657 and here are the results.

```
mcuee@mcuees-Mac-mini tools % git diff
diff --git a/src/jtagmkII.c b/src/jtagmkII.c
index 0aecb7ce..7ad73c84 100644
--- a/src/jtagmkII.c
+++ b/src/jtagmkII.c
@@ -87,6 +87,9 @@ struct pdata
   /* Major firmware version (needed for Xmega programming) */
   unsigned int fwver;
 
+  /* Remember if the last execution was a memory write */
+  int taint_write;
+
 #define FLAGS32_INIT_SMC      1 // Part will undergo chip erase
 #define FLAGS32_WRITE         2 // At least one write operation specified
   // Couple of flag bits for AVR32 programming
@@ -142,6 +145,7 @@ static int jtagmkII_paged_write(const PROGRAMMER *pgm, const AVRPART *p, const A
                                 unsigned int addr, unsigned int n_bytes);
 static unsigned char jtagmkII_mtype(const PROGRAMMER *pgm, const AVRPART *p, unsigned long addr);
 static unsigned int jtagmkII_memaddr(const PROGRAMMER *pgm, const AVRPART *p, const AVRMEM *m, unsigned long addr);
+static int jtagmkII_updi_term_keep_alive(const PROGRAMMER *pgm, const AVRPART *p_unused);
 
 // AVR32
 #define ERROR_SAB 0xFFFFFFFF
@@ -850,6 +854,7 @@ static int jtagmkII_chip_erase(const PROGRAMMER *pgm, const AVRPART *p) {
   if (!(p->prog_modes & (PM_PDI | PM_UPDI)))
       pgm->initialize(pgm, p);
 
+  PDATA(pgm)->taint_write = 1;
   return 0;
 }
 
@@ -1105,6 +1110,13 @@ static int jtagmkII_program_disable(const PROGRAMMER *pgm) {
   if (!PDATA(pgm)->prog_enabled)
     return 0;
 
+  // Delay early termination to ensure memory rewrite operations.
+  if (PDATA(pgm)->taint_write) {
+    pmsg_notice2("jtagmkII_program_disable(): Delay after writing: ");
+    jtagmkII_updi_term_keep_alive(pgm, /* unused */ (AVRPART*)NULL);
+    jtagmkII_updi_term_keep_alive(pgm, /* unused */ (AVRPART*)NULL);
+  }
+
   buf[0] = CMND_LEAVE_PROGMODE;
   pmsg_notice2("jtagmkII_program_disable(): "
     "Sending leave progmode command: ");
@@ -1128,6 +1140,7 @@ static int jtagmkII_program_disable(const PROGRAMMER *pgm) {
     return -1;
   }
 
+  PDATA(pgm)->taint_write  = 0;
   PDATA(pgm)->prog_enabled = 0;
   (void)jtagmkII_reset(pgm, 0x01);
 
@@ -2012,6 +2025,7 @@ static int jtagmkII_paged_write(const PROGRAMMER *pgm, const AVRPART *p, const A
   free(cmd);
   serial_recv_timeout = otimeout;
 
+  PDATA(pgm)->taint_write = 1;
   return n_bytes;
 }
 
@@ -2102,6 +2116,7 @@ static int jtagmkII_paged_load(const PROGRAMMER *pgm, const AVRPART *p, const AV
   }
   serial_recv_timeout = otimeout;
 
+  PDATA(pgm)->taint_write = 0;
   return n_bytes;
 }
 
@@ -2290,6 +2305,7 @@ retry:
     *value = resp[1];
 
   free(resp);
+  PDATA(pgm)->taint_write = 0;
   return 0;
 
 fail:
@@ -2400,6 +2416,7 @@ retry:
   }
 
   free(resp);
+  PDATA(pgm)->taint_write = 1;
   return 0;
 
 fail:

mcuee@mcuees-Mac-mini tools % ./test-avrdude -e "./avrdude_issue1649" -v -d 2 -p "-c nanoevery -p m4809 -P usb:2341:0058 -r" 
Testing ./avrdude_issue1649 version 7.2-20240204 (28589ca6)
Prepare "-c nanoevery -p m4809 -P usb:2341:0058 -r" and press 'enter' or 'space' to continue. Press any other key to skip
✅   3.438 s: fuse access: clear, set and read eesave fuse bit
✅   3.468 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
✅   3.621 s: chip erase
✅  10.536 s: flash -U write/verify holes_rjmp_loops_49152B.hex
✅   4.417 s: flash -T write/verify holes_rjmp_loops_49152B.hex
✅   3.612 s: eeprom check whether programmer can flip 0s to 1s
✅   3.600 s: eeprom -U write/verify holes_pack_my_box_256B.hex
✅   3.713 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_256B.hex
✅   4.564 s: chip erase and spot check flash is actually erased
✅   3.492 s: spot check eeprom is erased, too
✅   3.612 s: usersig -T/-U write/read random_data_64B.bin

mcuee@mcuees-Mac-mini tools % ./test-avrdude -e "./avrdude_issue1649" -v -d 0 -p "-c nanoevery -p m4809 -P usb:2341:0058 -r" 
Testing ./avrdude_issue1649 version 7.2-20240204 (28589ca6)
Prepare "-c nanoevery -p m4809 -P usb:2341:0058 -r" and press 'enter' or 'space' to continue. Press any other key to skip
✅   3.443 s: fuse access: clear, set and read eesave fuse bit
✅   3.449 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
✅   3.604 s: chip erase
✅  10.539 s: flash -U write/verify holes_rjmp_loops_49152B.hex
✅   4.396 s: flash -T write/verify holes_rjmp_loops_49152B.hex
✅   3.569 s: eeprom check whether programmer can flip 0s to 1s
✅   3.560 s: eeprom -U write/verify holes_pack_my_box_256B.hex
✅   3.699 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_256B.hex
✅   4.539 s: chip erase and spot check flash is actually erased
✅   3.459 s: spot check eeprom is erased, too
✅   3.605 s: usersig -T/-U write/read random_data_64B.bin

mcuee@mcuees-Mac-mini tools % ./test-avrdude -e "./avrdude_issue1649" -v -d 0 -p "-c nanoevery -p m4809 -P usb:2341:0058 -r" 
Testing ./avrdude_issue1649 version 7.2-20240204 (28589ca6)
Prepare "-c nanoevery -p m4809 -P usb:2341:0058 -r" and press 'enter' or 'space' to continue. Press any other key to skip
✅   3.435 s: fuse access: clear, set and read eesave fuse bit
✅   3.449 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
✅   3.588 s: chip erase
✅  10.542 s: flash -U write/verify holes_rjmp_loops_49152B.hex
✅   4.401 s: flash -T write/verify holes_rjmp_loops_49152B.hex
✅   3.576 s: eeprom check whether programmer can flip 0s to 1s
✅   3.573 s: eeprom -U write/verify holes_pack_my_box_256B.hex
✅   3.701 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_256B.hex
✅   4.561 s: chip erase and spot check flash is actually erased
✅   3.472 s: spot check eeprom is erased, too
✅   3.611 s: usersig -T/-U write/read random_data_64B.bin
```

askn37, 2024-02-04:
> I am sorry but I do not quite understand this issue. Please help to elaborate. Thanks

The updi_mode flag controls the operational state of ATSAMD11, which is responsible for programming. If false, behaves as USB serial and monitors 1200bps touch. If true, behaves as JTAG2UPDI variant.

Except for physical reset, the condition for changing this flag from true to false is that at least 500ms has passed since the execution of CMND_LEAVE_PROGMODE (that is, the "jtagmkII_program_disable(): Send Leave progmode command:" part of AVRDUDE)

Therefore, AVRDUDE requires a 500ms delay to re-enable her 1200bps touch.

mcuee, 2024-02-04:
@askn37 

I think you can create a PR based on your current patch. I am not good at programming so I will not be a good reviewer. But @stefanrueger is an excellent reviewer for codes. 

I am not a good reviewer for more complex code changes but I am a pretty good tester for the projects I am working on. :-)

Reference: an example from libusb project
https://github.com/libusb/libusb/pull/1406#issuecomment-1918633626

askn37, 2024-02-04:
This is the same idea as chanting a mantra before shutting down an old *UNIX machine. That was the landing spot.

```
sync; sync; sync
```

mcuee, 2024-02-04:
> This is the same idea as chanting a mantra before shutting down an old *UNIX machine. That was the landing spot.
> 
> ```
> sync; sync; sync
> ```

Haha, still the same for new Linux/BSD machines. :-)

I used old things like on the dumb terminals of DEC Ultrix (but MicroVMS/Vax VMS more often), Sun OS and Sun Solaris in the universities. In those cases, I did not get the chance to see "sync; sync; sync".

But for Linux, I played with old Linux distro back in 1998 on a DEC 486 machine and then later Intel Pentium 350/Celeron 400 PC for 1999 to 2002. Then I became a real Linux user since 2005 with Ubuntu 5.04 (still mainly using Ubuntu/Debian Linux now even though I play with other Linux distros like Fedora or Arch). I am OS neutural so I use Windows/macOS/Linux and play with BSDs and other alternative OS as part of testing for various open source projects like libusb, avrdude and OpenOCD.

mcuee, 2024-02-04:
> @askn37
>
> Wow, you really know the ins-and-outs of jtag2updi. Just wondering if you can provide a PR for avrdude to fix the issues with the various jtag2updi implementations out there.
>
> 1. official [...]

FYI, no issues with Nano 4808 with git main (no need any patches).

```
mcuee@mcuees-Mac-mini tools % pwd
/Users/mcuee/build/avr/avrdude_test/avrdude_main/tools

mcuee@mcuees-Mac-mini tools % ./test-avrdude -v -d 0 -p "-c jtag2updi -p m4808 -P /dev/cu.usbserial-22440"     
Testing avrdude version 7.2-20240201 (571a8063)
Prepare "-c jtag2updi -p m4808 -P /dev/cu.usbserial-22440" and press 'enter' or 'space' to continue. Press any other key to skip
✅   0.862 s: fuse access: clear, set and read eesave fuse bit
config eesave=eex_preserved # 1
✅   0.864 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
✅   1.014 s: chip erase
✅   4.812 s: flash -U write/verify holes_rjmp_loops_49152B.hex
✅   2.329 s: flash -T write/verify holes_rjmp_loops_49152B.hex
✅   0.957 s: eeprom check whether programmer can flip 0s to 1s
✅   0.953 s: eeprom -U write/verify holes_pack_my_box_256B.hex
✅   1.030 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_256B.hex
✅   2.492 s: chip erase and spot check flash is actually erased
✅   0.891 s: spot check eeprom is erased, too
✅   0.992 s: usersig -T/-U write/read random_data_64B.bin

mcuee@mcuees-Mac-mini tools % ./test-avrdude -v -d 1 -p "-c jtag2updi -p m4808 -P /dev/cu.usbserial-22440" 
Testing avrdude version 7.2-20240201 (571a8063)
Prepare "-c jtag2updi -p m4808 -P /dev/cu.usbserial-22440" and press 'enter' or 'space' to continue. Press any other key to skip
✅   0.860 s: fuse access: clear, set and read eesave fuse bit
config eesave=eex_preserved # 1
✅   0.889 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
✅   1.030 s: chip erase
✅   4.827 s: flash -U write/verify holes_rjmp_loops_49152B.hex
✅   2.348 s: flash -T write/verify holes_rjmp_loops_49152B.hex
✅   0.974 s: eeprom check whether programmer can flip 0s to 1s
✅   0.967 s: eeprom -U write/verify holes_pack_my_box_256B.hex
✅   1.036 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_256B.hex
✅   2.508 s: chip erase and spot check flash is actually erased
✅   0.906 s: spot check eeprom is erased, too
✅   1.004 s: usersig -T/-U write/read random_data_64B.bin

mcuee@mcuees-Mac-mini tools % ./test-avrdude -v -d 0 -p "-c jtag2updi -p m4808 -P /dev/cu.usbserial-22440" 
Testing avrdude version 7.2-20240201 (571a8063)
Prepare "-c jtag2updi -p m4808 -P /dev/cu.usbserial-22440" and press 'enter' or 'space' to continue. Press any other key to skip
✅   0.859 s: fuse access: clear, set and read eesave fuse bit
config eesave=eex_preserved # 1
✅   0.868 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
✅   1.014 s: chip erase
✅   4.811 s: flash -U write/verify holes_rjmp_loops_49152B.hex
✅   2.335 s: flash -T write/verify holes_rjmp_loops_49152B.hex
✅   0.959 s: eeprom check whether programmer can flip 0s to 1s
✅   0.961 s: eeprom -U write/verify holes_pack_my_box_256B.hex
✅   1.029 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_256B.hex
✅   2.488 s: chip erase and spot check flash is actually erased
✅   0.891 s: spot check eeprom is erased, too
✅   0.988 s: usersig -T/-U write/read random_data_64B.bin
```

mcuee, 2024-02-04:
@askn37 

And no regressions with your patch with Nano 4808.

```
mcuee@mcuees-Mac-mini tools % pwd                                                                                                  
/Users/mcuee/build/avr/avrdude_test/avrdude_sr/tools
mcuee@mcuees-Mac-mini tools % ls                                                                                                   
atdf-to-avrdude.xslt   avrdude_issue1649.conf get-dw-params.xsl      get-stk600-devices.xsl
avrdude.conf           bootloader-hash        get-hv-params.xsl      test-avrdude
avrdude_issue1649      build-mingw32.sh       get-stk600-cards.xsl   test_files

mcuee@mcuees-Mac-mini tools % ./test-avrdude -e "./avrdude_issue1649" -v -d 0 -p "-c jtag2updi -p m4808 -P /dev/cu.usbserial-22440"
Testing ./avrdude_issue1649 version 7.2-20240204 (28589ca6)
Prepare "-c jtag2updi -p m4808 -P /dev/cu.usbserial-22440" and press 'enter' or 'space' to continue. Press any other key to skip
✅   0.855 s: fuse access: clear, set and read eesave fuse bit
✅   0.869 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
✅   1.018 s: chip erase
✅   4.805 s: flash -U write/verify holes_rjmp_loops_49152B.hex
✅   2.331 s: flash -T write/verify holes_rjmp_loops_49152B.hex
✅   0.955 s: eeprom check whether programmer can flip 0s to 1s
✅   0.957 s: eeprom -U write/verify holes_pack_my_box_256B.hex
✅   1.028 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_256B.hex
✅   2.486 s: chip erase and spot check flash is actually erased
✅   0.887 s: spot check eeprom is erased, too
✅   0.986 s: usersig -T/-U write/read random_data_64B.bin

mcuee@mcuees-Mac-mini tools % ./test-avrdude -e "./avrdude_issue1649" -v -d 0 -p "-c jtag2updi -p m4808 -P /dev/cu.usbserial-22440"
Testing ./avrdude_issue1649 version 7.2-20240204 (28589ca6)
Prepare "-c jtag2updi -p m4808 -P /dev/cu.usbserial-22440" and press 'enter' or 'space' to continue. Press any other key to skip
✅   0.854 s: fuse access: clear, set and read eesave fuse bit
✅   0.870 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
✅   1.016 s: chip erase
✅   4.804 s: flash -U write/verify holes_rjmp_loops_49152B.hex
✅   2.330 s: flash -T write/verify holes_rjmp_loops_49152B.hex
✅   0.956 s: eeprom check whether programmer can flip 0s to 1s
✅   0.957 s: eeprom -U write/verify holes_pack_my_box_256B.hex
✅   1.027 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_256B.hex
✅   2.485 s: chip erase and spot check flash is actually erased
✅   0.888 s: spot check eeprom is erased, too
✅   0.987 s: usersig -T/-U write/read random_data_64B.bin

mcuee@mcuees-Mac-mini tools % ./test-avrdude -e "./avrdude_issue1649" -v -d 0 -p "-c jtag2updi -p m4808 -P /dev/cu.usbserial-22440"
Testing ./avrdude_issue1649 version 7.2-20240204 (28589ca6)
Prepare "-c jtag2updi -p m4808 -P /dev/cu.usbserial-22440" and press 'enter' or 'space' to continue. Press any other key to skip
✅   0.856 s: fuse access: clear, set and read eesave fuse bit
✅   0.868 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
✅   1.016 s: chip erase
✅   4.815 s: flash -U write/verify holes_rjmp_loops_49152B.hex
✅   2.332 s: flash -T write/verify holes_rjmp_loops_49152B.hex
✅   0.960 s: eeprom check whether programmer can flip 0s to 1s
✅   0.964 s: eeprom -U write/verify holes_pack_my_box_256B.hex
✅   1.034 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_256B.hex
✅   2.494 s: chip erase and spot check flash is actually erased
✅   0.888 s: spot check eeprom is erased, too
✅   0.986 s: usersig -T/-U write/read random_data_64B.bin
```

[10 more replies]

PR #1658 (2024-02-05): Update test script
Fixes #1649