avrdudes/avrdude #1653: libelf deprecation on macOS

iMichka, 2024-02-01:
Hello

I am a maintainer of the Homebrew project and wanted to give a small heads up: we would like to deprecate libelf on macOS.

The upstream `libelf` website is gone, the remaining source code can only be through archive.org ... (since at least 2019, which was 5 years ago ...). avrdude is one of the last projects that is using `libelf` in macOS.

Are there any plans for a replacement solution on your side?

mcuee, 2024-02-02:
@iMichka 

What is the replacement library recommended by Homebrew project?

mcuee, 2024-02-02:
As of now, I do not know of any plans to use an alternative library for the avrdude project.

For Windows MSVC build, we have a local version of libelf.
https://github.com/avrdudes/libelf

mcuee, 2024-02-02:
Google says a potential "replacement" of libelf is elfutils. But then Homebrew says it is Linux only.
https://formulae.brew.sh/formula/elfutils
https://github.com/Homebrew/homebrew-core/blob/1cde8401d8be5f28d74a402afe67fd52ac0575ed/Formula/e/elfutils.rb

mcuee, 2024-02-02:
MSYS2 mingw uses legacy libelf.
https://packages.msys2.org/package/mingw-w64-x86_64-libelf

Debian -- using elfutils to provide libelf
https://packages.debian.org/search?keywords=libelf

FreeBSD/OpenBSD -- using elftoolchain to provide libelf
https://wiki.freebsd.org/LibElf
https://man.openbsd.org/elf
https://sourceforge.net/p/elftoolchain/wiki/Home/

NetBSD seems to be still using legacy libelf.
https://ftp.netbsd.org/pub/NetBSD/NetBSD-current/pkgsrc/devel/libelf/index.html
https://pkgsrc.se/devel/libelf

mcuee, 2024-02-02:
Looks like there is no solution for macOS now. So I think this may not be an easy task.

dl8dtl, 2024-02-02:
Macports have a port of elftoolchain.
FreeBSD has a port of elfutils, so it's not quite clear why it is said to be "Linux only".
Other than this: the usual reason for deprecating something is that there's a better replacement available, rather than something not being developed anymore. There are many things that are no longer being developed since they are fairly complete, and just don't need further development. Take TaylorUUCP for an example: every MacOS these days still comes with it. ;-)

stefanrueger, 2024-02-03:
Thanks for the advance warning @iMichka. I agree with @dl8dtl that it seems unfortunate to deprecate elflib without an obvious replacement. AVR elf files won't have changed in the last decade or so, so that a library reading them doesn't need to be "new".

@mcuee @MCUdude Is is possible to drop the reliance of MacOS on libelf before  the new AVRDUDE release next week?

mcuee, 2024-02-03:
@stefanrueger 

I think we should stick to libelf on macOS with avrdude 7.3 release.

Then we can explore the alternative for avrdude 7.4 release.

mcuee, 2024-02-03:
> > Macports have a port of elftoolchain.
> 
> We need to check if it works for the purpose of avrdude though. If it works, then the solution is to ask Homebrew to provide elftoolchain.

I just took a look at elftoolchain macports and it is not in a good shape -- failed under Apple Silicon and almost no users (2 downloads in 30 days).
https://ports.macports.org/port/elftoolchain/details/

macports libelf is quite healthy (no issues with Apple Silicon, 641 downloads in 30 days)
https://ports.macports.org/port/libelf/details/

I think we can rule out the use of elftoolchain provided libelf as a candidate to replace libelf.

mcuee, 2024-02-03:
As for elfutils, I do not see much efforts to port it to macOS. Google only finds one with the last commit date on 2019-Dec-24.
https://github.com/kdab-hotspot/elfutils

mcuee, 2024-02-07:
There is of course another option to bundle libelf with avrdude.

For example, we may use the following libelf for Windows and make it compatible with macOS.
https://github.com/avrdudes/libelf

MCUdude, 2024-02-08:
@iMichka I have brew/avrdude related question. Avrdude 7.3 has a new dependency; libserialport. It's being used for automatic serial port discovery. However, Brew doesn't build Avrdude with libserialport. How can we change this?

```
% avrdude -curclock -patmega328p -Pch340
avrdude error: avrdude built without libserialport support; please compile again with libserialport installed
avrdude warning: serial adapter ch340 not connected to host
avrdude OS error: cannot open port ch340: No such file or directory
avrdude error: unable to open port ch340 for programmer urclock
avrdude error: avrdude built without libserialport support; please compile again with libserialport installed

avrdude done.  Thank you.
```

mcuee, 2024-02-09:
@MCUdude 

The forumla needs to be updated -- it is missing libserialport dependancy.
```
  #need to add the following line
  depends_on "libserialport"
````

Reference:
https://github.com/Homebrew/homebrew-core/blob/ca4f6490a40087a4789c8f8e934a21ed3a3bc233/Formula/a/avrdude.rb

```
  depends_on "cmake" => :build
  depends_on "hidapi"
  depends_on "libftdi"
  depends_on "libusb"
  depends_on "libusb-compat"
  #need to add the following line
  depends_on "libserialport"

  uses_from_macos "bison" => :build
  uses_from_macos "flex" => :build

  on_macos do
    # https://github.com/avrdudes/avrdude/issues/1653
    depends_on "libelf" => :build
  end

  on_linux do
    depends_on "elfutils"
    depends_on "readline"
  end
```

You may want to raise an issue here.
https://github.com/Homebrew/homebrew-core/issues

BTW, I usually use HEAD so I did not notice this issue.

mcuee, 2024-02-09:
Link to Homebrew issue here.
* https://github.com/Homebrew/homebrew-core/issues/106902

> `libelf` - has two dependents `avrdude` and `dynamips`. We've already migrated both of these to `elfutils` on Linux but we can't do so on macOS because `elfutils` is Linux-only.

mcuee, 2024-02-24:
From the following discussion.
* https://github.com/GNS3/dynamips/issues/142#issuecomment-1943097307

> Dynamips already points users to elfutils-libelf-devel on Redhat based systems, so it's already a compatible replacement.
Why not have a formula for elfutils-libelf in Homebrew?
> 
> The whole of elfutils might be hard due to linux-only stuff, but the libelf in elfutils by itself should be doable.
Then Homebrew can deprecate libelf in favor of elfutils-libelf.

mcuee, 2024-02-28:
@MCUdude 

Do you have the right for the following repo? If yes, please add me there as well. Thanks.
https://github.com/avrdudes/libelf

dl8dtl, 2024-02-28:
> @MCUdude
> 
> Do you have the right for the following repo? If yes, please add me there as well. Thanks. https://github.com/avrdudes/libelf

I added you all.

mcuee, 2024-03-02:
@mariusgreuel has done a good job that avrdude-libelf actually builds under macOS. In fact it builds on Windows, Linux and macOS with CMake.
* https://github.com/avrdudes/libelf/issues/1#issuecomment-1974750670

mcuee, 2024-03-17:
Remove this from avrdude 7.4 milestone since there is no active development on this topic.

flaviojs, 2024-05-04:
If the only issue is having source code available, then this page looks like a new good reference:
https://fossies.org/linux/misc/old/libelf-0.8.13.tar.gz/

stefanrueger, 2024-08-01:
Sooo, we are planning to release AVRDUDE 8.0 sometime soon-ish. Does anyone (ideally with a Mac!) have a plan how to deal with the libelf deprecation on macOS? @MCUdude @mcuee is it a case of installing libelf source for the Mac and linking to it statically? Just to be clear I won't be able to do that but is anyone else?

mcuee, 2024-08-02:
> Sooo, we are planning to release AVRDUDE 8.0 sometime soon-ish. Does anyone (ideally with a Mac!) have a plan how to deal with the libelf deprecation on macOS? @MCUdude @mcuee is it a case of [...]

My take is to close this issue as unplanned.

Worst case is just to change our CMake script to use our own libelf repo.
https://github.com/avrdudes/libelf/issues/1

stefanrueger, 2025-05-04:
Sooo, we are planning to release AVRDUDE 8.1 sometime soon-ish. Does anyone (ideally with a Mac!) have a plan how to deal with the libelf deprecation on macOS? @MCUdude @mcuee is it a case of installing libelf source for the Mac and linking to it statically? Just to be clear I won't be able to do that but is anyone else?

I am fine with both of @mcuee's suggestions:
 - Closing as unplanned 
 - Compiling in our own [libelf repo](https://github.com/avrdudes/libelf/issues/1)
 
I feel either way we should do something about it.

mcuee, 2025-05-04:
My recommendation is to just close this one as unplanned.

mcuee, 2025-05-12:
Close this issue for now.