avrdudes/avrdude #1655: SerialUPDI - unable to write USERROW memory on AVR64DU28

dbuchwald, 2024-02-03:
First reported by @MCUdude in discussion #1473.

This issue is related to SerialUPDI programmer and first ever tested DU chip, and it's pretty easy to replicate:

```
$ avrdude -cserialupdi -pavr64du32 -P ch340 -T 'write userrow 0 "The quick brown fox jumpes over the lazy dog"' -vvvv

avrdude: Version 7.2-20230720
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is /usr/local/etc/avrdude.conf
         User configuration file is /Users/hans/.avrduderc

         Using port            : /dev/cu.usbserial-1420
         Using programmer      : serialupdi
avrdude: opening serial port ...
avrdude: sending 1 bytes [0x00]
avrdude: ser_send: . [00]
avrdude: ser_recv: . [00]
         AVR Part              : AVR64DU32
         Programming modes     : UPDI, SPM

         Memory            Size  Pg size    Offset
         -----------------------------------------
         eeprom             256        1    0x1400 
         flash            65536      512  0x800000 
         fuses               16        1    0x1050 
         fuse0/wdtcfg         1        1    0x1050 
         fuse1/bodcfg         1        1    0x1051 
         fuse2/osccfg         1        1    0x1052 
         fuse5/syscfg0        1        1    0x1055 
         fuse6/syscfg1        1        1    0x1056 
         fuse7/codesize       1        1    0x1057 
         fuse8/bootsize       1        1    0x1058 
         fusea/pdicfg         2        1    0x105a 
         lock                 4        1    0x1040 
         prodsig/sigrow     128      128    0x1080 
         signature            3        1    0x1080 
         tempsense            4        1    0x1084 
         sernum              16        1    0x1090 
         bootrow            256      256    0x1100 
         userrow/usersig    512      512    0x1200 
         io                4160        1         0 
         sram              8192        1    0x6000 
         sib                 32        1         0 

         Variants             Package  F max   T range        V range       
         -------------------------------------------------------------------
         AVR64DU32-VQFN/TQFP  QFP32    32 MHz  [-40 C, 85 C]  [1.8 V, 5.5 V]

         Programmer Type       : serialupdi
         Description           : SerialUPDI
avrdude: STCS 0x08 to address 0x03
avrdude: sending 3 bytes [0x55, 0xc3, 0x08]
avrdude: ser_send: U [55] . [c3] . [08]
avrdude: ser_recv: U [55] . [c3] . [08]
avrdude: STCS 0x80 to address 0x02
avrdude: sending 3 bytes [0x55, 0xc2, 0x80]
avrdude: ser_send: U [55] . [c2] . [80]
avrdude: ser_recv: U [55] . [c2] . [80]
avrdude: LDCS from 0x00
avrdude: sending 2 bytes [0x55, 0x80]
avrdude: ser_send: U [55] . [80]
avrdude: ser_recv: U [55] . [80]
avrdude: ser_recv: 0 [30]
avrdude: received 1 bytes [0x30]
avrdude: UDPI init OK
avrdude: UPDI link initialization OK
avrdude: LDCS from 0x0B
[... 458 lines cut ...]
avrdude: sending reset request
avrdude: STCS 0x59 to address 0x08
avrdude: sending 3 bytes [0x55, 0xc8, 0x59]
avrdude: ser_send: U [55] . [c8] Y [59]
avrdude: ser_recv: U [55] . [c8] Y [59]
avrdude: sending release reset request
avrdude: STCS 0x00 to address 0x08
avrdude: sending 3 bytes [0x55, 0xc8, 0x00]
avrdude: ser_send: U [55] . [c8] . [00]
avrdude: ser_recv: U [55] . [c8] . [00]
avrdude: STCS 0x0C to address 0x03
avrdude: sending 3 bytes [0x55, 0xc3, 0x0c]
avrdude: ser_send: U [55] . [c3] . [0c]
avrdude: ser_recv: U [55] . [c3] . [0c]

avrdude done.  Thank you.
```

Root cause of the issue is the fact that so far all the USERROW memories were 64 bytes or smaller, and the implementations so far were based on the sequence of "set pointer; repeat <number_of_bytes>; write byte and increase pointer", which was fine as long as these memories were 256 bytes or less (this is limitation of the `repeat` command which takes single byte as a parameter.

Most probably the fix would be to change the USERROW write operation to "set pointer; repeat <number_of_bytes/2>; write 2-byte word and increase pointer", but this could introduce regression to all the old parts, so this needs to be tested carefully.

It would also be very helpful to have access to the datasheet of the chip in question, it might actually contain some guidelines for handling such a large memory.

@stefanrueger, @mcuee, @askn37 - do you think I should add any additional information here?

dbuchwald, 2024-02-03:
@MCUdude I have prepared preliminary fix, which should not introduce any regression for previously tested chips:

https://github.com/dbuchwald/avrdude/tree/issue_1655

Can you check if it fixes issue on AVR64DU28?

Basically the change is that if the USERROW memory is larger than 256 bytes then word-based write operation is performed instead of the previously used byte-based write.

I have also tested word-based write of USERROW against AVR16EA28 and it seems to have worked just fine, so maybe some day we could simplify code by changing to word-based write for all the UPDI AVRs, but I wouldn't introduce this so late into release cycle.

askn37, 2024-02-03:
> do you think I should add any additional information here?

I agree that you are absolutely right.

The maximum amount of memory that can be written with one set of repeated commands is 256 words, which is consistent with all known datasheet descriptions. At least I haven't read anything else written. And this is probably the upper limit that can be achieved with his current UPDI system. If a page size of 1024 bytes appears in the future, I think the specifications will probably be expanded or some kind of improvement will be proposed at that time.

MCUdude, 2024-02-03:
I'm away for the weekend, but will be back and can continue testing on Sunday evening.

But is the userrow size 64 or 512 bytes?

dbuchwald, 2024-02-03:
> I'm away for the weekend, but will be back and can continue testing on Sunday evening.

Sure, sorry, didn't mean to rush you.

> But is the userrow size 64 or 512 bytes?

For a minute I was hoping it was 64 bytes but no, after inspecting atpack and `pymcuprog` source I came to the conclusion that the size of USERROW memory for AVR64DU32 and AVR64DU28 is, indeed, 512 bytes.

askn37, 2024-02-04:
It is stated on page 3 to 4 of the brief note.

https://askn37.github.io/documents/AVR-DU-Product-Brief-DS40002328.pdf

[image]

Even in this document, it is unlikely that you are confusing it with bit amount.

askn37, 2024-02-04:
The following products have been registered on Mouser and DigiKey.

FBP-AVR-DU-ISO26262
Development Software ISO 26262 Functional Safety Basic Package for the AVR DU MCU Family
Factory Special Order: Obtain a quote to verify the current price, lead-time and ordering requirements of the manufacturer.
https://www.mouser.fi/ProductDetail/Microchip-Technology/FBP-AVR-DU-ISO26262?qs=mELouGlnn3ezslVP1LToMA%3D%3D

It appears to be a development kit to obtain ISO 26262 approval. I had no idea it would be possible to pre-order items like this before the individual chips went on sale.

From this, we can infer that an automotive-grade derivative of AVR-DU also exists. This suggests that there may be signatures that differ from the usual products, such as the ATtiny416auto example.

MCUdude, 2024-02-04:
@dbuchwald your suggested fix did the trick! 🚀 

```
$ ./test-avrdude -p "-c serialupdi -pavr64du32 -P ch340" -e /Users/hans/Downloads/avrdude-terminal/src/avrdude
Testing /Users/hans/Downloads/avrdude-terminal/src/avrdude version 7.2-20230720
Prepare "-c serialupdi -pavr64du32 -P ch340" and press 'enter' or 'space' to continue. Press any other key to skip
✅   0.533 s: fuse access: clear, set and read eesave fuse bit
✅   0.504 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
✅   0.504 s: chip erase
✅   3.934 s: flash -U write/verify holes_rjmp_loops_65536B.hex
✅   2.277 s: flash -T write/verify holes_rjmp_loops_65536B.hex
✅   0.576 s: eeprom check whether programmer can flip 0s to 1s
✅   2.986 s: eeprom -U write/verify holes_pack_my_box_256B.hex
✅   5.504 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_256B.hex
✅   2.379 s: chip erase and spot check flash is actually erased
✅   0.783 s: spot check eeprom is erased, too
✅   2.239 s: usersig -T/-U write/read random_data_512B.bin
```

stefanrueger, 2024-02-05:
Brilliant! Thanks all round for ironing out this problem and confirming it has done so!

stefanrueger, 2024-02-05:
@dbuchwald Could you do the honours and submit a PR, please?

dbuchwald, 2024-02-05:
@MCUdude thanks for testing it. I'm really glad it worked!

@stefanrueger PR is ready here: https://github.com/avrdudes/avrdude/pull/1659

closed 2024-02-05 by stefanrueger