avrdudes/avrdude #1664: auto-tools scripts improvement -- update configure.ac to reflect libreadline status

mcuee, 2024-02-06:
As of now, for auto-tools build, the presence of libreadline is not correct. It always says libreadline is not present even though libreadline is present and the final binary is linked to libreadline.

I tried to fix it last time but it caused issues with macOS terminal mode and I have to revert the following PR.
* https://github.com/avrdudes/avrdude/pull/1280

stefanrueger, 2024-02-07:
@mcuee Thanks for proactively starting issues on things we'd like improved. Can we please use fewer issues: we can summarise what needs improved, eg, have *one* issue for improving autotools build (listing the 4 aspects to improve) and *one* issue for improving documentation. It is unlikely we attract a number of people working in parallel on these issues (and for the autotools this is definitely not advisable). Also the new issue on jtag2updi improvement is kinda non-descript. I'd remove that for the time being or make it specific.

Ideally we never grow over a page of issues and discussions so we don't lose sight over things on the second page

mcuee, 2024-02-07:
@stefanrueger 

Done.

mcuee, 2024-02-07:
@stefanrueger 

I have trimmed down the number of issues and discussions to one page. Feel free to close those issues/discussions you think can be closed.

mcuee, 2024-02-11:
The following issue should be present in autotools build as well.
* https://github.com/avrdudes/avrdude/issues/1683

ndim, 2024-02-11:
> The following issue should be present in autotools build as well.
> 
>     * [avrintel.h missing from installation #1683](https://github.com/avrdudes/avrdude/issues/1683)

I will reflect in autoconf whatever comes out of the discussion over there.

mcuee, 2024-04-26:
Remove the bug label now that the remaining portions are for enhancement.

mcuee, 2024-08-23:
I think we can close this issue.

closed 2024-08-23 by mcuee