avrdudes/avrdude #1679: github actions improvement

mcuee, 2024-02-07:
1) Apple Silicon macOS build -- we have to wait github infrastructure to be ready first
* https://github.com/avrdudes/avrdude/discussions/1677

2) Deploy action -- to include MSYS2 mingw64/mingw32 binaries upon formal release like 7.3.

3) Deploy action for Arduino-packing github action upon formal release like 7.3.

mcuee, 2024-02-07:
Deploy action -- to include MSYS2 mingw64/mingw32 binaries upon formal release like 7.3.

Currently the deploy github action only upload Windows MSVC binaries.
https://github.com/avrdudes/avrdude/blob/main/.github/workflows/deploy.yml
https://github.com/avrdudes/avrdude/actions/runs/7813933855

The mingw32/mingw64 github action binaries will not have the correct version number, but rather one commit after.

mcuee, 2024-02-07:
As for the binaries generated by arduino-packing script, as of now, the script does not deal with releases yet.So the version number will not matching -- one commit after the release tag.
https://github.com/avrdudes/avrdude/blob/main/.github/workflows/arduino_packing.yml

@umbynos 
The [avrdude-packing script in this repo](https://github.com/avrdudes/avrdude/blob/main/.github/workflows/arduino_packing.yml) is a cut-down version of[ your more detailed script.](https://github.com/arduino/avrdude-packing/blob/main/.github/workflows/release.yml) One missing feature is to deal with the tagged release (say 7.3 release). Just wondering if you can help here. Thanks.

mcuee, 2024-02-09:
@umbynos 

As of now, the action is only executed upon git push to git main, but not release tagging.
https://github.com/avrdudes/avrdude/blob/main/.github/workflows/arduino_packing.yml

Your original version is okay but I do not know how to adapt it to avrdude git main.
https://github.com/arduino/avrdude-packing/blob/main/.github/workflows/release.yml

mcuee, 2024-02-17:
Related issue:
* https://github.com/arduino/avrdude-packing/issues/31
* https://github.com/arduino/crossbuild/issues/13

mcuee, 2024-02-17:
> As of now, the action is only executed upon git push to git main, but not release tagging.
> https://github.com/avrdudes/avrdude/blob/main/.github/workflows/arduino_packing.yml

@umbynos 
I have created a new PR to address the above issue. But I am not so sure if this is the right solution or not. Please review. Thanks.
* https://github.com/avrdudes/avrdude/pull/1690

ndim, 2024-02-20:
> 4) It may be good to add github actions for BSDs.

GitHub only has Ubuntu, Windows and OSX workers.

However, Cirrus CI has a free offer for Open Source Projects which includes CI builds on FreeBSD.

mcuee, 2024-02-20:
> > 4. It may be good to add github actions for BSDs.
> 
> GitHub only has Ubuntu, Windows and OSX workers.
> 
> However, Cirrus CI has a free offer for Open Source Projects which includes CI builds on FreeBSD.

There are some changes now. Please refer to the following discussion.
* https://github.com/libusb/libusb/issues/1465

mcuee, 2024-08-09:
I have documented steps to build libserialport for the Arduino cross-build Docker container. However, macOS support is missing (difficult to cross build libserialport under the container).

@ndim Just wondering if you can take a look here. Thanks.
https://github.com/arduino/avrdude-packing/issues/31

mcuee, 2024-08-09:
Even if we cannot get macOS libserialport cross-build to work, we may still want to improve the following two github actions to install libserialport for Linux and mingw32 cross build. But I am not so sure if we can do this before avrdude 8.0 release or not.

https://github.com/avrdudes/avrdude/blob/main/.github/workflows/arduino_packing.yml
https://github.com/avrdudes/avrdude/blob/main/.github/workflows/arduino_packing_release.yml

Worst case, I can do manually cross build, just like avrdude 7.3 release.

mcuee, 2024-08-31:
avrdude 8.0 release github action build:
https://github.com/avrdudes/avrdude/actions/runs/10541592535 (Deploy)
https://github.com/avrdudes/avrdude/actions/runs/10541592513 (avrdude_packing_release).

The binaries are uploaded to the release page.
https://github.com/avrdudes/avrdude/releases/tag/v8.0

mcuee, 2024-09-02:
@ndim and @umbynos 

I think we may need to improve the macOS crossbuild githun action to remove the following warnings.

Example run:
https://github.com/avrdudes/avrdude/actions/runs/10649351558/job/29519411601
```
[ 96%] Building C object src/CMakeFiles/avrdude.dir/whereami.c.o
cd /__w/avrdude/avrdude/avrdude/build/src && /opt/osxcross/target/bin/o64-clang -DCONFIG_DIR=\"/usr/local/etc\" -I/__w/avrdude/avrdude/avrdude/build/src -I/__w/avrdude/avrdude/avrdude -I/__w/avrdude/avrdude/avrdude/build -I/__w/avrdude/avrdude/avrdude/src  -I/opt/lib/x86_64-apple-darwin13/include -pthread -framework Foundation -framework IOKit -framework Cocoa -framework Security -DHAVE_USB_H   -Wall -Wextra -Wno-unused-parameter -Wno-format-zero-length -std=gnu11 -o CMakeFiles/avrdude.dir/whereami.c.o   -c /__w/avrdude/avrdude/avrdude/src/whereami.c
clang: warning: -framework Foundation: 'linker' input unused [-Wunused-command-line-argument]
clang: warning: -framework IOKit: 'linker' input unused [-Wunused-command-line-argument]
clang: warning: -framework Cocoa: 'linker' input unused [-Wunused-command-line-argument]
clang: warning: -framework Security: 'linker' input unused [-Wunused-command-line-argument]
[ 97%] Linking C executable avrdude
cd /__w/avrdude/avrdude/avrdude/build/src && /usr/bin/cmake -E cmake_link_script CMakeFiles/avrdude.dir/link.txt --verbose=1
/opt/osxcross/target/bin/o64-clang -I/opt/lib/x86_64-apple-darwin13/include -pthread -framework Foundation -framework IOKit -framework Cocoa -framework Security -DHAVE_USB_H  -L/opt/lib/x86_64-apple-darwin13/lib/ CMakeFiles/avrdude.dir/main.c.o CMakeFiles/avrdude.dir/developer_opts.c.o CMakeFiles/avrdude.dir/whereami.c.o  -o avrdude  libavrdude.a -lm /opt/lib/x86_64-apple-darwin13/lib/libelf.a /opt/lib/x86_64-apple-darwin13/lib/libusb.a /opt/lib/x86_64-apple-darwin13/lib/libusb-1.0.a /opt/lib/x86_64-apple-darwin13/lib/libhidapi.a /opt/lib/x86_64-apple-darwin13/lib/libftdi1.a /opt/lib/x86_64-apple-darwin13/lib/libreadline.a /opt/lib/x86_64-apple-darwin13/lib/libncurses.a 
make[2]: Leaving directory '/__w/avrdude/avrdude/avrdude/build'
```

mcuee, 2024-09-02:
https://github.com/avrdudes/avrdude/blob/main/.github/workflows/arduino_packing.yml
https://github.com/avrdudes/avrdude/blob/main/.github/workflows/arduino_packing_release.yml

The following four seem to be for linking only.
```
-framework Foundation -framework IOKit -framework Cocoa -framework Security 
```

ndim, 2024-09-02:
That appears to be stuff you probably need for every proper Macos program. Those are unused for building a CLI program like avrdude. Not sure what we can do there except disable the warning about unused arguments.

mcuee, 2024-09-02:
> That appears to be stuff you probably need for every proper Macos program. Those are unused for building a CLI program like avrdude. Not sure what we can do there except disable the warning about unused arguments.

I believe this is correct.
https://github.com/mcuee/avrdude/commit/5c0f8333ba4ca71324f55c42d3d63fd37651e50f

However, I have not figured how to cross build for macOS yet to verify the correctness of the above patch. For Linux/Windows I am good.

ndim, 2024-09-02:
Come to think about it... `windows-latest`, `macos-latest`, `ubuntu-latest`: None of those say anything about the processor architecture.

mcuee, 2025-05-22:
Closing this for now.

closed 2025-05-22 by mcuee