avrdudes/avrdude #1697: AVR109 auto reset functionality

MCUdude, 2024-02-21:
First pointed out [here](https://github.com/avrdudes/avrdude/discussions/1022#discussioncomment-8539144):

> My bootloader doesn't need an IO pin to enter bootloader mode. It automatically enters bootloader mode after the MCU resets itself and return to application after 500ms to 4s (the compiled hex has a 1 second delay), so the first upload is free, since there is no code in the application section. The second upload, you need to press the reset button. AVR109 doesn't issue an DTR or RTS signal on serial, so auto upload is not possible with avr109.

We should consider adding DTR/RTS toggling to the `avr109` programming option as we do with `urclock`, `arduino`, and `wiring`, as this would make it easier to enter bootloader mode using a simple auto-reset circuit.

MCUdude, 2024-02-22:
I've pushed a development branch here:
https://github.com/MCUdude/avrdude/tree/avr109-autoreset

However, I've not been able to find a bootloader that with auto reset, and doesn't work without it.
I've tried Xboot, and if I set `USE_ENTER_UART = yes` in the `arduino328p.conf.mk` file, it jumps straight into the bootloader even though no auto reset. And if I use `no`, I'm not able to enter bootloader mode, not even with auto reset present.

stefanrueger, 2024-02-25:
Your development branch looks neat and tidy
 - Does the usleep(250 * 1000) really need to be that long? 50 ms is plenty to discharge the reset cap (should work even for C = 10 µF and R = 5k). `-c arduino` needs that long sleep b/c the optiboot bootloader needs a 200 ms pause between two successive calls. `-c xbee` might need a long pause perhaps b/c it involves sending packets over the air, but the standard AVR109 board?
 - Would it not be better to keep default behaviour at *not* plucking the reset line?  I just imagine someone's board might blow up if DTR is suddenly toggled
 - Documentation would be cool

stefanrueger, 2024-07-21:
@MCUdude Do you want to pursue AVR109 auto reset further?

MCUdude, 2024-07-21:
I don't quite remember why I didn't continue working on this. But I remember using the Xboot bootloader (that uses the AVR109 protocol) and for some reason it would "just upload" even though there where no reset pulse. Does there a different AVR109 bootloader I can test with that would require manual or auto reset?

MCUdude, 2024-07-21:
I've done a little research, and I'd like to explain a little.

I've been using the [Xboot bootloader](https://github.com/alexforencich/xboot/) for testing the AVR109 protocol. This bootloader works great, but AFAIK it doesn't support the usual "wait and look for UART data" after a hardware reset. This makes it impossible to upload new programs just by connecting the DTR line to the RST pin through a 100nF cap (with a 10k pullup and HV protection diode). *If someone knows another AVR109 implementation that does support auto-reset or manual reset before upload (similar to the Optiboot bootloader), please let me know.*

I have an almost-ready PR. I haven't found a bootloader it works on yet, but the DTR line behaves the same way as when using `-c arduino`.

Another thing to take into account is that the DTR line is pulled down when the connection is established, even though I'm not touching it in the Avrdude source code. I think this is just normal handshake pin behaviour.

@stefanrueger do you want me to create a PR to add the same DTR toggling as `-carduino` adds. The DTR/RTS double pulse can be removed (single pulse instead) with `-xno_autoreset`. Alternatively, we can have it disabled by default, and enable it using `-xautoreset`.

### Single pulse, zoomed view of the DTR and RST lines (using `-xno_autoreset`):
[image]

### Double pulse, full view of the DTR and RST lines:
[image]

stefanrueger, 2024-07-22:
@MCUdude Thanks for the investigation.

How *does* the AVR109 bootloader decide to do bootloading? Does it wait for a specific pin to be pulled up/down at reset?

And, yes, depending on the OS/drivers sometimes DTR/RST is toggled on serial device open. That then *just works*, and AVRDUDE does not need to create an auto reset pulse.

It would be cool if the relevant programmers that have auto reset (urclock/arduino etc) could switch auto-reset off. And if those that didn't have auto-reset were to have an option to switch it on. It's unsystematic, but least there are no behaviour changes *unless* explicitly requested.

mcuee, 2024-07-22:
> How _does_ the AVR109 bootloader decide to do bootloading? Is bootloading triggered by a specific pin (other than reset) being pulled up/down when the AVR comes out of reset?

The official Microchip/Atmel AVR109 bootloader uses a port pin to enter bootloader mode.
https://ww1.microchip.com/downloads/en/Appnotes/doc1644.pdf

mcuee, 2024-07-22:
But there are different options for xboot: `USE_ENTER_DELAY` or `USE_ENTER_PIN`, or some other options.
https://github.com/alexforencich/xboot

MCUdude, 2024-07-22:
> How does the AVR109 bootloader decide to do bootloading? 

When using Xboot, I've only been able to enter bootloader mode by pulling a pin low, and then resetting the microcontroller. When just resetting the microcontroller, the program starts immediately.

> But there are different options for xboot: USE_ENTER_DELAY or USE_ENTER_PIN.¨

@mcuee I haven't gotten `USE_ENTER_DELAY` to work. Or at least it doesn't change the fact that the program starts immediately after a hardware reset.

> It would be cool if the relevant programmers that have auto reset (urclock/arduino etc) could switch auto-reset off

I'll see what I can do.

closed 2024-07-31 by stefanrueger