avrdudes/avrdude #1708: LinuxGPIO or SPI don't work on NVIDIA Orin Nx

ltidswell, 2024-02-25:
I am trying to program an Arduino Nano using AVRDUDE via the ICSP header on the Nano and the MOSI, MISO, and SCK pins on the 40 pin expansion header on the Orin Nx. I am using the Seeed Studio base board with the Orin Nx 16GB (https://www.seeedstudio.com/reComputer-J4012-p-5586.html) and have Jetpack 5.1.1 installed. Am using version 7.2 of AVRDUDE and followed these install instructions: https://github.com/avrdudes/avrdude/wiki/Building-AVRDUDE-for-Linux

Confirmed the Arduino & wires work by testing it on a Pi. Am trying to run:
`sudo avrdude -v -p m328p -c linuxgpio`

The naming convention for GPIO pins on the NVIDA Orin (also on the Xavier NX, possibly other NVIDIA devices too) is not gpioPINNUM like on other devices, the pins I am using and what they're called within /sys/class/gpio are:
`389 --> PG.06
481 --> PZ.03
483 --> PZ.05
482 --> PZ.04`
[image]

I modified linuxgpio.c to accept this naming convention and am managing to export the first pin but am now getting this error:
`avrdude main() error: unable to open port gpiochip0 for programmer linuxgpio`
I used _gpioinfo_ to confirm the pins I am using are in gpiochip0. 

I get a similar error trying to open the SPI port. 

Any idea on how to fix this / is there support for using AVRDUDE on these devices?
Note: programming works via USB connection but in my application I can't use this.

mcuee, 2024-02-25:
@SebKuzminsky 

Just wondering if you can help here since you are the author of PR #1299. 

* #1299

mcuee, 2024-02-25:
@s-wakaba 

Just wondering if you can help here as well since you seem to be familiar with linuxspi.
* https://github.com/avrdudes/avrdude/pull/1100

mcuee, 2024-04-14:
@ltidswell 

Are you using sysfs and not libgpiod? If that is the case, please try libgpiod to see if that works.

Reference: please refer to this discussion, once the user switches to libgpiod, the problem got resolved.
* https://github.com/avrdudes/avrdude/discussions/1737

mcuee, 2024-04-28:
@ltidswell 

Please help to use libgpiod and try latest git main. Thanks.

stefanrueger, 2024-05-08:
@ltidswell Just checking where you are with this issue?

ltidswell, 2024-05-08:
@stefanrueger Found the issue to be the SPI pins don't output enough current (20uA) on the Nvidia SOMs. Selecting pins that output 20mA (I2C pins, datasheet: https://files.seeedstudio.com/products/NVIDIA/reComputer-J401x-datasheet.pdf) works using linuxGPIO but still had to manually convert to Nvidia GPIO names. Am attempting to use a non-inverting buffer on the SPI pins to see if that works. @mcuee will try git main when I get a chance but hoping using a buffer will fix the issue.

stefanrueger, 2024-05-09:
Thanks for reporting back. This might help others with the same issue.

>  manually convert to Nvidia GPIO names

If you have an idea how git main could help please tell us in more detail. Better still submit a PR, eg, where the GPIO naming scheme is a -x command line parameter or a parameter in the avrdude.conf entry.

mcuee, 2024-08-09:
@ltidswell 

Any updates on this issue?

ltidswell, 2024-08-11:
@mcuee I fixed the issue by only using high current pins (https://github.com/avrdudes/avrdude/issues/1708#issuecomment-2101591077). Am yet to try other solutions

John-194, 2024-08-30:
@ltidswell
I am also trying to program Arduino via Jetson GPIO. Could you share how you modified linuxgpio.c to accept its naming convention?

closed 2025-04-15 by stefanrueger