avrdudes/avrdude #1713: Regression: EEPROM issue with official STK500 V1 FW -- avrdude 7.1 is okay but not 7.2 and 7.3

mcuee, 2024-03-02:
Strange that this comes up again during my benchmark testing of STK500v1.

Using a STK500 v1 clone with official V1 FW on ATmega8535.

```
mcuee@UbuntuSwift3 ~/build/avr/avrdude_main/tools (main)$
 ./test-avrdude -b -d 0 -v -p "-c stk500v1 -P /dev/ttyUSB1 -p m328p Ubuntu_Linux_20.04" 
Testing avrdude version 7.3

The benchmark shows the wall clock time of avrdude carrying out the following tasks
  - Fl-ewv: erase flash, then write a sketch with three sections separated by holes, and verify
  - Fl-r: read the entire flash memory and write to file
  - EE-wv: write data to EEPROM with two data sections separated by a hole, and verify
  - EE-r: read the entire EEPROM memory and write to file

The sketch and data payload is roughly one sixth of the respective memory size

| | `-c pgm` | `-p part` | Fl-ewv | Fl-r | EE-wv | EE-r | Comments |
|:-:|--:|--:|--:|--:|--:|--:|:--|
Prepare "-c stk500v1 -P /dev/ttyUSB1 -p m328p Ubuntu_Linux_20.04" and press 'enter' or 'space' to continue. Press any other key to skip
✅   4.644 s: flash -U write/verify holes_rjmp_loops_32768B.hex
✅   9.876 s: flash -U read all flash
❌   2.517 s: eeprom -U write/verify holes_pack_my_box_1024B.hex (failed command below)
$ avrdude -qq -c stk500v1 -P /dev/ttyUSB1 -p m328p Ubuntu_Linux_20.04 -Ueeprom:w:./test_files/holes_pack_my_box_1024B.hex
avrdude warning: verification mismatch
        device 0xff != input 0x50 at addr 0x0082 (error)
avrdude error: verification mismatch
✅   1.841 s: eeprom -U read all
|❌|stk500v1|m328p| 4.64 s| 9.88 s|error| 1.84 s| Ubuntu_Linux_20.04|

One or more AVRDUDE "-c stk500v1 -P /dev/ttyUSB1 -p m328p Ubuntu_Linux_20.04" tests failed. Do you want to retry this particular test? (y/n): n

mcuee@UbuntuSwift3 ~/build/avr/avrdude_bin $
 ./avrdude73 -C ./avrdude73.conf -c stk500v1 -P /dev/ttyUSB1 -p m328p -U eeprom:w:holes_pack_my_box_1024B.hex:i
avrdude73: AVR device initialized and ready to accept instructions
avrdude73: device signature = 0x1e950f (probably m328p)

avrdude73: processing -U eeprom:w:holes_pack_my_box_1024B.hex:i
avrdude73: reading input file holes_pack_my_box_1024B.hex for eeprom
           with 252 bytes in 2 sections within [0x82, 0x37d]
           using 64 pages and 4 pad bytes
avrdude73: writing 252 bytes eeprom ...
Writing | ################################################## | 100% 0.56 s 
avrdude73: 252 bytes of eeprom written
avrdude73: verifying eeprom memory against holes_pack_my_box_1024B.hex
Reading | ################################################## | 100% 0.08 s 
avrdude73 warning: verification mismatch
          device 0xff != input 0x50 at addr 0x0082 (error)
avrdude73 error: verification mismatch

avrdude73 done.  Thank you.

mcuee@UbuntuSwift3 ~/build/avr/avrdude_bin $
 ./avrdude72 -C ./avrdude72.conf -c stk500v1 -P /dev/ttyUSB1 -p m328p -U eeprom:w:holes_pack_my_box_1024B.hex:i
avrdude72: AVR device initialized and ready to accept instructions
avrdude72: device signature = 0x1e950f (probably m328p)

avrdude72: processing -U eeprom:w:holes_pack_my_box_1024B.hex:i
avrdude72: reading input file holes_pack_my_box_1024B.hex for eeprom
           with 252 bytes in 2 sections within [0x82, 0x37d]
           using 64 pages and 4 pad bytes
avrdude72: writing 252 bytes eeprom ...
Writing | ################################################## | 100% 0.57 s 
avrdude72: 252 bytes of eeprom written
avrdude72: verifying eeprom memory against holes_pack_my_box_1024B.hex
Reading | ################################################## | 100% 0.08 s 
avrdude72 warning: verification mismatch
          device 0xff != input 0x50 at addr 0x0082 (error)
avrdude72 error: verification mismatch

avrdude72 done.  Thank you.

mcuee@UbuntuSwift3 ~/build/avr/avrdude_bin $
 ./avrdude71 -C ./avrdude71.conf -c stk500v1 -P /dev/ttyUSB1 -p m328p -U eeprom:w:holes_pack_my_box_1024B.hex:i

avrdude71: AVR device initialized and ready to accept instructions
avrdude71: device signature = 0x1e950f (probably m328p)
avrdude71: reading input file holes_pack_my_box_1024B.hex for eeprom
           with 252 bytes in 2 sections within [0x82, 0x37d]
           using 64 pages and 4 pad bytes
avrdude71: writing 252 bytes eeprom ...

Writing | ################################################## | 100% 0.58 s 

avrdude71: 252 bytes of eeprom written
avrdude71: verifying eeprom memory against holes_pack_my_box_1024B.hex

Reading | ################################################## | 100% 0.08 s 

avrdude71: 252 bytes of eeprom verified

avrdude71 done.  Thank you.
```

mcuee, 2024-03-02:
History of stk500.c
https://github.com/avrdudes/avrdude/commits/main/src/stk500.c

avrdude 7.1 release was on 2023-Jan-09.
avrdude 7.2 release was on 2023-Jul-20.

stefanrueger, 2024-04-13:
> I will try git bisect

@mcuee any intell on what's going on?

mcuee, 2024-04-14:
> > I will try git bisect
> 
> @mcuee any intell on what's going on?

Sorry I have not done anything on this issue yet. I will try to carry out the analysis next week.

mcuee, 2024-04-15:
Symptom:

```
mcuee@UbuntuSwift3 ~/build/avr/avrdude_bin $ sudo ./avrdude71 -C ./avrdude71.conf -c usbasp -p m328p -U eeprom:w:the_quick_brown_fox_128B.hex:i

avrdude71: AVR device initialized and ready to accept instructions
avrdude71: device signature = 0x1e950f (probably m328p)
avrdude71: reading input file the_quick_brown_fox_128B.hex for eeprom
           with 128 bytes in 1 section within [0, 0x7f]
           using 32 pages and 0 pad bytes
avrdude71: writing 128 bytes eeprom ...

Writing | ################################################## | 100% 1.34 s 

avrdude71: 128 bytes of eeprom written
avrdude71: verifying eeprom memory against the_quick_brown_fox_128B.hex

Reading | ################################################## | 100% 0.05 s 

avrdude71: 128 bytes of eeprom verified

avrdude71 done.  Thank you.

mcuee@UbuntuSwift3 ~/build/avr/avrdude_bin $ avrdude -c stk500v1 -P /dev/ttyUSB0 -p m328p -U eeprom:w:holes_pack_my_box_1024B.hex:i
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e950f (probably m328p)

avrdude: processing -U eeprom:w:holes_pack_my_box_1024B.hex:i
avrdude: reading input file holes_pack_my_box_1024B.hex for eeprom
         with 252 bytes in 2 sections within [0x82, 0x37d]
         using 64 pages and 4 pad bytes
avrdude: writing 252 bytes eeprom ...
Writing | ################################################## | 100% 1.11 s 
avrdude: 252 bytes of eeprom written
avrdude: verifying eeprom memory against holes_pack_my_box_1024B.hex
Reading | ################################################## | 100% 0.16 s 
avrdude warning: verification mismatch
        device 0xff != input 0x50 at addr 0x0082 (error)
avrdude error: verification mismatch

avrdude done.  Thank you.

mcuee@UbuntuSwift3 ~/build/avr/avrdude_bin $ sudo ./avrdude71 -C ./avrdude71.conf -c usbasp -p m328p -qqt
avrdude> dump eeprom
0000  54 48 45 20 51 55 49 43  4b 20 42 52 4f 57 4e 20  |THE QUICK BROWN |
0010  46 4f 58 20 4a 55 4d 50  53 20 4f 56 45 52 20 54  |FOX JUMPS OVER T|
0020  48 45 20 4c 41 5a 59 20  44 4f 47 2e 20 54 48 45  |HE LAZY DOG. THE|
0030  20 51 55 49 43 4b 20 42  52 4f 57 4e 20 46 4f 58  | QUICK BROWN FOX|
0040  20 4a 55 4d 50 53 20 4f  56 45 52 20 54 48 45 20  | JUMPS OVER THE |
0050  4c 41 5a 59 20 44 4f 47  2e 20 54 48 45 20 51 55  |LAZY DOG. THE QU|
0060  49 43 4b 20 42 52 4f 57  4e 20 46 4f 58 20 4a 55  |ICK BROWN FOX JU|
0070  4d 50 53 20 4f 56 45 52  20 54 48 45 20 4c 41 5a  |MPS OVER THE LAZ|
0080  ff ff ff 41 43 4b 20 4d  59 20 42 4f 58 20 57 49  |...ACK MY BOX WI|
0090  54 48 20 46 49 56 45 20  44 4f 5a 45 4e 20 4c 49  |TH FIVE DOZEN LI|
00a0  51 55 4f 52 20 4a 55 47  53 50 41 43 4b 20 4d 59  |QUOR JUGSPACK MY|
00b0  20 42 4f 58 20 57 49 54  48 20 46 49 56 45 20 44  | BOX WITH FIVE D|
00c0  4f 5a 45 4e 20 4c 49 51  55 4f 52 20 4a 55 47 53  |OZEN LIQUOR JUGS|
00d0  50 41 43 4b 20 4d 59 20  42 4f 58 20 57 49 54 48  |PACK MY BOX WITH|
00e0  20 46 49 56 45 20 44 4f  5a 45 4e 20 4c 49 51 55  | FIVE DOZEN LIQU|
00f0  4f 52 20 4a 55 47 53 50  41 43 4b 20 4d 59 20 42  |OR JUGSPACK MY B|

avrdude> dump
0100  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0110  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0120  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0130  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0140  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0150  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0160  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0170  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
[... 20 lines cut ...]
02a0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
02b0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
02c0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
02d0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
02e0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
02f0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude> dump
0300  54 48 20 46 49 56 45 20  44 4f 5a 45 4e 20 4c 49  |TH FIVE DOZEN LI|
0310  51 55 4f 52 20 4a 55 47  53 50 41 43 4b 20 4d 59  |QUOR JUGSPACK MY|
0320  20 42 4f 58 20 57 49 54  48 20 46 49 56 45 20 44  | BOX WITH FIVE D|
0330  4f 5a 45 4e 20 4c 49 51  55 4f 52 20 4a 55 47 53  |OZEN LIQUOR JUGS|
0340  50 41 43 4b 20 4d 59 20  42 4f 58 20 57 49 54 48  |PACK MY BOX WITH|
0350  20 46 49 56 45 20 44 4f  5a 45 4e 20 4c 49 51 55  | FIVE DOZEN LIQU|
0360  4f 52 20 4a 55 47 53 50  41 43 4b 20 4d 59 20 42  |OR JUGSPACK MY B|
0370  4f 58 20 57 49 54 48 20  46 49 56 45 20 44 ff ff  |OX WITH FIVE D..|
0380  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0390  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
03a0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
03b0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
03c0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
03d0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
03e0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
03f0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
```

mcuee, 2024-04-15:
Zoom in to see which git commit got the issue.
Commit https://github.com/avrdudes/avrdude/commit/2b04a2e4a79a2ed162dd951cfb30dfed496fae5f is bad.
```
mcuee@UbuntuSwift3 ~/build/avr/avrdude_bin $ ./avrdude_2b04a2e -C avrdude_2b04a2e.conf -c stk500v1 -P /dev/ttyUSB0 -p m328p -U eeprom:w:holes_pack_my_box_1024B.hex:i

avrdude_2b04a2e: AVR device initialized and ready to accept instructions
avrdude_2b04a2e: device signature = 0x1e950f (probably m328p)
avrdude_2b04a2e: reading input file holes_pack_my_box_1024B.hex for eeprom
                 with 252 bytes in 2 sections within [0x82, 0x37d]
                 using 64 pages and 4 pad bytes
avrdude_2b04a2e: writing 252 bytes eeprom ...

Writing | ################################################## | 100% 0.56 s 

avrdude_2b04a2e: 252 bytes of eeprom written
avrdude_2b04a2e: verifying eeprom memory against holes_pack_my_box_1024B.hex

Reading | ################################################## | 100% 0.08 s 

avrdude_2b04a2e warning: verification mismatch
                device 0xff != input 0x50 at addr 0x0082 (error)
avrdude_2b04a2e error: verification mismatch

avrdude_2b04a2e done.  Thank you.
```

mcuee, 2024-04-15:
Commit https://github.com/avrdudes/avrdude/commit/65686c7fed377d2eac8b54949c4cacf8bd675635 is not OK.
```
mcuee@UbuntuSwift3 ~/build/avr/avrdude_bin $ ./avrdude_65686c7 -C avrdude_65686c7.conf -c stk500v1 -P /dev/ttyUSB0 -p m328p -U eeprom:w:holes_pack_my_box_1024B.hex:i

avrdude_65686c7: AVR device initialized and ready to accept instructions
avrdude_65686c7: device signature = 0x1e950f (probably m328p)
avrdude_65686c7: reading input file holes_pack_my_box_1024B.hex for eeprom
                 with 252 bytes in 2 sections within [0x82, 0x37d]
                 using 64 pages and 4 pad bytes
avrdude_65686c7: writing 252 bytes eeprom ...

Writing | ################################################## | 100% 0.56 s 

avrdude_65686c7: 252 bytes of eeprom written
avrdude_65686c7: verifying eeprom memory against holes_pack_my_box_1024B.hex

Reading | ################################################## | 100% 0.08 s 

avrdude_65686c7 warning: verification mismatch
                device 0xff != input 0x50 at addr 0x0082 (error)
avrdude_65686c7 error: verification mismatch

avrdude_65686c7 done.  Thank you.
```

mcuee, 2024-04-15:
Commit https://github.com/avrdudes/avrdude/commit/cb5f5956fc46020408709111ab88f385dfa932fa  is GOOD.
```
mcuee@UbuntuSwift3 ~/build/avr/avrdude_bin $ ./avrdude -c usbasp -p m328p -T "erase eeprom"
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e950f (probably m328p)

avrdude: processing -T erase eeprom
Caching | ################################################## | 100% 0.38 s 
avrdude: synching cache to device ... 
Writing | ################################################## | 100% 2.72 s 

avrdude done.  Thank you.

mcuee@UbuntuSwift3 ~/build/avr/avrdude_bin $ ./avrdude_cb5f595 -C avrdude_cb5f595.conf -c stk500v1 -P /dev/ttyUSB0 -p m328p -U eeprom:w:holes_pack_my_box_1024B.hex:i

avrdude_cb5f595: AVR device initialized and ready to accept instructions
avrdude_cb5f595: device signature = 0x1e950f (probably m328p)
avrdude_cb5f595: reading input file holes_pack_my_box_1024B.hex for eeprom
                 with 252 bytes in 2 sections within [0x82, 0x37d]
                 using 64 pages and 4 pad bytes
avrdude_cb5f595: writing 252 bytes eeprom ...

Writing | ################################################## | 100% 0.58 s 

avrdude_cb5f595: 252 bytes of eeprom written
avrdude_cb5f595: verifying eeprom memory against holes_pack_my_box_1024B.hex

Reading | ################################################## | 100% 0.08 s 

avrdude_cb5f595: 252 bytes of eeprom verified

avrdude_cb5f595 done.  Thank you.
```

stefanrueger, 2024-04-15:
Brilliant analysis, @mcuee. Please try adding in git main
``` diff
diff --git a/src/avrdude.conf.in b/src/avrdude.conf.in
index 78372b05..dfa46c67 100644
--- a/src/avrdude.conf.in
+++ b/src/avrdude.conf.in
@@ -10877,6 +10877,7 @@ part # m328
         page_size          = 4;
         min_write_delay    = 3600;
         max_write_delay    = 3600;
+        readback           = 0xff 0xff;
         mode               = 0x41;
         delay              = 20;
         blocksize          = 4;
```

We may need a solution something akin to PR  #1511

mcuee, 2024-04-15:
@stefanrueger 

Great. Yes ths issue is resolved with your patch.

```
mcuee@UbuntuSwift3 ~/build/avr/avrdude_bin $ ./avrdude -c usbasp -p m328p -T "erase eeprom"
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e950f (probably m328p)

avrdude: processing -T erase eeprom
Caching | ################################################## | 100% 0.41 s 
avrdude: synching cache to device ... 
Writing | ################################################## | 100% 2.73 s 

avrdude done.  Thank you.

mcuee@UbuntuSwift3 ~/build/avr/avrdude_bin $ ./avrdude_git -C avrdude_issue1713.conf -c stk500v1 -P /dev/ttyUSB0 -p m328p -U eeprom:w:holes_pack_my_box_1024B.hex:i
avrdude_git: AVR device initialized and ready to accept instructions
avrdude_git: device signature = 0x1e950f (probably m328p)

avrdude_git: processing -U eeprom:w:holes_pack_my_box_1024B.hex:i
avrdude_git: reading input file holes_pack_my_box_1024B.hex for eeprom
             with 252 bytes in 2 sections within [0x82, 0x37d]
             using 64 pages and 4 pad bytes
avrdude_git: writing 252 bytes eeprom ...
Writing | ################################################## | 100% 1.14 s 
avrdude_git: 252 bytes of eeprom written
avrdude_git: verifying eeprom memory against holes_pack_my_box_1024B.hex
Reading | ################################################## | 100% 0.16 s 
avrdude_git: 252 bytes of eeprom verified

avrdude_git done.  Thank you.

mcuee@UbuntuSwift3 ~/build/avr/avrdude_bin $ ./avrdude_git -C avrdude_issue1713.conf -c stk500v1 -P /dev/ttyUSB0 -p m328p -U eeprom:w:the_quick_brown_fox_1024B.hex 
avrdude_git: AVR device initialized and ready to accept instructions
avrdude_git: device signature = 0x1e950f (probably m328p)

avrdude_git: processing -U eeprom:w:the_quick_brown_fox_1024B.hex:i
avrdude_git: reading input file the_quick_brown_fox_1024B.hex for eeprom
             with 1024 bytes in 1 section within [0, 0x3ff]
             using 256 pages and 0 pad bytes
avrdude_git: writing 1024 bytes eeprom ...
Writing | ################################################## | 100% 4.35 s 
avrdude_git: 1024 bytes of eeprom written
avrdude_git: verifying eeprom memory against the_quick_brown_fox_1024B.hex
Reading | ################################################## | 100% 0.62 s 
avrdude_git: 1024 bytes of eeprom verified

avrdude_git done.  Thank you.
```

mcuee, 2024-04-21:
PR #1753 is good and it fixes the issue. Tested with ATmega328P and ATmega2560.
* #1753

closed 2024-04-25 by stefanrueger