avrdudes/avrdude #1716: EE-Prom readout

MartinKES, 2024-03-13:
Using a AT90CA128 on a interactiv Session: "avrdude -c stk500 -p AT90CAN128 -P [\\.\com9](file://./com9) -t"

Readout offset 8, 9 are always 0xFF but should be 0

MCUdude, 2024-03-13:
Avrdude is most likely using an older avrdude.conf file that contains errors.
Please post the entire Avrdude output with the `-v` flag as well.

mcuee, 2024-03-13:
@MartinKES 
Most likely an old version of avrdude is used.

Are you using Arduino As ISP as the programmer? Please use avrdude 7.3 release and `-c arduino_as_isp` and try again.
* https://github.com/avrdudes/avrdude/issues/1337
* https://github.com/avrdudes/avrdude/pull/1383

ndim, 2024-03-13:
Question to @MCUdude: An older `avrdude.conf` in the sense of older than the version of `avrdude` executable being used, or an older `avrdude.conf` in the sense of version consistent with the `avrdude` executable being used but containing unfixed bugs?

MCUdude, 2024-03-13:
> Question to @MCUdude: An older avrdude.conf in the sense of older than the version of avrdude executable being used, or an older avrdude.conf in the sense of version consistent with the avrdude executable being used but containing unfixed bugs?

A newer Avrdude executable is being used with an older avrdude.conf file, most likely bundled or built for an older Avrdude version.

> In version 7.3 the same result

Can you please share the _entire_ output, including the command you ran? It's impossible to say what's going on here. From what I know, the baud rate is incorrect. But that's impossible for me to know because you didn't include the Avrdude command.

MartinKES, 2024-03-13:
C:\Daten\Test\AvrDude\V7_3>avrdude -c stk500 -p AT90CAN128 -P \\.\com9 -t
avrdude: successfully opened stk500v2 device; please use -c stk500v2
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9781 (probably c128)

avrdude: processing -t interactive terminal
avrdude> dump ee 0 128
Reading | ################################################## | 100% 0.47 s
0000  01 03 00 00 00 00 00 00  ff ff ff ff ff ff ff ff  |................|
0010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0020  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0030  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0040  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0050  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0060  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0070  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
avrdude> quit

avrdude done.  Thank you.

C:\Daten\Test\AvrDude\V7_3>avrdude -c arduino_as_isp -p AT90CAN128 -P \\.\com9 -t
avrdude error: programmer is not responding
avrdude warning: attempt 1 of 10: not in sync: resp=0x00
avrdude warning: attempt 2 of 10: not in sync: resp=0xe1
avrdude warning: attempt 3 of 10: not in sync: resp=0xe5
avrdude warning: attempt 4 of 10: not in sync: resp=0xcb
avrdude warning: attempt 5 of 10: not in sync: resp=0x2b
avrdude warning: attempt 6 of 10: not in sync: resp=0xe1
avrdude warning: attempt 7 of 10: not in sync: resp=0xe5
avrdude warning: attempt 8 of 10: not in sync: resp=0xcb
avrdude warning: attempt 9 of 10: not in sync: resp=0x2b
avrdude warning: attempt 10 of 10: not in sync: resp=0xe1
avrdude error: unable to open port \\.\com9 for programmer arduino_as_isp
avrdude error: avrdude built without libserialport support; please compile again with libserialport installed

avrdude done.  Thank you.

mcuee, 2024-03-13:
@MartinKES 

In this case, your programmer is not "Arduino as ISP", but rather using `-c stk500v2`. What is your prorgrammer used?

Please follow the FAQ here when reporting an issue. Thanks.
https://github.com/avrdudes/avrdude/wiki/FAQ#how-to-report-an-issue-to-avrdude-project
**Tell us what exactly the problem is and how you came to experience it**
**- OS**
**- Programmer**
- Part
- How host, programmer and part are connected/rigged up
- AVRDUDE binary version used (in bash avrdude -v 2>&1 | grep -i version)
**- Full command line including all options**
**- Output of the command**
- Please try to use the latest released version
- You may be requested to try out git main as well
- Sometimes it also helps us if you upload a file with the -vv or -vvvv output
**- Sometimes it also helps us if you post the hex file used (if you are okay to share the hex file)**
- What you would have expected to happen

MCUdude, 2024-03-13:
> In this case, your programmer is not "Arduino as ISP", but rather using -c stk500v2

It was you that suggested `arduino_as_isp`, not him. He states that "Readout offset 8, 9 are always 0xFF but should be 0", but Avrdude reports something else.

```
avrdude: processing -t interactive terminal
avrdude> dump ee 0 128
Reading | ################################################## | 100% 0.47 s
0000  01 03 00 00 00 00 00 00  ff ff ff ff ff ff ff ff  |................|
0010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0020  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0030  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0040  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0050  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0060  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0070  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
avrdude> quit
```

@MartinKES how can you be sure that the actual EEPROM values stored at addresses 8 and 9 should be 0? What makes you think Avrdude reports incorrect data?

MartinKES, 2024-03-13:
@MCUdude 
- IAR Workbench 
- myAVRProgTool

Byte 0 Major Version
Byte 1 Minor Version
Byte 2-5 DWord actual Error Flags
Byte 6-9 DWord ored ErrorFlags

avrdude: processing -t interactive terminal
avrdude> dump ee 8 2
Reading | ################################################## | 100% 0.03 s
0008  ff ff                                             |..              |
avrdude> dump ee 8 2
Reading | ################################################## | 100% 0.00 s
0008  ff ff                                             |..              |
avrdude> dump ee 8 2
Reading | ################################################## | 100% 0.00 s
0008  ff ff                                             |..              |
avrdude> dump ee 8 2
Reading | ################################################## | 100% 0.00 s
0008  ff ff                                             |..              |
avrdude> dump ee 8 2
Reading | ################################################## | 100% 0.00 s
0008  ff ff                                             |..              |
avrdude> dump eeprom 8 2
Reading | ################################################## | 100% 0.00 s
0008  ff ff                                             |..              |
avrdude> dump eeprom 7 4
Reading | ################################################## | 100% 0.03 s
0007  00 ff ff ff                                       |....            |
avrdude> quit

at offset 182 are the next Databytes (Parameters):

Reading | ################################################## | 100% 0.50 s
0008  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0018  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0028  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0038  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0048  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0058  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0068  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0078  ff ff ff ff ff ff ff ff  76 19 10 19 00 00 00 00  |........v.......|

MartinKES, 2024-03-13:
Output from atprogram --> mySmartUSB light:
F:\Atmel\atprog\atbackend>atprogram -t stk500 -c COM9 -d at90can128 -i ISP read -ee -s 128 -o 0
Firmware check OK
:1000000001030000000000000000FFFFFFFFFFFFF2
:10001000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF0
:10002000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFE0
:10003000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFD0
:10004000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFC0
:10005000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFB0
:10006000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFA0
:10007000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF90
:00000001FF
Programmer I.O.

MCUdude, 2024-03-13:
Can you try to read the EEPROM using -U? Like so:

`C:\Daten\Test\AvrDude\V7_3>avrdude -c stk500 -p AT90CAN128 -P \\.\com9 -U eeprom:r:-:i`

MartinKES, 2024-03-13:
avrdude: processing -U eeprom:r:-:i
avrdude: reading eeprom memory ...
Reading | ################################################## | 100% 13.86 s
avrdude: writing output file <stdout>
:200000000103000000000000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF4
:20002000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFE0
:20004000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFC0
:20006000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFA0
:200080007619101900000000000000000000000000000000000000000000000000000000A8
:2000A000000000000000000000000000000000000000000000000000000000000000000040
:2000C000000000000000000000000000000000000000000000000000000000000000000020
:2000E000000000000000000000000000000000000000000000000000000000000000000000
:200100000000000000000000000000000000000000000000000000000000000000000000DF

mcuee, 2024-03-13:
> My programmer is mySmartUSB ([www.myAVR.de](http://www.myAVR.de)) light.

The good thing is that you have another programmer Atmel ICE and confirms that there is an issue here.

Now there could be two possibilities.
1) avrdude has a bug with `-c stk500v2` for AT90CAN128 chip when reading EEPROM
2) mySmartUSB is not compatible with official STK500v2 FW in terms of EEPROM reading.

We need to check official STK500v2 now to which one is true.

@MCUdude 
I could not find my AT90CAN128 board now. Please help here. Thanks.

MartinKES, 2024-03-14:
I have a STK500 (very old) on my desk. 

so please wait on hour

MartinKES, 2024-03-14:
I have found a AvrISP with serial link

C:\Daten\Test\AvrDude\V7_3>avrdude -p c128 -c stk500v2 -P\\.\com5 -t

avrdude> dump ee 0 128
Reading | ################################################## | 100% 0.50 s
0000  01 03 00 00 00 00 00 00  00 00 ff ff ff ff ff ff  |................|
0010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0020  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0030  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0040  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0050  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0060  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0070  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
avrdude> dump ee 256 128
Reading | ################################################## | 100% 0.51 s
0100  76 19 10 19 00 00 00 00  00 00 00 00 00 00 00 00  |v...............|
0110  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
0120  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
0130  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
0140  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
0150  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
0160  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
0170  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
avrdude>

Its oll Ok - so my SmartUSB Programmer should be the Problem

Sorry for the inconvenience
Martin Heumann

MartinKES, 2024-03-14:
Done with AvrISP (serial):

C:\Daten\Test\AvrDude\V7_3>avrdude -p c128 -c stk500v2 -P\\.\com5 -vvvv -T 'read eeprom 0 16'

avrdude: Version 7.3
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is C:\Daten\Test\AvrDude\V7_3\avrdude.conf

         Using port            : \\.\com5
         Using programmer      : stk500v2
avrdude: ser_send: . [1b] . [01] . [00] . [01] . [0e] . [01] . [14]
avrdude: ser_recv: . [1b]
avrdude: ser_recv: . [01]
avrdude: ser_recv: . [00]
avrdude: ser_recv: . [0b]
avrdude: ser_recv: . [0e]
avrdude: ser_recv: . [01]
avrdude: ser_recv: . [00]
avrdude: ser_recv: . [08]
avrdude: ser_recv: A [41]
avrdude: ser_recv: V [56]
avrdude: ser_recv: R [52]
avrdude: ser_recv: I [49]
avrdude: ser_recv: S [53]
avrdude: ser_recv: P [50]
avrdude: ser_recv: _ [5f]
avrdude: ser_recv: 2 [32]
avrdude: ser_recv: t [74]
avrdude: stk500v2_getsync(): found AVRISP programmer
         AVR Part              : AT90CAN128
         Programming modes     : ISP, HVPP, JTAG, JTAGmkI, SPM

         Memory         Size  Pg size
         ----------------------------
         eeprom         4096        8
         flash        131072      256
         efuse             1        1
         hfuse             1        1
         lfuse             1        1
         lock              1        1
         signature         3        1
         calibration       1        1
         io              224        1
         sram           4096        1

         Variants          Package  F max   T range         V range
         -----------------------------------------------------------------
         AT90CAN128        N/A      20 MHz  [-40 C,  85 C]  [4.5 V,   N/A]
         AT90CAN128-15AZ   TQFP64   16 MHz  [-40 C, 125 C]  [2.7 V, 5.5 V]
         AT90CAN128-16AI   TQFP64   16 MHz  [-40 C,  85 C]  [2.7 V, 5.5 V]
         AT90CAN128-16AU   TQFP64   16 MHz  [-40 C,  85 C]  [2.7 V, 5.5 V]
         AT90CAN128-16AUR  TQFP64   16 MHz  [-40 C,  85 C]  [2.7 V, 5.5 V]
         AT90CAN128-16MI   QFN64    16 MHz  [-40 C,  85 C]  [2.7 V, 5.5 V]
         AT90CAN128-16MU   QFN64    16 MHz  [-40 C,  85 C]  [2.7 V, 5.5 V]
         AT90CAN128-16MUR  QFN64    16 MHz  [N/A,     N/A]  [2.7 V, 5.5 V]

         Programmer Type       : STK500V2
         Description           : Atmel STK500 version 2.x firmware
         Programmer model      : AVRISP
avrdude: ser_send: . [1b] . [02] . [00] . [02] . [0e] . [03] . [90] . [86]
avrdude: ser_recv: . [1b]
avrdude: ser_recv: . [02]
avrdude: ser_recv: . [00]
avrdude: ser_recv: . [03]
avrdude: ser_recv: . [0e]
avrdude: ser_recv: . [03]
avrdude: ser_recv: . [00]
avrdude: ser_recv: . [0f]
[... 77 lines cut ...]
avrdude: ser_recv: . [09]
avrdude: ser_recv: . [00]
avrdude: ser_recv: . [04]
avrdude: ser_recv: . [0e]
avrdude: ser_recv: . [1b]
avrdude: ser_recv: . [00]
avrdude: ser_recv: . [81]
avrdude: ser_recv: . [00]
avrdude: ser_recv: . [82]
Reading | ################################################## | 100% 0.08 s
avrdude: device signature = 0x1e9781 (probably c128)

avrdude: processing -T 'read
avrdude do_cmd() [term.c:2373] error: (cmd) command 'read is invalid
avrdude: ser_send: . [1b] . [0a] . [00] . [03] . [0e] . [11] . [01] . [01] . [0d]
avrdude: ser_recv: . [1b]
avrdude: ser_recv: . [0a]
avrdude: ser_recv: . [00]
avrdude: ser_recv: . [02]
avrdude: ser_recv: . [0e]
avrdude: ser_recv: . [11]
avrdude: ser_recv: . [00]
avrdude: ser_recv: . [0c]

avrdude done.  Thank you.

closed 2024-03-14 by MartinKES

mcuee, 2024-03-14:
@MartinKES 

Please do the same with your SmartUSB Programmer. There might be a way to support your programmer if we can differentiate the two in terms of USB VID/PID and the firmware differences when reading the EEPROM.

MCUdude, 2024-03-14:
> error: (cmd) command 'read is invalid

@MartinKES  It looks like your shell doesn't line single quotes. can you try with double quotes instead, for both programmers?

```
avrdude -p c128 -c stk500v2 -P\\.\com5 -vvvv -T "read eeprom 0 16"
```

MartinKES, 2024-03-14:
Done with AvrDude 7.3 --> mySmartUSB light 
... Initialisierung
avrdude> read eeprom 0 16
Reading |                                                    | 0% 0.00 s
avrdude: ser_send: . [1b] . [10] . [00] . [05] . [0e] . [06] . [00] . [00] . [00] . [00] . [06]
avrdude: ser_recv: . [1b]
avrdude: ser_recv: . [10]
avrdude: ser_recv: . [00]
avrdude: ser_recv: . [02]
avrdude: ser_recv: . [0e]
avrdude: ser_recv: . [06]
avrdude: ser_recv: . [00]
avrdude: ser_recv: . [01]
avrdude: ser_send: . [1b] . [11] . [00] . [04] . [0e] . [16] . [00] . [08] . [a0] . [be]
avrdude: ser_recv: . [1b]
avrdude: ser_recv: . [11]
avrdude: ser_recv: . [00]
avrdude: ser_recv: . [0b]
avrdude: ser_recv: . [0e]
avrdude: ser_recv: . [16]
avrdude: ser_recv: . [00]
avrdude: ser_recv: . [01]
avrdude: ser_recv: . [03]
avrdude: ser_recv: . [00]
avrdude: ser_recv: . [00]
avrdude: ser_recv: . [00]
avrdude: ser_recv: . [00]
avrdude: ser_recv: . [00]
avrdude: ser_recv: . [00]
avrdude: ser_recv: . [00]
avrdude: ser_recv: . [1b]
Reading | ######################                             | 43% 0.04 s
avrdude: ser_send: . [1b] . [12] . [00] . [05] . [0e] . [06] . [00] . [00] . [00] . [08] . [0c]
avrdude: ser_recv: . [1b]
avrdude: ser_recv: . [12]
avrdude: ser_recv: . [00]
avrdude: ser_recv: . [02]
avrdude: ser_recv: . [0e]
avrdude: ser_recv: . [06]
avrdude: ser_recv: . [00]
avrdude: ser_recv: . [03]
avrdude: ser_send: . [1b] . [13] . [00] . [04] . [0e] . [16] . [00] . [08] . [a0] . [bc]
avrdude: ser_recv: . [1b]
avrdude: ser_recv: . [13]
avrdude: ser_recv: . [00]
avrdude: ser_recv: . [0b]
avrdude: ser_recv: . [0e]
avrdude: ser_recv: . [16]
avrdude: ser_recv: . [00]
avrdude: ser_recv: . [ff]
avrdude: ser_recv: . [ff]
avrdude: ser_recv: . [ff]
avrdude: ser_recv: . [ff]
avrdude: ser_recv: . [ff]
avrdude: ser_recv: . [ff]
avrdude: ser_recv: . [ff]
avrdude: ser_recv: . [ff]
avrdude: ser_recv: . [00]
avrdude: ser_recv: . [1b]
Reading | ################################################## | 100% 0.08 s
0000  01 03 00 00 00 00 00 00  ff ff ff ff ff ff ff ff  |................|
avrdude>

MartinKES, 2024-03-15:
@mcuee 
"avrdude -p c128 -c stk500v2 -P com9 -U ee:r:EeProm:i"  with mySmartUSB light
[eeprom.txt](https://github.com/avrdudes/avrdude/files/14614835/eeprom.txt)
[eeprom.txt](https://github.com/avrdudes/avrdude/files/14614841/eeprom.txt)
"since you mention that IAR Embedded Workbench seems to work fine." that was with AtmelIce Interface

mcuee, 2024-03-15:
>"since you mention that IAR Embedded Workbench seems to work fine." that was with AtmelIce Interface.

I see, 

Then it is clear that the issue is with `mySmartUSB light` prorgammer, no matter in the normal CLI mode or the Terminal mode.

mcuee, 2024-03-15:
@MartinKES 

Since myAVRProgTool works fine, If you really want to get your programmer working with avrdude, you probably have to reverse engineering the FW differences between official STK500v2 and `mySmartUSB light` programmer.

Probably not really worth the effort since you have other well supported programmers.

mcuee, 2024-03-15:
Change the title so that it is easier for future references.

MartinKES, 2024-03-15:
What is a good alternative for a programmer (USB to ISP) without drivers?

mcuee, 2024-03-15:
> What is a good alternative for a programmer (USB to ISP) without drivers?

There is no such thing as "without drivers" for Windows. Rather you want to use built-in drivers like WinUSB or USB HID.

Cheap and good one: well supported by avrdude: USBASP
You do need to flash the right FW in order to use WinUSB driver out of box.
https://github.com/nerdralph/usbasp

Reference:
https://github.com/avrdudes/avrdude/wiki/FAQ#i-am-new-to-avrdude-and-i-do-not-have-a-programmer-what-are-the-recommended-programmers-to-be-used-with-avrdude

rsinformatik, 2025-08-06:
I had the same issue. An firmware update of the mySmartUSB light programmer solved it.  (https://shop.myavr.de/index.php?sp=download.sp.php&suchwort=dl197)