avrdudes/avrdude #1718: avr_{read,write}_mem fail to initialize progress reporting

dl8dtl, 2024-03-18:
This occurred to me in the course of working on #1714.
Both functions don't initialize progress reporting, so it works only once.

https://github.com/avrdudes/avrdude/pull/1714/commits/296883cf1aa69ca80bb47068d1215b94dd379f73

stefanrueger, 2024-03-19:
Oh, the lack of initiating progress reports may have been deliberate! At some point the AVRDUDE project felt it had far too many progress reports (eg, reading all the one-byte fuses) and the inititiation of reporting was moved *outside* avr_{read,write}() to give the caller agency over whether or not progress should be reported. Here an example:
https://github.com/avrdudes/avrdude/blob/34366a6b9d4b489099e0cd9853d1e60098b44a41/src/update.c#L406-L410

The progress reporting system only ever reports progress if initiated beforehand, otherwise reporting calls just return.

dl8dtl, 2024-03-19:
Hmm ok.

Yes, I noticed that "normal" avrdude CLI now had two times "Reading". So I withdraw this PR (but we could keep the discussion).

Too bad, I tried to set up `avr_read_mem` and `avr_write_mem` as kind-of generic memory handling functions for the Python wrapper. Maybe I can find a way to have the Python wrapper silently initialize progress reports.

The above looks a bit hacky to me anyway: the very first call to `avr_read_mem` would still report progress, even for small memory sizes. It seems to me that a flag explicitly telling whether progress reports are desired were a better solution.

stefanrueger, 2024-03-19:
> the very first call to avr_read_mem would still report progress, even for small memory sizes

That would be a bug (not sure why that wasn't noticed in the AVRDUDE implementation); I'm likely out of bounds for the next week, so unable to look into this soon.

> looks a bit hacky to me

Could be a case of improving the documentation
https://github.com/avrdudes/avrdude/blob/34366a6b9d4b489099e0cd9853d1e60098b44a41/src/avr.c#L1625-L1646
or, as you say, have an explicit flag (and document).

dl8dtl, 2024-03-21:
For the time being, I worked around it in the Python wrapper by prepending the respective call to `report_progress` in the wrapper. Still, I think a cleaner solution would be worth the while.

closed 2024-03-29 by stefanrueger