avrdudes/avrdude #1719: Old `avrdude_message()` can go away

dl8dtl, 2024-03-18:
I just noticed that it's not used anymore anywhere.

stefanrueger, 2024-03-27:
Do you want the new `avrdude_message2()` be renamed to `avrdude_message()` once the old one is removed?

mcuee, 2024-03-27:
> Do you want the new `avrdude_message2()` be renamed to `avrdude_message()` once the old one is removed?

Hmm, I think it is better not to do that since it will cause confusion. 

We can add a note in `avrdude_message2()` to mention that there was  `avrdude_message()` function  in the history but that was removed since it is no longer in use.

dl8dtl, 2024-03-29:
> We can add a note in `avrdude_message2()` to mention that there was `avrdude_message()` function in the history but that was removed since it is no longer in use.

I did so on Stefan's PR.

PR #1730 (2024-03-29): Remove unused avrdude_message()
Fixes #1719
commit 85162436 (2024-03-29): Merge pull request #1730 from stefanrueger/remove_avrdude_message

Remove unused avrdude_message()

Fixes #1719
closed 2024-03-29 by dl8dtl